raspberrypi/linux #3306: ca-certificates-java does not install

volkerjaenisch, 2019-10-29:
Failed to install java on a rpi-zerow

Installing ca-certificates-java via apt leads to:

```
Errors were encountered while processing:
 ca-certificates-java
E: Sub-process /usr/bin/dpkg returned an error code (1)
Setting up ca-certificates-java (20190405) ...
cp: -r not specified; omitting directory '/etc/ssl/certs/java/cacerts'
```

This should not happen.

**To reproduce**
apt-get install ca-certificates-java

**Expected behaviour**
THe package should install cleanly

**Actual behaviour**
THe package fails

**System**
System Information
------------------

Raspberry Pi Zero W Rev 1.1
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

Linux raspberrypi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux
Revision	: 9000c1
Serial		: 000000001edcf469
Model		: Raspberry Pi Zero W Rev 1.1
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Sep 24 2019 17:37:19 
Copyright (c) 2012 Broadcom
version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (clean) (release) (start)

alloc failures:     0
compactions:        0
legacy block fails: 0

Filesystem information
----------------------
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root        7345680 2067840   4943820  30% /
devtmpfs          217212       0    217212   0% /dev
tmpfs             221540       0    221540   0% /dev/shm
tmpfs             221540    3152    218388   2% /run
tmpfs               5120       4      5116   1% /run/lock
tmpfs             221540       0    221540   0% /sys/fs/cgroup
/dev/mmcblk0p1    258095   53032    205064  21% /boot
tmpfs              44308       0     44308   0% /run/user/1000

Filename				Type		Size	Used	Priority
/var/swap                               file		102396	7168	-2

Package version information
[... 440 lines cut ...]
[   18.041607] usbcore: registered new interface driver brcmfmac
[   18.319850] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[   18.320066] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[   18.321270] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
[   23.829998] random: crng init done
[   23.830026] random: 7 urandom warning(s) missed due to ratelimiting
[   27.714446] uart-pl011 20201000.serial: no DMA platform data
[   28.721661] 8021q: 802.1Q VLAN Support v1.8
[   30.604740] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[   30.604770] brcmfmac: power management disabled
[   30.612035] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   32.396097] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[   32.801690] Bluetooth: Core ver 2.22
[   32.801855] NET: Registered protocol family 31
[   32.801866] Bluetooth: HCI device and connection manager initialized
[   32.801894] Bluetooth: HCI socket layer initialized
[   32.801908] Bluetooth: L2CAP socket layer initialized
[   32.801962] Bluetooth: SCO socket layer initialized
[   32.828577] Bluetooth: HCI UART driver ver 2.3
[   32.828596] Bluetooth: HCI UART protocol H4 registered
[   32.833534] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   32.837128] Bluetooth: HCI UART protocol Broadcom registered
[   33.643817] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   33.643831] Bluetooth: BNEP filters: protocol multicast
[   33.643858] Bluetooth: BNEP socket layer initialized

XECDesign, 2019-10-29:
Oracle Java and OpenJDK do not support ARMv6. I am not sure what the last version that did was, but I think it was 8. OpenJDK Zero may also work, but I haven't tested it. Both are available from the repo.

You'll need to uninstall whtarever jre/jdk you've got on there first.

JamesH65, 2020-02-14:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

closed 2020-11-18 by pelwell

JsBergbau, 2022-02-06:
> Oracle Java and OpenJDK do not support ARMv6

That is not exactly true. It is the server vm that isn't supported, but client VM is supported. Here https://www.azul.com/downloads/?version=java-11-lts&architecture=arm-32-bit-hf&package=jdk you find Java / JDK 11 for Raspberry PI Zero W.

Here https://github.com/JsBergbau/OpenJDK-Raspberry-Pi-Zero-W-armv6/blob/main/README.md you find OpenJDK 17 for armv6. It runs so far flawlessly signal-cli.

XECDesign, 2022-02-07:
> Here https://github.com/JsBergbau/OpenJDK-Raspberry-Pi-Zero-W-armv6/blob/main/README.md you find OpenJDK 17 for armv6 including the build instructions. It runs so far flawlessly signal-cli.

Thanks, I'll see if I can make the Debian package build with those options.

JsBergbau, 2022-02-07:
Thanks. Build instructions for OpenJDK 11 should also be the same. I suggest starting with OpenJDK 17 and if you need help, open an issue in the Repo and I'll have a look. When you've managed building OpenJDK 17 then we can have a look at OpenJDK 11.

XECDesign, 2022-02-17:
OpenJDK 17 seems to work, but I'm not having any luck with 11:

```
$ java
=============== DEBUG MESSAGE: Atomic load(jlong) unsupported on this platform ================

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting ((null)), id 0xe0000000, Internal Error (../../src/hotspot/cpu/arm/macroAssembler_arm.cpp:1629)]

...

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[error occurred during error reporting (printing all threads), id 0xe0000000, Internal Error (../../src/hotspot/cpu/arm/macroAssembler_arm.cpp:1629)]

=============== DEBUG MESSAGE: Atomic store(jlong) unsupported on this platform ================

[Too many errors, abort]
Aborted
```

OpenJDK 17 is in the 'untested' repo component for now.

JsBergbau, 2022-02-17:
Thanks for your info. I'm really happy that compiling JDK17 worked.

I didn't try to compile JDK11 so far because I used JDK 11 provided by Azul https://www.azul.com/downloads/?version=java-11-lts&architecture=arm-32-bit-hf&package=jdk

What options did you use to compile JDK 11?

XECDesign, 2022-02-17:
From the build log:
```
cd build && MAKE_VERBOSE=y QUIETLY= LOG=debug IGNORE_OLD_CONFIG=true LIBFFI_LIBS=-lffi_pic JOBS=9 bash ../configure \
        --with-jvm-variants=client,server --with-boot-jdk=/usr/lib/jvm/java-11-openjdk-armhf --disable-precompiled-headers --with-jvm-features=shenandoahgc --with-extra-cflags='-Wdate-time -D_FORTIFY_SOURCE=2 -g -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -fno-stack-protector -Wno-deprecated-declarations -Wdate-time -D_FORTIFY_SOURCE=2' --with-extra-cxxflags='-Wdate-time -D_FORTIFY_SOURCE=2 -g -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -fno-stack-protector -Wno-deprecated-declarations -fpermissive -Wno-deprecated-declarations' --with-extra-ldflags='-Xlinker -z -Xlinker relro -Xlinker -Bsymbolic-functions' \
        --with-abi-profile=armv6-vfp-hflt --enable-ccache --with-vendor-name='Raspbian' --with-vendor-url='Unknown' --with-vendor-bug-url='Unknown' --with-vendor-vm-bug-url='Unknown' --with-version-pre='' --with-version-build='9' --with-version-opt='post-Raspbian-1~deb11u1+rpt1' --with-copyright-year=2022  --with-debug-level=release --with-native-debug-symbols=internal --enable-unlimited-crypto --with-zlib=system --with-giflib=system --with-libpng=system --with-libjpeg=system --with-lcms=system --with-pcsclite=system --disable-warnings-as-errors --disable-javac-server  --with-harfbuzz=system --with-stdc++lib=dynamic --with-num-cores=9
```

JsBergbau, 2022-02-17:
I've noticed `--with-jvm-variants=client,server`. Since I've only build the client variante for OpenJDK17: Can you also provide the options for your working OpenJDK 17?

XECDesign, 2022-02-17:
Pretty much the same for OpenJDK17, just missing the `--with-abi-profile=armv6-vfp-hflt`. OpenJDK11 behaves the same with or without it under Raspbian's build environment, I was just seeing if it made a difference.

```
cd build && MAKE_VERBOSE=y QUIETLY= LOG=info,cmdlines IGNORE_OLD_CONFIG=true LIBFFI_LIBS=-lffi_pic JOBS=9 bash ../configure \
        --with-jvm-variants=client,server --with-boot-jdk=/usr/lib/jvm/java-17-openjdk-armhf --disable-precompiled-headers --with-jvm-features= --with-extra-cflags='-Wdate-time -D_FORTIFY_SOURCE=2 -g -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -fno-stack-protector -Wno-deprecated-declarations -Wdate-time -D_FORTIFY_SOURCE=2' --with-extra-cxxflags='-Wdate-time -D_FORTIFY_SOURCE=2 -g -ffile-prefix-map=/<<PKGBUILDDIR>>=. -fstack-protector-strong -Wformat -fno-stack-protector -Wno-deprecated-declarations -fpermissive -Wno-deprecated-declarations' --with-extra-ldflags='-Xlinker -z -Xlinker relro -Xlinker -Bsymbolic-functions' \
        --with-jni-libpath=/usr/lib/arm-linux-gnueabihf/jni:/lib/arm-linux-gnueabihf:/usr/lib/arm-linux-gnueabihf:/usr/lib/jni:/lib:/usr/lib --enable-ccache --disable-jtreg-failure-handler --disable-warnings-as-errors --with-vendor-name='Raspbian' --with-vendor-url='http://www.raspbian.org/' --with-vendor-bug-url='https://bugs.launchpad.net/raspbian/+filebug' --with-vendor-vm-bug-url='https://bugs.launchpad.net/raspbian/+filebug' --with-vendor-name='Raspbian' --with-vendor-url='Unknown' --with-vendor-bug-url='Unknown' --with-vendor-vm-bug-url='Unknown' --with-version-pre='' --with-version-build='8' --with-version-opt='Raspbian-1~deb11u1+rpt1' --with-copyright-year=2022  --with-debug-level=release --with-native-debug-symbols=internal --enable-unlimited-crypto --with-zlib=system --with-giflib=system --with-libpng=system --with-libjpeg=system --with-lcms=system --with-pcsclite=system --disable-warnings-as-errors --disable-javac-server  --with-harfbuzz=system --with-stdc++lib=dynamic --with-num-cores=9
```

Building with server enabled as well for people who want it, but have an additional patch to jvm.cfg to default to client.

JsBergbau, 2022-02-18:
I don't see the `--with-sysroot=` option, so this seems another build process and JDK-11 has no sysroot described in their help section.
I get JDK 11 configured by
```
 sh ./configure --openjdk-target=arm-linux-gnueabihf --with-sysroot=/home/pi/sysroot-armhf --with-abi-profile=armv6-vfp-hflt --with-num-cores=12 --with-memory-size=14000 --with-native-debug-symbols=zipped --with-jvm-variants=client --with-x=/home/pi/sysroot-armhf/usr --with-freetype=bundled
```

However when trying `make images` I get a lot of errors
```
make images
Building target 'images' in configuration 'linux-arm-normal-client-release'
Warning: No mercurial configuration present and no .src-rev
gmake[3]: *** Keine Regel vorhanden, um das Ziel „/home/pi/jdk-jdk-11-28/build/linux-arm-normal-client-release/buildtools/langtools_tools_classes/_the.BUILD_TOOLS_LANGTOOLS.vardeps“,
  benötigt von „/home/pi/jdk-jdk-11-28/build/linux-arm-normal-client-release/buildtools/langtools_tools_classes/_the.BUILD_TOOLS_LANGTOOLS_batch“, zu erstellen.  Schluss.
gmake[3]: *** Es wird auf noch nicht beendete Prozesse gewartet....
gmake[3]: *** Keine Regel vorhanden, um das Ziel „/home/pi/jdk-jdk-11-28/build/linux-arm-normal-client-release/make-support/vardeps/make/ReleaseFile.gmk/create-info-file.vardeps“,
  benötigt von „/home/pi/jdk-jdk-11-28/build/linux-arm-normal-client-release/jdk/release“, zu erstellen.  Schluss.
gmake[2]: *** [make/Main.gmk:344: release-file] Fehler 2
gmake[2]: *** Es wird auf noch nicht beendete Prozesse gewartet....
gmake[2]: *** [make/Main.gmk:73: buildtools-langtools] Fehler 2
gmake[3]: *** Keine Regel vorhanden, um das Ziel „/home/pi/jdk-jdk-11-28/build/linux-arm-normal-client-release/make-support/vardeps/make/ModuleWrapper.gmk/java.base/ORDERED_CFG_VARIANTS.vardeps“,
  benötigt von „/home/pi/jdk-jdk-11-28/build/linux-arm-normal-client-release/support/modules_libs/java.base/jvm.cfg“, zu erstellen.  Schluss.
gmake[3]: *** Es wird auf noch nicht beendete Prozesse gewartet....
gmake[3]: *** Keine Regel vorhanden, um das Ziel „/home/pi/jdk-jdk-11-28/build/linux-arm-normal-client-release/hotspot/variant-client/tools/jvmti/_the.BUILD_JVMTI_TOOLS.vardeps“,
  benötigt von „/home/pi/jdk-jdk-11-28/build/linux-arm-normal-client-release/hotspot/variant-client/tools/jvmti/_the.BUILD_JVMTI_TOOLS_batch“, zu erstellen.  Schluss.
gmake[2]: *** [make/Main.gmk:250: hotspot-client-gensrc] Fehler 2
gmake[2]: *** [make/Main.gmk:162: java.base-copy] Fehler 2

ERROR: Build failed for target 'images' in configuration 'linux-arm-normal-client-release' (exit code 2)

No indication of failed target found.
Hint: Try searching the build log for '] Error'.
Hint: See doc/building.html#troubleshooting for assistance.

make[1]: *** [/home/pi/jdk-jdk-11-28/make/Init.gmk:305: main] Fehler 2
make: *** [/home/pi/jdk-jdk-11-28/make/Init.gmk:186: images] Fehler 2
```
That seems to be a known problem https://www.mail-archive.com/bug-make@gnu.org/msg12006.html

Since azul provides a working JDK11 for armv6 https://www.azul.com/downloads/?version=java-11-lts&architecture=arm-32-bit-hf&package=jdk I would write them. Hopefully they help the raspberry Pi foundation.

Otherwise I consider JDK 11 a minor issue, because there is a working JDK11 available for Raspberry PI Zero provided by Azul. But there is currently no official JDK 17 available, so getting JDK 17 in the stable tree of apt would be a great improvement.

XECDesign, 2022-02-18:
We build almost all of our packages natively, so a sysroot is not needed.

I didn't see the issue you're running into, so it seems to be fixed or worked around by Debian's packaging.

> Since azul provides a working JDK11 for armv6 https://www.azul.com/downloads/?version=java-11-lts&architecture=arm-32-bit-hf&package=jdk I would write them. Hopefully they help the raspberry Pi foundation.

We've explored that avenue already. No luck.

Java issues in general seem to take a disproportionate amount of time to look into. At the end of the day, if OpenJDK devs don't want to support it, we can't do much about it without sacrificing more important work. In the meantime, Azul builds are available for people who require it.

Once the OpenJDK 17 build has had more testing, I can switch the 'full' images across so that everything works out of the box on armv6 pis, but I'll leave OpenJDK 11 as is.

JsBergbau, 2022-02-18:
> We build almost all of our packages natively, so a sysroot is not needed.

The commands look like that. What systems are you using to build then, since you configured 9 cores to use for the build? I've tried to use qemu with armv6 but with my limited knowledge I didn't manage to configure more than 512 MB RAM, because then it complained that Raspberry Pi also doesn't have more RAM.

> I didn't see the issue you're running into, so it seems to be fixed or worked around by Debian's packaging.

For cross compiling I'm on Debian bullseye and there are no updates available. BTW I'm very glad that you're using Debian and not Ubuntu because with Ubuntu there are so many updates and a lot more kernel updates which require reboot. At least for fully x64 systems I use.

> We've explored that avenue already. No luck.

This is a real pity and unfortunately does not speak for the company.

> Java issues in general seem to take a disproportionate amount of time to look into. At the end of the day, if OpenJDK devs don't want to support it, we can't do much about it without sacrificing more important work. 

For me Java is some ancient technology. It is very slow consumes a lot of memory and there are now much better alternatives available like Rust, Go or even python. Java claim(ed)s to run on Billions of devices but its so hard to get it to run on armv6. This is also very contradictory.
But since there are still a lot of programs based on Java you still need it.

What version of OpenJDK did you try to compile? Azul provides 11.0.14.1+1 which is probably based on https://github.com/openjdk/jdk/releases/tag/jdk-11%2B14 
Maybe that compiles and runs flawlessly? Perhaps in the latest version `--with-abi-profile=armv6-vfp-hflt` is not used because of a bug and that leads to that problem.

But if that takes to much ressources just leave it and focus on more important problems.

JsBergbau, 2022-02-21:
Just for the record, I've also tried to crosscompile 11.0.14 https://github.com/openjdk/jdk/archive/refs/tags/jdk-11+14.tar.gz
Did also not work. I'm also giving up trying to compile JDK 11.

> OpenJDK 17 should make its way into the repo some time this week, I think.

Thanks, that is great news.

maknon, 2022-06-26:
is there a chance to backport openjdk-17-jdk to buster

JsBergbau, 2022-07-03:
There is OpenJDK for Buster https://github.com/JsBergbau/OpenJDK-Raspberry-Pi-Zero-W-armv6