raspberrypi/linux #3308: python idle & idle3 doesn't launch in raspberrypi

0xh0n3y, 2019-10-29:
pi@raspberrypi:~ $ idle
Traceback (most recent call last):
  File "/usr/bin/idle", line 3, in <module>
    from idlelib.PyShell import main
  File "/usr/lib/python2.7/idlelib/PyShell.py", line 18, in <module>
    from code import InteractiveInterpreter
  File "/usr/lib/python2.7/code.py", line 10, in <module>
    from codeop import CommandCompiler, compile_command
  File "/usr/lib/python2.7/codeop.py", line 1
SyntaxError: Non-ASCII character '\x88' in file /usr/lib/python2.7/codeop.py on line 1, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details

pi@raspberrypi:~ $ idle3
from: can't read /var/mail/idlelib.PyShell
/usr/bin/idle3: line 6: syntax error: unexpected end of file

pelwell, 2019-10-29:
It works for me. I think your filesystem is corrupted - try a fresh installation.

0xh0n3y, 2019-10-29:
Which raspberrypi version is best for Raspberry Pi A+

0xh0n3y, 2019-10-29:
I'm using 2016 edition of RaspbeanOs

JamesH65, 2019-10-29:
The answer is almost always "The latest one". All our OS releases are backwards compatible.

0xh0n3y, 2019-10-29:
But what's the latest build of rapbeanOS should i use .Earlier when i flash the sd card with latest RaspbeanOs it doesn't support the Raspberry Pi 1 A+ because the RaspbeanOs is of 64-bit x64 architecture and the raspberry pi is of x86 architecture. Will latest RaspbeanOs available in 32 bit.

JamesH65, 2019-10-29:
The latest one from this page. https://www.raspberrypi.org/downloads/raspbian/

All our Pi OS's are 32bit at this stage. And as I said, all our OS releases are backwards compatible, so will work on ALL Pi models.

0xh0n3y, 2019-10-29:
Please share the link of latest RaspbeanOs compartable Raspberry Pi 1 Model A+ 512MB RAM

pelwell, 2019-10-29:
James just did. Download the latest from the page - one image supports all Pis.

0xh0n3y, 2019-10-30:
Fixed.

closed 2019-10-30 by 0xh0n3y