raspberrypi/linux #3309: How can I force raspberry4 to use EHCI instead of XHCI?

likang, 2019-10-29:
A hardware we used is incompatible with XHCI, I'm wondering how can I disable that since raspberry does not have BIOS.

pelwell, 2019-10-29:
> A hardware we used is incompatible with XHCI

What does that mean? XHCI defines a standard register interface to a USB3 controller, but as Wikipedia say, "The xHCI architecture was designed to support all USB speeds, including SuperSpeed (5 Gbit/s) and future speeds, under a single driver stack."

Your device on the USB bus shouldn't care how the USB host controller is programmed, only which versions of USB it supports (and whether it is compliant to those standards).

likang, 2019-10-29:
Thank you @pelwell , I have some problems on connect lots of USB device to raspberry4 (on raspberry 3b+ is OK), based on this article [WHY CAN'T I CONNECT MORE USB 3.0 DEVICES TO MY SYSTEM?](https://acroname.com/blog/why-cant-i-connect-more-usb-30-devices-my-system)  seems I have to disable "USB 3.0" mode, so I want to change my question to "can I use USB2.0 mode on Raspberry 4?"

P33M, 2019-10-29:
How many is "lots of USB devices"? Please post the output of `lsusb -t`.

What are "some problems"? Please post the output of `dmesg` after you have seen these problems occur.

The article you linked confuses hub tree depth with number of xHCI slots. Both are independent limitations of the hardware - hub tree depth is constrained by the USB specification, the number of simultaneously connected devices is constrained by the xHCI hardware implementation. The hardware used on Pi 4 has a maximum of 32 slots.

likang, 2019-10-29:
I plugged 1 device(contains 16 endpoints) on each USB port,  so totally I should get 4x16=64 endpoints, but now only 36 or 37 endpoints can be found.

`lsusb -t`

```pi@raspi:~ $ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 81, If 0, Class=Hub, Driver=hub/7p, 480M
        |__ Port 2: Dev 66, If 0, Class=Hub, Driver=hub/7p, 480M
            |__ Port 3: Dev 72, If 6, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 3: Dev 72, If 4, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 3: Dev 72, If 2, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 3: Dev 72, If 0, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 3: Dev 72, If 7, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 3: Dev 72, If 5, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 3: Dev 72, If 3, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 3: Dev 72, If 1, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 1: Dev 68, If 7, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 1: Dev 68, If 5, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 1: Dev 68, If 3, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 1: Dev 68, If 1, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 1: Dev 68, If 6, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 1: Dev 68, If 4, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 1: Dev 68, If 2, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 1: Dev 68, If 0, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 4: Dev 74, If 1, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 4: Dev 74, If 6, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 4: Dev 74, If 4, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 4: Dev 74, If 2, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 4: Dev 74, If 0, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 4: Dev 74, If 7, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 4: Dev 74, If 5, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 4: Dev 74, If 3, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 2: Dev 70, If 2, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 2: Dev 70, If 0, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 2: Dev 70, If 7, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 2: Dev 70, If 5, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 2: Dev 70, If 3, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 2: Dev 70, If 1, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 2: Dev 70, If 6, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 2: Dev 70, If 4, Class=Communications, Driver=cdc_acm, 12M
        |__ Port 3: Dev 76, If 0, Class=Hub, Driver=hub/7p, 480M
            |__ Port 1: Dev 77, If 0, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 1: Dev 77, If 1, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 1: Dev 77, If 2, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 1: Dev 77, If 3, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 1: Dev 77, If 4, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 1: Dev 77, If 5, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 1: Dev 77, If 6, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 1: Dev 77, If 7, Class=CDC Data, Driver=cdc_acm, 12M
        |__ Port 4: Dev 67, If 0, Class=Hub, Driver=hub/7p, 480M
            |__ Port 4: Dev 75, If 6, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 4: Dev 75, If 4, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 4: Dev 75, If 2, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 4: Dev 75, If 0, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 4: Dev 75, If 7, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 4: Dev 75, If 5, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 4: Dev 75, If 3, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 4: Dev 75, If 1, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 2: Dev 71, If 5, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 2: Dev 71, If 3, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 2: Dev 71, If 1, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 2: Dev 71, If 6, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 2: Dev 71, If 4, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 2: Dev 71, If 2, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 2: Dev 71, If 0, Class=Communications, Driver=cdc_acm, 12M
            |__ Port 2: Dev 71, If 7, Class=CDC Data, Driver=cdc_acm, 12M
            |__ Port 3: Dev 73, If 0, Class=Communications, Driver=cdc_acm, 12M
[... 609 lines cut ...]
[   75.190618] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 10.
[   75.410304] xhci_hcd 0000:01:00.0: Setup ERROR: setup address command for slot 10.
[   75.630115] usb 1-1.4.4: device not accepting address 57, error -22
[   75.669309] usb 1-1.4-port4: attempt power cycle
[   76.399945] usb 1-1.3.4: new full-speed USB device number 58 using xhci_hcd
[... cut ...]
[   79.587112] rpivid-mem feb20000.h264-decoder: rpivid-h264mem initialised: Registers at 0xfeb20000 length 0x0000ffff
[   79.587599] rpivid-mem feb30000.vp9-decoder: rpivid-vp9mem initialised: Registers at 0xfeb30000 length 0x0000ffff
[   79.602099] vc_sm_cma: module is from the staging directory, the quality is unknown, you have been warned.
[   79.604305] bcm2835_vc_sm_cma_probe: Videocore shared memory driver
[   79.604322] [vc_sm_connected_init]: start
[   79.605240] [vc_sm_connected_init]: installed successfully```

P33M, 2019-10-29:
Interesting. The xhci controller is rejecting an ADDRESS_DEVICE command that assigns an Enabled device slot to a particular Device. The controller is erroring out with "device slot not enabled".

This is weird because the xhci specification requires the host software to retrieve a slot ID from the controller and enable it prior to assigning a device address to it. Linux does this, and we can infer that the first command succeeded (no error message for "enable context command"). So why is the second command failing?

P33M, 2019-10-29:
What CDC devices are you using?

JamesH65, 2020-02-13:
Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

closed 2020-02-13 by JamesH65

JesseMCFC, 2020-04-19:
Actually having the same issue. Anyone that was able to find an answer?

JesseMCFC, 2020-04-19:
> Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

Please as I am having the same issue. Anyone that found a solution?

JesseMCFC, 2020-04-19:
> Closing due to lack of activity. Please request to be reopened if you feel this issue is still relevant.

Please. 

I am having about the same issue. Getting the following error with ‘dmesg’:

`[  808.960110] hub 1-1.4.7.7:1.0: USB hub found
[  808.960214] hub 1-1.4.7.7:1.0: 7 ports detected
[  809.270141] usb 1-1.4.7.7.1: new full-speed USB device number 74 using xhci_hcd
[  809.395200] usb 1-1.4.7.7.1: New USB device found, idVendor=04e2, idProduct=1414, bcdDevice= 0.03
[  809.395214] usb 1-1.4.7.7.1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  809.506135] usb 1-1.4.7.7.2: new full-speed USB device number 75 using xhci_hcd
[  809.631231] usb 1-1.4.7.7.2: New USB device found, idVendor=04e2, idProduct=1414, bcdDevice= 0.03
[  809.631246] usb 1-1.4.7.7.2: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[  809.633005] usb 1-1.4.7.7.2: Not enough bandwidth for new device state.
[  809.633872] usb 1-1.4.7.7.2: can't set config #1, error -28 `

And just like the original poster said, 36 are loaded fine but the other 28 are not working. Seems like the same issue.

andriisemenov, 2021-01-29:
Facing the same issue. Any update?

laf0rge, 2022-01-07:
I would also like to request re-opening this issue.  XHCI-only controllersa are a nightmare for any but the most trivial USB use cases.  They hide a lot of the important bits in proprietary firmware, and that firmware tends to be more buggy than the corresponding Linux kernel code for EHCI/OHCI/UICH (e.g. in terms of isochronous bandwidth allocation).

So any way how we can disable XHCI (and loose superspeed) while maintaining the much more flexible and interoperable EHCI linux kernel driver feature set would be a big improvement.

P33M, 2022-01-07:
Your question doesn't make any sense. There are no EHCI interfaces on a Pi 4.

laf0rge, 2022-01-07:
> Your question doesn't make any sense. There are no EHCI interfaces on a Pi 4.

Ok, then the logical choice is to buy devices with actual EHCI controllers in them.  At least there one is not subject to arbitrary bugs (which one cannot fix) like broken isochronous bandwidth computation, arbitrary limits on number of endpoints, devices, ...

P33M, 2022-01-07:
I am willing to entertain arguments as to why EHCI (a hardware specification mandating a particular software interface) is better than XHCI (a hardware specification mandating another particular software interface) for the same underlying hardware implementation.

laf0rge, 2022-02-13:
> I am willing to entertain arguments as to why EHCI (a hardware specification mandating a particular software interface) is better than XHCI (a hardware specification mandating another particular software interface) for the same underlying hardware implementation.

Since you're asking: The problem is the level of interface combined with non-open-source bits:  OHCI/UHCI/EHCI are closer to the hardware, while XHCI is at a higher level of abstraction, delegating more parts to the controller hard-wired logic or firmware.

There are bugs in XHCI controllers that we cannot fix due to those parts being in hardware/firmware (for example, miscalculating the available isochronous bandwidth).  In OHCI/UHCI/EHCI those bugs don't exist, as those parts are done inside the kernel.  And even if such bugs existed, they could be fixed by anyone by altering the Linux kernel source code.

sanitariu, 2023-12-20:
I have similar problems and will change my PI4 for PI3 :)