raspberrypi/linux #3310: Pi Zero fails to enumerate Realtek 8152B ethernet adapters

FD-, 2019-10-29:
It seems the Raspberry Pi Zero has problems with USB ethernet adapters that use the Realtek 8152B chip. More specifically, the adapter cannot be enumerated in a considerable fraction of boots, if the adapter is connected before the Pi boots. When enumeration fails, the kernel boot log show:

````
[    2.405793] usb 1-1: new high-speed USB device number 2 using dwc_otg
[    2.693484] usb 1-1: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.00
[    2.710902] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    2.727159] usb 1-1: Product: USB2.0 HUB
[    2.745895] hub 1-1:1.0: USB hub found
[    2.760023] hub 1-1:1.0: 4 ports detected
[    3.193396] usb 1-1.2: new high-speed USB device number 3 using dwc_otg
[    8.443463] usb 1-1.2: device descriptor read/64, error -110
[    8.763439] usb 1-1.2: device descriptor read/64, error -71
[    9.593715] usb 1-1.2: new high-speed USB device number 4 using dwc_otg
[    9.873452] usb 1-1.2: device descriptor read/64, error -71
[   10.203686] usb 1-1-port2: attempt power cycle
[   11.043456] usb 1-1.2: new full-speed USB device number 5 using dwc_otg
[   11.603445] usb 1-1.2: device not accepting address 5, error -32
[   11.903474] usb 1-1.2: new full-speed USB device number 6 using dwc_otg
[   12.343458] usb 1-1.2: device not accepting address 6, error -32
[   12.343724] usb 1-1-port2: unable to enumerate USB device
````

<br>
<details><summary>Full dmesg (click to expand)</summary>
<p>

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.75+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1270 Tue Sep 24 18:38:54 BST 2019
[    0.000000] CPU: ARMv6-compatible processor [410fb767] revision 7 (ARMv7), cr=00c5387d
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT nonaliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi Zero Rev 1.3
[    0.000000] Memory policy: Data cache writeback
[    0.000000] cma: Reserved 8 MiB at 0x1b400000
[    0.000000] On node 0 totalpages: 114688
[    0.000000]   Normal zone: 1008 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 114688 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x8c/0x49c with crng_init=0
[    0.000000] pcpu-alloc: s0 r0 d32768 u32768 alloc=1*32768
[    0.000000] pcpu-alloc: [0] 0 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 113680
[    0.000000] Kernel command line: coherent_pool=1M bcm2708_fb.fbwidth=1824 bcm2708_fb.fbheight=984 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:86:98:31 vc_mem.mem_base=0x1ec00000 vc_mem.mem_size=0x20000000  console=ttyAMA0,115200 console=tty1 root=PARTUUID=6c586e13-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 434424K/458752K available (6973K kernel code, 635K rwdata, 2080K rodata, 464K init, 797K bss, 16136K reserved, 8192K cma-reserved)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xdc800000 - 0xff800000   ( 560 MB)
                   lowmem  : 0xc0000000 - 0xdc000000   ( 448 MB)
                   modules : 0xbf000000 - 0xc0000000   (  16 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (6975 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   ( 464 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 636 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 798 kB)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=1, Nodes=1
[    0.000000] ftrace: allocating 25197 entries in 74 pages
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000032] sched_clock: 32 bits at 1000kHz, resolution 1000ns, wraps every 2147483647500ns
[    0.000094] clocksource: timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 1911260446275 ns
[    0.000220] bcm2835: system timer (irq = 27)
[    0.000637] Console: colour dummy device 80x30
[    0.001269] console [tty1] enabled
[    0.001350] Calibrating delay loop... 697.95 BogoMIPS (lpj=3489792)
[    0.060398] pid_max: default: 32768 minimum: 301
[    0.060968] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.061027] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.062466] CPU: Testing write buffer coherency: ok
[    0.063861] Setting up static identity map for 0x8200 - 0x8238
[... 859 lines cut ...]
**Used kernel**: Linux raspberrypi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux (also tried with older kernels)
**Tested Pis**: Pi Zero, Pi Zero W
**Power supplies**: Tested 2 different supplies + powering hubs (if they had a barrel connector socket)
**Connection**: RTL8152B connected to Pi via USB OTG adapter (tested 3 different ones) and optional hub (tested 3 different chips + no hub).

### What I have tried:

I tried a bunch of different USB-related kernel settings and USB quirks, all to no avail. Please note I could only apply the quirks to the hub that sat between Pi and RTL8152B, since obviously the kernel can only apply quirks after it knows what device it’s dealing with, ie enumeration succeeded.

* Use the dwc2 driver via a overlay in /boot/config.txt
* dwc_otg.speed=1
* usbcore.old_scheme_first=1
* usbcore.initial_descriptor_timeout=10000
* USB_QUIRK_NO_LPM on hub between Pi and RTL8152B (via usbcore.quirks)
* USB_QUIRK_RESET_RESUME on hub (via usbcore.quirks)
* USB_QUIRK_RESET on hub (via usbcore.quirks)
* USB_QUIRK_DELAY_INIT on hub (via usbcore.quirks)
* USB_QUIRK_LINEAR_FRAME_INTR_BINTERVAL on hub (via usbcore.quirks)
* USB_QUIRK_LINEAR_UFRAME_INTR_BINTERVAL on hub (via usbcore.quirks)
* USB_QUIRK_HUB_SLOW_RESET on hub (via usbcore.quirks)
* I didn't try any dwc_otg.fiq_* flags, since even the dwc2 module didn't work, and that one doesn't include any of the fiq code IIRC.

**Please let me know if there are any more kernel flags or experiments you want me to try!**

If you're willing to investigate this issue, I can send you one of these failing adapters. Just tell me what address to send it to.

FD-, 2019-10-30:
An additional note: I have one RTL8152B adapter (included in a hub, design differs from any of the others) that does work reliably. I already visually compared the PCBs, but couldn't find any difference that sticked out. Anyway, the majority of RTL8152B adapter designs shows the reported flaw with the Pi Zero, while they work fine with any other computer I tried.

melyux, 2020-08-28:
I have a RTL8152 and tried cold booting (overlay FS, so it's fine), but didn't have any enumeration failures yet. Have you gotten this error recently on the newest Raspbian?

narspt, 2022-03-13:
@FD- : did you ever find a solution/workaround for this?  I'm having same exact problem on a Pi Zero also.