raspberrypi/linux #3312: UART1 can not be made available on CM3/CM3+ running 4.19.66 built in master branch

stevep-korg, 2019-10-29:
UART1 can not be made available on CM3/CM3+ running 4.19.66 built in master branch, but it works fine in 4.14.112.

I noticed this in my own Yocto build, as well as in Raspbian Buster.  UART1 is not available when using the correct config.txt settings.  It works fine in 4.14.112 however.

UART1 should be easily available via a minor tweak to config.txt

1. Build linux-raspberrypi 4.14.112 in master branch with the proper settings for uart1 in config.txt, and making sure 8250 NR UARTS is 1.
2.  Install and boot the system
Result: uart1 is functional on ttyS0. Observed:
```
Linux Korg 4.14.112 #1 SMP PREEMPT Wed Sep 4 14:36:53 UTC 2019 aarch64 GNU/Linux
[    0.000000] Kernel command line: cma=256M 8250.nr_uarts=1 bcm2708_fb.fbwidth=720 bcm2708_fb.fbheight=480 bcm2708_fb.fbswap=1 smsc95xx.macaddr=B8:27:EB:55:00:EC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fsm_mask=0x0 dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.147861] bcm2835-aux-uart 3f215040.serial: irq not found - -517
[    0.171640] 3f215040.serial: ttyS0 at MMIO 0x0 (irq = 151, base_baud = 61440000) is a 16550
```

3. Build linux-raspberrypi 4.19.66 in master branch with the proper settings for uart1 in config.txt (same as for 4.14.112 above), and making sure 8250 NR UARTS is 1.
4. Install and boot the system
Result: uart1 fails to register. Observed:
```
Linux Korg 4.19.66 #1 SMP PREEMPT Fri Sep 6 04:46:37 UTC 2019 aarch64 GNU/Linux
[    0.000000] Kernel command line: coherent_pool=1M cma=256M 8250.nr_uarts=1 smsc95xx.macaddr=B8:27:EB:55:00:EC vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.fiq_fsm_mask=0x0 dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait
[    0.184922] bcm2835-aux-uart 3f215040.serial: unable to register 8250 port - -22
[    0.184970] bcm2835-aux-uart: probe of 3f215040.serial failed with error -22
```
I expected that uart1 would work in 4.19.66 as well as it did in 4.14.112 with the same proper settings and overlays as specified in config.txt and cmdline.txt

from config.txt
```
# CONFIGURE UART0 FOR USE
#  Tell Linux not to drive serial Console on GPIO pins 14 & 15
enable_uart=1
#  Disable Bluetooth (frees up UART0 and connects UART0 TXD0 and RXD0 pins to GPIO pins 14 & 15, respectively)
dtoverlay=pi3-disable-bt
# CONFIGURE UART1 FOR MIDI USE, connecting UART1 TXD1 and RXD1 pins to GPIO pins 32 & 33, respectively
dtoverlay=uart1,txd1_pin=32,rxd1_pin=33
# midi-uart1: fake out tty driver so that baudrate 38400 yields 31250 for UART1
dtoverlay=midi-uart1
```

cmdline.txt:
```
dwc_otg.fiq_fsm_mask=0x0 dwc_otg.lpm_enable=0  root=/dev/mmcblk0p2 rootfstype=ext4 rootwait 
```

stevep-korg, 2019-12-04:
Is anyone else noticing that the UART1 is not initializing correctly in Raspbian Buster?  My team really wishes to try 4.19.66 (we've been looking at 4.14.112), but we can't consider it if UART1 can't be made available.

pelwell, 2019-12-04:
The -22 error (-EINVAL) is significant, and appears to be caused by the use of the uart1-midi overlay. Try without it and confirm that the UART is functional, then close this issue and open a new one with a more accurate title.

stevep-st, 2020-02-02:
@pelwell Thank you for your input on this.  I am no longer with my previous organization, so I don't have a CM3 or CM3+ card/daughter card test bed on which to do more work.  That said, I will try do some work here to see if I can recreate this problem on a Raspberry Pi 3B, which I do have available to me.  I'll let you know how that goes.  If it exhibits the same problem, I'll close this issue and open a new one as you requested, with a different focus.

stevep-st, 2020-02-07:
I'm unable to research this issue any further at this time, because I just re-discovered that my RPi3 only has the ability to bring one of the UART TxRx pairs out onto the header, at GPIO pins 15 and 16.  The scenario for this issue involves trying to use **both** uarts with MIDI overlays.  I no longer have access to a CM3 or CM3+.

I'd be thankful if someone with a CM3 or CM3+ could try to use both UARTS with the MIDI overlays in Rasbian Buster and see if the -EINVAL error still occurs, then write up a new issue if the problem can be localized to loading the MIDI overlay(s), as @pelwell described above.

closed 2020-02-07 by stevep-st