raspberrypi/linux #3318: WARNING: CPU: 2 PID: 1922 at net/wireless/sme.c:752 __cfg80211_connect_result+0x3c0/0x410 [cfg80211]

HinTak, 2019-11-01:
This seems to need looking at, from dmesg:

```
Oct 30 13:38:17 pirecorder kernel: [1399728.779188] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
Oct 30 13:38:17 pirecorder kernel: [1399728.779448] ------------[ cut here ]------------
Oct 30 13:38:17 pirecorder kernel: [1399728.779724] WARNING: CPU: 2 PID: 1922 at net/wireless/sme.c:752 __cfg80211_connect_result+0x3c0/0x410 [cfg80211]
Oct 30 13:38:17 pirecorder kernel: [1399728.779731] Modules linked in: fuse binfmt_misc bnep hci_uart btbcm serdev bluetooth ecdh_generic spidev spi_bcm2835 i2c_dev 8021q garp stp llc evdev brcmfmac
 brcmutil sha256_generic cfg80211 rfkill raspberrypi_hwmon hwmon bcm2835_codec(C) bcm2835_v4l2(C) snd_bcm2835(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_common videobuf2_dma_contig videobuf2_vmalloc
 videobuf2_memops videobuf2_v4l2 videobuf2_common i2c_bcm2835 videodev snd_soc_bcm2835_i2s media vc_sm_cma(C) uio_pdrv_genirq uio fixed snd_soc_wm8960(O) snd_soc_ac108(O) snd_soc_seeed_voicecard(O) 
snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd ip_tables x_tables ipv6
Oct 30 13:38:17 pirecorder kernel: [1399728.780035] CPU: 2 PID: 1922 Comm: kworker/u8:0 Tainted: G        WC O      4.19.75-v7+ #1270
Oct 30 13:38:17 pirecorder kernel: [1399728.780040] Hardware name: BCM2835
Oct 30 13:38:17 pirecorder kernel: [1399728.780250] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
Oct 30 13:38:17 pirecorder kernel: [1399728.780286] [<80111fcc>] (unwind_backtrace) from [<8010d544>] (show_stack+0x20/0x24)
Oct 30 13:38:17 pirecorder kernel: [1399728.780304] [<8010d544>] (show_stack) from [<80819bc0>] (dump_stack+0xd4/0x118)
Oct 30 13:38:17 pirecorder kernel: [1399728.780323] [<80819bc0>] (dump_stack) from [<80120a88>] (__warn+0x104/0x11c)
Oct 30 13:38:17 pirecorder kernel: [1399728.780340] [<80120a88>] (__warn) from [<80120bd8>] (warn_slowpath_null+0x50/0x58)
Oct 30 13:38:17 pirecorder kernel: [1399728.780543] [<80120bd8>] (warn_slowpath_null) from [<7f3c174c>] (__cfg80211_connect_result+0x3c0/0x410 [cfg80211])
Oct 30 13:38:17 pirecorder kernel: [1399728.780935] [<7f3c174c>] (__cfg80211_connect_result [cfg80211]) from [<7f3913ec>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
Oct 30 13:38:17 pirecorder kernel: [1399728.781322] [<7f3913ec>] (cfg80211_process_wdev_events [cfg80211]) from [<7f391488>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
Oct 30 13:38:17 pirecorder kernel: [1399728.781706] [<7f391488>] (cfg80211_process_rdev_events [cfg80211]) from [<7f38b2d4>] (cfg80211_event_work+0x24/0x2c [cfg80211])
Oct 30 13:38:17 pirecorder kernel: [1399728.781908] [<7f38b2d4>] (cfg80211_event_work [cfg80211]) from [<8013bf64>] (process_one_work+0x170/0x458)
Oct 30 13:38:17 pirecorder kernel: [1399728.781925] [<8013bf64>] (process_one_work) from [<8013c2a8>] (worker_thread+0x5c/0x5a4)
Oct 30 13:38:17 pirecorder kernel: [1399728.781942] [<8013c2a8>] (worker_thread) from [<80142594>] (kthread+0x138/0x168)
Oct 30 13:38:17 pirecorder kernel: [1399728.781959] [<80142594>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
Oct 30 13:38:17 pirecorder kernel: [1399728.781967] Exception stack(0x98cadfb0 to 0x98cadff8)
Oct 30 13:38:17 pirecorder kernel: [1399728.781977] dfa0:                                     00000000 00000000 00000000 00000000
Oct 30 13:38:17 pirecorder kernel: [1399728.781988] dfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Oct 30 13:38:17 pirecorder kernel: [1399728.781999] dfe0: 00000000 00000000 00000000 00000000 00000013 00000000
Oct 30 13:38:17 pirecorder kernel: [1399728.782007] ---[ end trace d8e9be4c61d2c574 ]---
```

pelwell, 2019-11-01:
Please give us the usual information:
1. What model of Pi are you using?
2. Which OS/distribution?
3. What was the Pi doing when the problem occurred?
4. How frequently do you see the problem?
5. Can you reproduce it at will, and if so, how?

HinTak, 2019-11-01:
I have got a pile of these, seem to start with`brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)` .

1. OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
2. Raspbian GNU/Linux 10 (buster)
3. Every one of them seems to be led by `brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)`, according to `grep -A 1 bss kern.log*`

```
kern.log:Oct 30 11:45:58 pirecorder kernel: [1392990.038059] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
kern.log-Oct 30 11:45:58 pirecorder kernel: [1392990.038315] ------------[ cut here ]------------
--
kern.log:Oct 30 12:13:51 pirecorder kernel: [1394662.976791] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
kern.log-Oct 30 12:13:51 pirecorder kernel: [1394662.976899] ------------[ cut here ]------------
--
kern.log:Oct 30 12:47:55 pirecorder kernel: [1396707.366019] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
kern.log-Oct 30 12:47:55 pirecorder kernel: [1396707.366277] ------------[ cut here ]------------
--
kern.log:Oct 30 12:58:30 pirecorder kernel: [1397341.528189] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
kern.log-Oct 30 12:58:30 pirecorder kernel: [1397341.528296] ------------[ cut here ]------------
--
kern.log:Oct 30 13:11:52 pirecorder kernel: [1398143.800567] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
kern.log-Oct 30 13:11:52 pirecorder kernel: [1398143.800671] ------------[ cut here ]------------
--
kern.log:Oct 30 13:38:17 pirecorder kernel: [1399728.779188] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
kern.log-Oct 30 13:38:17 pirecorder kernel: [1399728.779448] ------------[ cut here ]------------
--
kern.log:Oct 30 13:51:24 pirecorder kernel: [1400516.057684] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
kern.log-Oct 30 13:51:24 pirecorder kernel: [1400516.057940] ------------[ cut here ]------------
--
kern.log:Oct 30 16:24:08 pirecorder kernel: [1409680.342042] brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)
kern.log-Oct 30 16:24:08 pirecorder kernel: [1409680.342552] ------------[ cut here ]------------
```

4. As above. The Pi was up for about 2 weeks continuously, that happened towards the end of the 2 week period. (I needed to reboot for an unrelated reason).

5. No; saw it intermittently; I'll write again if I see them again.

lategoodbye, 2019-11-03:
Please provide more information about your Wifi configuration (AP, STA, Adhoc, ...).

HinTak, 2019-11-06:
When the message happened, the device was running in STA mode, via something like this:

```
wpa_passphrase AP_Name password > /tmp/wpa.conf
sudo wpa_supplicant -B  -c /tmp/wpa.conf -i wlan0
sudo dhclient wlan0
```

scyto, 2019-11-11:
I am also getting regular "0x3c0/0x410 [cfg80211]" erros after my pi4 has been running for a while - i.e. i am not doing anything when it stops working, it is in dead of night.  messages.log shows multiple, usually causing the pi to dissapear - my APs think that the pi is continuing to attempts to connect (i can see the attempts in a event UI)

Pi4
using WLAN0
rasbian latest from site.
i can't make it happen - just does happen

```
Nov 10 03:03:54 genmon kernel: [125997.788087] ------------[ cut here ]------------
Nov 10 03:03:54 genmon kernel: [125997.788183] WARNING: CPU: 0 PID: 6804 at net/wireless/sme.c:752 __cfg80211_connect_result+0x3c0/0x410 [cfg80211]
Nov 10 03:03:54 genmon kernel: [125997.788187] Modules linked in: 8021q garp stp llc rtc_ds1307 brcmfmac brcmutil sha256_generic cfg80211 raspberrypi_hwmon hwmon rfkill snd_bcm2835(C) snd_pcm snd_timer snd bcm2835_codec(C) v4l2_mem2mem bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) i2c_bcm2835 videobuf2_dma_contig v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) rpivid_mem uio_pdrv_genirq fixed uio i2c_dev ip_tables x_tables ipv6
Nov 10 03:03:54 genmon kernel: [125997.788333] CPU: 0 PID: 6804 Comm: kworker/u8:2 Tainted: G        WC        4.19.75-v7l+ #1270
Nov 10 03:03:54 genmon kernel: [125997.788336] Hardware name: BCM2835
Nov 10 03:03:54 genmon kernel: [125997.788389] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
Nov 10 03:03:54 genmon kernel: [125997.788410] [<c0212d10>] (unwind_backtrace) from [<c020d530>] (show_stack+0x20/0x24)
Nov 10 03:03:54 genmon kernel: [125997.788418] [<c020d530>] (show_stack) from [<c097fb20>] (dump_stack+0xd4/0x118)
Nov 10 03:03:54 genmon kernel: [125997.788426] [<c097fb20>] (dump_stack) from [<c0222330>] (__warn+0x104/0x11c)
Nov 10 03:03:54 genmon kernel: [125997.788434] [<c0222330>] (__warn) from [<c0222480>] (warn_slowpath_null+0x50/0x58)
Nov 10 03:03:54 genmon kernel: [125997.788489] [<c0222480>] (warn_slowpath_null) from [<bf41f6fc>] (__cfg80211_connect_result+0x3c0/0x410 [cfg80211])
Nov 10 03:03:54 genmon kernel: [125997.788593] [<bf41f6fc>] (__cfg80211_connect_result [cfg80211]) from [<bf3ef39c>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
Nov 10 03:03:54 genmon kernel: [125997.788694] [<bf3ef39c>] (cfg80211_process_wdev_events [cfg80211]) from [<bf3ef438>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
Nov 10 03:03:54 genmon kernel: [125997.788792] [<bf3ef438>] (cfg80211_process_rdev_events [cfg80211]) from [<bf3e92d4>] (cfg80211_event_work+0x24/0x2c [cfg80211])
Nov 10 03:03:54 genmon kernel: [125997.788846] [<bf3e92d4>] (cfg80211_event_work [cfg80211]) from [<c023db40>] (process_one_work+0x170/0x458)
Nov 10 03:03:54 genmon kernel: [125997.788854] [<c023db40>] (process_one_work) from [<c023de84>] (worker_thread+0x5c/0x5a4)
Nov 10 03:03:54 genmon kernel: [125997.788860] [<c023de84>] (worker_thread) from [<c0244170>] (kthread+0x138/0x168)
Nov 10 03:03:54 genmon kernel: [125997.788866] [<c0244170>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
Nov 10 03:03:54 genmon kernel: [125997.788870] Exception stack(0xeb15bfb0 to 0xeb15bff8)
Nov 10 03:03:54 genmon kernel: [125997.788874] bfa0:                                     00000000 00000000 00000000 00000000
Nov 10 03:03:54 genmon kernel: [125997.788879] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Nov 10 03:03:54 genmon kernel: [125997.788883] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000
Nov 10 03:03:54 genmon kernel: [125997.788893] ---[ end trace 77c2645e50b9a632 ]---
```

scyto, 2019-11-11:
FYI all my AP equipment is ubiquiti unifi APs of various types fwiw

scyto, 2019-11-12:
I have tried different APs and i tried turning off unscheduled power delivery and the feature that encourages devices to connect on 5ghz.

Is there a way to force the pi4 to only connect on 2.4ghz?

doganmurat, 2019-11-13:
Same result here.
Tried with kernel 4.14 & 4.19.

scyto, 2019-11-13:
I have moved my pie onto a separate 2.4ghz SSID operating in 802.11b legacy mode.
Still happening.

I only have two settings left to test:

1. disable GTK re-keying
2. disable CCMP (aka switch back to TKIP 

I have a 100ft outdoor rate cable coming this week so i can attach in situ and trying doing the trace on the wifi.

scyto, 2019-11-13:
@doganmurat what AP WiFi equipment are you using?

doganmurat, 2019-11-14:
I am using a custom built Linux.
Kernel version 4.4.112.

I don’t think it is AP related. It is same as for different APs.
In my case message appears when connecting an AP and when I unplugged device.

pelwell, 2019-11-14:
Interestingly, I've just seen this on 5.4-rc7 on a Pi 4, but so far only once. There were no other obvious symptoms, just an unexpected warning.

scyto, 2019-11-14:
Thanks for info, for me when this happens all traffic stops flowing to the pi - whats interesting is it is still associating, i can see it move between APs.  I have a long outdoor rated Ethernet cable coming i will use to attach to the device and see if i can get a trace of the issue.

scyto, 2019-11-19:
ok, ethernet cable attached and i set ```root@genmon:~# echo 1 >/sys/kernel/debug/tracing/events/cfg80211/enable``
 anything else i need to do?

scyto, 2019-11-19:
Ok, seems the last few failures have always had this in the messages (could my unit just be plain old faulty?)
```
Nov 19 13:44:00 genmon kernel: [  151.181781] ------------[ cut here ]------------
Nov 19 13:44:00 genmon kernel: [  151.181812] WARNING: CPU: 3 PID: 646 at drivers/firmware/raspberrypi.c:211 get_throttled_show+0x68/0x6c
Nov 19 13:44:00 genmon kernel: [  151.181822] deprecated, use hwmon sysfs instead
Nov 19 13:44:00 genmon kernel: [  151.181831] Modules linked in: 8021q garp stp llc rtc_ds1307 brcmfmac brcmutil sha256_generic cfg80211 rfkill snd_bcm2835(C) snd_pcm snd_timer snd raspberrypi_hwmon bcm2835_codec(C) hwmon v4l2_mem2mem bcm2835_v4l2(C) videobuf2_dma_contig bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common i2c_bcm2835 videodev media vc_sm_cma(C) rpivid_mem uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6
Nov 19 13:44:00 genmon kernel: [  151.182188] CPU: 3 PID: 646 Comm: python Tainted: G         C        4.19.75-v7l+ #1270
Nov 19 13:44:00 genmon kernel: [  151.182195] Hardware name: BCM2835
Nov 19 13:44:00 genmon kernel: [  151.182222] [<c0212d10>] (unwind_backtrace) from [<c020d530>] (show_stack+0x20/0x24)
Nov 19 13:44:00 genmon kernel: [  151.182241] [<c020d530>] (show_stack) from [<c097fb20>] (dump_stack+0xd4/0x118)
Nov 19 13:44:00 genmon kernel: [  151.182260] [<c097fb20>] (dump_stack) from [<c0222330>] (__warn+0x104/0x11c)
Nov 19 13:44:00 genmon kernel: [  151.182277] [<c0222330>] (__warn) from [<c02223a0>] (warn_slowpath_fmt+0x58/0x74)
Nov 19 13:44:00 genmon kernel: [  151.182295] [<c02223a0>] (warn_slowpath_fmt) from [<c0810f14>] (get_throttled_show+0x68/0x6c)
Nov 19 13:44:00 genmon kernel: [  151.182315] [<c0810f14>] (get_throttled_show) from [<c06d8374>] (dev_attr_show+0x2c/0x5c)
Nov 19 13:44:00 genmon kernel: [  151.182336] [<c06d8374>] (dev_attr_show) from [<c0437658>] (sysfs_kf_seq_show+0x9c/0x108)
Nov 19 13:44:00 genmon kernel: [  151.182354] [<c0437658>] (sysfs_kf_seq_show) from [<c0435e68>] (kernfs_seq_show+0x34/0x38)
Nov 19 13:44:00 genmon kernel: [  151.182371] [<c0435e68>] (kernfs_seq_show) from [<c03d86dc>] (seq_read+0x1d0/0x4bc)
Nov 19 13:44:00 genmon kernel: [  151.182386] [<c03d86dc>] (seq_read) from [<c0436c84>] (kernfs_fop_read+0x134/0x1a4)
Nov 19 13:44:00 genmon kernel: [  151.182401] [<c0436c84>] (kernfs_fop_read) from [<c03afd54>] (__vfs_read+0x48/0x16c)
Nov 19 13:44:00 genmon kernel: [  151.182417] [<c03afd54>] (__vfs_read) from [<c03aff14>] (vfs_read+0x9c/0x168)
Nov 19 13:44:00 genmon kernel: [  151.182432] [<c03aff14>] (vfs_read) from [<c03b055c>] (ksys_read+0x74/0xe8)
Nov 19 13:44:00 genmon kernel: [  151.182446] [<c03b055c>] (ksys_read) from [<c03b05e8>] (sys_read+0x18/0x1c)
Nov 19 13:44:00 genmon kernel: [  151.182461] [<c03b05e8>] (sys_read) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
Nov 19 13:44:00 genmon kernel: [  151.182470] Exception stack(0xecffffa8 to 0xecfffff0)
Nov 19 13:44:00 genmon kernel: [  151.182483] ffa0:                   00001000 b3a08454 00000010 b3a08454 00001000 00000000
Nov 19 13:44:00 genmon kernel: [  151.182495] ffc0: 00001000 b3a08454 00000010 00000003 00001001 b6e426d8 00001001 000005e8
Nov 19 13:44:00 genmon kernel: [  151.182505] ffe0: 00000002 afbfa188 00000000 b6dbf8bc
Nov 19 13:44:00 genmon kernel: [  151.182515] ---[ end trace efec930a47963852 ]---
```

scyto, 2019-11-20:
issue did not resolve with pi zero
i am thinking SD card or powersupply possibly.....

lategoodbye, 2019-11-20:
The last warning is unrelated and advises you to use the new hwmon interface for under voltage detection.

scyto, 2019-11-20:
@lategoodbye thanks Stefan, thats good to know. I think this means my unit is silently dying in some way. It cannot service any IPv4 traffic but it does maintain wifi connection (which is very strange) and its DHCP address.  I guess i need to open a different issue for this.... ? i am stumped.

HinTak, 2019-12-02:
I just have one of these again, and this was more preceded by `brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)` . Also, since I last posted, there are a few Buster updates, so the latest message is from up-to-date Buster, just before I reboot. 

```
[  180.702832] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[  180.702848] brcmfmac: power management disabled
[  360.631937] rpi_firmware_get_throttled: 7 callbacks suppressed
[  360.631944] Under-voltage detected! (0x00050005)
[  364.791971] rpi_firmware_get_throttled: 7 callbacks suppressed
[  364.791979] Voltage normalised (0x00000000)
[  365.559593] ------------[ cut here ]------------
[  365.559865] WARNING: CPU: 2 PID: 44 at net/wireless/sme.c:752 __cfg80211_connect_result+0x3c0/0x410 [cfg80211]
[  365.559873] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic spidev spi_bcm2835 i2c_dev 8021q garp stp llc evdev brcmfmac brcmutil sha256_generic cfg80211 rfkill raspberrypi_hwmon hwmon bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem snd_bcm2835(C) bcm2835_mmal_vchiq(C) v4l2_common i2c_bcm2835 videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev snd_soc_bcm2835_i2s media vc_sm_cma(C) fixed uio_pdrv_genirq uio snd_soc_wm8960(O) snd_soc_ac108(O) snd_soc_seeed_voicecard(O) snd_soc_simple_card_utils snd_soc_core snd_compress snd_pcm_dmaengine snd_pcm snd_timer snd ip_tables x_tables ipv6
[  365.560164] CPU: 2 PID: 44 Comm: kworker/u8:1 Tainted: G         C O      4.19.75-v7+ #1270
[  365.560169] Hardware name: BCM2835
[  365.560373] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  365.560408] [<80111fcc>] (unwind_backtrace) from [<8010d544>] (show_stack+0x20/0x24)
[  365.560426] [<8010d544>] (show_stack) from [<80819bc0>] (dump_stack+0xd4/0x118)
[  365.560445] [<80819bc0>] (dump_stack) from [<80120a88>] (__warn+0x104/0x11c)
[  365.560461] [<80120a88>] (__warn) from [<80120bd8>] (warn_slowpath_null+0x50/0x58)
[  365.560667] [<80120bd8>] (warn_slowpath_null) from [<7f45874c>] (__cfg80211_connect_result+0x3c0/0x410 [cfg80211])
[  365.561079] [<7f45874c>] (__cfg80211_connect_result [cfg80211]) from [<7f4283ec>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
[  365.561464] [<7f4283ec>] (cfg80211_process_wdev_events [cfg80211]) from [<7f428488>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
[  365.561848] [<7f428488>] (cfg80211_process_rdev_events [cfg80211]) from [<7f4222d4>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[  365.562051] [<7f4222d4>] (cfg80211_event_work [cfg80211]) from [<8013bf64>] (process_one_work+0x170/0x458)
[  365.562068] [<8013bf64>] (process_one_work) from [<8013c2a8>] (worker_thread+0x5c/0x5a4)
[  365.562086] [<8013c2a8>] (worker_thread) from [<80142594>] (kthread+0x138/0x168)
[  365.562103] [<80142594>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[  365.562111] Exception stack(0xb9fb5fb0 to 0xb9fb5ff8)
[  365.562121] 5fa0:                                     00000000 00000000 00000000 00000000
[  365.562133] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  365.562143] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  365.562229] ---[ end trace 6d3c5e400672f63a ]---
[  365.563302] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```

As you can see, I see to have 3 minutes where wlan0 was struggling to get DHCP, and right after the oops, it seems to be successful! Strange.

scyto, 2019-12-03:
@HinTak what DHCP server are you using?

HinTak, 2019-12-03:
> @HinTak what DHCP server are you using?

It is a small commercial firewall box, running some form of embedded linux, I think.

HinTak, 2019-12-17:
Just noticed that I have a bunch of them again. Strangely enough, they appear every 1/2 hour or so in the last 24 hours, but not for about two weeks.

```
$ zcat messages.2.gz  | grep -a cut
Dec  2 09:27:49 pirecorder kernel: [  365.559593] ------------[ cut here ]------------
Dec  2 16:42:54 pirecorder kernel: [23340.981203] ------------[ cut here ]------------
Dec  2 19:04:32 pirecorder kernel: [31839.147011] ------------[ cut here ]------------
Dec  3 07:55:08 pirecorder kernel: [78075.687821] ------------[ cut here ]------------
$ cat messages  | grep -a cut
Dec 15 10:10:44 pirecorder kernel: [1123020.412896] ------------[ cut here ]------------
Dec 15 10:24:27 pirecorder kernel: [1123843.336597] ------------[ cut here ]------------
Dec 15 11:00:44 pirecorder kernel: [1126020.627473] ------------[ cut here ]------------
Dec 15 11:19:59 pirecorder kernel: [1127175.903059] ------------[ cut here ]------------
Dec 15 12:20:14 pirecorder kernel: [1130790.202034] ------------[ cut here ]------------
Dec 15 13:41:01 pirecorder kernel: [1135637.509789] ------------[ cut here ]------------
Dec 15 14:30:31 pirecorder kernel: [1138607.796885] ------------[ cut here ]------------
Dec 15 15:30:32 pirecorder kernel: [1142208.754651] ------------[ cut here ]------------
Dec 15 15:55:21 pirecorder kernel: [1143698.041457] ------------[ cut here ]------------
Dec 15 16:17:55 pirecorder kernel: [1145051.316137] ------------[ cut here ]------------
Dec 15 16:38:04 pirecorder kernel: [1146260.593193] ------------[ cut here ]------------
Dec 15 16:52:22 pirecorder kernel: [1147118.864111] ------------[ cut here ]------------
Dec 15 18:08:15 pirecorder kernel: [1151672.169536] ------------[ cut here ]------------
Dec 15 19:30:35 pirecorder kernel: [1156611.490457] ------------[ cut here ]------------
Dec 15 20:03:58 pirecorder kernel: [1158614.766242] ------------[ cut here ]------------
Dec 15 20:30:14 pirecorder kernel: [1160191.047035] ------------[ cut here ]------------
Dec 15 22:55:23 pirecorder kernel: [1168899.402231] ------------[ cut here ]------------
Dec 15 23:04:53 pirecorder kernel: [1169469.664175] ------------[ cut here ]------------
Dec 15 23:20:14 pirecorder kernel: [1170390.588088] ------------[ cut here ]------------
Dec 16 00:30:08 pirecorder kernel: [1174584.898085] ------------[ cut here ]------------
Dec 16 05:30:12 pirecorder kernel: [1192589.341378] ------------[ cut here ]------------
Dec 16 08:29:11 pirecorder kernel: [1203327.719007] ------------[ cut here ]------------
```

```
$ uptime
 16:11:10 up 15 days,  5:57,  1 user,  load average: 0.16, 0.07, 0.01
```

The machine has been up since Dec 2.

normanr, 2020-06-19:
btw, #3671 looks similar (I see both __cfg80211_connect_result warnings and error -52's)

cyberplant, 2020-06-27:
The same is happening to me. RPi4, connects to wifi but is not getting DHCP. I have another one and it works fine. I even swapped SD cards and still failing on this one.
The funny thing is.. if I configure IP and route manually, everything works perfectly! 
But something is not fine as I have these errors on the kernel.. 

FTR: This RPi4 was working fine, I installed ubuntu 64 bits and then my problems started.. ubuntu wasn't able to use wifi, I thought it was something on ubuntu so I came back to raspbian 32 bits, and now I'm having these issues. Maybe ubuntu changed something on the firmware and/or eeprom?

xyklex, 2020-07-20:
Hi, just adding a bit more context to this thread in case it helps, I build a RPi4 image using Yocto. 

Using only the onboard WiFi interface it boots and get IP addresses without the kernel error
```
$ uname -a
Linux raspberrypi4 4.19.126-v7l #1 SMP Wed Jun 3 14:46:19 UTC 2020 armv7l GNU/Linux

$ dhclient --version
isc-dhclient-4.4.2

$ wpa_supplicant -v
wpa_supplicant v2.9
Copyright (c) 2003-2019, Jouni Malinen <j@w1.fi> and contributors
```

But I also added to the image an external WiFi driver for a D-Link adapter (DWA-181). Device is recognized correctly without errors, my next step was using `wpa_supplicant` to authenticate against the network using `wpa_cli` and is were the error appeared
```
root@raspberrypi4:/etc# wpa_cli -i wlan1
wpa_cli v2.9
Copyright (c) 2004-2019, Jouni Malinen <j@w1.fi> and contributors

This software may be distributed under the terms of the BSD license.
See README for more details.

Interactive mode

<3>CTRL-EVENT-SCAN-RESULTS
<3>Trying to associate with 18:4b:0d:26:f6:18 (SSID='TestNetwork' freq=2437 MHz)
[  995.036504] ------------[ cut here ]------------
[  995.041409] WARNING: CPU: 0 PID: 175 at net/wireless/sme.c:756 __cfg80211_connect_result+0x3b0/0x414 [cfg80211]
[  995.051707] Modules linked in: xt_conntrack ipt_MASQUERADE nf_conntrack_netlink nfnetlink xt_addrtype iptable_filter iptable_nat nf_nat_ipv4 nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_tables x_tables br_netfilter bridge stp llc o
verlay ipv6 88x2bu(O) brcmfmac brcmutil vc4 bcm2835_codec(C) snd_soc_core cfg80211 snd_compress v3d bcm2835_v4l2(C) snd_pcm_dmaengine v4l2_mem2mem gpu_sched snd_pcm bcm2835_mmal_vchiq(C) rfkill videobuf2_dma_contig v4l2_common videobuf2_vmalloc snd_timer videobuf2_memops videobuf2_v4l2 raspberrypi_hwmon snd hwmon videobuf2_common videodev vc_sm_cma(C) media rpivid_mem uio_pdrv_genirq uio
[  995.107836] CPU: 0 PID: 175 Comm: kworker/u8:2 Tainted: G        WC O      4.19.126-v7l #1
[  995.116224] Hardware name: BCM2835
[  995.119814] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[  995.125761] [<c0212fc8>] (unwind_backtrace) from [<c020d4b8>] (show_stack+0x20/0x24)
[  995.133631] [<c020d4b8>] (show_stack) from [<c0a1197c>] (dump_stack+0xd4/0x118)
[  995.141060] [<c0a1197c>] (dump_stack) from [<c0222850>] (__warn.part.0+0xcc/0xe8)
[  995.148663] [<c0222850>] (__warn.part.0) from [<c0222a04>] (warn_slowpath_null+0x54/0x5c)
[  995.157104] [<c0222a04>] (warn_slowpath_null) from [<bf2938f4>] (__cfg80211_connect_result+0x3b0/0x414 [cfg80211])
[  995.167874] [<bf2938f4>] (__cfg80211_connect_result [cfg80211]) from [<bf26353c>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
[  995.180482] [<bf26353c>] (cfg80211_process_wdev_events [cfg80211]) from [<bf2635e0>] (cfg80211_process_rdev_events+0x48/0xa0 [cfg80211])
[  995.193174] [<bf2635e0>] (cfg80211_process_rdev_events [cfg80211]) from [<bf25d2e0>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[  995.204952] [<bf25d2e0>] (cfg80211_event_work [cfg80211]) from [<c023e058>] (process_one_work+0x200/0x4e0)
[  995.214759] [<c023e058>] (process_one_work) from [<c023f124>] (worker_thread+0x4c/0x568)
[  995.222977] [<c023f124>] (worker_thread) from [<c0244c18>] (kthread+0x140/0x170)
[  995.230489] [<c0244c18>] (kthread) from [<c02010ac>] (ret_from_fork+0x14/0x28)
[  995.237819] Exception stack(0xdb47ffb0 to 0xdb47fff8)
[  995.242947] ffa0:                                     00000000 00000000 00000000 00000000
[  995.251249] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[  995.259550] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[  995.266431] ---[ end trace dd1f92f69abf6ccb ]---
<3>Associated with 18:4b:0d:26:f6:18
<3>CTRL-EVENT-EAP-STARTED EAP authentication started
<3>CTRL-EVENT-EAP-STATUS status='started' parameter=''
<3>CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=25
<3>CTRL-EVENT-EAP-STATUS status='accept proposed method' parameter='PEAP'
<3>CTRL-EVENT-EAP-METHOD EAP vendor 0 method 25 (PEAP) selected
<3>CTRL-EVENT-EAP-SUCCESS EAP authentication completed successfully (based on lower layer success)
[  997.091042] IPv6: ADDRCONF(NETDEV_CHANGE): wlan1: link becomes ready
<3>PMKSA-CACHE-ADDED 18:4b:0d:26:f6:18 0
<3>WPA: Key negotiation completed with 18:4b:0d:26:f6:18 [PTK=CCMP GTK=CCMP]
<3>CTRL-EVENT-CONNECTED - Connection to 18:4b:0d:26:f6:18 completed [id=0 id_str=]

> > list_network
network id / ssid / bssid / flags
0       TestNetwork      any     [CURRENT]
> quit
```

Then I proceed to request a IP address with
```
$ dhclient wlan1
```
and it finished without errors and the interface works after that.

The only issue is that I can only enable one interface at a time, if I enable both at boot, only the first one that gets configured will work, the other not, no matter if is the onboard device or the external one.

ToddHwang-sw, 2020-07-20:
Hi xyklex

The same thing happens when I tried with "TPLink USB dongle" which is based on Realtek 8192eu controller. 

Not sure why "bss" pointer keep pointing NULL, even after the driver runs through successful association. 

This is very nervous patch just simply to hide this error in net/wireless/sme.c . 

 755 
 756         /* if (WARN_ON(!cr->bss)) -- Todd */
 757         if (!cr->bss)
 758                 return;
 759 

** However in my case, it was not related to overlapped initialization of two WiFi devices at the same time...

kucharskim, 2020-09-02:
I'm facing the same problem. Simple `ifdown wlan0` and `ifup wlan0` brings connectivity back again. It's on Raspberry Pi 3 Model B Rev 1.2 (hardware BCM2835, revision a02082) running on kernel `5.4.51-v7+ #1333 SMP Mon Aug 10 16:45:19 BST 2020`

Access point on other end is actually another RPi with hostapd setup.

Per earlier comments I just added `echo 1 >/sys/kernel/debug/tracing/events/cfg80211/enable` to `post-up` network configuration for wlan0.

```
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.4.51-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1333 SMP Mon Aug 10 16:45:19 BST 2020
[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=10c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Rev 1.2
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] Reserved memory: created CMA memory pool at 0x3a400000, size 64 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id shared-dma-pool
[    0.000000] On node 0 totalpages: 255488
[    0.000000]   Normal zone: 2246 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 255488 pages, LIFO batch:63
[    0.000000] percpu: Embedded 20 pages/cpu s49856 r8192 d23872 u81920
[    0.000000] pcpu-alloc: s49856 r8192 d23872 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 253242
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 snd_bcm2835.enable_compat_alsa=0 snd_bcm2835.enable_hdmi=1 snd_bcm2835.enable_headphones=1 bcm2708_fb.fbwidth=656 bcm2708_fb.fbheight=416 bcm2708_fb.fbdepth=16 bcm2708_fb.fbswap=1 vc_mem.mem_base=0x3f000000 vc_mem.mem_size=0x3f600000  dwc_otg.lpm_enable=0 console=ttyAMA0,115200 console=tty1 root=PARTUUID=a3d79e90-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 932248K/1021952K available (8192K kernel code, 689K rwdata, 2552K rodata, 1024K init, 825K bss, 24168K reserved, 65536K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28275 entries in 56 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] random: get_random_bytes called from start_kernel+0x328/0x500 with crng_init=0
[    0.000000] arch_timer: cp15 timer(s) running at 19.20MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000007] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000022] Switching to timer-based delay loop, resolution 52ns
[    0.000291] Console: colour dummy device 80x30
[    0.000878] printk: console [tty1] enabled
[    0.000947] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000995] pid_max: default: 32768 minimum: 301
[    0.001391] Mount-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.001437] Mountpoint-cache hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.002824] Disabling memory control group subsystem
[    0.002945] CPU: Testing write buffer coherency: ok
[    0.003525] CPU0: thread -1, cpu 0, socket 0, mpidr 80000000
[    0.004481] Setting up static identity map for 0x100000 - 0x10003c
[    0.004698] rcu: Hierarchical SRCU implementation.
[    0.005396] smp: Bringing up secondary CPUs ...
[    0.006528] CPU1: thread -1, cpu 1, socket 0, mpidr 80000001
[    0.007792] CPU2: thread -1, cpu 2, socket 0, mpidr 80000002
[    0.008936] CPU3: thread -1, cpu 3, socket 0, mpidr 80000003
[    0.009088] smp: Brought up 1 node, 4 CPUs
[    0.009176] SMP: Total of 4 processors activated (153.60 BogoMIPS).
[    0.009205] CPU: All CPU(s) started in HYP mode.
[    0.009229] CPU: Virtualization extensions available.
[    0.010349] devtmpfs: initialized
[    0.026723] VFP support v0.3: implementor 41 architecture 3 part 40 variant 3 rev 4
[    0.027015] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.027073] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.030203] pinctrl core: initialized pinctrl subsystem
[    0.031416] NET: Registered protocol family 16
[    0.035706] DMA: preallocated 1024 KiB pool for atomic coherent allocations
[    0.042039] hw-breakpoint: found 5 (+1 reserved) breakpoint and 4 watchpoint registers.
[    0.042080] hw-breakpoint: maximum watchpoint size is 8 bytes.
[    0.042329] Serial: AMBA PL011 UART driver
[    0.044304] bcm2835-mbox 3f00b880.mailbox: mailbox enabled
[    0.060091] raspberrypi-firmware soc:firmware: Attached to firmware from 2020-08-19 17:40, variant start_cd
[... 308 lines cut ...]
[... cut ...]
[28502.483880] Exception stack(0xb579ffb0 to 0xb579fff8)
[28502.483899] ffa0:                                     00000000 00000000 00000000 00000000
[28502.483911] ffc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[28502.483922] ffe0: 00000000 00000000 00000000 00000000 00000013 00000000
[28502.483934]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80142df4
[28502.483942]  r4:b6228e40 r3:80104648
[28502.483951] ---[ end trace 8b1d4d2f93a3b160 ]---
```

HinTak, 2020-09-02:
Hmm I just remember something - back in the days I worked on a wifi device driver (a long time ago, though the kernel tree still lists me as one of the maintainers - nothing needs changing for some years) - this looks a lot like one of those issues with driver' s states getting corrupted. Some of the states are being updated due to user actions like the user issuing an explicit network scan. Some are due to external events like traffic arriving. And the Internet states of the driver needs to be protected by mutexes if more than one such things are in transit (multiple CPUs doing different things simultaneously to the driver states). Anyway, the person who wrote the driver code should look at it...

kucharskim, 2020-09-04:
I don't see anything more interesting in dmesg after adding `echo 1 >/sys/kernel/debug/tracing/events/cfg80211/enable`. Below new entries in dmesg and logs from wpa supplicant:

```
[440596.243642] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[440906.808155] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[440944.501324] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[441264.304731] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[441265.830999] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[449948.107531] brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
[449951.354662] ieee80211 phy0: brcmf_update_bss_info: wl dtim_assoc failed (-52)
[449951.355045] ------------[ cut here ]------------
[449951.355350] WARNING: CPU: 0 PID: 24253 at net/wireless/sme.c:756 __cfg80211_connect_result+0x43c/0x4e8 [cfg80211]
[449951.355360] Modules linked in: tun sch_fq_codel ip6t_REJECT nf_reject_ipv6 ip6t_rt nft_chain_nat xt_MASQUERADE nf_nat nft_counter ipt_REJECT nf_reject_ipv4 xt_state xt_conntrack nf_conntrack nf_defrag_ipv4 xt_tcpudp nft_compat nf_tables nfnetlink ch341 usbserial brcmfmac brcmutil sha256_generic libsha256 raspberrypi_hwmon cfg80211 rfkill bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc vc_sm_cma(C) uio_pdrv_genirq uio fixed ip_tables x_tables ipv6 nf_defrag_ipv6
[449951.355578] CPU: 0 PID: 24253 Comm: kworker/u8:1 Tainted: G        WC        5.4.51-v7+ #1333
[449951.355583] Hardware name: BCM2835
[449951.355836] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[449951.355842] Backtrace: 
[449951.355865] [<8010d480>] (dump_backtrace) from [<8010d778>] (show_stack+0x20/0x24)
[449951.355877]  r6:b1764000 r5:00000000 r4:80d95cb8 r3:dfb3d0cc
[449951.355894] [<8010d758>] (show_stack) from [<808cdbe4>] (dump_stack+0xe0/0x124)
[449951.355909] [<808cdb04>] (dump_stack) from [<8012002c>] (__warn+0xec/0x104)
[449951.355923]  r8:000002f4 r7:00000009 r6:7f25e168 r5:00000000 r4:00000000 r3:dfb3d0cc
[449951.355937] [<8011ff40>] (__warn) from [<801200fc>] (warn_slowpath_fmt+0xb8/0xc0)
[449951.355950]  r9:7f25e168 r8:000002f4 r7:7f232ccc r6:00000009 r5:00000000 r4:80d04f48
[449951.356189] [<80120048>] (warn_slowpath_fmt) from [<7f232ccc>] (__cfg80211_connect_result+0x43c/0x4e8 [cfg80211])
[449951.356202]  r9:00000122 r8:b1765e54 r7:80d04f48 r6:00000000 r5:b754b80c r4:b9504004
[449951.356664] [<7f232890>] (__cfg80211_connect_result [cfg80211]) from [<7f1feb54>] (cfg80211_process_wdev_events+0x104/0x160 [cfg80211])
[449951.356676]  r8:b9504090 r7:b9504028 r6:b9504098 r5:b9504004 r4:b754b800
[449951.357136] [<7f1fea50>] (cfg80211_process_wdev_events [cfg80211]) from [<7f1febf0>] (cfg80211_process_rdev_events+0x40/0x98 [cfg80211])
[449951.357150]  r10:00000000 r9:00000100 r8:00000000 r7:b8ce0500 r6:b9442400 r5:b5464460
[449951.357155]  r4:b9504004
[449951.357617] [<7f1febb0>] (cfg80211_process_rdev_events [cfg80211]) from [<7f1f81b8>] (cfg80211_event_work+0x24/0x2c [cfg80211])
[449951.357624]  r5:b63cc300 r4:b54640e4
[449951.357866] [<7f1f8194>] (cfg80211_event_work [cfg80211]) from [<8013c2e4>] (process_one_work+0x17c/0x4b4)
[449951.357874]  r4:b54640e4 r3:7f1f8194
[449951.357889] [<8013c168>] (process_one_work) from [<8013cb4c>] (worker_thread+0x54/0x5b4)
[449951.357903]  r10:b63cc300 r9:b1764038 r8:80d03d00 r7:b9442418 r6:00000088 r5:b63cc314
[449951.357908]  r4:b9442400
[449951.357923] [<8013caf8>] (worker_thread) from [<80142f30>] (kthread+0x13c/0x168)
[449951.357936]  r10:b554c61c r9:a1607e74 r8:8013caf8 r7:b63cc300 r6:00000000 r5:b554cf00
[449951.357941]  r4:b554c600
[449951.357955] [<80142df4>] (kthread) from [<801010ac>] (ret_from_fork+0x14/0x28)
[449951.357961] Exception stack(0xb1765fb0 to 0xb1765ff8)
[449951.357971] 5fa0:                                     00000000 00000000 00000000 00000000
[449951.357983] 5fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[449951.357994] 5fe0: 00000000 00000000 00000000 00000000 00000013 00000000
[449951.358007]  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:80142df4
[449951.358014]  r4:b554cf00 r3:80104648
[449951.358051] ---[ end trace 8b1d4d2f93a3b161 ]---
[449951.358977] IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
[465105.566914] ieee80211 phy0: brcmf_update_bss_info: wl dtim_assoc failed (-52)
[465105.567107] ------------[ cut here ]------------
[465105.567403] WARNING: CPU: 0 PID: 24696 at net/wireless/sme.c:756 __cfg80211_connect_result+0x43c/0x4e8 [cfg80211]
[465105.567414] Modules linked in: tun sch_fq_codel ip6t_REJECT nf_reject_ipv6 ip6t_rt nft_chain_nat xt_MASQUERADE nf_nat nft_counter ipt_REJECT nf_reject_ipv4 xt_state xt_conntrack nf_conntrack nf_defrag_ipv4 xt_tcpudp nft_compat nf_tables nfnetlink ch341 usbserial brcmfmac brcmutil sha256_generic libsha256 raspberrypi_hwmon cfg80211 rfkill bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev mc vc_sm_cma(C) uio_pdrv_genirq uio fixed ip_tables x_tables ipv6 nf_defrag_ipv6
[465105.567630] CPU: 0 PID: 24696 Comm: kworker/u8:2 Tainted: G        WC        5.4.51-v7+ #1333
[465105.567634] Hardware name: BCM2835
[465105.567879] Workqueue: cfg80211 cfg80211_event_work [cfg80211]
[465105.567886] Backtrace: 
[... cut ...]
Sep 04 06:54:51 rpi-0058 wpa_supplicant[25565]: wlan0: Trying to associate with SSID 'rpi_22436'
Sep 04 06:54:52 rpi-0058 wpa_supplicant[25565]: wlan0: Associated with b8:27:eb:02:9c:93
Sep 04 06:54:52 rpi-0058 wpa_supplicant[25565]: wlan0: CTRL-EVENT-CONNECTED - Connection to b8:27:eb:02:9c:93 completed [id=0 id_str=]
Sep 04 06:54:52 rpi-0058 wpa_supplicant[25565]: wlan0: CTRL-EVENT-SUBNET-STATUS-UPDATE status=0
Sep 04 06:54:52 rpi-0058 wpa_supplicant[25565]: wlan0: CTRL-EVENT-REGDOM-CHANGE init=COUNTRY_IE type=COUNTRY alpha2=US
```

kucharskim, 2020-09-04:
In the middle of wpa supplicant logs, workaround kicks in, when ifdown/ifup wlan0 is triggered.

CalebCarroll, 2020-12-12:
I've managed to trigger the "wl dtim_assoc failed (-52)" / "net/wireless/sme.c:756 __cfg80211_connect_result" event 100% of the time on RPi 3 Model B+, RPi 4 Model B using this method:

1) allow the client to connect to the AP
2) change only the WPA2 PSK on the AP (do not change SSID)
3) client reports "carrier lost" then "carrier acquired" then "carrier lost" (expected because the WPA2 PSK on the AP has been changed)
4) change the WPA2 PSK on the AP back to the original value
5) client reports "carrier acquired" + "kernel ... brcmfmac: brcmf_update_bss_info: wl dtim_assoc failed (-52)"

In my wpa_supplicant.conf file I have additional parameters:

**autoscan=periodic:10
disable_scan_offload=1**

for documentation on what these parameters do: [https://w1.fi/wpa_supplicant/devel/structwpa__config.html](https://w1.fi/wpa_supplicant/devel/structwpa__config.html)

If the **disable_scan_offload** parameter (which makes wpa_supplicant to do the scanning rather than the driver) is not set then changing the WPA2 PSK on the AP results in the client only making one reconnection attempt.  The client will report "carrier acquired" then "carrier lost".  When the WPA2 PSK on the AP is changed back to the original value, the client will never attempt to connect again.

ngrayson, 2022-11-15:
I'm also getting this problem on my pi zero W running as a PiHole. I'm going to try a more formal Power Supply rather than a phone charging block and see if it fixes it as per [this reddit post](https://www.reddit.com/r/pihole/comments/igzigg/pi_hole_keeps_crashing/).

ralfizzle, 2023-01-31:
I am seeing this same issue. It appears that the driver crashes sometimes on startup and then this causes wpa_supplicant to associate with `00:00:00:00:00:00`

```
kern  :info  : brcmfmac: brcmf_cfg80211_set_power_mgmt: power save enabled
kern  :info  : brcmfmac: brcmf_cfg80211_set_power_mgmt: power save disabled
kern  :info  : 8021q: 802.1Q VLAN Support v1.8
kern  :info  : bcmgenet fd580000.ethernet: configuring instance for external RGMII (RX delay)
kern  :info  : bcmgenet fd580000.ethernet eth0: Link is Down
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
kern  :err   : ieee80211 phy0: brcmf_cfg80211_escan_handler: Buffer is too small: ignoring
[... cut ...]
kern  :warn  : 9fa0:                                     00000000 00000000 00000000 00000000
kern  :warn  : 9fc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
kern  :warn  : 9fe0: 00000000 00000000 00000000 00000000 00000013 00000000
kern  :warn  :  r10:00000000 r9:00000000 r8:00000000 r7:00000000 r6:00000000 r5:c0247370
kern  :warn  :  r4:c20414c0
kern  :warn  : ---[ end trace a961a1aea1fd8e02 ]---
kern  :info  : IPv6: ADDRCONF(NETDEV_CHANGE): wlan0: link becomes ready
```