raspberrypi/linux #3319: linux 4.19 with ubuntu 19.10(arm64) wayland/x.org desktop artefacts on rpi4

xlazom00, 2019-11-03:
I am testing ubunu 19.10 with raspberry original 4.19 arm64 kernel on rpi4
and there are some artefacts

**Actual behaviour**
1. on login screen there are many random dots(I have monitor with resolution 2560x1440)
2. gnome shell work really nice btw I can recommend it! but here are some artefacts when you for example moving terminal from left to right and back

**To reproduce**
update this images
https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/releases
to ubuntu 19.10
and install ubuntu-desktop-minimal
I have same artefacts with latest mesa master

**Expected behaviour**
1. no random dots on ubuntu login screen
2. no artefatcs when I am moving with windows

**System**
rpi4 with linux raspberry pi 4.19.81  arm 64
and ubuntu 19.10 also arm64
gnome in wayland mode

btw I can recommend ubuntu 19.10 desktop 
latest gnome is working fine on rpi 4
except this artefact
It will be nice to have working firefox with opengl and wayland

JamesH65, 2019-11-04:
It seems very unlikely we would have the manpower to debug someone else's problem - this seems to be a problem with Ubuntu or one of the third party desktop/subsystems you are using. Unless you can replicate this os Raspbian, I doubt we could spend any time on it. Have you tried the Canonical forums?

Askmewho, 2019-11-07:
Does wayland solves the lovely screen tearing we have on raspbian?

xlazom00, 2019-11-07:
@Askmewho yes,but now there are new gfx artefacts :(

xlazom00, 2019-11-07:
@JamesH65 upgrade to debian testing
install mutter
an run mutter  as wayland server
ten run gnome-terminal
and move windows from side to side up and down
it is maybe bug in mesa vc4 drivers I don't know
But I have same artefacts with mesa master
it looks like sometimes it show old part of memory but only on like 100 lines and another 100lines

Askmewho, 2019-11-08:
Ohh, if wayland resolve the tearing, rpi team should focus wayland support as main target... they are too many people complaining about tearing while they dont know its x11 related.. its horrible... hope to see this resolved so i can test wayland decently on ubuntu.

xlazom00, 2019-11-08:
I forget one thing I have attached 2k monitor like 2560x1440

JamesH65, 2019-11-08:
@askmewho We haven't done any work on Wayland for about 4 years, if not more, and as far as I know, there is no intention to do anything more on it. It is something that the community could do - with the new mesa drivers rather than the previous proprietary  graphics, it should be a much 'simpler' task requiring much less specialist knowledge. But that is a guess, I'm no Wayland expert.

JamesH65, 2020-02-14:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

xlazom00, 2020-02-14:
I am on 4.19.97-v8-27 and still same results

JamesH65, 2020-02-14:
We no longer do any work on Wayland, so not sure what we can do here. Our work is on drm/Mesa and X.

xlazom00, 2020-02-14:
I am running mutter(in wayland) with latest mesa( git master ) 
on Pi 4

It will be nice if ubuntu 20.04 desktop(gnome shell) can run on pi4 without any artefacts I think

JamesH65, 2020-02-14:
Then you need to talk to Canonical, who support Ubuntu.

Askmewho, 2020-02-14:
Ive tried wayland on ubuntu gnome 64 bit.. an unofficial release with ton of fixes from github. It does work, I have not experienced any glitches, but as expected sone programs work bad like media playback (except vlc) maybe you should try that one. I didnt fell too much tearing, but there is a minor one, but it may not much difference from x11 on the same distro. For some reason, It have way less tearing than on raspbian, maybe the environment (but you should know its a bit laggy bc its heavy, even with he acceleration).

xlazom00, 2020-02-14:
As I said I have same behaviour with raspbian

Askmewho, 2020-02-14:
Use this one https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial 
For some reason.. way less tearing than on raspbian. But its just my experience.

xlazom00, 2020-02-14:
@Askmewho 
I am using exactly that :)
But I do update to ubuntu 19.10 as I want to test latest gnome.