raspberrypi/linux #3324: RPI4 - 64bit - Wifi pauses when USB3 device is plugged in.

powderluv, 2019-11-08:
**Describe the bug**

Wifi connectivity is lost when a USB3 device (such as Google Coral EDGE TPU or Nod's USB camera) is plugged. Unplugging the USB devices restores connectivity. Weirdly if there is a ethernet cable plugged in too then wifi connectivity is restored with the USB device still plugged in.  

I have tested with the latest 4.19 kernel:
commit eef78b714c3fbc635642332a2ce7d66e51f514d9 
Date:   Thu Nov 7 15:24:32 2019 +0000

    overlays: README: Remove trailing whitespace
    
    A stray space slipped through the net.

Userspace is from https://github.com/sakaki-/gentoo-on-rpi-64bit v1.51 (so we have a full 64bit userspace). 

**To reproduce**

Install the image from https://github.com/sakaki-/gentoo-on-rpi-64bit onto your RPi4. 
Use nmtui to configure and use your Wifi. 
Ping the device from another machine while you plug in a USB3 device (such as Google Coral or Nod Rover). Notice Wifi becomes unreachable. Remove the USB device and Wifi will be restored. 

Also filed an issue at https://github.com/sakaki-/gentoo-on-rpi-64bit/issues/120

[1194970.642530] usb 2-1: new SuperSpeed Gen 1 USB device number 13 using xhci_hcd
[1194970.663389] usb 2-1: New USB device found, idVendor=1a6e, idProduct=089a, bcdDevice= 1.00
[1194970.663406] usb 2-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[1195129.257710] usb 2-1: USB disconnect, device number 13

PING 192.168.1.94 (192.168.1.94): 56 data bytes
64 bytes from 192.168.1.94: icmp_seq=0 ttl=64 time=87.477 ms
64 bytes from 192.168.1.94: icmp_seq=1 ttl=64 time=291.265 ms
64 bytes from 192.168.1.94: icmp_seq=2 ttl=64 time=257.235 ms
64 bytes from 192.168.1.94: icmp_seq=3 ttl=64 time=136.405 ms
64 bytes from 192.168.1.94: icmp_seq=4 ttl=64 time=14.718 ms
64 bytes from 192.168.1.94: icmp_seq=5 ttl=64 time=43.916 ms
64 bytes from 192.168.1.94: icmp_seq=6 ttl=64 time=20.869 ms
64 bytes from 192.168.1.94: icmp_seq=7 ttl=64 time=17.380 ms
64 bytes from 192.168.1.94: icmp_seq=8 ttl=64 time=15.561 ms
64 bytes from 192.168.1.94: icmp_seq=9 ttl=64 time=15.213 ms
64 bytes from 192.168.1.94: icmp_seq=10 ttl=64 time=45.855 ms
Request timeout for icmp_seq 11
Request timeout for icmp_seq 12
Request timeout for icmp_seq 13
Request timeout for icmp_seq 14
Request timeout for icmp_seq 15
Request timeout for icmp_seq 16
Request timeout for icmp_seq 17
Request timeout for icmp_seq 18
64 bytes from 192.168.1.94: icmp_seq=11 ttl=64 time=8012.002 ms
64 bytes from 192.168.1.94: icmp_seq=12 ttl=64 time=7011.518 ms
64 bytes from 192.168.1.94: icmp_seq=13 ttl=64 time=6093.284 ms

I have tried two different USB3 devices (one of which is the Google Coral Edge TPU) and both demonstrate the same behaviour. I checked if the USB is getting enumerated as an ethernet interface but that isn't the case.

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4B - 4GB
* Which OS and version (`cat /etc/rpi-issue`)?
Gentoo 64bit 

* Which firmware version (`vcgencmd version`)?

pi@pi64 ~ $ vcgencmd version
Aug 15 2019 12:05:53 
Copyright (c) 2012 Broadcom
version 9f8431fb7839c7f00f52b81f5822ddab2b31d0db (clean) (release) (start_x)

* Which kernel version (`uname -a`)?
pi@pi64 ~ $ uname -a
Linux pi64 4.19.81-v8-2711+ #1 SMP PREEMPT Fri Nov 8 02:47:04 GMT 2019 aarch64 GNU/Linux

P33M, 2019-11-08:
Poorly shielded USB3 devices will interfere with 2.4GHz ISM band signals. 

https://www.usb.org/sites/default/files/327216.pdf

There are two workarounds that are easy:
- Use a high-quality USB3 extension cable between the device and the Pi to move the source of interference away from the Pi's antenna
- Use a 5GHz band for Wifi

powderluv, 2019-11-08:
That was it. Though even an extension cable seemed to have problems switching to 5G on the AP side solved it. Thanks for the quick response.

closed 2019-11-08 by powderluv