raspberrypi/linux #3326: lan78xx: fix "enabled interrupts" warning

graysky2, 2019-11-09:
**Describe the bug**
Booting a RPi3B+ into kernel 5.3.8 consistently throws a warning in dmesg.

**To reproduce**
Boot into this kernel version and inspect dmesg.

**Expected behaviour**
No warning should be present.

**Actual behaviour**
Warning is triggered.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? RPi3B+
* Which OS and version (`cat /etc/rpi-issue`)? Arch ARM aarch64
* Which firmware version (`vcgencmd version`)? Not shipped with this kernel/architecture.
* Which kernel version (`uname -a`)? `Linux alarm 5.3.8-2-ARCH #1 SMP Sun Nov 3 14:59:20 UTC 2019 aarch64 GNU/Linux`

**Logs**
Link to [dmesg](https://gist.github.com/graysky2/07fa8b9a2ab0bff389557922250810db) log showing error.

**Additional context**
Discussion and proposed patches:
* https://patchwork.kernel.org/patch/10888797/
* https://www.spinics.net/lists/netdev/msg542290.html

lategoodbye, 2019-11-11:
This should be fixed now:
https://git.kernel.org/pub/scm/linux/kernel/git/next/linux-next.git/commit/?h=next-20191108&id=0a29ac5bd3a988dc151c8d26910dec2557421f64

lategoodbye, 2019-11-13:
@graysky2 Please test 5.3.11 and close this issue in case the issue is fixed.

graysky2, 2019-11-13:
@lategoodbye - I will later today, thanks.

closed 2019-11-13 by graysky2