raspberrypi/linux #3330: Bluetooth adapter reset when discovering & connecting at the same time

inwotep, 2019-11-12:
When performing bluetooth device discovery & connecting to a device at the same time, the bluetooth adapter oftentimes reports an HCI event "Hardware Error", which then results in the kernel resetting the adapter (power off/on)

From my tests, I am periodically performing discovery for 15s every 30s.  I then attempt to connect to each device and perform service discovery.   Eventually, one of these connection attempts will trigger the "Hardware Error".  These always occur after the same HCI command HCI_OP_LE_READ_REMOTE_FEATURES (sent in le_conn_complete_evt) times out:

Nov 12 10:36:10 scott-pi4 kernel: Bluetooth: hci1: command 0x2016 tx timeout
Nov 12 10:36:11 scott-pi4 kernel: Bluetooth: hci1: hardware error 0x03
Nov 12 10:40:26 scott-pi4 kernel: Bluetooth: hci1: command 0x2016 tx timeout
Nov 12 10:40:28 scott-pi4 kernel: Bluetooth: hci1: command 0x200c tx timeout
Nov 12 10:40:28 scott-pi4 kernel: Bluetooth: hci1: hardware error 0x03
Nov 12 10:42:40 scott-pi4 kernel: Bluetooth: hci1: command 0x2016 tx timeout
Nov 12 10:42:41 scott-pi4 kernel: Bluetooth: hci1: hardware error 0x03
Nov 12 10:42:51 scott-pi4 kernel: Bluetooth: hci1: command 0x2016 tx timeout
Nov 12 10:42:52 scott-pi4 kernel: Bluetooth: hci1: hardware error 0x03

Raspberry Pi 4 Model B Rev 1.1

pi@scott-pi4:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2019-10-11
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 4a37f04e974e1806551dda69afe0c3306360342d, stage2

pi@scott-pi4:~ $ vcgencmd version
Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)

pi@scott-pi4:~ $ uname -a
Linux scott-pi4 4.19.83-v7l+ #2 SMP Mon Nov 11 11:54:39 PST 2019 armv7l GNU/Linux
kernel rev: 3c235dcfe

I have attempted to use the latest firmware (7.45.189) for the 43455 from cypress, but no change.

[bt_logs.zip](https://github.com/raspberrypi/linux/files/3837753/bt_logs.zip)

inwotep, 2019-11-12:
The same issue also occurs on Raspberry Pi 3 Model B Rev 1.2:
Nov 12 11:36:20 scott-pi3 kernel: Bluetooth: hci0: command 0x2016 tx timeout
Nov 12 11:36:21 scott-pi3 kernel: Bluetooth: hci0: hardware error 0x03

pelwell, 2019-11-12:
Do you have a script that triggers the problem? And when you say "eventually", how long does it typically take to fail?

inwotep, 2019-11-12:
With the setup described above, it occurs within a few minutes.  I would say it occurs approximately every 5 minutes in my environment.  I'll try to put together a python script that can easily trigger the issue.

pelwell, 2019-11-12:
A script would be great. Does it rely on having any hardware there, i.e. is the connect part necessary? If so, how many devices are needed?

I'm hoping this is something we can hand over to Cypress, so the simpler the better.

inwotep, 2019-11-13:
Attached a python script which reliably reproduces the issue in my environment.  Included are logs from bluez, btsnoop, and the kernel.

This test program repeatedly performs discovery for 15 seconds while attempting to connect to devices which are discovered.  Connecting to the devices is necessary to reproduce the issue.  When left simply starting/stopping discovery every 15 seconds, no issues were observed.

[3330.zip](https://github.com/raspberrypi/linux/files/3838984/3330.zip)

inwotep, 2019-11-13:
I have slightly modified the test script to only attempt to connect to a single device repeatedly.  The issue is reproducible for a few different devices already.

pelwell, 2019-11-18:
After a few failed and inconclusive tests (the software stalled in some other way), it's failed in just under an hour with two test devices. If it's relatively repeatable for me too I'll pass the issue on to Cypress.

bschuste, 2020-08-25:
Do you have any update from Cypress on this issue?

travisjayday, 2023-12-31:
I'm experiencing the same issue (the same DMESG line happening right after the feature request on a new LE connection, the adapter getting stuck). However, I'm using an external Realtek Bluetooth USB adapter and running Debian in a VM. Therefore, this leads me to think it's an issue with linux / kernel / hci driver and not any specific hardware. Which is a bummer...