raspberrypi/linux #3331: 16-bit framebuffer only with the lastest kernel

BigMuscle85, 2019-11-13:
When using custom compiled kernel (4.19.81-v7+) or the latest beta kernel (via rpi-update), frame buffer has 16-bit depth only. Setting framebuffer_depth=32 makes no difference. Also X server displays 16-bit colors only. This does not happen when using the kernel 4.19.75-v7+.

I would expect 32-bit frame buffer by default or accepting framebuffer_depth=32 in /boot/config.txt.

- Raspberry Pi 3B+ with 7" touch display
- Raspbian Buster Lite
- Kernel version 4.19.81-v7+

```
pi@raspberrypi:~$ fbset

mode "800x480"
geometry 800 480 800 480 16
timings 0 0 0 0 0 0 0
rgba 5/11,6/5,5/0,0/16
endmode 
```

JamesH65, 2019-11-13:
I'm just looking into this. I can see what's happening, but not sure exactly what the appropriate action is.

JamesH65, 2019-11-14:
OK, a fix for this was just merged to our internal firmware repo. Will take a few days to reach rpi-update, then a bit longer to get in to apt.

popcornmix, 2019-11-14:
Fix is in latest rpi-update firmware.

BigMuscle85, 2019-11-15:
Thank you. I appreciate your quick response.
But how to get the fix to the custom compiled kernel? I pulled and built the latest code and the bug still persists there.

popcornmix, 2019-11-15:
It's a firmware rather than kernel fix. You need updated start.elf/fixup.dat.
Using [rpi-update](https://github.com/Hexxeh/rpi-update) is the easiest way of getting latest test firmware.

BigMuscle85, 2019-11-16:
Ok. I guess that plain rpi-update also replaces my custom compiled kernel and kernel modules. Is "sudo SKIP_KERNEL=1 rpi-update" enough to get this fix ?

popcornmix, 2019-11-18:
Yes, that will work. You just need the `start*.elf` / `fixup*.dat` files.

BigMuscle85, 2019-11-25:
Ok. I can confirm that this is fixed after running "sudo SKIP_KERNEL=1 rpi-update".

closed 2019-11-25 by popcornmix