raspberrypi/linux #3332: dwc_otg: armv7l kernel: rpi2/3: the dwc_otg will report i/o errors when copying a large-size file (300M~600M)

jason77-wang, 2019-11-14:
**Is this the right place for my bug report?**
Yes

**Describe the bug**
build an arm32 kernel with the default config (arch/arm/configs/bcm2711_defconfig), then booting this kernel on rpi2/3 with 1G physical memory, and plug a usb stick to the usb port, copy a large size file (300M-600M) to usb stick, the usb host driver dwc_otg will print I/O errors.

Please refer to: https://bugs.launchpad.net/ubuntu/+source/linux-raspi2/+bug/1852510

**To reproduce**
follow the steps as above

**Expected behaviour**
usb stick can work correctly on rpi2/3 under armv7l kernel with HIGHMEM and VMSPLIT_3G enabled.

**Actual behaviour**
usb stick can't work on rpi2/3 under the armv7l kernel with HIGHMEM and VMSPLIT_3G.

**System**
RPI2/3 boards with 1G physical ram

**Logs**
[ 176.129299] sd 0:0:0:0: [sda] tag#0 FAILED Result: hostbyte=DID_ERROR driverbyte=DRIVER_OK
[ 176.129326] sd 0:0:0:0: [sda] tag#0 CDB: Write(10) 2a 00 00 2e 24 b0 00 00 f0 00
[ 176.129349] blk_update_request: I/O error, dev sda, sector 3024048 op 0x1:(WRITE) flags 0x4000 phys_seg 15 prio class 0
[ 176.883968] usb 1-1.1.2: reset high-speed USB device number 8 using dwc_otg
[ 177.079960] usb 1-1.1.2: reset high-speed USB device number 8 using dwc_otg

**Additional context**
I found with the arm32 kernel,  if we enabled the HIGHMEM and VMSPLIT_3G (like arch/arm/configs/bcm2711_defconfig) in the kernel, some memory will map to highmem region on rpi2/3 with 1G physical memory, and the dwc_otg will not work anymore when copying a large-size file to usb stick.

pelwell, 2019-11-14:
bcm2711_defconfig is for Pi 4 (which uses the BCM2711 SoC). For Pi 2 and Pi 3 use bcm2709_defconfig.

jason77-wang, 2019-11-14:
If I use bcm2709_defconfig, then enable HIGHMEM and VMSPLIT_3G based on bcm2709_defconfig, the dwc_otg still can't work when copying a large-size file to usb stick.

And we plan to use a single arm32 kernel to support all rpi2/3/4 boards.

pelwell, 2019-11-14:
What I am saying is that by using bcm2711_defconfig, HIGHMEM or VMSPLIT_3G on Pi2 and Pi3 you are choosing an unsupported configuration.

jason77-wang, 2019-11-14:
Thanks for your reply.

Is it a known issue that Pi2/3 don't work with HIGHMEM and VMSPLIT_3G enabled? Apart from the dwc_otg, does any other device not work?

pelwell, 2019-11-14:
You might have more luck with the dwc2 driver.

jason77-wang, 2019-11-15:
dwc2 works well with HIGHMEM and VMSPLIT_3G, thanks.

Is it possible to enable dwc2 instead of dwc_otg for all rpi2/3/4 boards? If there is no known issues for dwc2, I plan to send a pull request to enable the dwc2.

thx.

pelwell, 2019-11-15:
dwc_otg is heavily optimised. Some users will experience a performance regression if dwc_otg became the default, otherwise we'd have dropped the downstream maintenance headache years ago (I've just spent some time working out why it was broken in 5.4). Don't waste  your time with the PR.

jason77-wang, 2019-11-17:
Probably I found the root cause, and tried to write a patch to fix it. After applying the patch, I don't see any failures when copying files to usb stick. Please take a look at the patch and help review it.
Thanks.
[patch.txt](https://github.com/raspberrypi/linux/files/3855209/patch.txt)

popcornmix, 2019-11-18:
ping @p33m

P33M, 2019-11-18:
The patch does seem to work around the decision to not initialise the buffer virtual address if the buffer/page is in highmem. dwc_otg uses the virtual address internally, notably to check if there is an alignment fault with the buffer start address - so either 1) the buffers must always be aligned to 4 byte boundaries (probably true if these are all disk transfer blocks, and thus page- or sector-aligned) or 2) a kernel VA mapping always exists by the time dwc_otg needs it to do internal processing.

I would say it's reasonable to assume we always have a kernel VA for the highmem page (otherwise how would copy_to_user work) - unless there's a specific gotcha then please submit a PR with this patch.

P33M, 2019-11-18:
Ah scratch that - the driver calls buf = __bus_to_virt(urb->transfer_dma) in the case where there's no virtual address in the URB, so the patch should be OK.

jason77-wang, 2019-11-18:
Thanks for your comment, the PR is sent.
https://github.com/raspberrypi/linux/pull/3341

P33M, 2020-01-14:
Dangling issue - the fix has been merged.

closed 2020-01-14 by P33M