raspberrypi/linux #3334: On Pi4, the USB mouse and keyboard cannot work after the system is newly started, but it works normally after re-plugging the usb receiver

pymumu, 2019-11-16:
**Describe the bug**
On Pi4, the USB mouse and keyboard cannot work after the system is newly started, but it works normally after re-plugging the usb receiver

Same keyboard works fine on Pi3B.

**To reproduce**
Insert the USB receiver, restart the system, after restarting, the USB keyboard and mouse do not work, re-plugging the USB receiver, the keyboard works fine.

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5
* Which firmware version (`vcgencmd version`)?
Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start
* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
 

**Logs**
[dmesg.log](https://github.com/raspberrypi/linux/files/3853832/dmesg.log)
after restart system:
```
[   22.892016] usb 1-1.4: device descriptor read/64, error -110
[   38.252002] usb 1-1.4: device descriptor read/64, error -110
[   38.471734] usb 1-1.4: new full-speed USB device number 5 using xhci_hcd
[   53.691992] usb 1-1.4: device descriptor read/64, error -110
[   69.052000] usb 1-1.4: device descriptor read/64, error -110
[   69.172147] usb 1-1-port4: attempt power cycle
[   69.831713] usb 1-1.4: new full-speed USB device number 6 using xhci_hcd
[   74.872692] usb 1-1.4: device descriptor read/8, error -110
[   80.073189] usb 1-1.4: device descriptor read/8, error -110
[   80.293008] usb 1-1.4: new full-speed USB device number 7 using xhci_hcd
[   85.353604] usb 1-1.4: device descriptor read/8, error -110
[   90.553839] usb 1-1.4: device descriptor read/8, error -110
[   90.674079] usb 1-1-port4: unable to enumerate USB device
```

replugin usb receiver, usb keyboard works fine
```
[ 2398.473958] usb 1-1.4: new full-speed USB device number 8 using xhci_hcd
[ 2398.613053] usb 1-1.4: New USB device found, idVendor=045e, idProduct=0745, bcdDevice= 6.56
[ 2398.613069] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[ 2398.613081] usb 1-1.4: Product: Microsoft® 2.4GHz Transceiver v8.0
[ 2398.613093] usb 1-1.4: Manufacturer: Microsoft
[ 2398.633766] input: Microsoft Microsoft® 2.4GHz Transceiver v8.0 as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.4/1-1.4:1.0/0003:045E:0745.0001/input/input0
```

**Additional context**
change another logitech usb mouse, there no problem, the microsoft keyboard have this issue.

JamesH65, 2019-11-16:
What power supply are you using? Do you get any low power warnings? (Yellow lightning bolt, error in dmesg?)

pymumu, 2019-11-16:
my power supply is 5V 3A, https://www.zmiusa.com/products/zmi-usb-c-pd-power-adapter
and no yellow lightning bolt or low power warnings in dmesg, i have already attached dmesg above.

P33M, 2019-11-21:
Reproducible with a `045e:07a5 Microsoft Corp. Wireless Receiver 1461C`

On a reboot with the device plugged directly into the Pi, the device will respond to Setup packets and correctly processes the SET_ADDRESS command, but never responds to the data stage of the GET_DESCRIPTOR(DEVICE) request.

A workaround is to use a hub between the Pi and the device. It doesn't matter if the hub is bus- or self-powered.

pymumu, 2019-11-21:
I don't have usb-hub right now, Is there any way to modify some configuration to solve the problem.

P33M, 2019-11-21:
It's unlikely that there is one. The wireless dongle that I have will even get stuck in the bad state without a reboot - if the dongle is plugged in while the Pi's USB port power is switched off (i.e. for the first 5-6 seconds during boot) then the same device hang occurs.

alexsmartens, 2020-01-23:
Is there any solution?

I'm having the same issue with Ubuntu 19.10 Sever and MS Surface mouse/keyboard. The boot time is also something like 3-5 min

rapscalyun, 2020-02-07:
Same issue here with this Non-microsoft USB keyboard:
Bus 001 Device 005: ID 04d9:4545 Holtek Semiconductor, Inc. Keyboard [Diatec Majestouch 2 Tenkeyless]

It's completly dead on boot up, just needs replugging.  I don't see anything unusual in the dmesg output.  Power supply is: Canakit 3A (https://www.canakit.com/raspberry-pi-4-power-supply.html)

mmlbam, 2020-03-15:
Same issue here with Rapoo 1800 wireless USB KB&Mouse on Pi4B with latest Raspbian buster image.
Device ID: 1bcf:05c2 Sunplus Innovation Technology Inc.
 
Any workaround?

ryukl, 2020-04-05:
same here microsoft comfort keybord mouse with Raspberry Pi 4B

59iosl30, 2020-04-13:
You can try this (see my last post):
https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=263848&p=1642282#p1642282

ryukl, 2020-05-03:
Thank you 59iosl30
I tried your last post method. But My Microsoft keyboard and mouse. It is still not work after reboot. I need unplug and plug again in my pi4

59iosl30, 2020-05-04:
This is a very strange issue. I needed hours to identify this problem (which does not occur on RPi3) but my last post/workaround works for me in combination with a mechanical keyboard from a different vendor. The keyboard was perfectly recognized at every boot on RPi4.

smelamud, 2020-05-26:
I also stumbled upon this issue with pi4 and two different Logitech mice and keyboards. They don't work after powerup, start working after removing/inserting the dongle and stop working again in several minutes. In the same time, another combined mouse+keyboard by Logitech works perfectly.

el1-x, 2020-06-13:
Add me to the list of users with this issue. Just purchase the MS 900 Desktop combo for my Pi4B and i have to reinsert the USB receiver after each boot.

timg236, 2020-06-13:
Please could you try temporarily adding sdhci.debug_quirks2=4 to cmdline.txt so see if the behaviour changes?

This prevents the voltage switch for UHS on the SD-CARD (which reduces performance) but has the effect of not requiring the PMIC to be reset during a reboot.

el1-x, 2020-06-13:
I added sdhci.debug_quirks2=4 to cmdline.txt as instructed. 
Keyboard does not operate post reboot.

timg236, 2020-06-13:
Thanks @el1-x 

If you are comfortable trying the beta version of the bootlaoder then the latest release (2020-06-12) toggles the USBB port power during reboot.

N.B. This is a beta release so you should only install this if you are already familiar with installing and reverting beta versions of the bootloader. 

Use the Raspberry Pi Imager to install version 2020-06-12
https://github.com/raspberrypi/rpi-eeprom/blob/master/releases.md

el1-x, 2020-06-13:
I will keep the beta bootloader in mind. I have not used a beta release, but happy to familiarise myself. 

I seems that usbcore.old_scheme_first=1 has worked, but i must say, i have no idea what it has done.

59iosl30, 2020-06-13:
In my case this only works in 64-bit mode:

/boot/config.txt:
arm_64bit=1

Since I added these two parameters I haven't any issues with the RPi4.

timg236, 2020-07-31:
I now have a Microsoft Wireless mouse and keyboard 900 and was able to reproduce the problem on a Pi 4 4GB. It seems that this USB device fails if there is a small glitch during the reset/reboot.
This is resolved by powering off USB port power for 1 second before booting. The latest beta now does this once, before starting the main boot-modes whether or not USB boot is selected (unless overridden via config.txt)

This resolves the issue for me and will be released via APT next week but if you are familiar with bootloader updates you can grab the binaries from Git
https://github.com/raspberrypi/rpi-eeprom/commit/d7b1a49a8356b09384817d362cbfb572e50770a8

pelupessy, 2020-08-08:
Same problem with DAS keyboard.

OS: retropie and ubuntu MATE (latest version)

timg236, 2020-08-08:
Please try with Raspberry Pi OS the latest beta bootloader. It works for me now

pelupessy, 2020-08-08:
Thanks! I am happy to confirm that installing the beta bootloader in retropie (it's based on raspberrypi os, after all) works! 

The usb keyboard is working fine in both Retropie and ubuntu MATE now.

timg236, 2020-08-08:
Thanks @pelupessy for confirming the fix. That helps to justify promoting this to stable as hopefully the last candidate for a the next production bootloader.

ChessSpider, 2020-09-05:
Updating to latest stable worked for me. Thanks everyone who worked on the fix

closed 2020-09-05 by pelwell