raspberrypi/linux #3335: hdparm not working correctly on PI4

pymumu, 2019-11-16:
**Describe the bug**
The `hdparm -I /dev/sda` is not working, similar harddisk command like `hdparm -y /dev/sda` also fails.

```
root@raspberrypi:/home/pi# hdparm -I /dev/sda

/dev/sda:
SG_IO: bad/missing sense data, sb[]:  70 00 05 00 00 00 00 0a 00 00 00 00 24 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00

ATA device, with non-removable media
Standards:
	Likely used: 1
Configuration:
	Logical		max	current
	cylinders	0	0
	heads		0	0
	sectors/track	0	0
	--
	Logical/Physical Sector size:           512 bytes
	device size with M = 1024*1024:           0 MBytes
	device size with M = 1000*1000:           0 MBytes 
	cache/buffer size  = unknown
Capabilities:
	IORDY not likely
	Cannot perform double-word IO
	R/W multiple sector transfer: not supported
	DMA: not supported
	PIO: pio0 
```

**To reproduce**
hdparm -I /dev/sda
hdparm -y /dev/sda
hdparm -C /dev/sda

**Expected behaviour**
Support set harddisk spin-down time.
On pi3, all things works fine

Run same command on pi3 with same USB hard disk, everything works fine:
```shell
root@raspberrypi:/home/pi# uname -a
Linux raspberrypi 4.19.75-v7+ #1270 SMP Tue Sep 24 18:45:11 BST 2019 armv7l GNU/Linux
root@raspberrypi:/home/pi# hdparm -I /dev/sda

/dev/sda:

ATA device, with non-removable media
	Model Number:       ST4000LM024-2AN17V                      
	Serial Number:      WCK2PTAP
	Firmware Revision:  0001    
	Transport:          Serial, ATA8-AST, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6, SATA Rev 3.0
Standards:
	Used: unknown (minor revision code 0x006d) 
	Supported: 10 9 8 7 6 5 
	Likely used: 10
Configuration:
	Logical		max	current
	cylinders	16383	16383
	heads		16	16
	sectors/track	63	63
	--
	CHS current addressable sectors:    16514064
	LBA    user addressable sectors:   268435455
	LBA48  user addressable sectors:  7814037168
	Logical  Sector size:                   512 bytes
	Physical Sector size:                  4096 bytes
	Logical Sector-0 offset:                  0 bytes
	device size with M = 1024*1024:     3815447 MBytes
	device size with M = 1000*1000:     4000787 MBytes (4000 GB)
	cache/buffer size  = unknown
	Form Factor: 2.5 inch
	Nominal Media Rotation Rate: 5526
Capabilities:
	LBA, IORDY(can be disabled)
	Queue depth: 32
	Standby timer values: spec'd by Standard, no device specific minimum
	R/W multiple sector transfer: Max = 16	Current = ?
	Advanced power management level: 254
	Recommended acoustic management value: 208, current value: 208
	DMA: mdma0 mdma1 *mdma2 udma0 udma1 udma2 udma3 udma4 udma5 udma6 
	     Cycle time: min=120ns recommended=120ns
	PIO: pio0 pio1 pio2 pio3 pio4 
	     Cycle time: no flow control=120ns  IORDY flow control=120ns
Commands/features:
	Enabled	Supported:
	   *	SMART feature set
	    	Security Mode feature set
	   *	Power Management feature set
	   *	Write cache
	   *	Look-ahead
	   *	Host Protected Area feature set
	   *	WRITE_BUFFER command
	   *	READ_BUFFER command
	   *	DOWNLOAD_MICROCODE
	   *	Advanced Power Management feature set
	    	Power-Up In Standby feature set
	   *	SET_FEATURES required to spinup after power up
	    	SET_MAX security extension
	   *	48-bit Address feature set
	   *	Device Configuration Overlay feature set
	   *	Mandatory FLUSH_CACHE
	   *	FLUSH_CACHE_EXT
	   *	SMART error logging
	   *	SMART self-test
	   *	General Purpose Logging feature set
	   *	WRITE_{DMA|MULTIPLE}_FUA_EXT
	   *	64-bit World wide name
	   *	WRITE_UNCORRECTABLE_EXT command
	   *	{READ,WRITE}_DMA_EXT_GPL commands
	   *	Segmented DOWNLOAD_MICROCODE
	    	unknown 119[6]
	   *	Gen1 signaling speed (1.5Gb/s)
	   *	Gen2 signaling speed (3.0Gb/s)
	   *	Gen3 signaling speed (6.0Gb/s)
	   *	Native Command Queueing (NCQ)
	   *	Phy event counters
	   *	READ_LOG_DMA_EXT equivalent to READ_LOG_EXT
	    	DMA Setup Auto-Activate optimization
	    	Device-initiated interface power management
	    	Software settings preservation
	    	unknown 78[7]
	   *	SMART Command Transport (SCT) feature set
	   *	SCT Write Same (AC2)
	   *	SCT Data Tables (AC5)
	    	unknown 206[7]
	    	unknown 206[12] (vendor specific)
	    	unknown 206[13] (vendor specific)
Security: 
	Master password revision code = 65534
		supported
	not	enabled
	not	locked
	not	frozen
	not	expired: security count
		supported: enhanced erase
	654min for SECURITY ERASE UNIT. 654min for ENHANCED SECURITY ERASE UNIT.
Logical Unit WWN Device Identifier: 5000c500b9af9856
	NAA		: 5
	IEEE OUI	: 000c50
	Unique ID	: 0b9af9856
Checksum: correct
```

**System**
* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
Pi4
* Which OS and version (`cat /etc/rpi-issue`)?
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5
* Which firmware version (`vcgencmd version`)?
Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start
* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

**Logs**
dmesg on PI4
```shell
[ 4683.050535] usb 2-1: new SuperSpeed Gen 1 USB device number 2 using xhci_hcd
[ 4683.081946] usb 2-1: New USB device found, idVendor=0bc2, idProduct=231a, bcdDevice= 7.08
[ 4683.081962] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 4683.081974] usb 2-1: Product: Expansion
[ 4683.081986] usb 2-1: Manufacturer: Seagate
[ 4683.081998] usb 2-1: SerialNumber: NAA3GRTM
[ 4683.114373] scsi host0: uas
[ 4683.116667] scsi 0:0:0:0: Direct-Access     Seagate  Expansion        0708 PQ: 0 ANSI: 6
[ 4683.120453] sd 0:0:0:0: Attached scsi generic sg0 type 0
[ 4689.456539] sd 0:0:0:0: [sda] 7814037167 512-byte logical blocks: (4.00 TB/3.64 TiB)
[ 4689.456546] sd 0:0:0:0: [sda] 4096-byte physical blocks
[ 4689.456979] sd 0:0:0:0: [sda] Write Protect is off
[ 4689.456985] sd 0:0:0:0: [sda] Mode Sense: 53 00 00 08
[ 4689.457472] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[ 4689.458243] sd 0:0:0:0: [sda] Optimal transfer size 33553920 bytes not a multiple of physical block size (4096 bytes)
[ 4689.632963]  sda: sda1 sda2
[ 4689.635493] sd 0:0:0:0: [sda] Attached SCSI disk
[ 4690.795363] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: (null)
```

same disk on pi3's dmsg log, this works fine
``` shell
[47035.952673] usb 1-1.4: new high-speed USB device number 6 using dwc_otg
[47036.163929] usb 1-1.4: New USB device found, idVendor=0bc2, idProduct=231a, bcdDevice= 7.08
[47036.163944] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[47036.163954] usb 1-1.4: Product: Expansion
[47036.163964] usb 1-1.4: Manufacturer: Seagate
[47036.163974] usb 1-1.4: SerialNumber: NAA3GRTM
[47036.164821] usb 1-1.4: The driver for the USB controller dwc_otg_hcd does not support scatter-gather which is
[47036.164833] usb 1-1.4: required by the UAS driver. Please try an other USB controller if you wish to use UAS.
[47036.164844] usb-storage 1-1.4:1.0: USB Mass Storage device detected
[47036.165459] scsi host0: usb-storage 1-1.4:1.0
[47037.223667] scsi 0:0:0:0: Direct-Access     Seagate  Expansion        0708 PQ: 0 ANSI: 6
[47037.224689] sd 0:0:0:0: Attached scsi generic sg0 type 0
[47041.142771] Voltage normalised (0x00000000)
[47041.911297] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[47041.911761] sd 0:0:0:0: [sda] 7814037167 512-byte logical blocks: (4.00 TB/3.64 TiB)
[47041.911774] sd 0:0:0:0: [sda] 4096-byte physical blocks
[47041.912268] sd 0:0:0:0: [sda] Write Protect is off
[47041.912281] sd 0:0:0:0: [sda] Mode Sense: 47 00 00 08
[47041.912855] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[47042.031770]  sda: sda1 sda2
[47042.035292] sd 0:0:0:0: [sda] Attached SCSI disk
[47043.667351] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: (null)
```

**Additional context**
same usb harddisk, on pi3 hdparm works file, on pi4 doesn't work.

P33M, 2019-11-20:
The Pi 4 uses UAS as the disk transport, not USB mass-storage which is used on all previous models of Pi.

I have a Sabrent USB3.0 adapter which exhibits the issue - on a Pi 4 `hdparm -y /dev/sda` returns the same error that you're seeing, but not on a Pi 3.

In both cases, I see that hdparm issues a ATA pass-through command with various parameters set, then requests an IN transfer to check the result. In the UAS case, the SCSI sense response is "illegal command" whereas in the mass storage case the mass-storage ATA passthrough response has a nonzero error field and the endpoint STALLs.

The command is failing on both Pis, presumably because the device doesn't support the command, but the UAS case is noisier about it.

P33M, 2019-11-20:
Can you try the steps here and report back: https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=245931

pymumu, 2019-11-21:
After adding quirks to /boot/cmdline.txt, problem solved, the hard disk works very well so far.

Thanks.

The quirks added to /boot/cmdline.txt
```shell
root@raspberrypi:/home/pi# cat /boot/cmdline.txt
console=serial0,115200 console=tty1 root=PARTUUID=d9b3f436-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles usb-storage.quirks=0bc2:231a:u
```

`hdparm -I /dev/sda` command result
```
root@raspberrypi:/home/pi# hdparm -I /dev/sda

/dev/sda:

ATA device, with non-removable media
        Model Number:       ST4000LM024-2AN17V
        Serial Number:      WCK2PTAP
        Firmware Revision:  0001
        Transport:          Serial, ATA8-AST, SATA 1.0a, SATA II Extensions, SATA Rev 2.5, SATA Rev 2.6, SATA Rev 3.0
Standards:
        Used: unknown (minor revision code 0x006d)
        Supported: 10 9 8 7 6 5
        Likely used: 10
Configuration:
        Logical         max     current
        cylinders       16383   16383
        heads           16      16
        sectors/track   63      63
        --
        CHS current addressable sectors:    16514064
        LBA    user addressable sectors:   268435455
        LBA48  user addressable sectors:  7814037168
        Logical  Sector size:                   512 bytes
        Physical Sector size:                  4096 bytes
        Logical Sector-0 offset:                  0 bytes
        device size with M = 1024*1024:     3815447 MBytes
        device size with M = 1000*1000:     4000787 MBytes (4000 GB)
        cache/buffer size  = unknown
        Form Factor: 2.5 inch
        Nominal Media Rotation Rate: 5526
Capabilities:
        LBA, IORDY(can be disabled)
        Queue depth: 32
        Standby timer values: spec'd by Standard, no device specific minimum
        R/W multiple sector transfer: Max = 16  Current = ?
        Advanced power management level: 254
        Recommended acoustic management value: 208, current value: 208
        DMA: mdma0 mdma1 *mdma2 udma0 udma1 udma2 udma3 udma4 udma5 udma6
             Cycle time: min=120ns recommended=120ns
        PIO: pio0 pio1 pio2 pio3 pio4
             Cycle time: no flow control=120ns  IORDY flow control=120ns
```

`dmesg` log
```
[  509.233313] usb 2-1: New USB device found, idVendor=0bc2, idProduct=231a, bcdDevice= 7.08
[  509.233328] usb 2-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[  509.233341] usb 2-1: Product: Expansion
[  509.233353] usb 2-1: Manufacturer: Seagate
[  509.233365] usb 2-1: SerialNumber: NAA3GRTM
[  509.236617] usb 2-1: UAS is blacklisted for this device, using usb-storage instead
[  509.236738] usb 2-1: UAS is blacklisted for this device, using usb-storage instead
[  509.236757] usb-storage 2-1:1.0: USB Mass Storage device detected
[  509.237819] usb-storage 2-1:1.0: Quirks match for vid 0bc2 pid 231a: 800000
[  509.237953] scsi host0: usb-storage 2-1:1.0
[  510.242582] scsi 0:0:0:0: Direct-Access     Seagate  Expansion        0708 PQ: 0 ANSI: 6
[  510.243651] sd 0:0:0:0: Attached scsi generic sg0 type 0
[  515.687671] sd 0:0:0:0: [sda] Very big device. Trying to use READ CAPACITY(16).
[  515.688048] sd 0:0:0:0: [sda] 7814037167 512-byte logical blocks: (4.00 TB/3.64 TiB)
[  515.688062] sd 0:0:0:0: [sda] 4096-byte physical blocks
[  515.688813] sd 0:0:0:0: [sda] Write Protect is off
[  515.688827] sd 0:0:0:0: [sda] Mode Sense: 47 00 00 08
[  515.689545] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[  515.808437]  sda: sda1 sda2
[  515.813176] sd 0:0:0:0: [sda] Attached SCSI disk
[  517.042890] EXT4-fs (sda2): mounted filesystem with ordered data mode. Opts: (null)
```

P33M, 2020-01-14:
Closing as workaround implemented.

closed 2020-01-14 by P33M