raspberrypi/linux #3339: 5.4.y branch arm64 build error

djazo, 2019-11-17:
The file `drivers/pci/controller/pcie-brcmstb-bounce64.c`
has call to non-specified function `is_device_dma_coherent`.

Cross building on x86_64 with:
`make ARCH=arm64 CROSS_COMPILE=aarch64-linux-gnu-`

It should be `dev_is_dma_coherent` and `#include <linux/dma-noncoherent.h>` should be added to load the header.

popcornmix, 2019-11-18:
@djazo can you confirm with this change 5.4 worked fine with 64-bit kernel?

popcornmix, 2019-11-18:
I tested change it it built and booted. Added https://github.com/raspberrypi/linux/commit/88322f50733f8e1e55d2b5561a3911d292137267

djazo, 2019-11-18:
I've been running 5.4-rc7 now on rpi4 since my original report, so can also confirm that it's been running ok.

popcornmix, 2019-11-18:
Okay to close the issue now the patch is added?

djazo, 2019-11-18:
Yeah. Closing.

closed 2019-11-18 by djazo

popcornmix, 2019-11-18:
Thanks for report.