raspberrypi/linux #3340: SC16IS752 Interrupt bug?

Thomas-GeDaD, 2019-11-18:
I´m not sure if it is the right place for the Problem:
I use 3x SC16is752 devices on a Pi 4 with buster (4.19.75-v7l+ #1270) 
All 3 devices works if I only use 1 or 2 of them:
In config.txt:
works:
`
#dtoverlay=sc16is752-i2c,int_pin=13,addr=0x4A,xtal=14745600
dtoverlay=sc16is752-i2c,int_pin=12,addr=0x49,xtal=14745600
dtoverlay=sc16is752-i2c,int_pin=6,addr=0x48,xtal=14745600
`
works:
`
dtoverlay=sc16is752-i2c,int_pin=13,addr=0x4A,xtal=14745600
#dtoverlay=sc16is752-i2c,int_pin=12,addr=0x49,xtal=14745600
dtoverlay=sc16is752-i2c,int_pin=6,addr=0x48,xtal=14745600
`
works:
`
dtoverlay=sc16is752-i2c,int_pin=13,addr=0x4A,xtal=14745600
dtoverlay=sc16is752-i2c,int_pin=12,addr=0x49,xtal=14745600
#dtoverlay=sc16is752-i2c,int_pin=6,addr=0x48,xtal=14745600
`
with all configuration I get ttySC0-ttySC3 and can send an recieve data from a GPS (NMEA)
All works perfectly.
But If I use all three devices I get problems=>
`
dtoverlay=sc16is752-i2c,int_pin=13,addr=0x4A,xtal=14745600
dtoverlay=sc16is752-i2c,int_pin=12,addr=0x49,xtal=14745600
dtoverlay=sc16is752-i2c,int_pin=6,addr=0x48,xtal=14745600
`
After startup the Pi dmesg seems fine without any issue.
`[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.19.75-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1270 SMP Tue Sep 24 18:51:41 BST 2019
[    0.000000] CPU: ARMv7 Processor [410fd083] revision 3 (ARMv7), cr=30c5383d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: Raspberry Pi 4 Model B Rev 1.1
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 256 MiB at 0x000000001ec00000
[    0.000000] On node 0 totalpages: 504832
[    0.000000]   DMA zone: 1728 pages used for memmap
[    0.000000]   DMA zone: 0 pages reserved
[    0.000000]   DMA zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 308224 pages, LIFO batch:63
[    0.000000] random: get_random_bytes called from start_kernel+0xc0/0x4e8 with crng_init=0
[    0.000000] percpu: Embedded 17 pages/cpu s39488 r8192 d21952 u69632
[    0.000000] pcpu-alloc: s39488 r8192 d21952 u69632 alloc=17*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 503104
[    0.000000] Kernel command line: coherent_pool=1M 8250.nr_uarts=1 cma=64M cma=256M video=HDMI-A-1:1920x1080M@60,margin_left=48,margin_right=48,margin_top=48,margin_bottom=48 smsc95xx.macaddr=DC:A6:32:12:F9:C6 vc_mem.mem_base=0x3ec00000 vc_mem.mem_size=0x40000000  dwc_otg.lpm_enable=0 console=ttyS0,115200 console=tty1 root=PARTUUID=5b55e216-02 rootfstype=ext4 elevator=deadline fsck.repair=yes rootwait quiet splash plymouth.ignore-serial-consoles
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Memory: 1721828K/2019328K available (8192K kernel code, 661K rwdata, 2352K rodata, 2048K init, 850K bss, 35356K reserved, 262144K cma-reserved, 1232896K highmem)
[    0.000000] Virtual kernel memory layout:
                   vector  : 0xffff0000 - 0xffff1000   (   4 kB)
                   fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
                   vmalloc : 0xf0800000 - 0xff800000   ( 240 MB)
                   lowmem  : 0xc0000000 - 0xf0000000   ( 768 MB)
                   pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
                   modules : 0xbf000000 - 0xbfe00000   (  14 MB)
                     .text : 0x(ptrval) - 0x(ptrval)   (10208 kB)
                     .init : 0x(ptrval) - 0x(ptrval)   (2048 kB)
                     .data : 0x(ptrval) - 0x(ptrval)   ( 662 kB)
                      .bss : 0x(ptrval) - 0x(ptrval)   ( 851 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] ftrace: allocating 28124 entries in 83 pages
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] GIC: Using split EOI/Deactivate mode
[    0.000000] arch_timer: cp15 timer(s) running at 54.00MHz (phys).
[... 451 lines cut ...]
[  462.735074] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[  462.735689] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[  462.736202] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[  468.001605] i2c-bcm2835 fe804000.i2c: i2c transfer timed out
[  468.001631] sc16is7xx_port_irq: 16481 callbacks suppressed
[  468.001641] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[  469.041626] i2c-bcm2835 fe804000.i2c: i2c transfer timed out
[  469.041656] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[  470.081686] i2c-bcm2835 fe804000.i2c: i2c transfer timed out
[  470.081713] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[  470.082007] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[  470.082478] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[  470.082951] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[  470.083443] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[  470.083742] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[  470.083995] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[  470.084257] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[  473.011785] sc16is7xx_port_irq: 11588 callbacks suppressed
`
At this time there are physical no data on the UART.
This passes infinitely until i restart.
The amazing think is, that this only happened if I use ttySC0,ttySC1,ttySC4,ttySC5
But It can not be a physical problem because all devices in two pairs or alone worked....
I think it´s a driver Issue?
Thanks....

pelwell, 2019-11-18:
Those devices, or rather that device driver, has been the source of many problems over the years. We've found and fixed a number of bugs, but we can't offer more than time-permitting support on third-party  devices and driver software.

The kernel log doesn't give any clue as to how much data was successfully delivered before the failure occurred. Please try systematically sending single bytes to each port individually and report the outcomes.

One thing I do notices is that one of your IRQ GPIOs (6) naturally pulls high while the others (12, 13) pull low, and the overlay makes no attempt to set the pull states. Do you have any kind of external pull-ups? What does the datasheet say on the matter? After booting but before using the serial ports, try forcing the pulls to a consistent state on all three GPIOs using raspi-gpio:
```
$ raspi-gpio set 6,12,13 ip pu
```
or:
```
$ raspi-gpio set 6,12,13 ip pd
```
Does that change the behaviour?

Thomas-GeDaD, 2019-11-18:
Hi pellwell,
thanks for your fast answer.
I have seen this and checked this times before.
It have no changes. All device Interrupts have a external 1KOhm pullup as described in the datasheet.
If I read the interrupt GPIO´s all are high on the first moment.

What do you mean with sending single bytes to each port? Physically on the uart side?
This happened directly if i want to configurate the Port... There are no data on the uart side at this moment...

pelwell, 2019-11-18:
Please try watching the state of the IRQ lines using raspi-gpio while you try starting one of the interfaces:
```
$ while true; do raspi-gpio get 6,12,13; sleep 0.5; done
```
You can adjust the sleep time (in seconds) to watch with finer granularity.

Before initialising the UART all three should be high, then afterwards I would expect no more than one to go low, and only momentarilty so. If you don't see any of the lines flicker, try removing the sleep and logging to a file. You could also install Joan's PiScope (http://abyz.me.uk/rpi/pigpio/piscope.html) - I've not tried it myself, but it looks like it could be useful here if you don't have any other kind of logic analyser or oscilloscope.

Thomas-GeDaD, 2019-11-18:
I have done this. All Interrupt GPIO are all the Time high!

> GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 12: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 12: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 12: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 12: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 12: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 12: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 12: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 12: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 6: level=1 fsel=0 func=INPUT pull=UP
GPIO 12: level=1 fsel=0 func=INPUT pull=DOWN
GPIO 13: level=1 fsel=0 func=INPUT pull=DOWN

All Intterupt GPIO´s high before and after I configurate a UART Interface....
I recheck it later with a Oscilloscope.
In dmesg the same issue:

> [ 7178.643982] sc16is7xx_port_irq: 8282 callbacks suppressed
[ 7178.643995] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7178.645169] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[ 7178.646349] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7178.647754] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[ 7178.648933] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7178.650114] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[ 7178.651529] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7178.652714] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[ 7178.653906] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7178.655327] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[ 7183.653777] sc16is7xx_port_irq: 8292 callbacks suppressed
[ 7183.653790] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7183.654408] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[ 7183.655025] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7183.655641] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[ 7183.656258] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7183.656874] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[ 7183.657489] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7183.658105] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0
[ 7183.658721] sc16is7xx 1-004a: ttySC4: Unexpected interrupt: 0
[ 7183.659337] sc16is7xx 1-004a: ttySC5: Unexpected interrupt: 0

pelwell, 2019-11-18:
The scope will definitely tell you more (and more accurately). Start with the IRQ lines, then we can take it from there.

Thomas-GeDaD, 2019-11-18:
The Interrupt works correct.
But at measured i think I have found the Issue:
[image]
As you see in the Addresstable for 0x4A I use Vdd and SCL (1001010) sometimes the Adress works and sometimes are not. I think there is a problem at startup. The SC16Is752 needs the I2C line for addressing in this case. And there something seems to work wrong. If I use a adress like 1001100 with Vdd and VSS all works fine??
Is the I2C all time availible direct at sturtup? Has anybody ever use it with adressing trough the SCL / SDA line?
Funny....

Thomas-GeDaD, 2019-11-18:
If I start the Pi together with the SC16IS752 it does not work. If I start the Pi and then Power up the SC16IS752 all works fine. The SC16IS752 seems to need the I2C runs before it starts...

Thomas-GeDaD, 2019-11-18:
By the way I think it´s an issue that all devices displayed as configurated and recognized:
[image]
But not availible:

closed 2019-11-18 by Thomas-GeDaD

EJM321, 2022-03-22:
I'd like to re-open this issue as I'm experiencing exactly the same problem as Thomas-GeDaD did, but with only x2 devices (SC16IS752).  I find that individually (i.e only one overlay enabled at a time in config.txt), each of the two devices work as expected, yet when I enable both overlays neither of them works (not TX or RX).  

I have 1k pullups on the interrupt pins.  Both devices seem to initialise correctly as I'm seeing "UU" at both addresses using i2cdetect.   I have other hardware on the bus too as you can see below, yet they have been behaving just fine.  
GPIO expander at 0x20
GPIO expander at 0x22
PWM/LED driver at 0x40

ls /dev yields the expected ttySC0, ttySC1, ttySC2, ttySC3

     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: 20 -- 22 -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: 40 -- -- -- -- -- -- -- UU -- -- -- UU -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: 70 -- -- -- -- -- -- --

I have tried:
- different I2C addresses (e.g 0x48 and 0x4d, 0x48 and 0x49, 0x48 and 0x4c). 
- different I2C speeds (200kHz,100kHz, 70kHz)
- different stty -F baud rate speeds 115200 and 9600 
- different rpi gpios (0x48 on GPIO24, 0x48 on GPIO6 and 0x4d on GPIO5)

RPI model: 3b
uname -a: Linux raspberrypi 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux

Any help/tips/ideas on this would be much appreciated!

---------------
Overlays I'm using, with config.txt parameters as needed:
/dts-v1/;
/plugin/;

/ {
	compatible = "brcm,bcm2835";

	fragment@0 {
		target = <&i2c1>;

		frag1: __overlay__ {
			#address-cells = <1>;
			#size-cells = <0>;
			status = "okay";

			sc16is752: sc16is752@48 {
				compatible = "nxp,sc16is752";
				reg = <0x48>; // i2c address
				clocks = <&sc16is752_clk>;
				interrupt-parent = <&gpio>;
				interrupts = <24 0x2>; /* IRQ_TYPE_EDGE_FALLING */
				gpio-controller;
				#gpio-cells = <0>;
				i2c-max-frequency = <400000>;
				status = "okay";

				sc16is752_clk: sc16is752_clk {
					compatible = "fixed-clock";
					#clock-cells = <0>;
					clock-frequency = <14745600>;
				};
			};
		};
	};

	__overrides__ {
		int_pin = <&sc16is752>,"interrupts:0";
		addr = <&sc16is752>,"reg:0",<&sc16is752_clk>,"name";
		xtal = <&sc16is752_clk>,"clock-frequency:0";
	};
};
----------------
config.txt
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on,i2c_arm_baudrate=100000
#dtparam=i2s=on
dtoverlay=spi0-cs
dtoverlay=spi1-1cs,cs0_pin=18,cs0_spidev=off
#dtoverlay=spi1-1cs,cs0_spidev
dtparam=spi=on

# Uncomment this to enable the lirc-rpi module
#dtoverlay=lirc-rpi

# Additional overlays and parameters are documented /boot/overlays/README

# Enable audio (loads snd_bcm2835)
dtparam=audio=on

# Disable bluetooth
dtoverlay = pi3-disable-bt

# Ensure mini UART is restored
enable_uart = 1

# PiCAN2 CAN-Bus board
dtparam=spi=on
dtoverlay=mcp2515-can1-0,oscillator=16000000,interrupt=16
dtoverlay=spi-bcm2835-overlay

enable_uart=1

max_usb_current=1
hdmi_group=2
hdmi_mode=87
hdmi_cvt 800 480 60 6 0 0 0
hdmi_drive=1

dtoverlay=sc16is752-i2c-overlay,int_pin=6,addr=0x48
dtoverlay=sc16is752-i2c-overlay,int_pin=5,addr=0x4c

---------------------
Here is a dtc -Ifs /proc/device-tree > devicetree4.txt dump:
[devicetree4.txt](https://github.com/raspberrypi/linux/files/8320508/devicetree4.txt)

Thomas-GeDaD, 2022-03-22:
This is what i add to the boot config and it works fine since the fix in all versions until now:
dtoverlay=sc16is752-i2c,int_pin=13,addr=0x4c,xtal=14745600
Here you find my postinstall for a board:
https://github.com/Thomas-GeDaD/SignalK_raspberry_MCS/blob/master/postinstall.js

pelwell, 2022-03-22:
Thanks for contributing, Thomas - which kernel are you running?

EJM321, 2022-03-23:
Hi Pelwell, thanks for the advice, I was bit hesitant to update the OS as I have a lot of pre-configured/installed python modules that I use.  It would take a lot of time setting that all up again.  But being on buster it appears that I can update to bullseye (5.10) using these steps without having to start from scratch:

https://www.tomshardware.com/how-to/upgrade-raspberry-pi-os-to-bullseye-from-buster

I have backups of my data, so should be good to go.

Also, just to make sure, I'm using the SC16IS752 overlay from here: 

https://github.com/jason77-wang/eoan-rpi-pull/commit/9bf908c63524b3148620e91088aa866d1a48fec2

Is this the correct one to use?  It appears to be the most recent.

So in summary, I will:
- Update OS to 5.10
- Recompile (dts to dtb) overlay with -overlay omitted in the filename, i.e: sc16ic752-i2c
- Try again with both devices enabled in config.txt: 

dtoverlay=sc16is752-i2c,int_pin=6,addr=0x48
dtoverlay=sc16is752-i2c,int_pin=5,addr=0x4c

Might also explicitly use the xtal parameter: xtal=14745600 as Thomas-GeDaD has got it configured.

Hopefully I get some joy out of these changes.

EJM321, 2022-03-23:
Also Thomas, to get the three of the devices working, did you have to power up the SC16IS752 devices after the RPI booted?  Or was that not critical in the end?

Thomas-GeDaD, 2022-03-23:
We use several Kernels. We sell the MCS in a shop and there are many  >500 in the market. I have nerver heard about an issue of the i2c-Serial interfaces, so i think it works on every Kernel version after this, old issue.
@EJM321
Why you compile your own overlay? The sc16is752 is already intigrated. So you only have to load the dt-overlay in the config.txt? So if you start from stratch, only add the line in the config.txt and all works fine. Here you find an overview of our board:
https://github.com/Thomas-GeDaD/openplotter-MCS/blob/master/QuickGuide_MCS.pdf
And this is the load of the dt-overlay:
https://github.com/Thomas-GeDaD/SignalK_raspberry_MCS/blob/f439b5b7c65583c19eb5798ef4d2adb38cd768c5/postinstall.js#L39
This is the shematic:
[image]

EJM321, 2022-03-23:
Thanks for the updated feedback Thomas, I will give that a try.

So to make sure, your issues with multiple SC16IS752 devices resolved itself when you updated your OS from 4.19.x to 5.x ?

Thomas-GeDaD, 2022-03-23:
no there was no issue with 4.19. This issue here was based on the sc16is752. The chip has a problem, when there is at first VDD before the i2c is availible. then you get an addressconflict when you use the scl/sda line for adressing the chip. I have change the addresses of the board so i dont need a i2c line for addressing. only use the VDD/gnd for the adresspins!
Ther was an issue with the dt-overlay in 5.4.51. But this was solves. In all other Kernel versions after beginning of 2020 all works fine.
Here you find the issue of 5.4.51:
https://github.com/raspberrypi/linux/issues/3765

EJM321, 2022-03-23:
But that's a bit strange as for that very reason I avoided using those addresses that use the SDA and SCL lines in the first place.  So I'm also just using the VDD/GND for addressing (e.g 0x48, 0x4d, 0x4c etc), 

But still I'm experiencing this multiple device issue?  Anyway I will update the OS first and let you know how it went.

pelwell, 2022-03-23:
Thomas is correct in that you don't need to install any overlays - it is already present and verified to work.

EJM321, 2022-03-23:
Thanks Pelwell, I take note of that.