raspberrypi/linux #3342: Blank screen on display with native resolution of 1366 x 768

dluksza, 2019-11-20:
**Describe the bug**
[This monitor](https://www.ebay.co.uk/itm/Hannspree-HT161HNB-15-6-inch-LED-1366-x-768-12ms-Response-Speakers-HDMI/382903381495) is not working with Raspbian Little and Desktop.

**To reproduce**
Just start with a fresh copy of Raspbian Lite or Desktop.

**Expected behaviour**
Raspberry should be able to use any type of monitor.

**Actual behaviour**
After starting up raspberry it blinks a few times then shows _no signal_ and finally switches to _power saving mode_. No output from kernel buffer, nor frame buffer, nor X is visible. Upgrading to latest Raspbian version (together with firmware upgrade) doesn't fix this issue.

On the other side `kmscube` is able to initialize the display and renders colorful spinning cube.

I've manage to get kernel buffer working on Raspbian Little by editing `/boot/config.txt` and setting:
```
hdmi_group=2
hdmi_mode=16
```
it also works with `hdmi_mode=39` and `hdmi_mode=81`, but still X server is showing nothing.

Finally after stopping adding `xrandr --output HDM-1 --mode 1360x768` to `/root/.xsessionrc`, then stopping `lightdm` and executing `startx` I'm able to see window manager.

All of the above I need to do over ssh, since I can only see a black screen on Pi.

**System**
Here is my [raspinfo.txt](https://github.com/raspberrypi/linux/files/3871252/raspinfo.txt)

**Additional context**
I'm mostly interested in running `flutter-pi` on my Pi 4, which (I assume) uses direct access to DRI, more context about this issue can be found here https://github.com/ardera/flutter-pi/issues/16

carlonluca, 2019-12-12:
I had other issues with a display with that resolution: #3221. Maybe related?

popcornmix, 2023-01-24:
That resolution is a known issue. See: https://forums.raspberrypi.com/viewtopic.php?t=284866

The 2711 hardware cannot support odd horizontal blanking timings.

The kms driver should filter out that resolution which normally means it falls back to 1080p or 720p.

It may be possible to drive 1366x768 using custom timings (with even blanking periods), but there's no way to be sure if display will accept that.

But there's nothing more we can do as it's a hardware issue.