raspberrypi/linux #3344: Enable CONFIG_HID_STEAM

rodrigorc, 2019-11-22:
In a desktop PC, the Steam Controller is usually managed by the Steam Client, by Valve. But in the RPI the best option is to enable the hid-steam kernel module. It would be nice to have it enabled by default.

popcornmix, 2019-11-25:
@pelwell looks okay to me.
It's a tristate and the source for it looks fairly modest in size.

AnsisMalins, 2020-02-22:
Liked, favorited and upvoted. I'm too stupid to compile a custom kernel for this.

pelwell, 2020-02-24:
Apologies - this one slipped through the net. It's in the rpi-4.19.y tree now (2620ff62e7e747acac5ceaeae88491a34a174945) and will be included in future builds.

AnsisMalins, 2020-02-24:
Thank you. I'm not familiar with your process. When is the next build expected?

pelwell, 2020-02-24:
There is no fixed schedule - it depends on what has changed - but there is usually at least one per week made available via rpi-update. The Raspbian kernel package is updated much less frequently - every few months, perhaps - and there are typically two or three full images released per year.

rodrigorc, 2020-03-02:
This is already available for rpi-update. And it works perfectly.

Thank you.

closed 2020-03-02 by rodrigorc