raspberrypi/linux #3346: Using Apt-get disables WiFi connection temporarily

skylerspark, 2019-11-23:
**Describe the bug**
When I use apt-get instal, specifically the command `sudo apt-get install chromium-browser —yes` (Found on a tutorial to installing chromium on raspbian Jessie (I’m using raspbian buster))

**To reproduce**
Using Raspbian Buster on Raspberry pi1 and run the above command with a WiFi adapter plugged in.

**Expected behaviour**
It should download chromium and then install it, but instead after running, this wired x and lines icon comes up on my WiFi connection, and the chromium download fails.

Also, afterwards, I’m unable to connect to the internet any time after unless I reboot or close all applications and refresh everything

**System**

* Raspberry Pi 1
* OS: Latest Version of Raspbian Buster with Pixel Desktop addon.
* Firmware Ver: 6820edeee4ef3891b95fc01cf02a7abd7ca52f17
* Kernel Ver: `4.19.75+ #1270` armv6l

skylerspark, 2019-11-29:
I thunk it may also be to blame on my wifi adapter, but even then, im not sure

JamesH65, 2020-02-14:
I think it would be worth updating to the latest software to see if this issue is still present.  If so its more than likely a crash in the wireless driver code, so would need to know what wireless adapter you are using. A dmesg dump of any errors at the time of failure would also be useful