raspberrypi/linux #3349: NOOBS v3.2.1->RaspbianFull - RPi4 Kernel panic not syncing. Attempted to kill init exit code 0x00007f00

slrslr, 2019-11-26:
I have downloaded official torrent of the NOOBS v3.2.1, (after download, the downloaded files are checked by the torrent client if are complete). I extracted .zip archive, formated the SDHC using SD Formatter and copied over the files. Inserted SDHC card into a RPi 4B 4GB RAM and ran. It shown NOOBS boot menu where i selected Raspbian FULL (it was 3 days back), it installed and i clicked OK button. It rebooted and ended with errors
Kernel panic not syncing. Attempted to kill init exit code 0x00007f00

[image]

Sidenote: when i tried to flash Raspbian directly to SDHC, i had problem "[Caught <SEGV>, core dump failed, Freezing execution](https://github.com/raspberrypi/linux/issues/3345)"

Please let me know if i need to try anything (describe it exactly as i am a linux noob).

XECDesign, 2019-11-26:
This looks like an unreliable sd card or an issue with the power supply.

slrslr, 2019-12-05:
> This looks like an unreliable sd card or an issue with the power supply.

it is possible, because this time i have used Windows 10 disk formating tool to quick format **different SDHC card** as FAT32 and copied over noobs v3.2.1 files, installed light weight raspbian buster desktop successfully. Should this be closed or i should try something?

pelwell, 2019-12-05:
If everything is working and we have a plausible explanation for the original problem then I would say close the issue. You can always reopen it later without losing the history if it recurs.

closed 2019-12-05 by slrslr