raspberrypi/linux #3350: Pi4: Kernel 4.19.84/85 breaks v3d driver

mkreisl, 2019-11-26:
Theese two commits

https://github.com/raspberrypi/linux/commit/866a26769fb1ebfef4335b93a3b1352447f0446f#diff-b8449e22343b231189dcdc72f2252892
https://github.com/raspberrypi/linux/commit/eb0ba882d1f3e23c6d584a37c55f0aaddcb0df2f#diff-b8449e22343b231189dcdc72f2252892

stops v3d driver working

Got this error

```
Nov 26 14:36:46 kmxbilr2 kernel: [   66.129251] v3d fec00000.v3d: Failed to get clock (-2)
```

Well done guys :angry:

Reverting those two commits brings back v3d driver working

pelwell, 2019-11-26:
If you're building on arm64 (you helpfully don't say) then you're already running old code - try with https://github.com/raspberrypi/linux/commit/ff2c5910ce487c365654f13e4edb446ed1c0552c in there as well.

mkreisl, 2019-11-26:
> If you're building on arm64 (you helpfully don't say) then you're already running old code - try with [ff2c591](https://github.com/raspberrypi/linux/commit/ff2c5910ce487c365654f13e4edb446ed1c0552c) in there as well.

No, 32bit built only

pelwell, 2019-11-26:
It works for me with top-of-tree rpi-4.19.y (c078c64fecb325ee86da705b91ed286c90aae3f6) and with the tree as of eb0ba882d1f3e23c6d584a37c55f0aaddcb0df2f. Is your firmware up-to-date?

mkreisl, 2019-11-26:
Yes, it is

pelwell, 2019-11-26:
Sorry - that comment should have said "and with the tree as of ff2c591".

mkreisl, 2019-11-26:
I can confirm, works again

Please add those two settings to `bcm2709_defconfig`, that would have saved me a lot of time!

closed 2019-11-26 by mkreisl

pelwell, 2019-11-26:
bcm2709_defconfig is not meant for 2711. There's no warranty if you're using it off-label.