raspberrypi/linux #3351: Proper critical CPU temperature

lategoodbye, 2019-11-27:
**Describe the bug**
During upstream of the BCM2835 thermal driver, we needed to define a critical temperature. Due a misunderstanding the assumption was this temperature defines when to start thermal throttling, but this was wrong. The actual meaning is that the Linux kernel should do the last countermeasures before the hardware get damaged: a thermal shutdown.

**To reproduce**
I didn't reproduced it by myself, but it's obvious that the trip point definition of 80 °C in bcm283x.dtsi is wrong. But there is at least one [report](https://archlinuxarm.org/forum/viewtopic.php?f=65&p=62642&sid=766fdf02ff982a34beee0819e5ac0e02) in the Arch forum.

So the question what would be the right critical temperature? 87 °C?
Do we need some SoC specific values?

JamesH65, 2019-11-27:
HW won't get damaged until much MUCH higher than that - probably in excess of 125c. We attempt to keep temperatures below 85 (or whatever it is) for safety reasons - don't want users burning themselves on the chip. So what value you use depends on what you are warning for.

Some information here, not particularly comprehensive. I need to update that a bit.

https://www.raspberrypi.org/documentation/hardware/raspberrypi/frequency-management.md

JamesH65, 2019-11-27:
All that said, we limit the Pi to 85 (or at least try to) for various reasons. So I'd say anything over 85 is out of the intended range. I guess thermal inertia might get it a bit over that in a worst case, but that won't cause problems.

lategoodbye, 2019-11-30:
Okay, i send out a patch upstream:
http://lists.infradead.org/pipermail/linux-rpi-kernel/2019-November/008962.html

Thanks

closed 2019-12-05 by lategoodbye