raspberrypi/linux #3359: Enable BFQ I/O Scheduler

felixbuenemann, 2019-11-30:
It would be great if the Raspberry Pi kernel could enable the bfq I/O scheduler, since it gives superior responsiveness and fairer scheduling on slow devices like MMC, SD cards or HDDs.

The relevant options should be:

```
CONFIG_IOSCHED_BFQ=m
CONFIG_BFQ_GROUP_IOSCHED=y
```

Here are some relevant performance tests using bfq scheduler for blk-mq with SD cards:

[A solution to high latencies caused by I/O](https://elinux.org/images/1/1c/ELC-18-HighLat-Valente.pdf)

timg236, 2019-11-30:
Looks interesting, but don't see any numbers for Raspberry Pi. Before changing the default kernel for ALL Raspberry Pi users we would need to see benchmarks & real world tests for a variety of Pi models and hardware. Can it be configured in but only activated at runtime (assuming that it being idle is zero cost) ?
If people are 

If people are using this already then feel free to post data here

felixbuenemann, 2019-12-01:
> Looks interesting, but don't see any numbers for Raspberry Pi.

Having actual results from Raspberry Pi devices is a valid concern.

I've got a first gen model B, 3B and 4B, so I should be able to do some testing.

It'll probably take some time, since it's a rather busy time of the year for me…

> Can it be configured in but only activated at runtime (assuming that it being idle is zero cost) ?

Yes, it can be compiled as a module (as I suggested) and only enabled for some block devices using udev rules, eg. I could use the following to only enable it for spinning drives:

```sh
# set bfq scheduler for rotating disks
ACTION=="add|change", KERNEL=="sd[a-z]", ATTR{queue/rotational}=="1", ATTR{queue/scheduler}="bfq"
```

timg236, 2019-12-04:
It might be worth posting something on the forums to see if other people are using a feature. If there is a reasonable user base actively using a feature then that's help to justify enabling it for everyone.

mongaikan, 2023-01-30:
Enabling BFQ in the kernel config won't make it the default so I don't see what's the rational behind not enabling it?

pelwell, 2023-01-31:
Are you saying it's unreasonable to ask for justification for increasing the size of all our images?

Most feature requests are granted after a bit of back-and-forth, especially those which have no effect (apart from image size) when not enabled - we didn't expect three years to pass without feedback.

pelwell, 2023-01-31:
If this scheduler is so beneficial, it's odd that out of all the kernel defconfigs, only 22 platforms enable it, one of which is ARCH=arm and none of which are ARCH=arm64.

mongaikan, 2023-01-31:
> Are you saying it's unreasonable to ask for justification for increasing the size of all our images?
>
> Most feature requests are granted after a bit of back-and-forth, especially those which have no [...]

I was just wondering as a user why you would choose not to enable a scheduler that's been performing really well with single queue drives but it seems you don't know either.

> If this scheduler is so beneficial, it's odd that out of all the kernel defconfigs, only 22 platforms enable it, one of which is ARCH=arm and none of which are ARCH=arm64.

No idea, again, just a user. Why don't you figure it out? IIRC Archlinux ARM got it in theirs.
cf https://www.kernel.org/doc/html/latest/block/bfq-iosched.html if you're ever interested in the subject.

Aw and maybe take a good long holiday sometime soon? Sounds like you sorely need it.
Cheers

JamesH65, 2023-01-31:
If you want this, do the testing yourself. Not our job to test every option **other** people might need. We've always been happy to incorporate features with the appropriate testing and performance implications assessed and deemed acceptable.