raspberrypi/linux #3360: PI 3B+ : 8 UVCs can't work well on RP3B+

steven-hz, 2019-11-30:
6 uvc can works fine with 2 powered hubs, but when the num of uvc is 8 ,  RP3B+ will report :

> USB isochronous frame lost (-63). 

Then, the v4l2 api can't get the the random 2 video streams through  VIDIOC_DQBUF cmd of ioctl.

The output of lsusb -v and lsusb -t
```
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Couldn't open device, some information will be missing
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            4.14
  iManufacturer           3 
  iProduct                2 
  iSerial                 1 
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0 
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0 
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
```
```
pi@raspberrypi:~ $ lsusb -t
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 1: Dev 3, If 0, Class=Hub, Driver=hub/3p, 480M
            |__ Port 2: Dev 5, If 0, Class=Hub, Driver=hub/4p, 480M
                |__ Port 3: Dev 11, If 0, Class=Video, Driver=uvcvideo, 480M
                |__ Port 3: Dev 11, If 1, Class=Video, Driver=uvcvideo, 480M
                |__ Port 1: Dev 7, If 1, Class=Video, Driver=uvcvideo, 480M
                |__ Port 1: Dev 7, If 0, Class=Video, Driver=uvcvideo, 480M
                |__ Port 4: Dev 13, If 1, Class=Video, Driver=uvcvideo, 480M
                |__ Port 4: Dev 13, If 0, Class=Video, Driver=uvcvideo, 480M
                |__ Port 2: Dev 9, If 0, Class=Video, Driver=uvcvideo, 480M
                |__ Port 2: Dev 9, If 1, Class=Video, Driver=uvcvideo, 480M
            |__ Port 1: Dev 6, If 0, Class=Vendor Specific Class, Driver=lan78xx, 480M
        |__ Port 3: Dev 4, If 0, Class=Hub, Driver=hub/4p, 480M
            |__ Port 2: Dev 8, If 0, Class=Video, Driver=uvcvideo, 480M
            |__ Port 2: Dev 8, If 1, Class=Video, Driver=uvcvideo, 480M
            |__ Port 3: Dev 10, If 1, Class=Video, Driver=uvcvideo, 480M
            |__ Port 3: Dev 10, If 0, Class=Video, Driver=uvcvideo, 480M
            |__ Port 4: Dev 12, If 0, Class=Video, Driver=uvcvideo, 480M
            |__ Port 4: Dev 12, If 1, Class=Video, Driver=uvcvideo, 480M
```
Is there any suggestions about this issue?

BTW, kernel version is : 

> Linux  version 4.14.98-v7+

Thanks in advance.

pelwell, 2019-11-30:
As has been mentioned many times over the years, the USB controller used in Pis prior to the 4B wasn't designed for this kind of load, being targeted at mobile applications that would spend most of their time in device mode. The fact that it works as well as it does is down to a lot of serious optimisation work by @P33M, and unless there are specific, unexpected bugs, I don't think they'll be doing further development on it.

P33M, 2019-12-02:
This number of devices isn't going to be supportable on DWC hosts (>8 total simultaneously active endpoints). It probably won't even be supportable on an ideal USB2.0 host controller - that number of isochronous video endpoints is likely to completely saturate the available bandwidth.

Closing - not a supportable configuration.

closed 2019-12-02 by P33M