raspberrypi/linux #3361: "buggy DT: spidev listed directly in DT" warning is back again in latest kernel.

trejan, 2019-11-30:
The "buggy DT: spidev listed directly in DT" warning is back again in the latest kernel (4.19.86-v7l+ #1283 SMP Fri Nov 29 18:37:40 GMT 2019) that was originally patched in https://github.com/raspberrypi/linux/issues/1054 due to https://github.com/raspberrypi/linux/commit/e616dd603c296b88561ee394e30f6d2445960521#diff-ec1637a62b0a5dff379a8cc872db7aa8

pelwell, 2019-11-30:
Yeah, I spotted that yesterday while in the middle of something else. I don't agree with the upstream view of how spidev should be used, so I'll keep on disabling the warning.

pelwell, 2019-12-02:
The warning is suppressed again - see https://github.com/raspberrypi/linux/commit/a535fff2659f6949b031ec5bfe94813dc24e8b3e.

closed 2019-12-02 by trejan