raspberrypi/linux #3366: How to resolve "Max number of devices this xHCI host supports is 32." on Pi 4?

RockXiroz, 2019-12-02:
I saw a similar issue: https://github.com/raspberrypi/linux/issues/3309
My issue is a little bit different it, but the work around solution is the same: How can I force raspberry4 to use EHCI instead of XHCI?
I also checked the link, its solution also be "If you need more than 32 devices, do not use (and disable) USB3/xhci":
http://marc.merlins.org/perso/linux/post_2018-12-20_Getting-Around-USB3-xhci-32-Device-Limit-_Max-number-of-devices-this-xHCI-host-supports-is-32_.html

I have 45 USB devices need to connect to PI 4. PI itself has 6 USB devices. So the total number of USB is 51(45+6).
$ cat /sys/firmware/devicetree/base/model
Raspberry Pi 4 Model B Rev 1.1
Verision:
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
$ lsusb -t
/:  Bus 03.Port 1: Dev 1, Class=root_hub, Driver=dwc_otg/1p, 480M
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
        |__ Port 3: Dev 89, If 0, Class=Hub, Driver=hub/7p, 480M
            |__ Port 1: Dev 90, If 0, Class=Hub, Driver=hub/7p, 480M
                |__ Port 2: Dev 92, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 3: Dev 99, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
                |__ Port 4: Dev 115, If 1, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                |__ Port 4: Dev 115, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 480M
                |__ Port 5: Dev 120, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 6: Dev 125, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 2: Dev 91, If 0, Class=Hub, Driver=hub/7p, 480M
                |__ Port 6: Dev 113, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 2: Dev 94, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 7: Dev 123, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
                |__ Port 5: Dev 108, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 3: Dev 103, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
            |__ Port 3: Dev 93, If 0, Class=Hub, Driver=hub/7p, 480M
                |__ Port 7: Dev 127, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
                |__ Port 5: Dev 112, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 3: Dev 107, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
                |__ Port 6: Dev 117, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 2: Dev 97, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
            |__ Port 4: Dev 96, If 0, Class=Hub, Driver=hub/7p, 480M
                |__ Port 5: Dev 106, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 3: Dev 82, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
                |__ Port 6: Dev 111, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 2: Dev 101, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 7: Dev 121, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
            |__ Port 7: Dev 100, If 0, Class=Hub, Driver=hub/7p, 480M
                |__ Port 5: Dev 119, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 3: Dev 114, If 0, Class=Vendor Specific Class, Driver=usbfs, 480M
                |__ Port 6: Dev 124, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
                |__ Port 2: Dev 105, If 0, Class=Vendor Specific Class, Driver=ftdi_sio, 12M
        |__ Port 4: Dev 3, If 0, Class=Human Interface Device, Driver=usbhid, 1.5M

 $ lsusb
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0461:4de2 Primax Electronics, Ltd
Bus 001 Device 124: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 119: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 114: ID 0e8d:201c MediaTek Inc.
Bus 001 Device 105: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 100: ID 0424:2517 Standard Microsystems Corp. Hub
Bus 001 Device 121: ID 0e8d:201c MediaTek Inc.
Bus 001 Device 111: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 106: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 082: ID 0e8d:201c MediaTek Inc.
Bus 001 Device 101: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 096: ID 0424:2517 Standard Microsystems Corp. Hub
Bus 001 Device 127: ID 0e8d:201c MediaTek Inc.
Bus 001 Device 117: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 112: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 107: ID 0e8d:201c MediaTek Inc.
Bus 001 Device 097: ID 0403:6001 Future Technology Devices International, Ltd FT232 Serial (UART) IC
Bus 001 Device 093: ID 0424:2517 Standard Microsystems Corp. Hub
Bus 001 Device 123: ID 0e8d:201c MediaTek Inc.
[... 46 lines cut ...]
[275568.002068] usb 1-1.3.2.7: Manufacturer: alps
[275568.002079] usb 1-1.3.2.7: SerialNumber: LAFINT_V21D1T_Q026
[275568.100878] usb 1-1.3.7.6: new full-speed USB device number 124 using xhci_hcd
[275568.256684] usb 1-1.3.7.6: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[275568.256701] usb 1-1.3.7.6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[275568.256714] usb 1-1.3.7.6: Product: FT232R USB UART
[275568.256726] usb 1-1.3.7.6: Manufacturer: FTDI
[275568.256737] usb 1-1.3.7.6: SerialNumber: A106OZ6B
[275568.267376] ftdi_sio 1-1.3.7.6:1.0: FTDI USB Serial Device converter detected
[275568.267526] usb 1-1.3.7.6: Detected FT232RL
[275568.270626] usb 1-1.3.7.6: FTDI USB Serial Device converter now attached to ttyUSB15
[275568.271665] xhci_hcd 0000:01:00.0: Error while assigning device slot ID
[275568.271681] xhci_hcd 0000:01:00.0: Max number of devices this xHCI host supports is 32.
[275568.271694] usb 1-1.3.7-port7: couldn't allocate usb_device
[275568.340910] usb 1-1.3.1.6: new full-speed USB device number 125 using xhci_hcd
[275568.488563] usb 1-1.3.1.6: New USB device found, idVendor=0403, idProduct=6001, bcdDevice= 6.00
[275568.488571] usb 1-1.3.1.6: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[275568.488576] usb 1-1.3.1.6: Product: FT232R USB UART
[275568.488581] usb 1-1.3.1.6: Manufacturer: FTDI
[275568.488586] usb 1-1.3.1.6: SerialNumber: A106OY36
[275568.499165] ftdi_sio 1-1.3.1.6:1.0: FTDI USB Serial Device converter detected
[275568.499229] usb 1-1.3.1.6: Detected FT232RL
[275568.500392] usb 1-1.3.1.6: FTDI USB Serial Device converter now attached to ttyUSB16
[275568.500668] xhci_hcd 0000:01:00.0: Error while assigning device slot ID
[275568.500674] xhci_hcd 0000:01:00.0: Max number of devices this xHCI host supports is 32.

pelwell, 2019-12-02:
The XHCI driver is reporting the value exposed by the device via the standard HCSPARAMS1 register. That bit of the standard says:
> **Number of Device Slots (MaxSlots).** This field specifies the number of Device Context Structures and Doorbell Array entries this host controller can support.

It is conceivable that this field has been set by the VL805 firmware to an unnecessarily conservative value, but the fact that the device datasheet includes the value "20h" (32) makes me think it is a hard limitation of the controller.

P33M, 2019-12-02:
There is no "force USB controller into EHCI mode" workaround. The Pi 4 host controller is just that - an xHCI host controller.

Out of interest, to disambiguate your issue from the one you linked to, how many of the devices can you use simultaneously (up to the hardware limit of 32)?

RockXiroz, 2019-12-04:
Thanks @pelwell for supporting, it sounds like a hardware limitation. Does Raspberry Pi consider to develop a product with strong USB ability?

> Out of interest, to disambiguate your issue from the one you linked to, how many of the devices can you use simultaneously (up to the hardware limit of 32)?

@P33M, actually, we just try how many USB devices can Pi 4 connect to. We will try to looking for a solution.  If we still can't resolve the issue, we maybe use 2 piece of Pi 4.

pelwell, 2019-12-04:
I think Pi4 has a capable USB interface, especially for a $35 device, but maximum number of slots is something we can bear in mind for future products. Your use case is an unusual one, and it feels like the "gearing" is wrong - having to dedicate a whole USB slot to a single UART - but you may have no alternative. Your backup plan of adding a second Pi 4 sounds like a reasonable compromise.

AppApe, 2019-12-04:
> The XHCI driver is reporting the value exposed by the device via the standard HCSPARAMS1 register. That bit of the standard says:
>
>
> It is conceivable that this field has been set by the VL805 [...]

pelwell, 2019-12-04:
Pardon?

pelwell, 2019-12-04:
Pardon?

burtyb, 2019-12-11:
Using the controller on the USB-C connector you should be able to connect a few more using the DWC2 controller and if the (other) xHCI controller could be [coaxed into detecting devices](https://www.raspberrypi.org/forums/viewtopic.php?t=246348) maybe you'd get more there.

P33M, 2020-01-14:
The recommendation for using more than 32 attached USB devices (including hubs) on a Pi 4 is to use multiple Pi 4s. It is a hardware limit that is unlikely to ever change.

closed 2020-01-14 by P33M

andriisemenov, 2021-02-04:
Is there any other solution to this issue other than adding another raspberry? Any news on this?

andriisemenov, 2021-02-04:
Found a way to extend beyond 32 devices.
Added to /boot/config.txt:
dtoverlay=dwc2, dr_mode=host
otg_mode=1

Removed in cmdline.txt:
modules-load=dwc2,g_ether

After that I could connect to the USB C port of RPI4 a USB C -> USB A converter and a hub to it. The devices appear in the lsusb as expected under dwc2.

egwepas, 2021-02-11:
Hi @andriisemenov , do those option also raise the maximum number of devices on the "normal" usb 3 / 2 ports, or is it only for the USB C port ?

andriisemenov, 2021-02-12:
@egwepas, It does not extend the capacity of xhci hardware (native ports). It only enables the Type C to work as a host and since Type C port has it's own connection to the MPU it is possible to add more devices through type C.

egwepas, 2021-02-12:
Thanks; any idead of the limit of the theorical maximum number of devices over the USB-C port ?

andriisemenov, 2021-02-12:
Tried with 16, works

egwepas, 2021-02-12:
Thanks for the feedback

Pack3tL0ss, 2021-03-07:
@andriisemenov how are you powering the RPi while using the USB-C ports?  GPIO pins? PoE hat? 

I'm using a 3rd Party PoE hat (LoveRPi).  I'm able to get devices on the USB-C ports to show up, but only if I plug it in after the device has powered on.  If the hub on the USB-C port is plugged in and the device is rebooted, or PoE is cycled it won't power up.

Handri-Kosada, 2021-05-07:
> Found a way to extend beyond 32 devices.
> Added to /boot/config.txt:
> dtoverlay=dwc2, dr_mode=host
> otg_mode=1
> 
> Removed in /boot/cmdline.txt:
> modules-load=dwc2,g_ether
> 
> After that I could connect to the USB C port of RPI4 a USB C -> USB A converter and a hub to it. The devices appear in the lsusb as expected under dwc2.

does this work for ubuntu on raspberry pi 4?

juergh, 2021-05-10:
Wrong place to ask for Ubuntu support. Check /boot/firmware. If you encounter issues, please log a bug at https://bugs.launchpad.net/ubuntu/+source/linux-raspi/+filebug.