raspberrypi/linux #3367: question about"cross-compile raspberrypi kernel"

haixiansheng, 2019-12-03:
I cross-compile the linux kernel, there are some problem, sincerely hope someone can help answer these questions. My HOST computer is ubuntu 18.04.3, and the raspberrypi Linux kernel is 4.19. Use the raspberry/tools.  The compilation record is as follows:

In file included from net/bluetooth/l2cap_sock.c:35:0:
net/bluetooth/l2cap_sock.c: In function 'l2cap_sock_connect':
include/net/bluetooth/l2cap.h:837:12: error: inlining failed in call to always_inline 'l2cap_chan_connect': function body not available
net/bluetooth/l2cap_sock.c:126:6: error: called from here
scripts/Makefile.build:305: recipe for target 'net/bluetooth/l2cap_sock.o' failed
make[2]: *** [net/bluetooth/l2cap_sock.o] Error 1
make[2]: *** Waiting for unfinished jobs....
scripts/Makefile.build:441: recipe for target 'net/bluetooth' failed
make[1]: *** [net/bluetooth] Error 2
make[1]: *** Waiting for unfinished jobs....
  CC [M]  net/can/gw.o
  CC [M]  net/can/raw.o
  CC [M]  net/can/af_can.o
  CC [M]  net/can/proc.o
  LD [M]  net/can/can-gw.o
  LD [M]  net/can/can-raw.o
  LD [M]  net/can/can-bcm.o
  LD [M]  net/can/can.o
Makefile:944: recipe for target 'net' failed
make: *** [net] Error 2

Thanks a lot.

JamesH65, 2019-12-03:
Are you following the instructions on this page?

https://www.raspberrypi.org/documentation/linux/kernel/building.md

haixiansheng, 2019-12-04:
Thank you very much for reply, I finally found the problem, my linux kernel files are incomplete.

closed 2019-12-04 by haixiansheng