raspberrypi/linux #3368: any chance to get the single linux patches for kernel ?

artynet, 2019-12-04:
Hello @popcornmix @pelwell,

I was wondering if there is a way to "explode" the global merge patches like :

Merge remote-tracking branch 'stable/linux-4.19.y' into rpi-4.19.y

into the single ones, so to have a clear view of every single step. I am interested in splitting the merge of rpi-4.14.y branch too. Please let me know it there is a git one-liner to achieve the task....

Kind Regards

popcornmix, 2019-12-04:
I don't believe so. You might be better looking at a future branch (e.g. rpi-5.4.y) which is still rebased rather than merged.

popcornmix, 2019-12-04:
Technically something like:
git rebase --onto stable/linux-4.19.y stable/linux-4.19.y rpi-4.19.y

will produce a new tree of stable/linux-4.19.y with the rpi-4.19.y commits rebased onto it, but you are going to have many conflicts that need fixing up (many files have been changed by both upstream and downstream).

artynet, 2019-12-04:
@popcornmix I also thought so....is there any fork of the repository with the rebased patches instead of the global merge ? it would be very helpful to me since I need to tailor that specific releases from time to time and for this reason I won't mint to have all the set of patches....

popcornmix, 2019-12-04:
What you want doesn't exist and is quite a lot of work to create (see my second comment).

artynet, 2019-12-04:
> What you want doesn't exist and is quite a lot of work to create (see my second comment).

ok thank you, for helping me out anyway !

pelwell, 2019-12-04:
If what you want is to convert from:
```
A-B-M1-C-D-M2-E-F
    /      /
 a-b    c-d
```
to:
```
A-B-a-b-C-D-c-d-E-F
```
then it should be possible to write a script which, when pointed at a suitable branch and given a sensible starting point (i.e. a major Linux version - the upstream branches are full of merges, and you don't want to expand them all) does exactly that - following merges and cherry-picking - but it won't be trivial.

ehem, 2019-12-22:
I was attempting to get something similar too.  Plan was to turn everything into a patch series and then rebase onto my favored distribution's kernel.  Then I discovered Debian's 4.19.67 kernel source is pretty well unrelated to the kernel.org 4.19.67 (newest common ancestor was 4.10).

Depending on the goal two approaches present themselves.  First, start with a branch at c5a52cfb34759055912bd95cf6010b094520c94d, search backwards through git's history splitting out the merges from Greg Kroah-Hartman's tree (official Linux).

Another approach is to create a branch at c5a52cfb34759055912bd95cf6010b094520c94d and then rebase onto 312017a460d5ea31d646e7148e400e13db799ddc.  Once that is done every change between your new head and 312017a460d5ea31d646e7148e400e13db799ddc are your patches.  After I filtered out reverts I saw 855 patches.

ehem, 2019-12-22:
(liberal use of `gitk` or another visualizer is quite helpful for this operation)

popcornmix, 2020-05-15:
Closing as it's not something we will do.

closed 2020-05-15 by popcornmix