raspberrypi/linux #3369: RPi4: Viewsonic IFP-7550-2 is blanking when USB for touchscreen is connected

cprezzi, 2019-12-05:
I have a RPi4B (4GB) connected to a Viewsonic IFP7550-2 touchscreen which is working fine at 4k/30Hz. When I connect the USB cable for the touch function, the screen goes blank. When unplugging again, the desktop comes back but the user is logged out and needs to login again.

**To reproduce**
1. Boot Raspbian to desktop with only HDMI connected.
2. Connect touch USB (-> screen blanks immediately)
3. Unplug touch USB (-> screen comes back with login popup)

**Expected behaviour**
Touchscreen schould be usable and not blank the screen.
(When using a Win10 PC instead of the Pi, everything works as expected!)

**System**
```
pi@raspi-ifp7550:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
pi@raspi-ifp7550:~ $ cat /etc/rpi-issue
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5
pi@raspi-ifp7550:~ $ vcgencmd version
Sep  9 2019 18:49:23
Copyright (c) 2012 Broadcom
version a62a87c77be35ef577563092923cf86e3d8b9408 (clean) (release) (start_db)
pi@raspi-ifp7550:~ $ uname -a
Linux raspi-ifp7550 4.19.71-v7l+ #1262 SMP Mon Sep 9 18:36:37 BST 2019 armv7l GNU/Linux
```

pelwell, 2019-12-05:
Is it possible that the Pi4 has rebooted, rather than just logging out the user? Boot the Pi, wait a minute or two, then plug in the touchscreen. When you unplug it and log in, the value `uptime` reports will tell you whether or not it has rebooted.

cprezzi, 2019-12-05:
I am booting directly to the desktop with auto login, which means a reboot would not show the login dialog. Also, I can connect the USB for only a few seconds and the desktop with the login dialog comes back right after that. During this period, an SSH connection does not disconnect either.

JamesH65, 2019-12-05:
Weird one. Can you have the Pi plugged in to a different display when you plug in the USB to see what it might be doing? Does that display blank or does it just go to the login? Having "dmesg -w" running in a SSH session may also report something perhaps?

cprezzi, 2019-12-05:
Same happens when HDMI is instead connected to a PC monitor. Screen goes blank (with one unserline cursor on top left) and comes back when USB is unplugged.

timg236, 2019-12-05:
What happens if you attach a USB keyboard in this state and press CTRL-ALT-F1 .. FN to switch consoles?
The flashing cursor indicates that this isn't a display problem but USB is causing Raspbian to switch the console for X desktop for some unknown reason.

cprezzi, 2019-12-05:
Now that you say, I realized that the USB keyboard is blocked while the touch is connected (no key working and num led turns off) and it comes back when touch is disconnected.
I will test if I can run dmesg while touch is connected.

pelwell, 2019-12-05:
You can run dmesg after you've unplugged the touch screen - there's a large buffer to hold the kernel messages.

cprezzi, 2019-12-05:
Ok, I will do that.

cprezzi, 2019-12-06:
Ok, here is the "dmesg -w" trace from the time I connected the touch USB untill disconnect:
```
[   38.681709] usb 1-1.3: new high-speed USB device number 5 using xhci_hcd
[   38.812314] usb 1-1.3: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[   38.812331] usb 1-1.3: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[   38.812344] usb 1-1.3: Product: USB 2.0 Hub
[   38.814275] hub 1-1.3:1.0: USB hub found
[   38.814434] hub 1-1.3:1.0: 4 ports detected
[   39.231663] usb 1-1.3.4: new high-speed USB device number 6 using xhci_hcd
[   39.462040] usb 1-1.3.4: New USB device found, idVendor=1a40, idProduct=0101, bcdDevice= 1.11
[   39.462049] usb 1-1.3.4: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[   39.462054] usb 1-1.3.4: Product: USB 2.0 Hub
[   39.464491] hub 1-1.3.4:1.0: USB hub found
[   39.464566] hub 1-1.3.4:1.0: 4 ports detected
[   39.861711] usb 1-1.3.3: new full-speed USB device number 7 using xhci_hcd
[   40.099064] usb 1-1.3.3: New USB device found, idVendor=1ff7, idProduct=0f21, bcdDevice= 3.00
[   40.099084] usb 1-1.3.3: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[   40.099097] usb 1-1.3.3: Product: Touch Device
[   40.099110] usb 1-1.3.3: Manufacturer: CVTE
[   40.099121] usb 1-1.3.3: SerialNumber: 00000000070B
[   40.123288] input: CVTE Touch Device Mouse as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/0003:1FF7:0F21.0003/input/input2
[   40.123528] input: CVTE Touch Device Touchscreen as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/0003:1FF7:0F21.0003/input/input4
[   40.123809] hid-generic 0003:1FF7:0F21.0003: input,hiddev96,hidraw0: USB HID v1.11 Mouse [CVTE Touch Device] on usb-0000:01:00.0-1.3.3/input0
[   40.127120] input: CVTE Touch Device as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.1/0003:1FF7:0F21.0004/input/input5
[   40.127373] hid-generic 0003:1FF7:0F21.0004: input,hidraw2: USB HID v1.10 Device [CVTE Touch Device] on usb-0000:01:00.0-1.3.3/input1
[   40.133456] input: CVTE Touch Device as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.2/0003:1FF7:0F21.0005/input/input6
[   40.133692] input: CVTE Touch Device Touchscreen as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.2/0003:1FF7:0F21.0005/input/input7
[   40.133889] hid-generic 0003:1FF7:0F21.0005: input,hidraw3: USB HID v1.10 Device [CVTE Touch Device] on usb-0000:01:00.0-1.3.3/input2
[   40.137962] input: CVTE Touch Device as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.3/0003:1FF7:0F21.0006/input/input8
[   40.201838] hid-generic 0003:1FF7:0F21.0006: input,hidraw4: USB HID v1.11 Keyboard [CVTE Touch Device] on usb-0000:01:00.0-1.3.3/input3
[   40.446300] input: CVTE Touch Device as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.0/0003:1FF7:0F21.0003/input/input12
[   40.446638] hid-multitouch 0003:1FF7:0F21.0003: input,hiddev96,hidraw0: USB HID v1.11 Mouse [CVTE Touch Device] on usb-0000:01:00.0-1.3.3/input0
[   40.685740] input: CVTE Touch Device Pen as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.2/0003:1FF7:0F21.0005/input/input13
[   40.686455] input: CVTE Touch Device as /devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3.3/1-1.3.3:1.2/0003:1FF7:0F21.0005/input/input14
[   40.687533] hid-multitouch 0003:1FF7:0F21.0005: input,hidraw3: USB HID v1.10 Device [CVTE Touch Device] on usb-0000:01:00.0-1.3.3/input2
[   46.525791] usb 1-1.3: USB disconnect, device number 5
[   46.525800] usb 1-1.3.3: USB disconnect, device number 7
[   47.173266] usb 1-1.3.4: USB disconnect, device number 6
```

cprezzi, 2019-12-06:
Tell me, if you need the full trace (fromm boot).

cprezzi, 2020-02-13:
I was able to get the touchscreen working by changing to Gentoo Linux (https://github.com/sakaki-/gentoo-on-rpi-64bit). This tells me that the problem is located inside Raspbian.