raspberrypi/linux #3372: gpio-poweroff device tree overlay causes kernel panic on RPi 4

DivingIvan, 2019-12-06:
Adding the gpio-poweroff device tree overlay to the /boot/config.txt file results in a kernel panic when the Pi is shutdown using 'sudo shutdown -h 0'.

I have added the following 2 lines to the /boot/config.txt file:-

**dtoverlay=gpio-poweroff,gpiopin=18,active_low=0
dtoverlay=gpio-shutdown,gpio_pin=17,active_low=1,gpio_pull=up**

When the Pi is shutdown, a kernel panic occurs, and the Pi will not fully power down.

Removing the gpio-poweroff device tree overlay from the /boot/config.txt results in the Pi shutting down correctly and fulling powering off.

Model - Raspberry Pi 4 - 2GB

OS

Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5

Firmware version

Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)

Kernel version

Linux raspberrypi 4.19.75-v7l+ raspberrypi/firmware#1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux

pelwell, 2019-12-06:
[ Transferred to the linux repo, where it belongs. ]

Use of this overlay (and the gpio-poweroff mechanism) required that there is an external mechanism to remove the power in response to the GPIO being toggled. Failure to do so within the allowed window (3 seconds is the default) results in a kernel warning, followed by undefined behaviour.

Also, enabling this overlay prevents the usual reset and shut down from working - it absolutely requires that something switches off the power.

In other words, I think the overlay is behaving exactly as it should (but perhaps not how you want).

BHaberland, 2020-01-03:
Thanks for the explanation pelwell. I'm observing same as DivingIvan. 
The power would be turned off if the specified pin would be toggled. 
I have several RPi3B+ setups working fine using gpio-poweroff,gpiopin=4,active_low=1
The RPi4 is not pulling down gpiopin 4 or 26 (default for gpio-poweroff) or 18 (as DivingIvan)

pelwell, 2020-01-06:
@BHaberland Should I assume from the fact that you deleted your comment that you have resolved your problem? I've just tested gpio-poweroff on pin 4 (active_low=1) and pin 18 (active_low=0), and in both cases the GPIO was waggled as expected.

BHaberland, 2020-01-06:
Hi Phil,

Yes, the issue is resolved and was entirely on my side: The pullup of 1M 
(brown black green) was actually 15 Ohms (brown green black). Horrible 
mistake of mine.

Thanks for checking back,
Bert.

pelwell, 2020-01-06:
Now _that's_ what I call a pull-up!

amahoneyLIT, 2020-02-07:
Hi @pelwell, I was thinking to use gpio-poweroff such that a person would be instructed to switch off power with a rocker switch once it is safe to shut down (an LED is turned on with the specifed gpio pin). The 3 second delay causes a kernel panic (which I'm not sure if it matters; the LED stays on), so I thought I would use the `timeout-ms` parameter to make this long (e.g. 30 sec, or 300 sec) 

I see `timeout-ms` it documented here
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/Documentation/devicetree/bindings/power/reset/gpio-poweroff.txt

and the driver has a function to set it
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/power/reset/gpio-poweroff.c#L72
>	device_property_read_u32(&pdev->dev, "timeout-ms", &timeout);

but the overlay doesn't seem to have it
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/gpio-poweroff-overlay.dts

and not reflected in this doc
https://github.com/raspberrypi/firmware/blob/master/boot/overlays/README

Is there currently a way to use this feature? Thanks!

amahoneyLIT, 2020-02-07:
@pelwell  I tried your gpio-poweroff.dtbo file from here https://github.com/raspberrypi/firmware/issues/1069#issuecomment-438614600

I had to change the param from `timeout-ms` to `timeout_ms` but it works!

It would be great to see what the .dts file looks like so I could compile it myself too (and maybe learn more about overlays :)

pelwell, 2020-02-07:
@amahoneyLIT This is one of those situations where you throw together a new feature, make a version available for testing, and then it gets forgotten when there is no feedback.

The updated overlay is now in rpi-4.19.y - see here: https://github.com/raspberrypi/linux/blob/rpi-4.19.y/arch/arm/boot/dts/overlays/gpio-poweroff-overlay.dts

Thanks for the nudge.

JamesH65, 2020-02-13:
Can this be closed?

DivingIvan, 2020-02-14:
Hi James. Yes, this can now be closed. It was clearly my misunderstanding of how this is supposed to work, and I can see that it does indeed work as designed.

closed 2020-02-14 by pelwell