raspberrypi/linux #3377: Kernel 5.4.3 Compile issue

Dark-Sky, 2019-12-15:
5.4.3 Kernel fails to compile. 
Cross compile with with gcc 9.2
 
Steps:
git clone -b rpi-5.4.y --single-branch https://github.com/raspberrypi/linux.git
make -j8 ARCH=arm CROSS_COMPILE=arm-none-eabi- bcm2709_defconfig
make -j8 ARCH=arm CROSS_COMPILE=arm-none-eabi- menuconfig
make -j8 ARCH=arm CROSS_COMPILE=arm-none-eabi
 
It looks like it compiles ok until it gets to the end of stage 1 then I get this warning and it looks like it repeats for every module.
 
```
  GEN     .version
  CHK     include/generated/compile.h
  LD      vmlinux.o
  MODPOST vmlinux.o
WARNING: vmlinux.o (.noinit): unexpected non-allocatable section.
Did you forget to use "ax"/"aw" in a .S file?
Note that for example <linux/init.h> contains
section definitions for use in .S files.
 
  MODINFO modules.builtin.modinfo
  LD      .tmp_vmlinux1
  KSYM    .tmp_kallsyms1.o
  LD      .tmp_vmlinux2
  KSYM    .tmp_kallsyms2.o
  LD      vmlinux
```

Dark-Sky, 2019-12-18:
I cross compiled this today with gcc 7.3.1 and it compiled ok.

popcornmix, 2019-12-18:
Compiling okay for me (on 4.9.3).

Dark-Sky, 2019-12-18:
After I used another cross compiler I looked at my arch linux cross compiler in the repo and it also had arm-none-eabi-newlib installed. I remember way back installing it because after installing arm-none-eabi-gcc pacman recommended it in the post install recommends. Did not install it before that. Any way I am recompiling with arm-none-eabi-newlib removed and see what it does.

Dark-Sky, 2019-12-18:
It now cross compiles just fine using gcc 9.2 with arm-none-eabi-newlib removed. I guess I will not pay any attention to post package install recommends in the future. The problem was on my end.

closed 2019-12-18 by Dark-Sky

Dark-Sky, 2019-12-19:
@popcornmix I am really sorry to bother you with this again but I closed this issue too soon. The issue still exists trying to compile with gcc9. I guess when I recompiled yesterday I did it in the same terminal where I had exported the gcc 7.3.1 toolchain path. I am not used to exporting a path with the toolchain in the manjaro arm repo.

Here is what I know experimenting if it will help any:

gcc 9.2.0 will compile for aarch64 on this kernel but not 32 bit
gcc 8.3.1 is the last gcc version I have found that will compile 32 bit.

Again I am sorry for bugging you again on this.

popcornmix, 2019-12-19:
could you try building linux-5.4.y or linux-stable tree?
I don't imagine this issue has anything to do with the raspberry pi kernel, but is some incompatibility between upstream linux-5.4.y and gcc 9.2.0.

If it works with upstream linux tree we can look further at any downstream patches.

Dark-Sky, 2019-12-19:
The RPi linux-5.4.y is what I was doing in this thread. I am in the middle of another project right now but will give upstream a try later on today.

Thanks

Dark-Sky, 2019-12-19:
@popcornmix 

This is the rest of all I know:

I dowloaded the same gcc 9.2 toolchain version from developers.arm.com and it does the same trying to compile 32 bit with the warnings.

I looked back the last time I compiled for 32 bit and it was gcc 9.1 but that could or not could mean anything if the code has changed it is compiling.

I cloned upstream and checked out the v5.4 branch and compiled their master branch 5.5.0-rc2 and both branches did the same thing with the warnings. Intresting though they did not error out and finished compiling. Here is all the modules that gave the warnings when compiling upstream. I do not know but seems like there could be more with your tree. For the sake of brevity in posting I left out the  text after each warning. They were all the same as the text following the "WARNING: vmlinux.o" at the top: (Did you forget to use "ax"/"aw" in a .S file?)

```
  MODPOST vmlinux.o
WARNING: vmlinux.o (.noinit): unexpected non-allocatable section.
Did you forget to use "ax"/"aw" in a .S file?
Note that for example <linux/init.h> contains
section definitions for use in .S files.

  Building modules, stage 2.
  Kernel: arch/arm/boot/Image is ready
WARNING: fs/configfs/configfs.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/char/hw_random/rng-core.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/net/wireless/broadcom/brcm80211/brcmutil/brcmutil.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/net/wireless/broadcom/brcm80211/brcmfmac/brcmfmac.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/function/usb_f_acm.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/function/usb_f_serial.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/function/usb_f_obex.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/function/usb_f_ecm.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/function/usb_f_eem.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/function/usb_f_ecm_subset.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/function/usb_f_rndis.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/legacy/g_ether.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/legacy/g_serial.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/usb/gadget/libcomposite.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/media/common/videobuf2/videobuf2-common.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/bluetooth/hci_uart.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/staging/vc04_services/vchiq.o (.noinit): unexpected non-allocatable section.
WARNING: drivers/staging/vc04_services/bcm2835-camera/bcm2835-v4l2.o (.noinit): unexpected non-allocatable section.
  CC [M]  drivers/bluetooth/btbcm.mod.o
  CC [M]  drivers/char/hw_random/bcm2835-rng.mod.o
```