raspberrypi/linux #3380: System halt interrupted with ovelay gpio-poweroff enabled ?

bk4nt, 2019-12-17:
Hello,

I've an OLED attached and a corresponding service which nicely blanks that display and exits on a init 0 or init 6.

When I add `dtoverlay gpio-poweroff gpiopin=22 active_low` my OLED display isn't blanked anymore on shutdown or init 0. So I assume that the halt scripts are interrupted by a too early halt of the CPU when this feature is activated.

I saw also some differences on the green Act LED. Where I do see much more blinks after an init 0 when I do not activate gpio-poweroff feature. With gpio-poweroff active, I see only some short Act LED activities.

My setup seems correct. I added a LED on the gpio-poweroff GPIO, which turns on (low pin state) for an init 0. The line I did add to` /boot/config.txt` was:

`dtoverlay=gpio-poweroff,gpiopin=22,active_low`

And I'm using a basic install:

```pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"
```

BR

pelwell, 2019-12-17:
Unless you have external hardware to remove the power supply within three seconds of it toggling the GPIO, don't use gpio-poweroff.

bk4nt, 2019-12-17:
That is the point, I wanted to add extra hardware for full power off.

But when I enable the feature, I see / it seems halt scripts aren't fully executed anymore on init 0.

pelwell, 2019-12-18:
In a normal reboot or shutdown, the ARMs trigger a watchdog reset, having saved just enough state to determine the cause of the reset. The firmware then decides whether this is a poweroff or a reboot, flashing the ACT LED for a poweroff.

By enabling the gpio-poweroff module via the overlay you are replacing the usual watchdog reset -> power off mechanism with a GPIO signal and a pause, the idea being to indicate when the kernel is at a safe point to switch off. If execution continued then the system could be in an indeterminate state when the power was removed, so this is the correct behaviour.

bk4nt, 2019-12-18:
But it looks looks like the system runs longer, performs and ends more tasks for init 0 when gpio-poweroff feature isn't set.

If this is expected to work that way, I'll use serial port and TxD LED output signal instead of gpio-poweroff pin to trigger the external power off hardware.

pelwell, 2019-12-18:
I suspect it's the watchdog reset rather than any Linux code that triggers the display to blank.

bk4nt, 2019-12-18:
I do think the gpio-poweroff feature kicks in to early, whilst the Pi still is active, in the shutdown process.

I've now used an oscilloscope, to better see what happens. My OLED in a custom but basic addon, on SPI port, with a RESET pin via GPIO 25. When I do stop what drives my OLED, I see the RESET pin or GPIO 25 going down, this clearing the display.

My OLED driver is loaded and killed via a basic /etc/init.d script. By the way, on either init 0 or init 6, I do also see GPIO 25 going down during Linux shutdown process, and my OLED display gets so cleared before halt or reboot.

But if I add gpio-poweroff feature, on init 0, GPIO 25 doesn't go down, and by the way, my OLED display isn't cleared. So I do think the feature kicks in and stops the CPU before all the processes had a chance to exit on shutdown.

bk4nt, 2019-12-18:
I had now a look at some kernel codes. Seems kernel_power_off(void) calls machine_power_off(); (setting the GPIO poweroff pin state) after kernel_shutdown_prepare(SYSTEM_POWER_OFF); only...

https://github.com/raspberrypi/linux/blob/e2d2941326922b63d722ebc46520c3a2287b675f/kernel/reboot.c#L284

Then I connected a monitor to my Pi, I was using it headless via ssh only. An now, I see there that I get each time a kernel panic during shutdown process if I do add the gpio-poweroff feature (using `sudo dtoverlay gpio-poweroff gpiopin=22 active_low`). init 6 is ok, init 0 crashes on its way:

https://github.com/bk4nt/tests/blob/master/gpio-poweroff/DSC_1379.small.JPG

pelwell, 2019-12-18:
It only "panics" (actually it's only a warning) because you haven't turned off the power in the configured period - that is how gpio-poweroff works. You'll have to find something else if it isn't exactly what you want.

bk4nt, 2019-12-18:
Ok... so this panic is also expected.

But then why doesn't it allow all the running processes (including that for my OLED display) to exit cleanly before the CPU is stopped, like the system does for common halt? This is confusing then.

pelwell, 2019-12-19:
You'll need to provide a way of demonstrating that shutdown scripts are run with "sudo poweroff", but not if the gpio-poweroff overlay is used.

bk4nt, 2019-12-20:
Hello, issue might be closed.

I still don't know why my OLED driver behaves differently with or without the gpio-power overlay (seems to be related to any Python and Linux internals on shutdown...). But investigatng further, I found a fix, on my processes side.

The basic code I used, with which Linux shutdow gets delayed up to a very long delay, untill the process ends; for this, I had to add additionnal handler for SIGHUP:

```#!/usr/bin/env python

import RPi.GPIO as GPIO
import signal
import sys
import time

GPIO_LED   = 20

def signal_handler(sig, frame):
        GPIO.output(GPIO_LED, GPIO.HIGH)
        for y in range(0, int(10.0/0.05)):
                GPIO.output(GPIO_LED, GPIO.LOW)
                time.sleep(0.05)
                GPIO.output(GPIO_LED, GPIO.HIGH)
                time.sleep(0.05)
        GPIO.cleanup()
        sys.exit(0)

signal.signal(signal.SIGINT, signal_handler)
signal.signal(signal.SIGTERM, signal_handler)
signal.signal(signal.SIGHUP, signal_handler)

GPIO.setmode(GPIO.BCM)

GPIO.setup(GPIO_LED, GPIO.OUT)
GPIO.output(GPIO_LED, GPIO.HIGH)

while True:
        GPIO.output(GPIO_LED, GPIO.LOW)
        time.sleep(1)
        GPIO.output(GPIO_LED, GPIO.HIGH)
        time.sleep(1)
```

So I added now SIGHUP handlers in some custom scripts I use as services, and it works now like expected (gracefull shutdown).

---8<----------

I would now have an additionnal question, about the power GPIO signal itself. It seems a 100ms pulse is sent out if power down shall be reached, and that this pulse isn't sent out if the box should go to reboot. This can be noticed with a LED connected to that GPIO pin. But this isn't described so in Pi documentation, where I read mostly " a reboot will also cause the pin to go low":
```
Name:   gpio-poweroff
Info:   Drives a GPIO high or low on poweroff (including halt). Enabling this
        overlay will prevent the ability to boot by driving GPIO3 low.
Load:   dtoverlay=gpio-poweroff,<param>=<val>
Params: gpiopin                 GPIO for signalling (default 26)

        active_low              Set if the power control device requires a
                                high->low transition to trigger a power-down.
                                Note that this will require the support of a
                                custom dt-blob.bin to prevent a power-down
                                during the boot process, and that a reboot
                                will also cause the pin to go low.
```
https://raw.githubusercontent.com/raspberrypi/firmware/master/boot/overlays/README

The kernel documentation mentions that 100ms pulse activity for powerdown. So I assume that this 100ms pulse only is the trigger for power down, correct?

```
When the power-off handler is called, the gpio is configured as an
output, and drive active, so triggering a level triggered power off
condition. This will also cause an inactive->active edge condition, so
triggering positive edge triggered power off. After a delay of 100ms,
the GPIO is set to inactive, thus causing an active->inactive edge,
triggering negative edge triggered power off. After another 100ms
delay the GPIO is driver active again. If the power is still on and
the CPU still running after a 3000ms delay, a WARN_ON(1) is emitted.

```
https://www.kernel.org/doc/Documentation/devicetree/bindings/power/reset/gpio-poweroff.txt

pelwell, 2019-12-20:
The trigger is any change in the pin. The on-off-on sequence means that both edge -sensitive and level-sensitive circuits will be triggered.

bk4nt, 2019-12-20:
But a change (single change) occurs also for a reboot, for which I wouldn't expect the power to be cut...

I think I'll use and detect the on-off-on sequence for power off, as it doesn't exist for reboot.

pelwell, 2019-12-20:
If you don't want a glitch on reboot, do what the overlay README says and don't use active_low.