raspberrypi/linux #3386: Raspberry Pi 3+ problems with USB control transfers with ~32 bytes or more.

rafaelmartins, 2019-12-28:
**Describe the bug**
When issuing USB control transfers to low speed devices (`USBDEVFS_CONTROL` ioctl), the raspberry pi will usually hang if the data to write or read is around 32 bytes or more.

```
[ 3119.616154] usb 1-1.2: usbfs: USBDEVFS_CONTROL failed cmd dwtk rqt 192 rq 82 len 32 ret -110
```

The `-110` error is a timeout, but I debugged the USB device, and it is not getting any data. This is similar to an old gphoto2 issue I found in google, from long time ago: https://www.raspberrypi.org/forums/viewtopic.php?p=219980&sid=5fcfbe923c93bae7eba181deacc15f0f#p219980

It works totally fine with `dwc2` driver instead of `dwc_otg`, then it is probably a regression in your custom driver.

All of the logs below were produced with `rpi-update`d firmware:
```
 *** If no errors appeared, your firmware was successfully updated to 57d3b3ac6e893c8545b830a866091cdf8daf4961
```

**To reproduce**
I don't have easy steps to reproduce, because I found this bug while developing custom firmware, but I have `strace` logs, that you can find below.

**Expected behaviour**
This `strace` log snippet was generated using `dwc2`
```
futex(0x3e76d8, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x3e7640, FUTEX_WAKE_PRIVATE, 1)  = 1
ioctl(3, USBDEVFS_CONTROL, 0x1c34d64)   = 0
futex(0x3e76d8, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x3e7640, FUTEX_WAKE_PRIVATE, 1)  = 1
ioctl(3, USBDEVFS_CONTROL, 0x1c34d34)   = 1
futex(0x3e76d8, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x3e7640, FUTEX_WAKE_PRIVATE, 1)  = 1
write(2, "Verifying page 0x0000 (1/64) ..."..., 33) = 33
ioctl(3, USBDEVFS_CONTROL, 0x1c34d88)   = 32
futex(0x3e76d8, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x3e7640, FUTEX_WAKE_PRIVATE, 1)  = 1
ioctl(3, USBDEVFS_CONTROL, 0x1c34d58)   = 1
write(2, "Verifying page 0x0020 (2/64) ..."..., 33) = 33
ioctl(3, USBDEVFS_CONTROL, 0x1c34d88)   = 32
ioctl(3, USBDEVFS_CONTROL, 0x1c34d58)   = 1
```
This should be clear enough for someone with some USB knowledge, but I can provide full logs if needed, they are mostly a repetition of this.

**Actual behaviour**
This `strace` log snippet was generated using `dwc_otg`
```
ioctl(3, USBDEVFS_CONTROL, 0x2434d64)   = 0
futex(0x3e76d8, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x3e7640, FUTEX_WAKE_PRIVATE, 1)  = 1
ioctl(3, USBDEVFS_CONTROL, 0x2434d34)   = 1
futex(0x3e76d8, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x3e7640, FUTEX_WAKE_PRIVATE, 1)  = 1
write(2, "Verifying page 0x0000 (1/64) ..."..., 33) = 33
futex(0x3e76d8, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x3e7640, FUTEX_WAKE_PRIVATE, 1)  = 1
ioctl(3, USBDEVFS_CONTROL, 0x2434d88)   = -1 ETIMEDOUT (Connection timed out)
futex(0x3e76d8, FUTEX_WAKE_PRIVATE, 1)  = 1
futex(0x3e7640, FUTEX_WAKE_PRIVATE, 1)  = 1
mmap2(NULL, 262144, PROT_READ|PROT_WRITE, MAP_PRIVATE|MAP_ANONYMOUS, -1, 0) = 0x66306000
write(2, "Error: usb: connection timed out"..., 33) = 33
exit_group(1)                           = ?
+++ exited with 1 +++
```

Producing the error message:
```
[ 3119.616154] usb 1-1.2: usbfs: USBDEVFS_CONTROL failed cmd dwtk rqt 192 rq 82 len 32 ret -110
```

**System**
`raspinfo` output without `dwc2` enabled:
https://gist.github.com/rafaelmartins/d8f5e41a911d6091627979b8122565fd

**Additional context**
I verified the expected behaviour with other hardware, mostly amd64, to verify that the custom hardware firmware I'm developing isn't the culprit. As it works fine with dwc2, I'd say that it should be a regression.

Thanks.

P33M, 2019-12-28:
Please also add a full `sudo lsusb -v`.

Low-speed devices have a maximum packet size of 8 bytes. Clearly, multi-stage control transfers work because there's no issue reading back HID descriptors from keyboards/mice and these are >32 bytes.

What API call are you using to do the control transfer, and can you post a code example?

rafaelmartins, 2019-12-28:
> Please also add a full `sudo lsusb -v`.

https://gist.github.com/rafaelmartins/f2463aee873dce32dc6e51a7058d0b20

> Low-speed devices have a maximum packet size of 8 bytes. The data phase (in this case device to host) is split into multiple 8-byte transfers for larger control requests. Clearly, multi-stage control transfers work because there's no issue reading back HID descriptors from keyboards/mice and these are >32 bytes.

You are correct, but the USB device isn't getting even the first 8 bytes packet in the data phase. unfortunately my device either gets requests for 2, 4 bytes or >=32 bytes, then I don't know exactly which length causes the issue, but I'd say something around 8 or 9 bytes. I tried a lot of things, like claiming interface (which is not supposed to be required for raw control transfers), but in fact it works pretty well with `dwc2`.

> 
> What API call are you using to do the control transfer, and can you post a code example?

I'm using raw `ioctl`s: https://github.com/rafaelmartins/dwtk/blob/master/internal/usbfs/usbfs.go#L155-L183

I could also reproduce this problem using libusb.

rafaelmartins, 2019-12-28:
As this is not an HID device, maybe the `dwc_otg` driver was optimized for that, e.g. by expecting the sequence of requests that HID devices send, and getting confused when these raw, simple requests are found. I do not looked at `dwc_otg` code yet.

If there's a way I could bisect this issue, please let me know.

P33M, 2019-12-28:
In the first instance, the line
`[ 3119.616154] usb 1-1.2: usbfs: USBDEVFS_CONTROL failed cmd dwtk rqt 192 rq 82 len 32 ret -110`
(bmRequestType=0xc0) implies that this is a device-to-host transfer that isn't completing.

- Is the behaviour the same for device_to_host as well as host_to_device?
- Does the device always see the SETUP packet regardless of the size requested?
- Can your device count data stage requests, e.g. if two IN tokens were issued can it distinguish between them?
- What range of request lengths work and which don't - can you step in byte sizes from 1-32 on inbound and outbound control transfers?

rafaelmartins, 2019-12-28:
I'll need to write some test code to issue these requests. Will report results here when done.

rafaelmartins, 2019-12-28:
> In the first instance, the line
> `[ 3119.616154] usb 1-1.2: usbfs: USBDEVFS_CONTROL failed cmd dwtk rqt 192 rq 82 len 32 ret -110`
> (bmRequestType=0xc0) implies that this is a device-to-host transfer that isn't completing.
> 
>     * Is the behaviour the same for device_to_host as well as host_to_device?

No, it only happens on device to host (IN).

>     * Does the device always see the SETUP packet regardless of the size requested?

Yes, snippet from gdb, trying to read 32 bytes:

```
GNU gdb (crosstool-NG 1.24.0) 8.2.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_pc-linux-gnu --target=avr".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://github.com/rafaelmartins/avr-toolchain>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from firmware.elf...done.
(gdb) target remote :8000
Remote debugging using :8000
0x00000000 in __vectors ()
(gdb) hbreak usbFunctionRead 
Hardware assisted breakpoint 1 at 0x496: file main.c, line 373.
(gdb) c
Continuing.

Breakpoint 1, usbFunctionRead (data=0x800080 <usbTxBuf+1> "", len=8 '\b') at main.c:373
373	{
(gdb) print remaining 
$1 = 32
(gdb) 
```

`remaining` is basically `wLength` from the SETUP packet at this point. However, the host never read the data.

>     * Can your device count data stage requests, e.g. if two IN tokens were issued can it distinguish between them?

Not sure, I'm doing everything synchronously.

>     * What range of request lengths work and which don't - can you step in byte sizes from 1-32 on inbound and outbound control transfers?

It stops working at 12 bytes. Any read from 1 to 11 works fine.

rafaelmartins, 2020-01-02:
After a lot of testing, I found a computer with similar processing power as an rpi and could reproduce the issue, that turned out to be unrelated to the rpi itself. It is a timing issue that only happens when the host takes some time to request the data. I fixed it by adapting my firmware to use some internal buffers, then it should not be a problem anymore.

Feel free to close the issue, unless you want to investigate why `dwc2` works and `dwc_otg` doesn't. Or in other words, why `dwc2` is faster than `dwc_otg`.

Thanks and best regards

P33M, 2020-01-09:
dwc_otg has an artificial rate limit built in to stop flooding the host CPU with interrupts. It will do full/low-speed control and bulk transfers on a SOF boundary so there's at least 125uS of lag between requests.

closed 2020-01-09 by P33M

rafaelmartins, 2020-01-09:
> dwc_otg has an artificial rate limit built in to stop flooding the host CPU with interrupts. It will do full/low-speed control and bulk transfers on a SOF boundary so there's at least 125uS of lag between requests.

this explains it. thanks for the help debugging.