raspberrypi/linux #3389: System crash (Kernel Oops) during USB Bulk Transfer (Read) with Transfer Buffers >16k

localuser2, 2020-01-02:
### Description

A USB bulk read transfer with a transfer buffer size >16 KiB (16384 Bytes) results in frequent system crashes (Kernel Oops). The problem could to be connected to a strange behaviour of the Linux Kernel for ARM regarding USB read transfers: If the transfer buffer exceeds a length of 16KiB the USB read request seems to result always in multiple USB Request Blocks (URB). This is observable via Wireshark on all (for me accessible) ARM platforms. An amd64-Kernel of roughly the same version does not show this behaviour. 

### Steps to reproduce

The crashes seem to be independent from the kind of the USB device or the driver back-end. I was able to reproduce the problem with libusb0 as well as with libusb1 and at least two different USB controllers. However the generation of the RaspberryPi hardware seems to be relevant. We found the problem with a RasperryPi 3 (B+) the most current model RaspberryPi 4 seems to be not affected. To reproduce the crash, you will need some kind of USB hardware to read from. For the example I used a setup of two DIGITUS USB-to-Serial converters and a Null-Modem-Cable:

- connect one side of the Null-Modem-Cable to a Linux-PC
- become root (or set up a udev-rule) to be able to write to the serial interface
- write data to the serial interface (e.g. `while true;do echo BLABLUBB > /dev/ttyUSB0;done`)
- connect the other side of the Null-Modem-Cable to a RaspberryPi 3 (B+?)
- install `gcc` C compiler and package `libusb-1.0-0-dev`
- compile the attached code example: `gcc -lusb-1.0 -o usb-crash usb-crash.c`
- unload the FTDI kernel driver (e.g. `sudo rmmod ftdi-sio`) to be able to open the USB device
- become root (or set up a udev-rule) to be able to read from the serial interface
- execute example program `./usb-crash`

```c
#include <stdio.h>
#include <stdlib.h>

#include <libusb-1.0/libusb.h>

// if this define is changed to 16384 or a lesser value, the crashes disappear
#define BUFLEN 16384+1

static unsigned char buf[BUFLEN];

int main(void)
{
  int result = libusb_init(NULL);
  if (result < 0)
  {
    fprintf(stderr, "failed to initialize libusb\n");
    exit(1);
  }

  // this is the Product and Vendor ID of the DIGITUS USB-to-Serial converter (FTDI-based)
  // however, the problem seems to be reproducible with similar hardware -> edit if necessary
  libusb_device_handle *devh = libusb_open_device_with_vid_pid(NULL, 0x0403, 0x6001);

  if (!devh)
  {
    fprintf(stderr, "failed to open device\n");
    exit(1);
  }

  int actual_length;

  // this loop will crash the raspi with high probability
  for (int ix = 0; ix < 10000; ++ix)
  {
    result = libusb_claim_interface(devh, 0);
    if (result < 0)
    {
      fprintf(stderr, "libusb_usb_claim_interface error %d\n", result);
      exit(1);
    }

    // 129 is the read endpoint for bulk transfers of this device -> edit if necessary
    result = libusb_bulk_transfer(devh, 129, (unsigned char*)&buf, BUFLEN, &actual_length, 1000);
    if (result < 0)
    {
      fprintf(stderr, "libusb_bulk_transfer error %d\n", result);
      libusb_release_interface(devh, 0);
      exit(1);
    }

    printf("Iteration %d: Actually read %d bytes\n", ix, actual_length);
[... 684 lines cut ...]
[    5.964923] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Feb 27 2018 03:15:32 version 7.45.154 (r684107 CY) FWID 01-4fbe0b04
[    6.002760] random: crng init done
[    6.002775] random: 7 urandom warning(s) missed due to ratelimiting
[    7.386880] uart-pl011 3f201000.serial: no DMA platform data
[    7.461254] 8021q: 802.1Q VLAN Support v1.8
[    7.701852] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[    7.729112] IPv6: ADDRCONF(NETDEV_UP): wlan0: link is not ready
[    7.729147] brcmfmac: power management disabled
[    7.966567] IPv6: ADDRCONF(NETDEV_UP): eth0: link is not ready
[    7.966585] 8021q: adding VLAN 0 to HW filter on device eth0
[    7.975401] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   14.276568] Bluetooth: Core ver 2.22
[   14.276665] NET: Registered protocol family 31
[   14.276673] Bluetooth: HCI device and connection manager initialized
[   14.276696] Bluetooth: HCI socket layer initialized
[   14.276709] Bluetooth: L2CAP socket layer initialized
[   14.276750] Bluetooth: SCO socket layer initialized
[   14.297562] Bluetooth: HCI UART driver ver 2.3
[   14.297576] Bluetooth: HCI UART protocol H4 registered
[   14.297661] Bluetooth: HCI UART protocol Three-wire (H5) registered
[   14.297866] Bluetooth: HCI UART protocol Broadcom registered
[   14.544601] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   14.544611] Bluetooth: BNEP filters: protocol multicast
[   14.544631] Bluetooth: BNEP socket layer initialized
```

JamesH65, 2020-01-02:
@P33M This one's for you I think!!

P33M, 2020-01-03:
I can't reproduce. 2000+ iterations and counting. Is the baudrate important?

I strongly recommend getting a full stack trace and dmesg log via the onboard GPIO uart on GPIO 14 and 15 - add the parameters "enable_uart=1" and "dtoverlay=pi3-disable-bt" to /boot/config.txt and reboot. Uart is at 115200 baud by default.

After rebooting, enable kdgb: `echo ttyAMA0,115200 | sudo tee /sys/module/kgdboc/parameters/kgdboc`

When you get a crash you should drop into a kgdb prompt on the uart. Press spacebar to scroll through the stack trace then do "dmesg" and "btc" to capture the full output.

P33M, 2020-01-03:
I do see one odd message in dmesg though:
```
[   31.351907] dwc_otg: DEVICE:004 : update_urb_state_xfer_comp:751:trimming xfer length
```
But this happens only once.

P33M, 2020-01-03:
Aha, after messing with baudrates on both ends I finally get a crash:
```
pi@raspberrypi:~$ [  206.983853] Unable to handle kernel paging request at virtual address 0ee06801
[  206.991082] pgd = c03209db
[  206.993782] [0ee06801] *pgd=00000000
[  206.997357] Internal error: Oops: 5 [#1] SMP ARM
[  207.001967] Modules linked in: fuse 8021q garp stp llc brcmfmac brcmutil sha256_generic usbserial cfg80211 rfkill raspberrypi_hwmon hwmon snd_bcm2835(C) snd_pcm snd_timer snd bcm2835_v4l2(C) bcm2835_codec(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) v4l2_common videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq uio fixed i2c_dev ip_tables x_tables ipv6 [last unloaded: ftdi_sio]
[  207.041921] CPU: 0 PID: 1035 Comm: usb-crash Tainted: G         C        4.19.75-v7+ #1270
[  207.050184] Hardware name: BCM2835
[  207.053597] PC is at __kmalloc+0xac/0x288
[  207.057613] LR is at __kmalloc+0x60/0x288
[  207.061623] pc : [<80294ea0>]    lr : [<80294e54>]    psr: 20000093
[  207.067889] sp : a99d7bb0  ip : a99d7bb0  fp : a99d7bf4
[  207.073113] r10: b9c01e40  r9 : 80657540  r8 : 00488020
[  207.078339] r7 : b9c01e40  r6 : 0000002c  r5 : ad26d240  r4 : 0ee06801
[  207.084866] r3 : 00000000  r2 : 80c97a60  r1 : 396bd000  r0 : a99d7bb0
[  207.091394] Flags: nzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
[  207.098617] Control: 10c5383d  Table: 2984c06a  DAC: 00000055
[  207.104367] Process usb-crash (pid: 1035, stack limit = 0xead25408)
[  207.110637] Stack: (0xa99d7bb0 to 0xa99d8000)
[  207.114999] 7ba0:                                     80d8b568 80d8b588 80d04d70 00020c80
[  207.123182] 7bc0: a99d7c14 a99d7bd0 80294fb4 b96b0000 ad26d240 b97d3020 ad26d240 b9430400
[  207.131365] 7be0: 00000000 00000000 a99d7c04 a99d7bf8 80657540 80294e00 a99d7c1c a99d7c08
[  207.139548] 7c00: 806514e4 80657528 b96b0000 00000001 a99d7c4c a99d7c20 8064a5d4 806514c4
[  207.147731] 7c20: a99d7c78 8061b6ec a99d7c4c b6346480 ad26d240 80d04d48 b96b0000 b9430400
[  207.155913] 7c40: a99d7cac a99d7c50 8064c3c0 8064a588 00004000 00000000 00000000 00000001
[  207.164096] 7c60: 00000000 8083719c 00000000 b97d3000 00000000 b97d3020 60000093 cb577e12
[  207.172281] 7c80: 8061d0d8 b6346480 b9430400 80d04d48 00000000 00480020 b6346488 00000000
[  207.180464] 7ca0: a99d7d5c a99d7cb0 8061d860 8064c1f0 a99d7cd4 80d8bb40 00000000 80d8bb40
[  207.188647] 7cc0: 00000000 00000000 00000000 a99d7cd8 8074f98c 806eefb0 00000000 b936a064
[  207.196829] 7ce0: 00000001 806ed564 00000001 cb577e12 80e6572c a99d7e78 af031f80 00004000
[  207.205012] 7d00: 006000c0 80295024 00000002 00000000 a99d7d2c a99d7d20 8026729c 802420c8
[  207.213195] 7d20: a99d7d5c a99d7d30 80267314 cb577e12 80831d24 b6346480 00480020 00000002
[  207.221376] 7d40: 00000201 b96b3400 00000040 00000000 a99d7da4 a99d7d60 8061f444 8061d794
[  207.229557] 7d60: a99d7d8c a99d7d70 80201fd4 8020304c 00000002 00000000 af031f80 a99d7e78
[  207.237738] 7d80: af031f80 80e3ff68 ad26d940 af031f94 af031f90 00000000 a99d7e1c a99d7da8
[  207.245919] 7da0: 8062bc24 8061f198 a99d7e1c a99d7dc0 802c8998 802c8894 5e0f24c0 00000000
[  207.254100] 7dc0: 20228679 af031f90 00000000 00000000 00000000 00000001 00000000 00000000
[  207.262281] 7de0: 00000000 00000001 006c1108 b97d3000 b96b34ac 80d04d48 b96b3478 b96b34ac
[  207.270462] 7e00: b96b3400 00000009 802c550a af031f80 a99d7ed4 a99d7e20 8062c508 8062add4
[  207.278643] 7e20: ad37f000 00000026 00000000 00000026 a99d6000 a9a91240 a99d7e5c b92ce6f0
[  207.286824] 7e40: 006c1108 80168264 5e0f24c0 00000000 00000000 af384e40 a99d7e8c a99d7e68
[  207.295005] 7e60: 8082ac3c 808234f8 00000007 af384e40 af384e40 00000000 00008103 00000000
[  207.303185] 7e80: 00000001 00021050 00004000 00000000 00000000 00000000 00000000 00000000
[  207.311366] 7ea0: 006bfec8 cb577e12 8019a7d4 80d04d48 006c1108 b92ce6f0 b7e49780 00000009
[  207.319547] 7ec0: a99d6000 00000036 a99d7f6c a99d7ed8 802bfa98 8062bc58 af384e40 802fead4
[  207.327728] 7ee0: 00000001 af384e40 a99d7fa4 a99d7ef8 802fead4 802aca90 00000000 00000000
[  207.335908] 7f00: 00000000 00000026 00000001 00000000 00000000 00000001 00000000 00000000
[  207.344090] 7f20: 00000000 a99d7f30 b7e496c0 802cbd6c 006c1108 006c1108 00000000 cb577e12
[  207.352270] 7f40: a99d7f5c b7e49781 006c1108 b7e49780 802c550a 00000009 a99d6000 00000036
[  207.360452] 7f60: a99d7f94 a99d7f70 802c0224 802bf9e8 006c42f8 006c1108 006bfec8 00000000
[  207.368633] 7f80: 00000036 801011c4 a99d7fa4 a99d7f98 802c0264 802c01ec 00000000 a99d7fa8
[  207.376814] 7fa0: 80101000 802c0258 006c1108 006bfec8 00000009 802c550a 006c1108 006c42f8
[  207.384994] 7fc0: 006c1108 006bfec8 00000000 00000036 00000002 00000001 00000001 00000000
[  207.393175] 7fe0: 76eab0b4 7ea2d5b4 76e92608 76e0851c 60000010 00000009 00000000 00000000
[  207.401374] [<80294ea0>] (__kmalloc) from [<80657540>] (__DWC_ALLOC_ATOMIC+0x24/0x28)
[  207.409217] [<80657540>] (__DWC_ALLOC_ATOMIC) from [<806514e4>] (dwc_otg_hcd_qtd_create+0x2c/0x60)
[  207.418185] [<806514e4>] (dwc_otg_hcd_qtd_create) from [<8064a5d4>] (dwc_otg_hcd_urb_enqueue+0x58/0x204)
[  207.427676] [<8064a5d4>] (dwc_otg_hcd_urb_enqueue) from [<8064c3c0>] (dwc_otg_urb_enqueue+0x1dc/0x384)
[  207.436992] [<8064c3c0>] (dwc_otg_urb_enqueue) from [<8061d860>] (usb_hcd_submit_urb+0xd8/0x988)
[  207.445784] [<8061d860>] (usb_hcd_submit_urb) from [<8061f444>] (usb_submit_urb+0x2b8/0x4f4)
[... cut ...]
[  207.493069] [<802c0264>] (sys_ioctl) from [<80101000>] (ret_fast_syscall+0x0/0x28)
[  207.500636] Exception stack(0xa99d7fa8 to 0xa99d7ff0)
[  207.505691] 7fa0:                   006c1108 006bfec8 00000009 802c550a 006c1108 006c42f8
[  207.513872] 7fc0: 006c1108 006bfec8 00000000 00000036 00000002 00000001 00000001 00000000
[  207.522050] 7fe0: 76eab0b4 7ea2d5b4 76e92608 76e0851c
[  207.527107] Code: e7914002 e3540000 0a000067 e5973014 (e7943003)
[  207.533203] ---[ end trace 9f415fbd16d0a534 ]---
```
Someone's trampling freed memory, I would guess.

localuser2, 2020-01-03:
So it is a kernel bug? Do you have enough information or do I have to provide something else?

P33M, 2020-01-03:
Can you provide your GPIO uart crash logs so I can confirm I have hit the same bug?

localuser2, 2020-01-03:
Here it is (crash, dmesg, btc):
```
[   57.463534] Unable to handle kernel paging request at virtual address fb0070b1
[   57.471709] pgd = 06b30de9
[   57.475317] [fb0070b1] *pgd=00000000
[   57.479785] Internal error: Oops: 5 [#1] SMP ARM
[   58.484380] KGDB: Timed out waiting for secondary CPUs.

Entering kdb (current=0xb1598f00, pid 844) on processor 1 Oops: (null)
due to oops @ 0x80294ea0
CPU: 1 PID: 844 Comm: usb-crash Tainted: G         C        4.19.75-v7+ #1270
Hardware name: BCM2835
PC is at __kmalloc+0xac/0x288
LR is at __kmalloc+0x60/0x288
pc : [<80294ea0>]    lr : [<80294e54>]    psr: a0000093
sp : b165bbb0  ip : b165bbb0  fp : b165bbf4
r10: b9c01e40  r9 : 80657540  r8 : 00488020
r7 : b9c01e40  r6 : 0000002c  r5 : b788d500  r4 : fb0070b1
r3 : 00000000  r2 : 80c97a60  r1 : 396ce000  r0 : b165bbb0
Flags: NzCv  IRQs off  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 10c5383d  Table: 3165c06a  DAC: 00000055
CPU: 1 PID: 844 Comm: usb-crash Tainted: G         C        4.19.75-v7+ #1270
Hardware name: BCM2835
[<80111fcc>] (unwind_backtrace) from [<8010d544>] (show_stack+0x20/0x24)
[<8010d544>] (show_stack) from [<80819bc0>] (dump_stack+0xd4/0x118)
[<80819bc0>] (dump_stack) from [<80109d34>] (show_regs+0x1c/0x20)
[<80109d34>] (show_regs) from [<801da678>] (kdb_main_loop+0x3ec/0x828)
[<801da678>] (kdb_main_loop) from [<801dd6ec>] (kdb_stub+0x240/0x480)
[<801dd6ec>] (kdb_stub) from [<801d2ab8>] (kgdb_cpu_enter+0x4b4/0x868)
[<801d2ab8>] (kgdb_cpu_enter) from [<801d30f0>] (kgdb_handle_exception+0x12c/0x244)
more> 
[<801d30f0>] (kgdb_handle_exception) from [<801114e4>] (kgdb_notify+0x40/0x90)
[<801114e4>] (kgdb_notify) from [<80144078>] (notifier_call_chain+0x54/0x94)
[<80144078>] (notifier_call_chain) from [<801442a0>] (notify_die+0x58/0x78)
[<801442a0>] (notify_die) from [<8010d660>] (die+0x118/0x35c)
[<8010d660>] (die) from [<801168e4>] (__do_kernel_fault.part.0+0x74/0x84)
[<801168e4>] (__do_kernel_fault.part.0) from [<80116684>] (do_sect_fault+0x0/0x20)
[<80116684>] (do_sect_fault) from [<80837900>] (do_translation_fault+0x7c/0xc0)
[<80837900>] (do_translation_fault) from [<801166fc>] (do_DataAbort+0x58/0xf4)
[<801166fc>] (do_DataAbort) from [<80101934>] (__dabt_svc+0x54/0x80)
Exception stack(0xb165bb60 to 0xb165bba8)
bb60: b165bbb0 396ce000 80c97a60 00000000 fb0070b1 b788d500 0000002c b9c01e40
bb80: 00488020 80657540 b9c01e40 b165bbf4 b165bbb0 b165bbb0 80294e54 80294ea0
bba0: a0000093 ffffffff
[<80101934>] (__dabt_svc) from [<80294ea0>] (__kmalloc+0xac/0x288)
[<80294ea0>] (__kmalloc) from [<80657540>] (__DWC_ALLOC_ATOMIC+0x24/0x28)
[<80657540>] (__DWC_ALLOC_ATOMIC) from [<806514e4>] (dwc_otg_hcd_qtd_create+0x2c/0x60)
[<806514e4>] (dwc_otg_hcd_qtd_create) from [<8064a5d4>] (dwc_otg_hcd_urb_enqueue+0x58/0x204)
[<8064a5d4>] (dwc_otg_hcd_urb_enqueue) from [<8064c3c0>] (dwc_otg_urb_enqueue+0x1dc/0x384)
[<8064c3c0>] (dwc_otg_urb_enqueue) from [<8061d860>] (usb_hcd_submit_urb+0xd8/0x988)
more> 
[<8061d860>] (usb_hcd_submit_urb) from [<8061f444>] (usb_submit_urb+0x2b8/0x4f4)
[<8061f444>] (usb_submit_urb) from [<8062bc24>] (proc_do_submiturb+0xe5c/0xe84)
[<8062bc24>] (proc_do_submiturb) from [<8062c508>] (usbdev_ioctl+0x8bc/0x1acc)
[<8062c508>] (usbdev_ioctl) from [<802bfa98>] (do_vfs_ioctl+0xbc/0x804)
[<802bfa98>] (do_vfs_ioctl) from [<802c0224>] (ksys_ioctl+0x44/0x6c)
[<802c0224>] (ksys_ioctl) from [<802c0264>] (sys_ioctl+0x18/0x1c)
[<802c0264>] (sys_ioctl) from [<80101000>] (ret_fast_syscall+0x0/0x28)
Exception stack(0xb165bfa8 to 0xb165bff0)
bfa0:                   018c44c8 018c4440 00000009 802c550a 018c44c8 018d86a8
bfc0: 018c44c8 018c4440 00000000 00000036 00000002 00000001 00000001 00000000
bfe0: 76f270b4 7eebf5b4 76f0e608 76e8451c

[1]kdb> 

[1]kdb> 
[1]kdb> 
[1]kdb> dmesg
<6>[    0.000000] Booting Linux on physical CPU 0x0
<5>[    0.000000] Linux version 4.19.75-v7+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1270 SMP Tue Sep 24 18:45:11 BST 2019
[... 343 lines cut ...]
[<80101934>] (__dabt_svc) from [<80294ea0>] (__kmalloc+0xac/0x288)
[<80294ea0>] (__kmalloc) from [<80657540>] (__DWC_ALLOC_ATOMIC+0x24/0x28)
[<80657540>] (__DWC_ALLOC_ATOMIC) from [<806514e4>] (dwc_otg_hcd_qtd_create+0x2c/0x60)
[<806514e4>] (dwc_otg_hcd_qtd_create) from [<8064a5d4>] (dwc_otg_hcd_urb_enqueue+0x58/0x204)
[<8064a5d4>] (dwc_otg_hcd_urb_enqueue) from [<8064c3c0>] (dwc_otg_urb_enqueue+0x1dc/0x384)
[<8064c3c0>] (dwc_otg_urb_enqueue) from [<8061d860>] (usb_hcd_submit_urb+0xd8/0x988)
[<8061d860>] (usb_hcd_submit_urb) from [<8061f444>] (usb_submit_urb+0x2b8/0x4f4)
[<8061f444>] (usb_submit_urb) from [<8062bc24>] (proc_do_submiturb+0xe5c/0xe84)
[<8062bc24>] (proc_do_submiturb) from [<8062c508>] (usbdev_ioctl+0x8bc/0x1acc)
[<8062c508>] (usbdev_ioctl) from [<802bfa98>] (do_vfs_ioctl+0xbc/0x804)
more> 
[<802bfa98>] (do_vfs_ioctl) from [<802c0224>] (ksys_ioctl+0x44/0x6c)
[<802c0224>] (ksys_ioctl) from [<802c0264>] (sys_ioctl+0x18/0x1c)
[<802c0264>] (sys_ioctl) from [<80101000>] (ret_fast_syscall+0x0/0x28)
Exception stack(0xb165bfa8 to 0xb165bff0)
bfa0:                   018c44c8 018c4440 00000009 802c550a 018c44c8 018d86a8
bfc0: 018c44c8 018c4440 00000000 00000036 00000002 00000001 00000001 00000000
bfe0: 76f270b4 7eebf5b4 76f0e608 76e8451c
Stack traceback for pid 0
0xb9d44b00        0        0  1    2   I  0xb9d45958  swapper/2
Stack traceback for pid 112
0xb8dc6900      112        1  1    3   R  0xb8dc7758  systemd-journal
[<80831314>] (__schedule) from [<801de6a4>] (seccomp_run_filters+0x70/0x16c)
[1]kdb> 
```

P33M, 2020-01-03:
In your wireshark traces, what are the URB buffer sizes? I've added debug to dwc_otg and I only ever see 16kib transfer buffer sizes in the URBs that are given to the HCD.

P33M, 2020-01-03:
Findings so far:
- Slub debugging (redzone+poison) doesn't turn up anything, but since we're usually crashing in kmalloc that doesn't tell us much
- The dereferenced faulting "address" is always present in a captured USB bitstream, so read data from the device is getting written either off the end of a buffer or to a bogus buffer address
- I've not yet narrowed down why we're getting the trim warning, all the crashes I've had since I moved printks around haven't had a trim warning.
- For the crash to occur, "some number of" full-size packets (64 bytes) need to be transferred in a single URB. If I tune the data rate such that you always get short packets, the crash doesn't occur.

What's weird about ftdi devices is that they have a variable data aggregation window, set by the driver, and is dependent on baud rate. This means that the device will rate-limit its returned bulk updates (presumably to reduce IO processing load) so this is why only certain baud rates trigger the crash. What remains to be explained is why only certain sets of returned buffer lengths cause issues - I can transfer thousands of URBs without a crash if we are always returning filled URBs.

localuser2, 2020-01-03:
This is how a transfer of two bytes (reading) looks like with my Raspi 3B+ (stopping after one iteration):

[image]

which is odd, because there are two transfers host->device and two more device->host. So to transfer 2 bytes almost 20kbytes go over the line.

While I was debugging the crashes with our USB devices I found this strange behavior. Here are two more examples to clarify. Please note that this is another USB controller than in the DIGITUS converters.

Here four bytes are transferred on a Linux PC (amd64, Debian 10) with a transfer buffer size of 20kbyte:

[image]

Here is the same transfer with the same buffer size device on my Raspi 3B+:

[image]

On AMD64 only slightly more than 100 bytes go over the line. On Raspi 3B+ its 20k. It is the same on Raspi 4 but without the crashes.

It looks like the 20k buffer is transferred in two fragments whether there are enough data available or not. However, even with an 16kib+1 buffer 20k go over the line...

P33M, 2020-01-03:
usbmon traces capture both the submit() action and the complete() action. So each wire transfer has two points in the trace - the point at which the device driver requests a transfer, and one where the results of the transfer are returned to the driver. 

Requesting a transfer with a buffer of 1 byte from an endpoint that returns 64 byte packets is clearly nonsense - in the first instance we should work out why libusb does this on ARM platforms.

In the majority of cases it looks like dwc_otg is completing the urb with the 1-byte allocation with a zero-byte return length, which I suspect is why I didn't get crashes.

localuser2, 2020-01-03:
I do not think libusb is the problem here. As far as I know it is completely platform independent. All abstractions should be done behind the device paths `/dev/bus/usb` or `/proc/bus/usb`, right?
At least I found no `#ifdef ...arm...` or something like this. Besides, a userland library should never be able to crash the operating system.

P33M, 2020-01-03:
On Pi, something is deciding to split the 16385 byte libusb transfer (as requested by the application) into a 16k transfer buffer and a 1-byte transfer buffer. Similarly for the 20k transfer, you get a 16k transfer and a 4k transfer buffer. 

Whoever's splitting this transfer is not aware of the endpoint maxpacket size, which is a bug - you're giving an inadequately sized buffer to a DMA-capable device. I need to check to see if xhci-hcd outright rejects transfers like these, or if you're just "always lucky" with the Pi 4 controller.

localuser2, 2020-01-06:
On the Raspi it seems to be always two transfers in every direction. Even if the buffer is distinctively smaller than 16k. I tried with 4k. No crashes but four points in the Wireshark trace.

If I understand your last post right, is it possible that the problem also occurs with AMD64 platform if the buffer and the actual data length is greater than 16k? In contrast to the Pi, the URB size seems to be correlated with the actually transmitted data length here.

By the way: I also tried the USB-C interface (OTG) on the Raspi 4. It should be USB 2.0 (EHCI), right? No crashes.

P33M, 2020-01-06:
If I run the same test on a Pi 4 (xhci controller) and use a 16385 byte transfer, the transfer always completes with an error:
```
[216983.911031] usb 1-1.3: userurb ba804927, ep1 bulk-in, length 16385
[216984.114002] usb 1-1.3: urb complete
[216984.114013] usb 1-1.3: userurb ba804927, ep1 bulk-in, actual_length 16384 status -75
```
But, the submission was with a buffer size of 16385 bytes instead of 16k/1.

If I use a transfer size that's not a multiple of 64 and if an "overrun" condition occurs, the driver always returns -75 (EOVERFLOW -> device babble error). The URB is never split, even if I request a ridiculous buffer size like 1MiB+1.

I would say from a controller driver standpoint that a babble error is the correct response in this situation - USB bulk transfers don't *require* max packet sized transfers, and some upper protocol could want to transfer e.g. 16385 bytes exactly.

So, what's splitting the dwc_otg transfer into two URBs, and in the case where we could write off the end of a buffer (transfer space remaining < wMaxPacketSize) we should guard this with an overrun.

localuser2, 2020-01-06:
I can confirm that dwc2 is loaded on my Raspi 4. A module with name dwc-otg seems not to be existent. If dwc2 is unloaded, the USB hardware on USB-C port is not accessible anymore (lsusb).

P33M, 2020-01-06:
In the dwc2 case I see the same behaviour as dwc_otg - 16384 + 1 byte split URBs, plus multiple occurences of the message
```
[  455.395114] dwc2 3f980000.usb: dwc2_update_urb_state(): trimming xfer length
```
Which does an equivalent length munging operation to the one seen in dwc_otg.

P33M, 2020-01-06:
There's a bug in both dwc_otg and dwc2. In an IN split transaction, the host channel is set up for the max packet size of the endpoint because multiple individual host channel operations are required to do a split transaction. This unconditional overwrite has the side-effect of ignoring the remaining space in the URB's buffer for bulk, interrupt and control transfers (isoc is treated as an array of maxpacket-sized transfers).

https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/usb/dwc2/hcd.c#L1466
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/usb/host/dwc_otg/dwc_otg_hcd.c#L1861

For dwc2 and dwc_otg, the transfer length is correctly calculated (or at least constrained) at host channel init time.
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/usb/dwc2/hcd.c#L2842
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/usb/host/dwc_otg/dwc_otg_hcd.c#L1254

P33M, 2020-01-06:
For the other half of the unexplained behaviour - neither version of the dwc driver supports scatter-gather (sg_tablesize = 0). This means the flag USBDEVFS_CAP_BULK_SCATTER_GATHER isn't set when querying ioctl(..., USBDEVFS_GET_CAPABILITIES, ...), so libusb is submitting URBs split on a 16k boundary. 16k is a historical "limit" for the maximum size of a single user-initiated USB bulk transfer.

The reason that I saw the second 1-byte URB being returned with 0 bytes transferred is because it's marked as a "continuation" URB by usbfs, so when a short transfer is received on the first URB, the second URB is automatically unlinked before returning the results to userspace.

I have a tentative fix (which results in dwc_otg having the same error behaviour as xhci if the buffer gets overrun) but it's a bit hacky. The OTG core requires the transfer length register to be an integer multiple of the maximum packet size for the endpoint or you end up with a bogus length register on transfer completion, so I have to mangle both the transfer length and endpoint maxpacket fields in order to generate a device babble error.

localuser2, 2020-01-06:
So a buffer size of 32KiB would have been (accidentally) okay?

P33M, 2020-01-06:
Any buffer that did not have a length modulo 64 bytes could potentially corrupt memory, but I suspect you only stomped on something important if more than 32 bytes was received from the device in the last transfer of a sequence of bulk transfers (packet > cacheline width).

I'm now running your test  continuously, looping over libusb bulk request sizes from 16385 to 16449 bytes in length. I have the UARTs set to different baud rates so the mismatch of framing and bit position will be interpreted as random numbers of bytes on each read. It's not crashed yet with the fix applied...

localuser2, 2020-01-06:
Sounds good. However, I am pretty sure, the problem was fairly reproducible with the first transfer of 4 bytes (actual length) in our setup. If you have a fix, I can try if the problem is gone. It would be nice if you could provide a deb package containing the fixed kernel though...

pelwell, 2020-01-06:
That's a big ask - once the kernel has been tested sufficiently here it will get pushed to the firmware repo and made available via rpi-update. A package release will be weeks or months later.

localuser2, 2020-01-06:
I guess, rpi-update is fine too. I was not aware rpi-update is not using deb packages for updates. I'm just trying to avoid compiling the kernel myself.

JamesH65, 2020-01-06:
@localuser2 Takes less than 2hrs usually to download, change, and rebuild a kernel on Pi4.   https://www.raspberrypi.org/documentation/linux/kernel/building.md

localuser2, 2020-01-06:
I can try...

P33M, 2020-01-07:
Can you build and test the kernel from this branch? 

https://github.com/P33M/linux/tree/dwc-buffer-bounds-check

localuser2, 2020-01-07:
okay.

localuser2, 2020-01-07:
Bad news: I can still crash it.

To be sure, there was no mistake on my side, here is what I did:
```
$ git clone --depth=1 --branch=dwc-buffer-bounds-check https://github.com/P33M/linux.git
$ cd linux
$ KERNEL=kernel7
$ make bcm2711_defconfig
$ make -j4 zImage modules dtbs
$ sudo make modules_install
$ sudo cp arch/arm/boot/dts/*.dtb /boot/
$ sudo cp arch/arm/boot/dts/overlays/*.dtb* /boot/overlays/
$ sudo cp arch/arm/boot/dts/overlays/README /boot/overlays/
$ sudo cp arch/arm/boot/zImage /boot/$KERNEL.img
$ sudo reboot
$ uname -a
Linux smartnir 4.19.93-v7l+ #1 SMP Tue Jan 7 15:23:07 CET 2020 armv7l GNU/Linux
```
Then I run `usb-crash` several times which did not crash the Pi.

However, as mentioned before, there is another scenario where only four bytes of actual data are transmitted which lead to crashes. It is the initiation of communication of our USB devices (NIR spectrometers) with the device driver:

- one byte is transmitted to the device
- four bytes are read from the device (two status bytes and the the protocol version of the device)

I wrote a minimal example to isolate the issue which just reads the protocol version end exits. With 16KiB buffer no problem. With 20k buffer the system crashes after a couple of iterations.

I will provide a crash log.

localuser2, 2020-01-07:
Here it is (please note that the test program is a Mono assembly, so the process' name is `mono`): 

```
[  102.786161] Unable to handle kernel paging request at virtual address f7eb6031
[  102.793889] pgd = f45f1b53
[  102.797062] [f7eb6031] *pgd=80000000007003, *pmd=00000000
[  102.802953] Internal error: Oops: 206 [#1] SMP ARM

Entering kdb (current=0xec63ad00, pid 873) on processor 0 Oops: (null)
due to oops @ 0xc03be950
CPU: 0 PID: 873 Comm: mono Tainted: G         C        4.19.93-v7l+ #1
Hardware name: BCM2835
PC is at __kmalloc+0xa8/0x290
LR is at 0x112b
pc : [<c03be950>]    lr : [<0000112b>]    psr: a0000013
sp : ec469d60  ip : ec469d60  fp : ec469d9c
r10: 00000000  r9 : c0799fb4  r8 : ef001900
r7 : 00000e20  r6 : 006000c0  r5 : ef001900  r4 : f7eb6031
r3 : 00000000  r2 : c0e9e020  r1 : 2e8bd000  r0 : 00000000
Flags: NzCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment user
Control: 30c5383d  Table: 2cdab7c0  DAC: fffffffd
CPU: 0 PID: 873 Comm: mono Tainted: G         C        4.19.93-v7l+ #1
Hardware name: BCM2835
[<c0212d54>] (unwind_backtrace) from [<c020d368>] (show_stack+0x20/0x24)
[<c020d368>] (show_stack) from [<c09ada34>] (dump_stack+0xd8/0x11c)
[<c09ada34>] (dump_stack) from [<c0209d74>] (show_regs+0x1c/0x20)
[<c0209d74>] (show_regs) from [<c02defb4>] (kdb_main_loop+0x3a4/0x7c4)
[<c02defb4>] (kdb_main_loop) from [<c02e1fa8>] (kdb_stub+0x268/0x478)
[<c02e1fa8>] (kdb_stub) from [<c02d7618>] (kgdb_cpu_enter+0x470/0x7d0)
[<c02d7618>] (kgdb_cpu_enter) from [<c02d7bb0>] (kgdb_handle_exception+0xe8/0x22c)
more> 
[<c02d7bb0>] (kgdb_handle_exception) from [<c021229c>] (kgdb_notify+0x40/0x8c)
[<c021229c>] (kgdb_notify) from [<c0246000>] (notifier_call_chain+0x58/0x94)
[<c0246000>] (notifier_call_chain) from [<c02462a8>] (notify_die+0x58/0x78)
[<c02462a8>] (notify_die) from [<c020d4f0>] (die+0x184/0x374)
[<c020d4f0>] (die) from [<c02172dc>] (__do_kernel_fault.part.0+0x74/0x84)
[<c02172dc>] (__do_kernel_fault.part.0) from [<c0217378>] (do_DataAbort+0x0/0xe8)
[<c0217378>] (do_DataAbort) from [<c09ca420>] (do_translation_fault+0xd4/0x10c)
[<c09ca420>] (do_translation_fault) from [<c02173c8>] (do_DataAbort+0x50/0xe8)
[<c02173c8>] (do_DataAbort) from [<c0201934>] (__dabt_svc+0x54/0x80)
Exception stack(0xec469d10 to 0xec469d58)
9d00:                                     00000000 2e8bd000 c0e9e020 00000000
9d20: f7eb6031 ef001900 006000c0 00000e20 ef001900 c0799fb4 00000000 ec469d9c
9d40: ec469d60 ec469d60 0000112b c03be950 a0000013 ffffffff
[<c0201934>] (__dabt_svc) from [<c03be950>] (__kmalloc+0xa8/0x290)
[<c03be950>] (__kmalloc) from [<c0799fb4>] (proc_do_submiturb+0xd7c/0xea4)
[<c0799fb4>] (proc_do_submiturb) from [<c079b2b4>] (usbdev_ioctl+0x11d8/0x1b64)
[<c079b2b4>] (usbdev_ioctl) from [<c03ea298>] (do_vfs_ioctl+0xc0/0x7d4)
[<c03ea298>] (do_vfs_ioctl) from [<c03ea9f0>] (ksys_ioctl+0x44/0x70)
[<c03ea9f0>] (ksys_ioctl) from [<c03eaa34>] (sys_ioctl+0x18/0x1c)
[<c03eaa34>] (sys_ioctl) from [<c0201000>] (ret_fast_syscall+0x0/0x28)
Exception stack(0xec469fa8 to 0xec469ff0)
9fa0:                   024b1214 024c7490 00000009 802c550a 024b1214 024600e0
more> 
9fc0: 024b1214 024c7490 00000001 00000036 00000002 00000001 00000001 00004000
9fe0: b43aa0b4 be83fc3c b4391608 b6c3751c

[0]kdb> 

[0]kdb> 
[0]kdb> 
[0]kdb> dmesg
<6>[    0.000000] Booting Linux on physical CPU 0x0
<5>[    0.000000] Linux version 4.19.93-v7l+ (pi@smartnir) (gcc version 8.3.0 (Raspbian 8.3.0-6+rpi1)) #1 SMP Tue Jan 7 15:23:07 CET 2020
<6>[    0.000000] CPU: ARMv7 Processor [410fd034] revision 4 (ARMv7), cr=30c5383d
<6>[    0.000000] CPU: div instructions available: patching division code
<6>[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
<6>[    0.000000] OF: fdt: Machine model: Raspberry Pi 3 Model B Plus Rev 1.3
<6>[    0.000000] Memory policy: Data cache writealloc
<6>[    0.000000] cma: Reserved 8 MiB at 0x000000003ac00000
[... 351 lines cut ...]
Exception stack(0xec469fa8 to 0xec469ff0)
9fa0:                   024b1214 024c7490 00000009 802c550a 024b1214 024600e0
9fc0: 024b1214 024c7490 00000001 00000036 00000002 00000001 00000001 00004000
9fe0: b43aa0b4 be83fc3c b4391608 b6c3751c
Stack traceback for pid 336
0xecfa0000      336        1  1    1   R  0xecfa0e58  in:imklog
[<c09c3d38>] (__schedule) from [<c025bd14>] (enqueue_task_fair+0xf30/0x14ec)
[<c025bd14>] (enqueue_task_fair) from [<c02d7d98>] (kgdb_nmicallback+0xa4/0xac)
[<c02d7d98>] (kgdb_nmicallback) from [<c02121e8>] (kgdb_call_nmi_hook+0x38/0x3c)
more> 
[<c02121e8>] (kgdb_call_nmi_hook) from [<c02b5ea4>] (flush_smp_call_function_queue+0xc0/0x194)
[<c02b5ea4>] (flush_smp_call_function_queue) from [<c02b6c70>] (generic_smp_call_function_single_interrupt+0x1c/0x20)
[<c02b6c70>] (generic_smp_call_function_single_interrupt) from [<c02110d0>] (handle_IPI+0x290/0x3a0)
[<c02110d0>] (handle_IPI) from [<c02021e4>] (bcm2836_arm_irqchip_handle_irq+0xa4/0xa8)
[<c02021e4>] (bcm2836_arm_irqchip_handle_irq) from [<c0201d0c>] (__irq_usr+0x4c/0x60)
Exception stack(0xece7ffb0 to 0xece7fff8)
ffa0:                                     b6568d18 0a0a0a0a 20737365 62653766
ffc0: 00000000 00000000 00000000 686f3d6c 00001fa0 b6a57264 00000000 b6a46cc4
ffe0: 01010101 b65684a8 b6a46520 b6c7efb0 60000010 ffffffff
Stack traceback for pid 0
0xef144b00        0        0  1    2   I  0xef145958  swapper/2
Stack traceback for pid 106
0xedfdad00      106        1  1    3   R  0xedfdbb58  systemd-journal
[0]kdb>  
```

P33M, 2020-01-07:
Can you do another crash dump, but after doing:
```
 echo 64 | sudo tee /sys/module/usbcore/parameters/usbfs_snoop_max
 echo 1 | sudo tee /sys/module/usbcore/parameters/usbfs_snoop
```
This might make the crash less reliable, so if it stops crashing then replace the "64" above with "0". Also, what is the expected 4-byte response from the device?

localuser2, 2020-01-07:
The expected response in this scenario (which it is if it does not crash as well) is 0x31 0x60 0x00 0x10. It is possible that only the two first bytes are transferred if the device is not ready yet. In this case the read operation is multiple times repeated after some time (and at some point a `TimeoutException` is thrown).

With `usbfs_snoop_max=64` and  `usbfs_snoop=1` I was not able to crash it. However, I see a very strange behavior: The first transfer looks like expected. Then the device does not answer anymore. The status bytes 0x31 0x60 come through, but not the following 0x00 0x10, resulting in a communication timeout. If I unplug and plug-in the USB connector, it is working again. But only one time exactly. So there seems to be some kind of reset necessary.

With `usbfs_snoop_max=0` its the same. No crashes, no second transmission.

This is what is written to syslog ( `usbfs_snoop_max=0`):

First communication after plug-in (response 0x31 0x60 0x00 0x10):
```
Jan  7 18:12:21 smartnir kernel: [ 2001.328019] usb 1-1.3: opened by process 2073: mono
Jan  7 18:12:21 smartnir kernel: [ 2001.328789] usb 1-1.3: usbdev_do_ioctl: SUBMITURB
Jan  7 18:12:21 smartnir kernel: [ 2001.328802] usb 1-1.3: control urb: bRequestType=80 bRequest=06 wValue=0300 wIndex=0000 wLength=0022
Jan  7 18:12:21 smartnir kernel: [ 2001.328817] usb 1-1.3: userurb 06328580, ep0 ctrl-in, length 34
Jan  7 18:12:21 smartnir kernel: [ 2001.329152] usb 1-1.3: urb complete
Jan  7 18:12:21 smartnir kernel: [ 2001.329160] usb 1-1.3: userurb 06328580, ep0 ctrl-in, actual_length 4 status 0
Jan  7 18:12:21 smartnir kernel: [ 2001.329192] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.329201] usb 1-1.3: reap 06328580
Jan  7 18:12:21 smartnir kernel: [ 2001.329222] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.335811] usb 1-1.3: opened by process 2073: mono
Jan  7 18:12:21 smartnir kernel: [ 2001.335875] usb 1-1.3: usbdev_do_ioctl: SUBMITURB
Jan  7 18:12:21 smartnir kernel: [ 2001.335886] usb 1-1.3: control urb: bRequestType=80 bRequest=06 wValue=0301 wIndex=0000 wLength=00ff
Jan  7 18:12:21 smartnir kernel: [ 2001.335900] usb 1-1.3: userurb 9f7778c7, ep0 ctrl-in, length 255
Jan  7 18:12:21 smartnir kernel: [ 2001.340278] usb 1-1.3: urb complete
Jan  7 18:12:21 smartnir kernel: [ 2001.340287] usb 1-1.3: userurb 9f7778c7, ep0 ctrl-in, actual_length 20 status 0
Jan  7 18:12:21 smartnir kernel: [ 2001.340322] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.340329] usb 1-1.3: reap 9f7778c7
Jan  7 18:12:21 smartnir kernel: [ 2001.340348] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.345466] usb 1-1.3: opened by process 2073: mono
Jan  7 18:12:21 smartnir kernel: [ 2001.345529] usb 1-1.3: usbdev_do_ioctl: SUBMITURB
Jan  7 18:12:21 smartnir kernel: [ 2001.345540] usb 1-1.3: control urb: bRequestType=80 bRequest=06 wValue=0302 wIndex=0000 wLength=00ff
Jan  7 18:12:21 smartnir kernel: [ 2001.345554] usb 1-1.3: userurb 631d5161, ep0 ctrl-in, length 255
Jan  7 18:12:21 smartnir kernel: [ 2001.354026] usb 1-1.3: urb complete
Jan  7 18:12:21 smartnir kernel: [ 2001.354038] usb 1-1.3: userurb 631d5161, ep0 ctrl-in, actual_length 46 status 0
Jan  7 18:12:21 smartnir kernel: [ 2001.354068] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.354076] usb 1-1.3: reap 631d5161
Jan  7 18:12:21 smartnir kernel: [ 2001.354102] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.355000] usb 1-1.3: opened by process 2073: mono
Jan  7 18:12:21 smartnir kernel: [ 2001.355055] usb 1-1.3: usbdev_do_ioctl: SUBMITURB
Jan  7 18:12:21 smartnir kernel: [ 2001.355065] usb 1-1.3: control urb: bRequestType=80 bRequest=06 wValue=0303 wIndex=0000 wLength=00ff
Jan  7 18:12:21 smartnir kernel: [ 2001.355077] usb 1-1.3: userurb 631d5161, ep0 ctrl-in, length 255
Jan  7 18:12:21 smartnir kernel: [ 2001.359401] usb 1-1.3: urb complete
Jan  7 18:12:21 smartnir kernel: [ 2001.359411] usb 1-1.3: userurb 631d5161, ep0 ctrl-in, actual_length 18 status 0
Jan  7 18:12:21 smartnir kernel: [ 2001.359445] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.359456] usb 1-1.3: reap 631d5161
Jan  7 18:12:21 smartnir kernel: [ 2001.359476] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.617578] usb 1-1.3: opened by process 2073: mono
Jan  7 18:12:21 smartnir kernel: [ 2001.619089] usb 1-1.3: usbdev_do_ioctl: CLAIMINTERFACE
Jan  7 18:12:21 smartnir kernel: [ 2001.635309] usb 1-1.3: usbdev_do_ioctl: SUBMITURB
Jan  7 18:12:21 smartnir kernel: [ 2001.635325] usb 1-1.3: control urb: bRequestType=40 bRequest=00 wValue=0000 wIndex=0000 wLength=0000
Jan  7 18:12:21 smartnir kernel: [ 2001.635339] usb 1-1.3: userurb 4bee00e6, ep0 ctrl-out, length 0
Jan  7 18:12:21 smartnir kernel: [ 2001.635653] usb 1-1.3: urb complete
Jan  7 18:12:21 smartnir kernel: [ 2001.635662] usb 1-1.3: userurb 4bee00e6, ep0 ctrl-out, actual_length 0 status 0
Jan  7 18:12:21 smartnir kernel: [ 2001.635715] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.635723] usb 1-1.3: reap 4bee00e6
Jan  7 18:12:21 smartnir kernel: [ 2001.638284] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:12:21 smartnir kernel: [ 2001.640714] usb 1-1.3: usbdev_do_ioctl: SUBMITURB
[... cut ...]
Jan  7 18:13:25 smartnir kernel: [ 2065.872217] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:13:25 smartnir kernel: [ 2065.872222] usb 1-1.3: reap 42779ca4
Jan  7 18:13:25 smartnir kernel: [ 2065.872237] usb 1-1.3: usbdev_do_ioctl: DISCARDURB 928b44da
Jan  7 18:13:25 smartnir kernel: [ 2065.872247] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
Jan  7 18:13:25 smartnir kernel: [ 2065.872251] usb 1-1.3: reap 928b44da
Jan  7 18:13:25 smartnir kernel: [ 2065.872297] usb 1-1.3: usbdev_do_ioctl: REAPURBNDELAY
```

P33M, 2020-01-07:
So, aside from the lack of crashing, is the device behaviour different with the debug enabled? I.e. is the response returned from the device on the bulk transfer different?

One thing that stands out is that the lower 2 bytes of your crashing address (dud pointer) are xxxx6031, i.e. byte-swapped 0x31 0x60. If you crash the Pi multiple times, what is the faulting address in each case?

I have a sneaking suspicion that this is a different bug, triggered by the way that usbfs is reaping URBs when trying to clean up after itself.

localuser2, 2020-01-08:
The first (i.e. working) transfer works as expected with the debug enabled. The response is not different.

Regarding to the crashing address: This is indeed interesting and certainly no coincidence.

Here the addresses are for several crashes:

```
[   86.722894] Unable to handle kernel paging request at virtual address f7eb6031
...
[   60.836620] Unable to handle kernel paging request at virtual address f7eb6031
...
[   40.836554] Unable to handle kernel paging request at virtual address f7eb6031
...
[   44.087212] Unable to handle kernel paging request at virtual address f7eb6031
```
i.e. it is always the same address.

This test brought something other out though: It crashes always at the second iteration. If I disconnect the USB device after the first transfer and try again it does not crash. Something seems to be in an very inconsistent state after the first transfer which leads to the crash at the second transfer. Please note that the iterations of the test are not run in a loop within a process. The test program opens, claims, writes, reads, unclaims, closes, exits. Iterations are done at the shell level.

Disconcertingly the usbfs seems to have some kind of memory which is reset if the device is unplugged and reconnected.

In this light the former test (`usbfs_snoop`) seems to be more consistent. The usbfs is broken somehow after the first transfer. The only difference is that it does not crash with an Oops.

P33M, 2020-01-08:
Does your device always return bulk data, i.e. never NAKs transfers? My current working theory is that there's a race between freeing the URB's DMA buffer (as usbfs is busy reaping URBs) and an outstanding transfer that returns data. 

dwc_otg's unlink function doesn't cancel ongoing DMA (because force-disabling a channel doesn't work), so there's a potential hole if the buffer is freed and re-used before the core finishes whatever it was doing.

localuser2, 2020-01-08:
Yes, the device always returns at least 2 bytes bulk data.

P33M, 2020-01-08:
Can you try recompiling the branch (I've pushed another tentative fix)?

localuser2, 2020-01-08:
Now it is not crashing anymore. But it is still only one transfer possible. Then only the two status bytes come through resulting in a timeout.

With 16KiB buffer everything works as expected. While testing with 16k and 20k version of my test program I found another piece of the puzzle:

The timeout (without the current fix i.e. the crash) occurs even if the first transfer is a 16k transfer. Here some possible sequences to clarify:

- connect device
- 16k transfer -> okay
- 16k transfer -> okay
- 20k transfer -> timeout
- 16k transfer -> okay
...

-connect device
- 20k transfer -> okay
- 20k transfer -> timeout
- 16k transfer -> okay

It does not seem to matter what kind of transfer buffer length the first transfer is. The usbfs is always broken afterwards (regarding 20k transfers).

P33M, 2020-01-09:
Is there any way you can capture the full-speed bus traffic with an analyzer?

localuser2, 2020-01-09:
We only have a digital oscilloscope. Eventually we could capture the signal levels and send it to you. We do not have an decoder/logic analyzer at hand which is capable of the necessary sample rate.

Would a set of raw data be sufficient? Presumed we can export it in a useful format.

P33M, 2020-01-09:
Decoding a raw bitstream is going to be a non-starter. If you have a Saleae Logic Pro (8 or 16 channel) analyzer then it can decode ls/fs USB and export it in a reasonable format

Other than that, without representative hardware, I'm not going to be able to debug much further.

Given that the two changes fix definite memory corruption bugs, I think those should get pushed out for wider testing anyway.

localuser2, 2020-01-09:
We have a Saleae Logic, but not the Pro version. This will not be fast enough.

So I guess this is a dead end then? What restrictions are to be considered to work around the trapdoors? Buffer lengths <= 16KiB? Multiples of 64bytes?

Would es be possible to return an error if a dangerous buffer length is used?

lurch, 2020-01-17:
I dunno if it's relevant (maybe it was just a typo?), but I just noticed that in one of your earlier messages you had:
```
$ KERNEL=kernel7
$ make bcm2711_defconfig
```

AFAIK this is an invalid combination; as https://www.raspberrypi.org/documentation/linux/kernel/building.md notes, you want to use `KERNEL=kernel7; make bcm2709_defconfig` on a Pi3 or `KERNEL=kernel7l; make bcm2711_defconfig` on a Pi4.

pelwell, 2020-01-17:
That was probably a cut-and-paste error in compiling the report:
```
$ uname -a
Linux smartnir 4.19.93-v7l+ #1 SMP Tue Jan 7 15:23:07 CET 2020 armv7l GNU/Linux
```
The `smartnir`, `#1` and date suggests that this was indeed the result of a local build; had the script been followed with `KERNEL=kernel7` then it would not have overwritten the old `kernel7l.img`. The other explanation would be that `kernel7l.img` had been deleted from the card, causing the firmware to load `kernel7.img` as a fallback.

localuser2, 2020-01-17:
Yes. This was a copy-and-paste error. Sorry

Armavica, 2020-07-18:
Hello, I have a similar issue with a Raspberry Pi 3B+ that I also use to control a lab device by USB. I use `pyusb` with a buffer length of 1000. The kernel crashed twice in one week, following an upgrade of the Pi (it never crashed for this reason before). I can give more details as soon as I go back to the lab and reboot the Pi.

Armavica, 2020-07-21:
The problem started occurring after a [distribution upgrade](https://paste.fulltxt.net/I7CZm) (updating `deb http://raspbian.raspberrypi.org/raspbian/ stretch main contrib non-free rpi` to `buster`). It never happened before.
This is the [raspinfo](https://paste.fulltxt.net/I2ZHEc) of the affected configuration. I changed the buffer size from 1000 to 1024 (multiple of 64), which still resulted in a crash. I then updated the pi with `rpi-update` and am now running my application to see if it still crashes.