raspberrypi/linux #3391: vc4-fkms-v3d and framebuffer

ajefr, 2020-01-03:
Hello,

With the introduction of the new vc4-fkms-v3d with the RPI4, I encounter many problems with this new emulated framebuffer.

First trouble :
It's impossible to increase framebuffer size.
For example, if you define a 640x480 framebuffer/resolution in your config.txt, you canno't increase it to 800x600 by issuing this kind of command :
`fbset -xres 800`
give you
`ioctl FBIOPUT_VSCREENINFO: Invalid argument`
while reducing framebuffer works :
`fbset -xres 600`

Second trouble :
Issuing a command to change color depth also give an error :
`fbset -depth 16`
give the same invalid error
`ioctl FBIOPUT_VSCREENINFO: Invalid argument`

Third error
Pixel clock is not saved in the framebuffer information
`fbset -pixclock 1000`
give no error but the value is not applied

Tested on a RPI4B, buster, firmware 4.19.88-v7l+

Those options were essentials when you wanted to change resolutions on the fly : [#637](https://github.com/raspberrypi/firmware/issues/637)

Thanks

pelwell, 2020-01-03:
[ Moved to the linux repo ]

lategoodbye, 2020-01-04:
Sorry for hijacking this issue. Currently the mainline development of the RPi 4 requires a Serial-USB-Adapter. So at least framebuffer support for Linux & U-Boot would be a big improvement. Since the guys from Igalia didn't show a motivation to touch the required low level driver (power domain, clock) we stuck at this point.
It would be very helpful to get some pointers which downstream patches or at least which power domain & clocks are required for framebuffer support (just CLI).

joolswills, 2020-01-04:
From my understanding kms mode switching has to be done on a per application basis. There's no way of just changing the framebuffer mode in a terminal and have that information passed to an app you run. (Different under x since all apps will run under that)

@psyke83 and @cmitu did some work on this for retropie so we have mode switching built into sdl2 and selectable from our app launcher (runcommand) for rpi4 + fkms.

See https://github.com/RetroPie/RetroPie-Setup/pull/2900

And

https://github.com/RetroPie/SDL-mirror/pull/16

pelwell, 2020-01-10:
@lategoodbye If you have a specific feature request, can you open a separate issue?

lategoodbye, 2020-01-10:
@pelwell No, i'm fine this [this](https://github.com/lategoodbye/rpi-zero/issues/46#issuecomment-571971903) comment.