raspberrypi/linux #3397: Feature request: NFSv4.2 client support in kernel

adduty, 2020-01-08:
Could you please add CONFIG_NFS_V4_2=y to the kernel config?

NFSv4.2 adds sparse file support, which would help extend the life of SSD-backed NFS shares for those network booting the Raspberry Pi.

For those who are curious, see https://msystechnologies.com/blogs/what-is-new-in-nfs-v4-2/ for a more comprehensive list of features added in NFSv4.2.

pelwell, 2020-01-10:
How much does the kernel footprint grow with this feature added? Boot to as close to an idle system as you can, then run `free` on a kernel version with and without this patch, then post the results.

adduty, 2020-01-21:
I have booted to Raspbian Lite using a kernel compiled from the latest source, once without CONFIG_NFS_V4_2=y and once with it. Here are the results:

**Without:**
```
$ free
              total        used        free      shared  buff/cache   available
Mem:        1986012       63568     1848488       16816       73956     1833772
Swap:             0           0           0
```

**With:**
```
$ free
              total        used        free      shared  buff/cache   available
Mem:        1986012       62848     1849452       16816       73712     1834616
Swap:             0           0           0
```

**Difference in kernel image size:**
```
-rw-r--r--  1 root root  5737656 Jan 21 23:17 kernel7l_custom_42.img
-rw-r--r--  1 root root  5731288 Jan 21 20:11 kernel7l_custom.img
```

adduty, 2020-01-27:
@pelwell does this look acceptable?

pelwell, 2020-02-04:
That looks acceptable. See https://github.com/raspberrypi/linux/commit/3508d8f3326cca9cdffcff88190cb8ab315260bc.