raspberrypi/linux #3398: Unable to boot PI3B from external USB HD anymore

gorky, 2020-01-09:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
General boot failure when attempting to boot using an external USB Drive.

**To reproduce**
Two scenarios:
1) External USB HD Only:
* Image an SD Card with https://downloads.raspberrypi.org/raspbian_lite_latest (version September 2019 - full version archived at https://downloads.raspberrypi.org/raspbian/images/raspbian-2019-09-30/2019-09-26-raspbian-buster.zip)
* Set config.txt setting ```program_usb_boot_mode=1```
* Boot PI (it works)
* Powerdown PI
* Image external HD.
* Set config.txt setting ```program_usb_boot_mode=1```
* Plug HD into powered usb hub, plug Hub into PI
* Boot PI.
2) Boot on SD card, rootfs on External HD
* Set config.txt setting ```program_usb_boot_mode=1```
* Boot PI (it works)
* Powerdown PI
* Image external HD.
* update cmdline.txt file on SD card to load from root partition on external HD. (Note: balenaEtcher may set Disk UUIDs/PartitionUUIDs to the same value -- be sure to check and modify if required!)
* Insert SD & plug in External USB HD.
* Boot PI.

**Expected behaviour**
System boots

**Actual behaviour**
1)   First boot gets four Raspi's.  Second boot, no splash or Raspi's.  See https://github.com/raspberrypi/linux/issues/3126
2) See attached Screen

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? e.g. Pi3B+, PiZeroW
* Which OS and version (`cat /etc/rpi-issue`)?
* Which firmware version (`vcgencmd version`)?
* Which kernel version (`uname -a`)?

**Logs**
See screenshot of available output (Scenario 2)

**Additional context**
None currently available.
[image]

gorky, 2020-01-09:
Workaround:
Boot from SD, mount directories needing many updates (/var) and large storage space from external HD.

pelwell, 2020-01-09:
For scenario 2, try waiting a few minutes to see if further output appears.

gorky, 2020-01-09:
Went on a 20minute coffee break, then another 20minutes before I had this picture taken, uploaded, and added to this ticket.  Am I correct in thinking 40+ minutes should have been more than enough time?

pelwell, 2020-01-09:
2 minutes is a significant time during the boot process - there's usually some warning about waiting for the rootfs at around that point.

gorky, 2020-01-09:
Sorry, but that screen shot is all that there is.  Goes for the first 5 seconds, then the last message that you see on the screen 15seconds into the boot, then nothing.  Btw, I didn't realize that I had left it running after I opened this ticket -- it's STILL at that point.  No further progress. So, it's been sitting there now for some 20+ hours, with no change.  That screen is still the latest, sorry.

pelwell, 2020-01-10:
You might get some more clues with "initcall_debug" in cmdline.txt.

gorky, 2020-01-10:
No additional information.  Did I mess something up?
[cmdline.txt](https://github.com/raspberrypi/linux/files/4047053/cmdline.txt)
[image]