raspberrypi/linux #3401: kernel panic

js290, 2020-01-12:
**Is this the right place for my bug report?**
This repository contains the Linux kernel used on the Raspberry Pi. If you believe that the issue you are seeing is kernel-related, this is the right place. If not, we have other repositories for the GPU firmware at [github.com/raspberrypi/firmware](https://github.com/raspberrypi/firmware) and Raspberry Pi userland applications at [github.com/raspberrypi/userland](https://github.com/raspberrypi/userland). If you have problems with the Raspbian distribution packages, report them in the [github.com/RPi-Distro/repo](https://github.com/RPi-Distro/repo). If you simply have a question, then [the Raspberry Pi forums](https://www.raspberrypi.org/forums) are the best place to ask it.

**Describe the bug**
Add a clear and concise description of what you think the bug is.
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167099] ------------[ cut here ]------------
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167177] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:466 dev_watchdog+0x26c/0x288
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167190] NETDEV WATCHDOG: eth0 (dm9601): transmit queue 0 timed out
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167196] Modules linked in: sha256_generic cfg80211 rfkill 8021q garp stp llc nft_counter xt_tcpudp xt_state xt_conntrack nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_compat nf_tables nfnetlink sr9700 dm9601 raspberrypi_hwmon hwmon bcm2835_codec(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167417] CPU: 0 PID: 0 Comm: swapper Tainted: G         C        4.19.75+ #1270
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167424] Hardware name: BCM2835
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167472] [<c0017f54>] (unwind_backtrace) from [<c0014e9c>] (show_stack+0x20/0x24)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167510] [<c0014e9c>] (show_stack) from [<c06ba0b0>] (dump_stack+0x20/0x28)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167546] [<c06ba0b0>] (dump_stack) from [<c00244f0>] (__warn+0xf4/0x11c)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167570] [<c00244f0>] (__warn) from [<c002456c>] (warn_slowpath_fmt+0x54/0x74)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167599] [<c002456c>] (warn_slowpath_fmt) from [<c05e58ec>] (dev_watchdog+0x26c/0x288)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167646] [<c05e58ec>] (dev_watchdog) from [<c0076624>] (call_timer_fn+0x3c/0x120)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167674] [<c0076624>] (call_timer_fn) from [<c00767a4>] (expire_timers+0x9c/0xd8)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167700] [<c00767a4>] (expire_timers) from [<c0076a80>] (run_timer_softirq+0xa4/0x1ac)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167726] [<c0076a80>] (run_timer_softirq) from [<c000a294>] (__do_softirq+0x12c/0x344)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167761] [<c000a294>] (__do_softirq) from [<c0028eb8>] (irq_exit+0xe4/0x100)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167792] [<c0028eb8>] (irq_exit) from [<c0067374>] (__handle_domain_irq+0x60/0xb8)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167815] [<c0067374>] (__handle_domain_irq) from [<c000a050>] (bcm2835_handle_irq+0x28/0x48)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167834] [<c000a050>] (bcm2835_handle_irq) from [<c000999c>] (__irq_svc+0x5c/0x7c)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167845] Exception stack(0xc09e7ef0 to 0xc09e7f38)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167859] 7ee0:                                     c00115e4 00000000 40000093 40000093
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167878] 7f00: c09e6000 c09e9090 c0a76fb6 c09e9020 c0a84c80 dbfffa40 c09c0a30 c09e7f4c
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167896] 7f20: c09e941c c09e7f40 00000000 c00115e8 40000013 ffffffff
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167927] [<c000999c>] (__irq_svc) from [<c00115e8>] (arch_cpu_idle+0x30/0x40)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167957] [<c00115e8>] (arch_cpu_idle) from [<c06d5cf8>] (default_idle_call+0x34/0x48)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.167994] [<c06d5cf8>] (default_idle_call) from [<c00502b8>] (do_idle+0x94/0xf4)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.168027] [<c00502b8>] (do_idle) from [<c00505bc>] (cpu_startup_entry+0x1c/0x20)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.168063] [<c00505bc>] (cpu_startup_entry) from [<c06cffec>] (rest_init+0x7c/0x9c)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.168088] [<c06cffec>] (rest_init) from [<c0972ed4>] (start_kernel+0x3e8/0x49c)
Jan 12 00:18:06 raspberrypi kernel: [ 8378.168103] ---[ end trace 5d6e43a630ac3840 ]---

**To reproduce**
List the steps required to reproduce the issue.

**Expected behaviour**
Add a clear and concise description of what you expected to happen.

**Actual behaviour**
Add a clear and concise description of what actually happened.

**System**
System Information
------------------

Raspberry Pi Zero Rev 1.3
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

Linux raspberrypi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux
Revision        : 900093
Serial          : 000000001a48b5ea
Model           : Raspberry Pi Zero Rev 1.3
Throttled flag  : throttled=0x0
Camera          : supported=0 detected=0

Videocore information
---------------------

Sep 24 2019 17:37:19 
Copyright (c) 2012 Broadcom
[... 494 lines cut ...]
[   14.627519] bcm2835_mmal_vchiq: module is from the staging directory, the quality is unknown, you have been warned.
[   14.704098] bcm2835_v4l2: module is from the staging directory, the quality is unknown, you have been warned.
[   14.728331] bcm2835_codec: module is from the staging directory, the quality is unknown, you have been warned.
[   14.841631] bcm2835-codec bcm2835-codec: Device registered as /dev/video10
[   14.841724] bcm2835-codec bcm2835-codec: Loaded V4L2 decode
[   14.852073] bcm2835-codec bcm2835-codec: Device registered as /dev/video11
[   14.852136] bcm2835-codec bcm2835-codec: Loaded V4L2 encode
[   14.862817] bcm2835-codec bcm2835-codec: Device registered as /dev/video12
[   14.862876] bcm2835-codec bcm2835-codec: Loaded V4L2 isp
[   17.316185] dm9601 1-1.2:1.0 eth0: register 'dm9601' at usb-20980000.usb-1.2, Davicom DM96xx USB 10/100 Ethernet, m.m.m.m
[   17.316646] usbcore: registered new interface driver dm9601
[   17.331312] usbcore: registered new interface driver sr9700
[   22.425432] random: crng init done
[   22.425456] random: 7 urandom warning(s) missed due to ratelimiting
[   24.557142] 8021q: 802.1Q VLAN Support v1.8
[   25.574583] cfg80211: Loading compiled-in X.509 certificates for regulatory database
[   26.476602] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
[   26.949592] Adding 102396k swap on /var/swap.  Priority:-2 extents:1 across:102396k SSFS
[   26.967067] dm9601 1-1.2:1.0 eth0: link up, 100Mbps, full-duplex, lpa 0xFFFF

**Logs**
If applicable, add the relevant output from `dmesg` or similar.

**Additional context**
Add any other relevant context for the problem.

JamesH65, 2020-01-12:
Can you please list the steps required to replicate the issue, and describe the setup of your system.

js290, 2020-01-13:
I don't know how to replicate.  The pi was unresponsive, and after reboot I found the shared stack trace in the logs.  My Pi is running Pi Hole, but it has never crashed before.

dwrobel, 2020-01-18:
> Can you please list the steps required to replicate the issue, and describe the setup of your system.

I have a similar callstack from `4.19.83`:

```
[173036.430458] ------------[ cut here ]------------
[173036.435289] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:465 dev_watchdog+0x290/0x294
[173036.443781] NETDEV WATCHDOG: eth1 (ax88179_178a): transmit queue 0 timed out
[173036.451059] Modules linked in: tun xt_nat ipt_MASQUERADE nf_conntrack_netbios_ns nf_conntrack_broadcast xt_CT ip6t_REJECT nf_reject_ipv6 ip6t_rpfilter ipt_REJECT nf_reject_ipv4 xt_conntrack ebtable_nat ebtable_broute bridge stp llc ip6table_nat nf_nat_ipv6 ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat_ipv4 nf_nat iptable_mangle iptable_raw iptable_security nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 ip_set nfnetlink ebtable_filter ebtables ip6table_filter ip6_tables iptable_filter dm_mirror dm_region_hash dm_log dm_mod brcmfmac brcmutil cfg80211 ax88179_178a bcm2835_codec(C) v4l2_mem2mem rfkill bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) videobuf2_dma_contig v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common raspberrypi_hwmon videodev vc_sm_cma(C) media gpio_fan
[173036.526417]  hwmon uio_pdrv_genirq uio fixed zram sch_fq_codel zsmalloc fuse
[173036.535602] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        4.19.83-2.rpi.fc31.armv7hl #1
[173036.548687] Hardware name: BCM2835
[173036.554174] [<80112330>] (unwind_backtrace) from [<8010d594>] (show_stack+0x20/0x24)
[173036.566084] [<8010d594>] (show_stack) from [<80908224>] (dump_stack+0xc8/0x10c)
[173036.577768] [<80908224>] (dump_stack) from [<80125df0>] (__warn.part.0+0xc4/0xe0)
[173036.589713] [<80125df0>] (__warn.part.0) from [<80125e84>] (warn_slowpath_fmt+0x78/0x9c)
[173036.602407] [<80125e84>] (warn_slowpath_fmt) from [<807d11f8>] (dev_watchdog+0x290/0x294)
[173036.615328] [<807d11f8>] (dev_watchdog) from [<8019f308>] (call_timer_fn+0x3c/0x198)
[173036.627892] [<8019f308>] (call_timer_fn) from [<8019f550>] (expire_timers+0xec/0x14c)
[173036.640667] [<8019f550>] (expire_timers) from [<8019f668>] (run_timer_softirq+0xb8/0x1f0)
[173036.653979] [<8019f668>] (run_timer_softirq) from [<801023f0>] (__do_softirq+0x180/0x3e0)
[173036.667479] [<801023f0>] (__do_softirq) from [<8012bdc4>] (irq_exit+0xf8/0x134)
[173036.680390] [<8012bdc4>] (irq_exit) from [<801862c0>] (__handle_domain_irq+0x70/0xc0)
[173036.693957] [<801862c0>] (__handle_domain_irq) from [<801021a4>] (bcm2836_arm_irqchip_handle_irq+0x64/0xa4)
[173036.709527] [<801021a4>] (bcm2836_arm_irqchip_handle_irq) from [<801019bc>] (__irq_svc+0x5c/0x7c)
[173036.724305] Exception stack(0x80e01ef0 to 0x80e01f38)
[173036.732344] 1ee0:                                     00000000 395bc000 400e0093 400e0093
[173036.746346] 1f00: 80e00000 00000000 80e04df4 80e04e3c 80e97522 80b8a9ec 10c5387d 80e01f4c
[173036.760295] 1f20: ffffe000 80e01f40 80e00000 80109bcc 400e0013 ffffffff
[173036.769846] [<801019bc>] (__irq_svc) from [<80109bcc>] (arch_cpu_idle+0x34/0x4c)
[173036.782957] [<80109bcc>] (arch_cpu_idle) from [<80923c7c>] (default_idle_call+0x34/0x48)
[173036.796767] [<80923c7c>] (default_idle_call) from [<80158614>] (do_idle+0xec/0x168)
[173036.810119] [<80158614>] (do_idle) from [<8015895c>] (cpu_startup_entry+0x28/0x2c)
[173036.823355] [<8015895c>] (cpu_startup_entry) from [<8091d3fc>] (rest_init+0xb8/0xbc)
[173036.836802] [<8091d3fc>] (rest_init) from [<80d00fec>] (start_kernel+0x490/0x4c0)
[173036.850016] ---[ end trace 8a54cc574983428a ]---
```
It's from the headless system _(no hdmi, keyboard, etc. connected)_ which works as a router _(has two USB ethernet cards, both are: 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet)_.

It usually happened after a week of stable work.

Later on, I upgraded to:
```
# uname -a
Linux localhost.localdomain 4.19.93-1.rpi.fc31.armv7hl #1 SMP Tue Jan 7 19:57:20 CET 2020 armv7l armv7l armv7l GNU/Linux
```
and at the time of writing:
```
# uptime
 17:31:35 up 10 days, 18:24,  1 user,  load average: 0.00, 0.01, 0.00
```
I haven't observed it so far.

ablacksheep, 2020-01-21:
as @dwrobel said, it might be related

> It's from the headless system _(no hdmi, keyboard, etc. connected)_ which works as a router _(has two USB ethernet cards, both are: 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet)_.
> 
> It usually happened after a week of stable work.

I have a  very similar setup with 2x 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet. One card stopped responding after aprox 10 days with kernel:
`Linux ky015 4.19.93-v7l+ #1290 SMP Fri Jan 10 16:45:11 GMT 2020 armv7l GNU/Linux`

```
kernel: [987818.955728] ------------[ cut here ]------------
kernel: [987818.955779] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:466 dev_watchdog+0x294/0x298
kernel: [987818.955796] NETDEV WATCHDOG: enx00249b1e558a (ax88179_178a): transmit queue 0 timed out
kernel: [987818.955809] Modules linked in: nf_conntrack_netlink xt_multiport tun bridge 8021q garp stp llc rtc_ds1307 ax88179_178a brcmfmac brcmutil sha256_generic cfg80211 rfkill vc4 bcm2835_codec(C) drm_kms_helper v4l2_mem2mem v3d gpu_sched bcm2835_v4l2(C) bcm2835_mmal_vchiq(C) videobuf2_dma_contig v4l2_common videobuf2_vmalloc videobuf2_memops nf_log_ipv6 videobuf2_v4l2 videobuf2_common ip6t_REJECT nf_reject_ipv6 videodev media vc_sm_cma(C) drm drm_panel_orientation_quirks xt_hl raspberrypi_hwmon snd_soc_core hwmon ip6_tables snd_compress snd_pcm_dmaengine ip6t_rt syscopyarea sysfillrect sysimgblt fb_sys_fops snd_bcm2835(C) snd_pcm snd_timer snd i2c_bcm2835 rpivid_mem nf_log_ipv4 nf_log_common ipt_REJECT nf_reject_ipv4 uio_pdrv_genirq uio fixed xt_LOG nft_limit xt_limit xt_addrtype nft_chain_nat_ipv4 xt_nat
kernel: [987818.956016]  xt_tcpudp ipt_MASQUERADE nf_nat_ipv4 xt_conntrack nft_compat nft_counter nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables i2c_dev nfnetlink ip_tables x_tables ipv6
kernel: [987818.956090] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        4.19.88-v7l+ #1284
kernel: [987818.956099] Hardware name: BCM2835
kernel: [987818.956127] [<c0212e04>] (unwind_backtrace) from [<c020d5e0>] (show_stack+0x20/0x24)
kernel: [987818.956147] [<c020d5e0>] (show_stack) from [<c09afa48>] (dump_stack+0xe0/0x124)
kernel: [987818.956169] [<c09afa48>] (dump_stack) from [<c0222544>] (__warn+0x104/0x11c)
kernel: [987818.956187] [<c0222544>] (__warn) from [<c02225b4>] (warn_slowpath_fmt+0x58/0x74)
kernel: [987818.956205] [<c02225b4>] (warn_slowpath_fmt) from [<c08d5a7c>] (dev_watchdog+0x294/0x298)
kernel: [987818.956225] [<c08d5a7c>] (dev_watchdog) from [<c029b848>] (call_timer_fn+0x3c/0x198)
kernel: [987818.956243] [<c029b848>] (call_timer_fn) from [<c029ba90>] (expire_timers+0xec/0x14c)
kernel: [987818.956259] [<c029ba90>] (expire_timers) from [<c029bba8>] (run_timer_softirq+0xb8/0x1ec)
kernel: [987818.956275] [<c029bba8>] (run_timer_softirq) from [<c0202410>] (__do_softirq+0x190/0x3f0)
kernel: [987818.956292] [<c0202410>] (__do_softirq) from [<c02286cc>] (irq_exit+0xfc/0x120)
kernel: [987818.956310] [<c02286cc>] (irq_exit) from [<c0281d9c>] (__handle_domain_irq+0x70/0xc4)
kernel: [987818.956325] [<c0281d9c>] (__handle_domain_irq) from [<c0202244>] (gic_handle_irq+0x4c/0x88)
kernel: [987818.956340] [<c0202244>] (gic_handle_irq) from [<c02019bc>] (__irq_svc+0x5c/0x7c)
kernel: [987818.956350] Exception stack(0xc1001ed8 to 0xc1001f20)
kernel: [987818.956360] 1ec0:                                                       c0209ae4 00000000
kernel: [987818.956372] 1ee0: 40000093 40000093 c1004dbc c1000000 c1004e04 00000001 c109681a c10abf40
kernel: [987818.956384] 1f00: c0e67a38 c1001f34 c10051c4 c1001f28 00000000 c0209ae8 40000013 ffffffff
kernel: [987818.956406] [<c02019bc>] (__irq_svc) from [<c0209ae8>] (arch_cpu_idle+0x34/0x4c)
kernel: [987818.956425] [<c0209ae8>] (arch_cpu_idle) from [<c09ccd0c>] (default_idle_call+0x34/0x48)
kernel: [987818.956441] [<c09ccd0c>] (default_idle_call) from [<c0254a14>] (do_idle+0xec/0x16c)
kernel: [987818.956456] [<c0254a14>] (do_idle) from [<c0254d54>] (cpu_startup_entry+0x28/0x2c)
kernel: [987818.956472] [<c0254d54>] (cpu_startup_entry) from [<c09c63a0>] (rest_init+0xbc/0xc0)
kernel: [987818.956492] [<c09c63a0>] (rest_init) from [<c0e00fe0>] (start_kernel+0x4b4/0x4e4)
kernel: [987818.956509] ---[ end trace e4e030a1002f670f ]---

```

ablacksheep, 2020-02-13:
13:46:09 up **18 days**, 14:03,  1 user,  load average: 0,00, 0,01, 0,00

Same port crashed again, not sure if it is a pattern after 2 crashes, but so far it is the one plugged in the usb2 port that crash. The one pluggen in the usb3 seems to work fine. Might not be related, but i have to mention it. I can also add that the two interfaces have different traffic patterns, the one not crashing has continues traffic, and the port that crash has traffic spikes from a few times a day to some days between. Between the traffic spikes, there is basically no traffic.

```
Bus 002 Device 002: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0b95:1790 ASIX Electronics Corp. AX88179 Gigabit Ethernet
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

```
------------[ cut here ]------------
[1605418.628803] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:466 dev_watchdog+0x294/0x298
[1605418.628829] NETDEV WATCHDOG: enx00249b1e558a (ax88179_178a): transmit queue 0 timed out
[1605418.628843] Modules linked in: xt_recent nf_conntrack_netlink xt_multiport tun bridge 8021q garp stp llc rtc_ds1307 ax88179_178a brcmfmac brcmutil nf_log_ipv6 sha256_generic ip6t_REJECT nf_reject_ipv6 cfg80211 rfkill xt_hl vc4 ip6_tables ip6t_rt v3d raspberrypi_hwmon drm_kms_helper hwmon gpu_sched drm drm_panel_orientation_quirks snd_soc_core snd_bcm2835(C) snd_compress bcm2835_codec(C) snd_pcm_dmaengine v4l2_mem2mem bcm2835_v4l2(C) snd_pcm syscopyarea bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc snd_timer videobuf2_dma_contig nf_log_ipv4 nf_log_common sysfillrect snd sysimgblt vc_sm_cma(C) videobuf2_memops ipt_REJECT fb_sys_fops nf_reject_ipv4 videobuf2_v4l2 videobuf2_common xt_LOG videodev media i2c_bcm2835 rpivid_mem nft_limit uio_pdrv_genirq uio fixed xt_limit xt_addrtype nft_chain_nat_ipv4
[1605418.629058]  xt_nat xt_tcpudp ipt_MASQUERADE nf_nat_ipv4 xt_conntrack nft_compat nft_counter nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink i2c_dev ip_tables x_tables ipv6
[1605418.629146] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        4.19.93-v7l+ #1290
[1605418.629154] Hardware name: BCM2835
[1605418.629187] [<c0212e04>] (unwind_backtrace) from [<c020d5e0>] (show_stack+0x20/0x24)
[1605418.629210] [<c020d5e0>] (show_stack) from [<c09b05e8>] (dump_stack+0xe0/0x124)
[1605418.629237] [<c09b05e8>] (dump_stack) from [<c0222544>] (__warn+0x104/0x11c)
[1605418.629255] [<c0222544>] (__warn) from [<c02225b4>] (warn_slowpath_fmt+0x58/0x74)
[1605418.629273] [<c02225b4>] (warn_slowpath_fmt) from [<c08d664c>] (dev_watchdog+0x294/0x298)
[1605418.629294] [<c08d664c>] (dev_watchdog) from [<c029b8c4>] (call_timer_fn+0x3c/0x198)
[1605418.629315] [<c029b8c4>] (call_timer_fn) from [<c029bb0c>] (expire_timers+0xec/0x14c)
[1605418.629333] [<c029bb0c>] (expire_timers) from [<c029bc24>] (run_timer_softirq+0xb8/0x1ec)
[1605418.629351] [<c029bc24>] (run_timer_softirq) from [<c0202410>] (__do_softirq+0x190/0x3f0)
[1605418.629369] [<c0202410>] (__do_softirq) from [<c02286cc>] (irq_exit+0xfc/0x120)
[1605418.629387] [<c02286cc>] (irq_exit) from [<c0281e18>] (__handle_domain_irq+0x70/0xc4)
[1605418.629402] [<c0281e18>] (__handle_domain_irq) from [<c0202244>] (gic_handle_irq+0x4c/0x88)
[1605418.629420] [<c0202244>] (gic_handle_irq) from [<c02019bc>] (__irq_svc+0x5c/0x7c)
[1605418.629429] Exception stack(0xc1001ed8 to 0xc1001f20)
[1605418.629447] 1ec0:                                                       c0209ae4 00000000
[1605418.629462] 1ee0: 40000093 40000093 c1004dbc c1000000 c1004e04 00000001 c109685a c10abf80
[1605418.629475] 1f00: c0e67a38 c1001f34 c10051c4 c1001f28 00000000 c0209ae8 40000013 ffffffff
[1605418.629495] [<c02019bc>] (__irq_svc) from [<c0209ae8>] (arch_cpu_idle+0x34/0x4c)
[1605418.629513] [<c0209ae8>] (arch_cpu_idle) from [<c09cd884>] (default_idle_call+0x34/0x48)
[1605418.629532] [<c09cd884>] (default_idle_call) from [<c0254a5c>] (do_idle+0xec/0x16c)
[1605418.629548] [<c0254a5c>] (do_idle) from [<c0254d9c>] (cpu_startup_entry+0x28/0x2c)
[1605418.629565] [<c0254d9c>] (cpu_startup_entry) from [<c09c6f18>] (rest_init+0xbc/0xc0)
[1605418.629584] [<c09c6f18>] (rest_init) from [<c0e00fe4>] (start_kernel+0x4b8/0x4e8)
[1605418.629602] ---[ end trace 7ab266a8281fe475 ]---
```

ablacksheep, 2020-02-18:
Same again, so incr counter.

> Same port crashed again, not sure if it is a pattern after 2 crashes, but so far it is the one plugged in the usb2 port that crash. The one pluggen in the usb3 seems to work fine. Might not be [...]

Last known traffic on interface was Feb 14 19:34:55.
Crash happend when i tried to pass traffic today at Feb 18.
`15:49:29 up 3 days, 20:16, 1 user, load average: 0,00, 0,00, 0,00`

**Some wild guessing.. Could this be an usb power related issue?**
Something like the device deadlocks when power control is trying to suspend it?

```kernel: [332213.777314] ------------[ cut here ]------------
kernel: [332213.777361] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:466 dev_watchdog+0x294/0x298
kernel: [332213.777383] NETDEV WATCHDOG: enx00249b1e558a (ax88179_178a): transmit queue 0 timed out
kernel: [332213.777402] Modules linked in: nf_conntrack_netlink xt_multiport tun bridge 8021q garp stp llc rtc_ds1307 ax88179_178a nf_log_ipv6 ip6t_REJECT nf_reject_ipv6 vc4 xt_hl brcmfmac ip6_tables drm_kms_helper v3d brcmutil gpu_sched ip6t_rt drm raspberrypi_hwmon hwmon drm_panel_orientation_quirks sha256_generic snd_soc_core snd_compress snd_pcm_dmaengine cfg80211 snd_bcm2835(C) snd_pcm bcm2835_codec(C) nf_log_ipv4 snd_timer nf_log_common syscopyarea sysfillrect rfkill sysimgblt v4l2_mem2mem fb_sys_fops snd videobuf2_dma_contig bcm2835_v4l2(C) ipt_REJECT nf_reject_ipv4 bcm2835_mmal_vchiq(C) v4l2_common i2c_bcm2835 videobuf2_vmalloc xt_LOG videobuf2_memops videobuf2_v4l2 videobuf2_common videodev media vc_sm_cma(C) rpivid_mem nft_limit uio_pdrv_genirq uio xt_limit xt_addrtype nft_chain_nat_ipv4 xt_nat xt_tcpudp
kernel: [332213.777617]  ipt_MASQUERADE nf_nat_ipv4 xt_conntrack nft_compat nft_counter nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink i2c_dev ip_tables x_tables ipv6
kernel: [332213.777696] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        4.19.97-v7l+ #1294
kernel: [332213.777706] Hardware name: BCM2835
kernel: [332213.777738] [<c0212e04>] (unwind_backtrace) from [<c020d5e0>] (show_stack+0x20/0x24)
kernel: [332213.777761] [<c020d5e0>] (show_stack) from [<c09b15c8>] (dump_stack+0xe0/0x124)
kernel: [332213.777782] [<c09b15c8>] (dump_stack) from [<c0222544>] (__warn+0x104/0x11c)
kernel: [332213.777800] [<c0222544>] (__warn) from [<c02225b4>] (warn_slowpath_fmt+0x58/0x74)
kernel: [332213.777822] [<c02225b4>] (warn_slowpath_fmt) from [<c08d7624>] (dev_watchdog+0x294/0x298)
kernel: [332213.777844] [<c08d7624>] (dev_watchdog) from [<c029b8c4>] (call_timer_fn+0x3c/0x198)
kernel: [332213.777863] [<c029b8c4>] (call_timer_fn) from [<c029bb0c>] (expire_timers+0xec/0x14c)
kernel: [332213.777880] [<c029bb0c>] (expire_timers) from [<c029bc24>] (run_timer_softirq+0xb8/0x1ec)
kernel: [332213.777900] [<c029bc24>] (run_timer_softirq) from [<c0202410>] (__do_softirq+0x190/0x3f0)
kernel: [332213.777919] [<c0202410>] (__do_softirq) from [<c02286cc>] (irq_exit+0xfc/0x120)
kernel: [332213.777937] [<c02286cc>] (irq_exit) from [<c0281e18>] (__handle_domain_irq+0x70/0xc4)
kernel: [332213.777952] [<c0281e18>] (__handle_domain_irq) from [<c0202244>] (gic_handle_irq+0x4c/0x88)
kernel: [332213.777971] [<c0202244>] (gic_handle_irq) from [<c02019bc>] (__irq_svc+0x5c/0x7c)
kernel: [332213.777981] Exception stack(0xc1001ed8 to 0xc1001f20)
kernel: [332213.777993] 1ec0:                                                       c0209ae4 00000000
kernel: [332213.778007] 1ee0: 40000093 40000093 c1004dbc c1000000 c1004e04 00000001 c109689a c10abfc0
kernel: [332213.778023] 1f00: c0e67a38 c1001f34 c10051c4 c1001f28 00000000 c0209ae8 40000013 ffffffff
kernel: [332213.778043] [<c02019bc>] (__irq_svc) from [<c0209ae8>] (arch_cpu_idle+0x34/0x4c)
kernel: [332213.778061] [<c0209ae8>] (arch_cpu_idle) from [<c09ce864>] (default_idle_call+0x34/0x48)
kernel: [332213.778081] [<c09ce864>] (default_idle_call) from [<c0254a5c>] (do_idle+0xec/0x16c)
kernel: [332213.778098] [<c0254a5c>] (do_idle) from [<c0254d9c>] (cpu_startup_entry+0x28/0x2c)
kernel: [332213.778117] [<c0254d9c>] (cpu_startup_entry) from [<c09c7ef8>] (rest_init+0xbc/0xc0)
kernel: [332213.778139] [<c09c7ef8>] (rest_init) from [<c0e00fe4>] (start_kernel+0x4b8/0x4e8)
kernel: [332213.778156] ---[ end trace 0500650fdc7f74d3 ]---```

ablacksheep, 2020-02-18:
@JamesH65 
Created a close to identical testrig, it failed pretty quick. Same pattern here usb2 interfaced one crashed. Both interfaces has close to no traffic in this test rig(i suggest they have some initial dhcp client traffic), as i just connect the two usb interfaces together with a cable without sending any additional traffic over them.

However, when i did add the interfaces to different network namespaces, and sendt loads of traffic over them, then they stayed up healty.

[syslog-testrig2.txt](https://github.com/raspberrypi/linux/files/4220126/syslog-testrig2.txt)

To reproduce:
Raspberry Pi 4 Model B 4GB RAM
Get two adapters, plug one in the usb3 and one in the usb2 port. Connect a cable, so link is up. Wait.
Remote access to the testrig can be provided to the right person, if that person feels for helping debugging it. :)

Info:

```
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

Jan 22 2020 17:23:15 
Copyright (c) 2012 Broadcom
version acb03c7d18aa08f25f6419d696d0cc39761d3e62 (clean) (release) (start)

Linux raspberrypi 4.19.97-v7l+ #1293 SMP Wed Jan 22 17:16:14 GMT 2020 armv7l GNU/Linux

```

TimoErdmann, 2020-02-26:
I get a similar issue with a Raspberry Pi 3 B+

It runs as a headless system only connected to a webcam and an ehternet cable. It is powered via an POE Shield. 

`Feb 26 09:40:21 beta2-pi kernel: [20113.151890] ------------[ cut here ]------------
Feb 26 09:40:21 beta2-pi kernel: [20113.151964] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:466 dev_watchdog+0x294/0x298
Feb 26 09:40:21 beta2-pi kernel: [20113.151994] NETDEV WATCHDOG: eth0 (lan78xx): transmit queue 0 timed out
Feb 26 09:40:21 beta2-pi kernel: [20113.152004] Modules linked in: bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc evdev brcmfmac brcmutil snd_usb_audio snd_hwdep sha256_generic snd_usbmidi_lib snd_rawmidi uvcvideo snd_seq_device cfg80211 rfkill snd_bcm2835(C) raspberrypi_hwmon snd_pcm bcm2835_codec(C) v4l2_mem2mem snd_timer bcm2835_v4l2(C) snd videobuf2_dma_contig bcm2835_mmal_vchiq(C) v4l2_common videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common videodev vc_sm_cma(C) media rpi_poe_fan hwmon uio_pdrv_genirq uio fixed ip_tables x_tables ipv6
Feb 26 09:40:21 beta2-pi kernel: [20113.152208] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        4.19.102-v7+ #1295
Feb 26 09:40:21 beta2-pi kernel: [20113.152213] Hardware name: BCM2835
Feb 26 09:40:21 beta2-pi kernel: [20113.152242] [<801120c0>] (unwind_backtrace) from [<8010d5f4>] (show_stack+0x20/0x24)
Feb 26 09:40:21 beta2-pi kernel: [20113.152260] [<8010d5f4>] (show_stack) from [<80849c68>] (dump_stack+0xe0/0x124)
Feb 26 09:40:21 beta2-pi kernel: [20113.152279] [<80849c68>] (dump_stack) from [<80120c9c>] (__warn+0x104/0x11c)
Feb 26 09:40:21 beta2-pi kernel: [20113.152294] [<80120c9c>] (__warn) from [<80120d0c>] (warn_slowpath_fmt+0x58/0x74)
Feb 26 09:40:21 beta2-pi kernel: [20113.152311] [<80120d0c>] (warn_slowpath_fmt) from [<8076f218>] (dev_watchdog+0x294/0x298)
Feb 26 09:40:21 beta2-pi kernel: [20113.152329] [<8076f218>] (dev_watchdog) from [<80199044>] (call_timer_fn+0x3c/0x198)
Feb 26 09:40:21 beta2-pi kernel: [20113.152345] [<80199044>] (call_timer_fn) from [<8019928c>] (expire_timers+0xec/0x14c)
Feb 26 09:40:21 beta2-pi kernel: [20113.152359] [<8019928c>] (expire_timers) from [<801993a4>] (run_timer_softirq+0xb8/0x1ec)
Feb 26 09:40:21 beta2-pi kernel: [20113.152375] [<801993a4>] (run_timer_softirq) from [<80102410>] (__do_softirq+0x190/0x3f0)
Feb 26 09:40:21 beta2-pi kernel: [20113.152392] [<80102410>] (__do_softirq) from [<80126e24>] (irq_exit+0xfc/0x120)
Feb 26 09:40:21 beta2-pi kernel: [20113.152409] [<80126e24>] (irq_exit) from [<80180270>] (__handle_domain_irq+0x70/0xc4)
Feb 26 09:40:21 beta2-pi kernel: [20113.152424] [<80180270>] (__handle_domain_irq) from [<801021b4>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
Feb 26 09:40:21 beta2-pi kernel: [20113.152439] [<801021b4>] (bcm2836_arm_irqchip_handle_irq) from [<801019bc>] (__irq_svc+0x5c/0x7c)
Feb 26 09:40:21 beta2-pi kernel: [20113.152446] Exception stack(0x80d01ee8 to 0x80d01f30)
Feb 26 09:40:21 beta2-pi kernel: [20113.152458] 1ee0:                   80109ae4 00000000 40000093 40000093 80d04d70 80d00000
Feb 26 09:40:21 beta2-pi kernel: [20113.152471] 1f00: 80d04db8 00000001 80d8efbe babff9c0 80c64a38 80d01f44 80d0517c 80d01f38
Feb 26 09:40:21 beta2-pi kernel: [20113.152480] 1f20: 00000000 80109ae8 40000013 ffffffff
Feb 26 09:40:21 beta2-pi kernel: [20113.152497] [<801019bc>] (__irq_svc) from [<80109ae8>] (arch_cpu_idle+0x34/0x4c)
Feb 26 09:40:21 beta2-pi kernel: [20113.152515] [<80109ae8>] (arch_cpu_idle) from [<80866d54>] (default_idle_call+0x34/0x48)
Feb 26 09:40:21 beta2-pi kernel: [20113.152533] [<80866d54>] (default_idle_call) from [<80152e68>] (do_idle+0xec/0x16c)
Feb 26 09:40:21 beta2-pi kernel: [20113.152552] [<80152e68>] (do_idle) from [<801531a8>] (cpu_startup_entry+0x28/0x2c)
Feb 26 09:40:21 beta2-pi kernel: [20113.152570] [<801531a8>] (cpu_startup_entry) from [<80860400>] (rest_init+0xbc/0xc0)
Feb 26 09:40:21 beta2-pi kernel: [20113.152589] [<80860400>] (rest_init) from [<80c00fb0>] (start_kernel+0x484/0x4b4)
Feb 26 09:40:21 beta2-pi kernel: [20113.152620] ---[ end trace 6a98d99e7596fecd ]---
Feb 26 09:40:21 beta2-pi kernel: [20113.153022] WARN::dwc_otg_hcd_urb_dequeue:639: Timed out waiting for FSM NP transfer to complete on 2
Feb 26 09:40:21 beta2-pi kernel: [20113.154499] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 09:40:21 beta2-pi kernel: [20113.154638] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 09:40:21 beta2-pi kernel: [20113.154773] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 09:40:21 beta2-pi kernel: [20113.154914] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 09:40:21 beta2-pi kernel: [20113.155057] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 09:40:21 beta2-pi kernel: [20113.155195] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 09:40:21 beta2-pi kernel: [20113.155320] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
Feb 26 09:40:21 beta2-pi kernel: [20113.155772] lan78xx 1-1.1.1:1.0 eth0: kevent 0 may have been dropped
`

It becomes completly unresponsive over the network. The last message is repeated until a reboot by plugging it out and in again. 

 uname -a:
`Linux beta2-pi 4.19.102-v7+ #1295 SMP Thu Feb 6 15:43:59 GMT 2020 armv7l GNU/Linux`

ablacksheep, 2020-02-26:
> I get a similar issue with a Raspberry Pi 3 B+
> 
> It runs as a headless system only connected to a webcam and an ehternet cable. It is powered via an POE Shield.
> 

I suspect it is related.

It seems to have the usb2 bus in common. As far as i remember the pi3b+ ethernet is provided via onboard chip connected to the **USB 2.0 bus** and so far holds true for my p4 testing above. it is only the adapter connected to the usb2 bus that crashes.

ablacksheep, 2020-02-26:
Updated test-rig yesterday(25.02.2020), apt update && apt full-upgrade and rpi-update. Crashed some hours later..  adapter plugged in usb2 port.

```
Feb 26 04:50:17 raspberrypi kernel: [19287.339544] ------------[ cut here ]------------
Feb 26 04:50:17 raspberrypi kernel: [19287.339592] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:466 dev_watchdog+0x294/0x298
Feb 26 04:50:17 raspberrypi kernel: [19287.339619] NETDEV WATCHDOG: enx00249b5b748d (ax88179_178a): transmit queue 0 timed out
Feb 26 04:50:17 raspberrypi kernel: [19287.339634] Modules linked in: xt_multiport bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc sg brcmfmac ax88179_178a vc4 brcmutil drm_kms_helper sha256_generic snd_soc_core snd_compress cfg80211 v3d snd_pcm_dmaengine gpu_sched syscopyarea bcm2835_codec(C) sysfillrect rfkill bcm2835_v4l2(C) snd_bcm2835(C) v4l2_mem2mem drm bcm2835_mmal_vchiq(C) snd_pcm sysimgblt videobuf2_dma_contig v4l2_common videobuf2_vmalloc drm_panel_orientation_quirks videobuf2_memops snd_timer fb_sys_fops videobuf2_v4l2 raspberrypi_hwmon videobuf2_common hwmon snd videodev media vc_sm_cma(C) rpivid_mem uio_pdrv_genirq uio nf_log_ipv6 ip6t_REJECT nf_reject_ipv6 xt_hl ip6_tables ip6t_rt nf_log_ipv4 nf_log_common ipt_REJECT nf_reject_ipv4 xt_LOG nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nft_compat
Feb 26 04:50:17 raspberrypi kernel: [19287.339856]  nft_counter nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink ip_tables x_tables ipv6
Feb 26 04:50:17 raspberrypi kernel: [19287.339921] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        4.19.105-v7l+ #1296
Feb 26 04:50:17 raspberrypi kernel: [19287.339933] Hardware name: BCM2835
Feb 26 04:50:17 raspberrypi kernel: [19287.339961] [<c0212e04>] (unwind_backtrace) from [<c020d5e0>] (show_stack+0x20/0x24)
Feb 26 04:50:17 raspberrypi kernel: [19287.339981] [<c020d5e0>] (show_stack) from [<c09b63a8>] (dump_stack+0xe0/0x124)
Feb 26 04:50:17 raspberrypi kernel: [19287.340004] [<c09b63a8>] (dump_stack) from [<c0222548>] (__warn+0x104/0x11c)
Feb 26 04:50:17 raspberrypi kernel: [19287.340021] [<c0222548>] (__warn) from [<c02225b8>] (warn_slowpath_fmt+0x58/0x74)
Feb 26 04:50:17 raspberrypi kernel: [19287.340040] [<c02225b8>] (warn_slowpath_fmt) from [<c08dc098>] (dev_watchdog+0x294/0x298)
Feb 26 04:50:17 raspberrypi kernel: [19287.340065] [<c08dc098>] (dev_watchdog) from [<c029b95c>] (call_timer_fn+0x3c/0x198)
Feb 26 04:50:17 raspberrypi kernel: [19287.340083] [<c029b95c>] (call_timer_fn) from [<c029bba4>] (expire_timers+0xec/0x14c)
Feb 26 04:50:17 raspberrypi kernel: [19287.340098] [<c029bba4>] (expire_timers) from [<c029bcbc>] (run_timer_softirq+0xb8/0x1ec)
Feb 26 04:50:17 raspberrypi kernel: [19287.340118] [<c029bcbc>] (run_timer_softirq) from [<c0202410>] (__do_softirq+0x190/0x3f0)
Feb 26 04:50:17 raspberrypi kernel: [19287.340139] [<c0202410>] (__do_softirq) from [<c02286d0>] (irq_exit+0xfc/0x120)
Feb 26 04:50:17 raspberrypi kernel: [19287.340157] [<c02286d0>] (irq_exit) from [<c0281e54>] (__handle_domain_irq+0x70/0xc4)
Feb 26 04:50:17 raspberrypi kernel: [19287.340173] [<c0281e54>] (__handle_domain_irq) from [<c0202244>] (gic_handle_irq+0x4c/0x88)
Feb 26 04:50:17 raspberrypi kernel: [19287.340187] [<c0202244>] (gic_handle_irq) from [<c02019bc>] (__irq_svc+0x5c/0x7c)
Feb 26 04:50:17 raspberrypi kernel: [19287.340204] Exception stack(0xc1001ed8 to 0xc1001f20)
Feb 26 04:50:17 raspberrypi kernel: [19287.340219] 1ec0:                                                       c0209ae4 00000000
Feb 26 04:50:17 raspberrypi kernel: [19287.340232] 1ee0: 40000093 40000093 c1004dbc c1000000 c1004e04 00000001 c10968da c10abfc0
Feb 26 04:50:17 raspberrypi kernel: [19287.340244] 1f00: c0e67a38 c1001f34 c10051c4 c1001f28 00000000 c0209ae8 40000013 ffffffff
Feb 26 04:50:17 raspberrypi kernel: [19287.340260] [<c02019bc>] (__irq_svc) from [<c0209ae8>] (arch_cpu_idle+0x34/0x4c)
Feb 26 04:50:17 raspberrypi kernel: [19287.340279] [<c0209ae8>] (arch_cpu_idle) from [<c09d3644>] (default_idle_call+0x34/0x48)
Feb 26 04:50:17 raspberrypi kernel: [19287.340300] [<c09d3644>] (default_idle_call) from [<c0254a48>] (do_idle+0xec/0x16c)
Feb 26 04:50:17 raspberrypi kernel: [19287.340316] [<c0254a48>] (do_idle) from [<c0254d88>] (cpu_startup_entry+0x28/0x2c)
Feb 26 04:50:17 raspberrypi kernel: [19287.340333] [<c0254d88>] (cpu_startup_entry) from [<c09cccd8>] (rest_init+0xbc/0xc0)
Feb 26 04:50:17 raspberrypi kernel: [19287.340352] [<c09cccd8>] (rest_init) from [<c0e00fe4>] (start_kernel+0x4b8/0x4e8)
Feb 26 04:50:17 raspberrypi kernel: [19287.340369] ---[ end trace ca8f5cbe348fd389 ]---
```

ablacksheep, 2020-03-22:
Tested driver #2999  mentioned in #2942, so far it has given 17 days uptime with no fault. One thing i noticed is that they disable hub_initiated_lpm. They prob also does more stuff, but I havn't had time to look into that. 

For the upstream driver it might look like this is a pattern,
it seems like "transmit queue 0 timed out" typical happens when when usbcore or the devices initiate lpm and the device lock up because of failed or incorrect pre requirements before lpm can be achieved. I tested by incr time before lpm and by disabled lpm on usb2, and the timeout crashes stopped happening. However, without the crashes the device still locked up. It looked like something still activating the lpm state of the device, maybe the device itself?

If someone is annoyed by the issue with "ID 0b95:1790 ASIX Electronics Corp. AX88179" and the Pi4¿ It can simply be worked around by forcing usb2 device not to enter lpm by running an endless ping with standard 1 second interval. (make sure you ping something that leaves the correct interface.)

ablacksheep, 2020-04-15:
` 10:45:53 up 15 days, 13:19,  1 user,  load average: 0,02, 0,05, 0,00`
With no bad behavior on test rig after moving both adapters to the usb3 ports running original kernel shipped driver module. I will test moving 1 or both the modules back to usb2 ports. I expect it to crash fairly quick.

ablacksheep, 2020-05-13:
> I expect it to crash fairly quick.
I messed up the testrig, will have it back running during the weekend.

ablacksheep, 2020-06-03:
Tested with a newer kernel, still crashing. But I discovered that the crash of the device(AX88179) in the usb2.port03 **seemed** to be dependent on which of the usb3 ports the other device(AX88179) was plugged into. The device plugged in usb2 crashed(fast aprox6h.) when other device was plugged in usb3.port00. Moving the device in the usb3  back to usb3.port01 to check if device in usb2 still crashes, and observed pattern of not crashing in usb3.port01 was just a coincidence.

```
Linux raspberrypi 4.19.122-v7l-24.05.2020+ #1 SMP Sun May 24 13:13:51 BST 2020 armv7l GNU/Linux
Jun  3 10:30:58 raspberrypi kernel: [    0.000000] Booting Linux on physical CPU 0x0
Jun  3 15:50:43 raspberrypi kernel: [19177.332749] ------------[ cut here ]------------
Jun  3 15:50:43 raspberrypi kernel: [19177.332793] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:466 dev_watchdog+0x298/0x29c
Jun  3 15:50:43 raspberrypi kernel: [19177.332815] NETDEV WATCHDOG: enx00249b2c7dfb (ax88179_178a): transmit queue 0 timed out
Jun  3 15:50:43 raspberrypi kernel: [19177.332832] Modules linked in: xt_multiport bnep hci_uart btbcm serdev bluetooth ecdh_generic 8021q garp stp llc ax88179_178a brcmfmac brcmutil vc4 drm_kms_helper v3d gpu_sched bcm2835_codec(C) cfg80211 rfkill v4l2_mem2mem bcm2835_v4l2(C) videobuf2_dma_contig bcm2835_mmal_vchiq(C) v4l2_common raspberrypi_hwmon videobuf2_vmalloc hwmon videobuf2_memops videobuf2_v4l2 drm videobuf2_common drm_panel_orientation_quirks videodev snd_soc_core media snd_bcm2835(C) vc_sm_cma(C) snd_compress snd_pcm_dmaengine snd_pcm syscopyarea sysfillrect snd_timer sysimgblt fb_sys_fops snd ip6t_REJECT nf_reject_ipv6 rpivid_mem uio_pdrv_genirq uio xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nft_compat nft_counter nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp
Jun  3 15:50:43 raspberrypi kernel: [19177.333054]  nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nf_tables nfnetlink ip_tables x_tables ipv6
Jun  3 15:50:43 raspberrypi kernel: [19177.333111] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        4.19.122-v7l-24.05.2020+ #1
Jun  3 15:50:43 raspberrypi kernel: [19177.333120] Hardware name: BCM2835
Jun  3 15:50:43 raspberrypi kernel: [19177.333151] [<c0212d5c>] (unwind_backtrace) from [<c020d370>] (show_stack+0x20/0x24)
Jun  3 15:50:43 raspberrypi kernel: [19177.333171] [<c020d370>] (show_stack) from [<c09c0034>] (dump_stack+0xd8/0x11c)
Jun  3 15:50:43 raspberrypi kernel: [19177.333192] [<c09c0034>] (dump_stack) from [<c02224d8>] (__warn.part.3+0xcc/0xe8)
Jun  3 15:50:43 raspberrypi kernel: [19177.333210] [<c02224d8>] (__warn.part.3) from [<c022256c>] (warn_slowpath_fmt+0x78/0x94)
Jun  3 15:50:43 raspberrypi kernel: [19177.333227] [<c022256c>] (warn_slowpath_fmt) from [<c08e6e88>] (dev_watchdog+0x298/0x29c)
Jun  3 15:50:43 raspberrypi kernel: [19177.333250] [<c08e6e88>] (dev_watchdog) from [<c029c048>] (call_timer_fn+0x3c/0x1a4)
Jun  3 15:50:43 raspberrypi kernel: [19177.333269] [<c029c048>] (call_timer_fn) from [<c029c2a8>] (expire_timers+0xf8/0x168)
Jun  3 15:50:43 raspberrypi kernel: [19177.333286] [<c029c2a8>] (expire_timers) from [<c029c3cc>] (run_timer_softirq+0xb4/0x1e4)
Jun  3 15:50:43 raspberrypi kernel: [19177.333302] [<c029c3cc>] (run_timer_softirq) from [<c02023f8>] (__do_softirq+0x188/0x410)
Jun  3 15:50:43 raspberrypi kernel: [19177.333324] [<c02023f8>] (__do_softirq) from [<c0228520>] (irq_exit+0xf8/0x134)
Jun  3 15:50:43 raspberrypi kernel: [19177.333346] [<c0228520>] (irq_exit) from [<c02825dc>] (__handle_domain_irq+0x70/0xc4)
Jun  3 15:50:43 raspberrypi kernel: [19177.333364] [<c02825dc>] (__handle_domain_irq) from [<c0202234>] (gic_handle_irq+0x4c/0x88)
Jun  3 15:50:43 raspberrypi kernel: [19177.333380] [<c0202234>] (gic_handle_irq) from [<c02019bc>] (__irq_svc+0x5c/0x7c)
Jun  3 15:50:43 raspberrypi kernel: [19177.333389] Exception stack(0xc1001ed0 to 0xc1001f18)
Jun  3 15:50:43 raspberrypi kernel: [19177.333400] 1ec0:                                     c0209ab4 00000000 40000093 40000093
Jun  3 15:50:43 raspberrypi kernel: [19177.333412] 1ee0: ffffe000 c1004db8 c1004e00 00000001 00000001 c109729a c0b94234 c1001f2c
Jun  3 15:50:43 raspberrypi kernel: [19177.333427] 1f00: c1000000 c1001f20 00000000 c0209ab8 40000013 ffffffff
Jun  3 15:50:43 raspberrypi kernel: [19177.333447] [<c02019bc>] (__irq_svc) from [<c0209ab8>] (arch_cpu_idle+0x34/0x4c)
Jun  3 15:50:43 raspberrypi kernel: [19177.333471] [<c0209ab8>] (arch_cpu_idle) from [<c09dbcf8>] (default_idle_call+0x40/0x48)
Jun  3 15:50:43 raspberrypi kernel: [19177.333493] [<c09dbcf8>] (default_idle_call) from [<c02549c0>] (do_idle+0x124/0x164)
Jun  3 15:50:43 raspberrypi kernel: [19177.333509] [<c02549c0>] (do_idle) from [<c0254cc0>] (cpu_startup_entry+0x28/0x2c)
Jun  3 15:50:43 raspberrypi kernel: [19177.333523] [<c0254cc0>] (cpu_startup_entry) from [<c09d537c>] (rest_init+0xb8/0xbc)
Jun  3 15:50:43 raspberrypi kernel: [19177.333541] [<c09d537c>] (rest_init) from [<c0e01000>] (start_kernel+0x4b0/0x4e0)
Jun  3 15:50:43 raspberrypi kernel: [19177.333559] ---[ end trace 3831669ada16d78b ]---

```

ablacksheep, 2020-06-04:
> Moving the device in the usb3 back to usb3.port01 to check if device in usb2 still crashes, and observed pattern of not crashing in usb3.port01 was just a coincidence.

It was just a was a coincidence. It keeps crashing. :)

shaohaoz, 2020-06-11:
I've had similar issue on RPI3B+. Any updates?

pelwell, 2020-06-11:
What do you mean by "similar issue"? What external hardware are you using, and what does the kernel log show?

ablacksheep, 2020-08-02:
`Linux raspberrypi 5.4.51-v7l+ #1327 SMP Thu Jul 23 11:04:39 BST 2020 armv7l GNU/Linux`

```

[   11.591708] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   11.642833] ax88179_178a 1-1.4:1.0 enx00249b2c7dfb: ax88179 - Link status is: 1
[   11.649467] IPv6: ADDRCONF(NETDEV_CHANGE): enx00249b2c7dfb: link becomes ready
[   11.675800] ax88179_178a 2-1:1.0 enx00249b5b748d: ax88179 - Link status is: 1
[   11.682343] IPv6: ADDRCONF(NETDEV_CHANGE): enx00249b5b748d: link becomes ready

[15255.045089] ax88179_178a 2-1:1.0 enx00249b5b748d: ax88179 - Link status is: 0
[19342.353100] ------------[ cut here ]------------
[19342.353148] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x310/0x314
[19342.353170] NETDEV WATCHDOG: enx00249b2c7dfb (ax88179_178a): transmit queue 0 timed out
[19342.353187] Modules linked in: xt_multiport bnep hci_uart btbcm bluetooth ecdh_generic ecc 8021q garp stp llc brcmfmac brcmutil ax88179_178a sha256_generic libsha256 vc4 cfg80211 cec rfkill v3d gpu_sched drm_kms_helper raspberrypi_hwmon drm bcm2835_codec(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_v4l2(C) videobuf2_dma_contig drm_panel_orientation_quirks bcm2835_mmal_vchiq(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_soc_core videobuf2_common snd_compress snd_bcm2835(C) snd_pcm_dmaengine snd_pcm videodev mc snd_timer vc_sm_cma(C) snd syscopyarea sysfillrect rpivid_mem sysimgblt fb_sys_fops ip6t_REJECT nf_reject_ipv6 uio_pdrv_genirq uio xt_hl ip6_tables ip6t_rt ipt_REJECT nf_reject_ipv4 nft_limit xt_limit xt_addrtype xt_tcpudp xt_conntrack nft_compat nft_counter nf_conntrack_netbios_ns nf_conntrack_broadcast nf_nat_ftp nf_nat nf_conntrack_ftp nf_conntrack nf_defrag_ipv4 nf_tables nfnetlink ip_tables x_tables ipv6 nf_defrag_ipv6
[19342.353518] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.51-v7l+ #1327
[19342.353526] Hardware name: BCM2711
[19342.353534] Backtrace: 
[19342.353565] [<c020d46c>] (dump_backtrace) from [<c020d768>] (show_stack+0x20/0x24)
[19342.353579]  r6:c1200000 r5:00000000 r4:c129c8f8 r3:3b427126
[19342.353600] [<c020d748>] (show_stack) from [<c0a39a24>] (dump_stack+0xe0/0x124)
[19342.353618] [<c0a39944>] (dump_stack) from [<c0221c50>] (__warn+0xec/0x104)
[19342.353631]  r8:000001c0 r7:00000009 r6:c0e375c8 r5:00000000 r4:c1201cd4 r3:3b427126
[19342.353648] [<c0221b64>] (__warn) from [<c0221cec>] (warn_slowpath_fmt+0x84/0xc0)
[19342.353660]  r9:c0e375c8 r8:000001c0 r7:c09476cc r6:00000009 r5:c0e375e0 r4:c1204f88
[19342.353675] [<c0221c6c>] (warn_slowpath_fmt) from [<c09476cc>] (dev_watchdog+0x310/0x314)
[19342.353687]  r9:001d0e50 r8:d6aa3800 r7:00000000 r6:c1203d00 r5:d6aa3aa8 r4:00000000
[19342.353704] [<c09473bc>] (dev_watchdog) from [<c02a05bc>] (call_timer_fn+0x40/0x180)
[19342.353716]  r8:c09473bc r7:00000100 r6:c1200000 r5:d6aa3aa8 r4:eff0c440
[19342.353732] [<c02a057c>] (call_timer_fn) from [<c02a170c>] (run_timer_softirq+0x288/0x654)
[19342.353744]  r9:00000000 r8:d6aa3aa8 r7:c1200000 r6:001d0e50 r5:c1201dac r4:eff0c440
[19342.353758] [<c02a1484>] (run_timer_softirq) from [<c020249c>] (__do_softirq+0x1a4/0x418)
[19342.353770]  r10:00000004 r9:00000282 r8:ef848800 r7:00000100 r6:c1200000 r5:00000001
[19342.353780]  r4:c1203084
[19342.353799] [<c02022f8>] (__do_softirq) from [<c0227d54>] (irq_exit+0x100/0x110)
[19342.353812]  r10:0000014e r9:c1200000 r8:ef848800 r7:00000001 r6:00000000 r5:00000000
[19342.353822]  r4:c10a82e4
[19342.353840] [<c0227c54>] (irq_exit) from [<c028302c>] (__handle_domain_irq+0x70/0xc4)
[19342.353862] [<c0282fbc>] (__handle_domain_irq) from [<c02022b8>] (gic_handle_irq+0x4c/0x88)
[19342.353874]  r8:f0815000 r7:f0814000 r6:c1201ed0 r5:f081400c r4:c1205a14 r3:c1201ed0
[19342.353888] [<c020226c>] (gic_handle_irq) from [<c0201a3c>] (__irq_svc+0x5c/0x7c)
[19342.353898] Exception stack(0xc1201ed0 to 0xc1201f18)
[19342.353909] 1ec0:                                     c0209b94 00000000 60000093 c021c160
[19342.353921] 1ee0: c1204fb4 c1200000 c1204ffc 00000001 c12a32ed c106ca48 0000014e c1201f2c
[19342.353936] 1f00: c12053ac c1201f20 00000000 c0209b98 60000013 ffffffff
[19342.353951]  r8:c12a32ed r7:c1201f04 r6:ffffffff r5:60000013 r4:c0209b98 r3:3b427126
[19342.353973] [<c0209b64>] (arch_cpu_idle) from [<c0a5aa4c>] (default_idle_call+0x34/0x48)
[19342.353989] [<c0a5aa18>] (default_idle_call) from [<c0255b00>] (do_idle+0xec/0x170)
[19342.354002] [<c0255a14>] (do_idle) from [<c0255e60>] (cpu_startup_entry+0x28/0x2c)
[19342.354015]  r8:ffffffff r7:c12b5680 r6:c106ca48 r5:c12b56d8 r4:000000ce r3:c1201f48
[19342.354031] [<c0255e38>] (cpu_startup_entry) from [<c0a53e54>] (rest_init+0xbc/0xc0)
[19342.354047] [<c0a53d98>] (rest_init) from [<c1000bec>] (arch_call_rest_init+0x18/0x1c)
[19342.354056]  r4:c1204f88 r3:c109d270
[19342.354068] [<c1000bd4>] (arch_call_rest_init) from [<c10010d4>] (start_kernel+0x4e4/0x518)
[19342.354081] [<c1000bf0>] (start_kernel) from [<00000000>] (0x0)
[19342.354097] ---[ end trace e35d0439ee9b8b60 ]---

```

paralin, 2020-08-02:
I have the same `handle_IPI` / `cpu_startup_entry` failure on linux kernel 5.4.51 33115317f0006e15418396acef7bd6f9dc99b91e and firmware bd816dbac723e04f8f8b06bb2e16d767cda7692c

syphr42, 2020-08-29:
I've got a similar issue on Raspberry Pi 3B+. On Stretch everything is fine. I installed fresh with the same config on Buster (using Ansible for configuration) and it seemed like the network would randomly go down. I managed to find a way to reproduce it by running `sudo apt-get update` where apt calls go through a proxy located on the local network. This causes the stack trace every time. If I comment out the proxy, everything is fine with the apt update.

Other scenarios cause the same stack trace, but I have not been able to reproduce them consistently. While using tcpdump over SSH this error happens sometimes. Sometimes while tailing a log file over SSH it happens. Sometimes an apt update on another node on the network causes it. Other than the scenario above, nothing seems to consistently cause it.

The PI is setup as a router with the onboard NIC plus a TrendNET USB ethernet adapter. The onboard is on the LAN side and the USB ethernet adapter is on the WAN side. 802.1Q VLAN tagging on the LAN side with virtual interfaces as well.

```
$ uname -r
5.4.51-v7+
```

```
kernel: [ 4777.047109] ------------[ cut here ]------------
kernel: [ 4777.047169] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x310/0x314
kernel: [ 4777.047184] NETDEV WATCHDOG: wan0 (ax88179_178a): transmit queue 0 timed out
kernel: [ 4777.047196] Modules linked in: cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc nf_log_ipv6 ip6_tables nft_chain_nat xt_nat xt_MASQUERADE nf_nat xt_HL nf_log_ipv4 nf_log_common nft_limit nft_counter xt_LOG xt_limit xt_mac xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv4 nft_compat nf_tables nfnetlink ax88179_178a evdev brcmfmac brcmutil raspberrypi_hwmon sha256_generic libsha256 cfg80211 rfkill bcm2835_codec(C) bcm2835_isp(C) bcm2835_v4l2(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_bcm2835(C) videobuf2_common vc_sm_cma(C) snd_pcm snd_timer videodev snd mc 8021q garp stp llc uio_pdrv_genirq uio fixed ip_tables x_tables ipv6 nf_defrag_ipv6
kernel: [ 4777.047470] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.51-v7+ #1333
kernel: [ 4777.047475] Hardware name: BCM2835
kernel: [ 4777.047479] Backtrace: 
kernel: [ 4777.047498] [<8010d480>] (dump_backtrace) from [<8010d778>] (show_stack+0x20/0x24)
kernel: [ 4777.047508]  r6:80d00000 r5:00000000 r4:80d95cb8 r3:96b3c785
kernel: [ 4777.047522] [<8010d758>] (show_stack) from [<808cdbe4>] (dump_stack+0xe0/0x124)
kernel: [ 4777.047536] [<808cdb04>] (dump_stack) from [<8012002c>] (__warn+0xec/0x104)
kernel: [ 4777.047548]  r8:000001c0 r7:00000009 r6:80b0f6a4 r5:00000000 r4:80d01cec r3:96b3c785
kernel: [ 4777.047560] [<8011ff40>] (__warn) from [<801200c8>] (warn_slowpath_fmt+0x84/0xc0)
kernel: [ 4777.047572]  r9:80b0f6a4 r8:000001c0 r7:807db124 r6:00000009 r5:80b0f6bc r4:80d04f48
kernel: [ 4777.047587] [<80120048>] (warn_slowpath_fmt) from [<807db124>] (dev_watchdog+0x310/0x314)
kernel: [ 4777.047598]  r9:0006d4d8 r8:b3eb8800 r7:00000000 r6:80d03d00 r5:b3eb8aa8 r4:00000000
kernel: [ 4777.047615] [<807dae14>] (dev_watchdog) from [<8019be6c>] (call_timer_fn+0x40/0x180)
kernel: [ 4777.047626]  r8:807dae14 r7:00000100 r6:80d00000 r5:b3eb8aa8 r4:b6b25440
kernel: [ 4777.047639] [<8019be2c>] (call_timer_fn) from [<8019cfbc>] (run_timer_softirq+0x288/0x654)
kernel: [ 4777.047650]  r9:00000000 r8:b3eb8aa8 r7:80d00000 r6:0006d4d8 r5:80d01dc4 r4:b6b25440
kernel: [ 4777.047666] [<8019cd34>] (run_timer_softirq) from [<8010249c>] (__do_softirq+0x1a4/0x418)
kernel: [ 4777.047678]  r10:00000004 r9:00000082 r8:b645e000 r7:00000100 r6:80d00000 r5:00000001
kernel: [ 4777.047682]  r4:80d03084
kernel: [ 4777.047697] [<801022f8>] (__do_softirq) from [<80126130>] (irq_exit+0x100/0x110)
kernel: [ 4777.047708]  r10:b73ff900 r9:80d00000 r8:b645e000 r7:00000001 r6:00000000 r5:00000000
kernel: [ 4777.047713]  r4:80ca42e4
kernel: [ 4777.047726] [<80126030>] (irq_exit) from [<80181200>] (__handle_domain_irq+0x70/0xc4)
kernel: [ 4777.047738] [<80181190>] (__handle_domain_irq) from [<80102228>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
kernel: [ 4777.047750]  r8:80d9b18d r7:80d01f0c r6:ffffffff r5:60000013 r4:00000000 r3:80d01ed8
kernel: [ 4777.047761] [<801021c8>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
kernel: [ 4777.047775] Exception stack(0x80d01ed8 to 0x80d01f20)
kernel: [ 4777.047785] 1ec0:                                                       80109b94 00000000
kernel: [ 4777.047796] 1ee0: 60000093 8011a520 80d04f68 80d00000 80d04fb0 00000001 80d9b18d 80c69a40
kernel: [ 4777.047807] 1f00: b73ff900 80d01f34 80d0535c 80d01f28 00000000 80109b98 60000013 ffffffff
kernel: [ 4777.047813]  r4:80109b98 r3:96b3c785
kernel: [ 4777.047830] [<80109b64>] (arch_cpu_idle) from [<808eea44>] (default_idle_call+0x34/0x48)
kernel: [ 4777.047843] [<808eea10>] (default_idle_call) from [<80153bc0>] (do_idle+0xec/0x170)
kernel: [ 4777.047854] [<80153ad4>] (do_idle) from [<80153f20>] (cpu_startup_entry+0x28/0x2c)
kernel: [ 4777.047865]  r8:00000000 r7:80c69a40 r6:80dac5c0 r5:80dac618 r4:000000ce r3:80d01f50
kernel: [ 4777.047877] [<80153ef8>] (cpu_startup_entry) from [<808e7e68>] (rest_init+0xbc/0xc0)
kernel: [ 4777.047890] [<808e7dac>] (rest_init) from [<80c00bd8>] (arch_call_rest_init+0x18/0x1c)
kernel: [ 4777.047897]  r4:80d04f48 r3:80c99260
kernel: [ 4777.047907] [<80c00bc0>] (arch_call_rest_init) from [<80c010a8>] (start_kernel+0x4cc/0x500)
kernel: [ 4777.047917] [<80c00bdc>] (start_kernel) from [<00000000>] (0x0)
kernel: [ 4777.047953] ---[ end trace 9a5a0325809920e6 ]---
```

syphr42, 2020-08-29:
As an additional data point, I have an older 10/100 USB ethernet adapter with a Realtek chip and the same crash occurs.

```
Bus 001 Device 004: ID 0bda:8150 Realtek Semiconductor Corp. RTL8150 Fast Ethernet Adapter
```

```
kernel: [  336.076919] ------------[ cut here ]------------
kernel: [  336.076977] WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x310/0x314
kernel: [  336.077002] NETDEV WATCHDOG: lan0 (lan78xx): transmit queue 0 timed out
kernel: [  336.077015] Modules linked in: cmac bnep hci_uart btbcm bluetooth ecdh_generic ecc nf_log_ipv6 ip6_tables nft_chain_nat xt_nat xt_MASQUERADE nf_nat xt_HL nf_log_ipv4 nf_log_common nft_limit nft_counter xt_LOG xt_limit xt_mac xt_tcpudp xt_conntrack nf_conntrack nf_defrag_ipv4 nft_compat nf_tables nfnetlink evdev rtl8150 brcmfmac brcmutil sha256_generic libsha256 cfg80211 raspberrypi_hwmon rfkill bcm2835_codec(C) bcm2835_v4l2(C) bcm2835_isp(C) v4l2_mem2mem bcm2835_mmal_vchiq(C) videobuf2_dma_contig snd_bcm2835(C) videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 snd_pcm videobuf2_common snd_timer videodev snd mc vc_sm_cma(C) 8021q garp stp llc fixed uio_pdrv_genirq uio ip_tables x_tables ipv6 nf_defrag_ipv6
kernel: [  336.077310] CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.51-v7+ #1333
kernel: [  336.077315] Hardware name: BCM2835
kernel: [  336.077319] Backtrace: 
kernel: [  336.077341] [<8010d480>] (dump_backtrace) from [<8010d778>] (show_stack+0x20/0x24)
kernel: [  336.077352]  r6:80d00000 r5:00000000 r4:80d95cb8 r3:96b3c785
kernel: [  336.077367] [<8010d758>] (show_stack) from [<808cdbe4>] (dump_stack+0xe0/0x124)
kernel: [  336.077381] [<808cdb04>] (dump_stack) from [<8012002c>] (__warn+0xec/0x104)
kernel: [  336.077393]  r8:000001c0 r7:00000009 r6:80b0f6a4 r5:00000000 r4:80d01cec r3:96b3c785
kernel: [  336.077404] [<8011ff40>] (__warn) from [<801200c8>] (warn_slowpath_fmt+0x84/0xc0)
kernel: [  336.077415]  r9:80b0f6a4 r8:000001c0 r7:807db124 r6:00000009 r5:80b0f6bc r4:80d04f48
kernel: [  336.077432] [<80120048>] (warn_slowpath_fmt) from [<807db124>] (dev_watchdog+0x310/0x314)
kernel: [  336.077452]  r9:00000e18 r8:b5ea2000 r7:00000000 r6:80d03d00 r5:b5ea22a8 r4:00000000
kernel: [  336.077471] [<807dae14>] (dev_watchdog) from [<8019be6c>] (call_timer_fn+0x40/0x180)
kernel: [  336.077486]  r8:807dae14 r7:00000100 r6:80d00000 r5:b5ea22a8 r4:b6b25440
kernel: [  336.077504] [<8019be2c>] (call_timer_fn) from [<8019cfbc>] (run_timer_softirq+0x288/0x654)
kernel: [  336.077518]  r9:00000000 r8:b5ea22a8 r7:80d00000 r6:00000e18 r5:80d01dc4 r4:b6b25440
kernel: [  336.077535] [<8019cd34>] (run_timer_softirq) from [<8010249c>] (__do_softirq+0x1a4/0x418)
kernel: [  336.077551]  r10:00000004 r9:00000082 r8:b645e000 r7:00000100 r6:80d00000 r5:00000001
kernel: [  336.077562]  r4:80d03084
kernel: [  336.077580] [<801022f8>] (__do_softirq) from [<80126130>] (irq_exit+0x100/0x110)
kernel: [  336.077596]  r10:b73ff900 r9:80d00000 r8:b645e000 r7:00000001 r6:00000000 r5:00000000
kernel: [  336.077605]  r4:80ca42e4
kernel: [  336.077624] [<80126030>] (irq_exit) from [<80181200>] (__handle_domain_irq+0x70/0xc4)
kernel: [  336.077638] [<80181190>] (__handle_domain_irq) from [<80102228>] (bcm2836_arm_irqchip_handle_irq+0x60/0xa4)
kernel: [  336.077652]  r8:80d9b18d r7:80d01f0c r6:ffffffff r5:60000013 r4:00000000 r3:80d01ed8
kernel: [  336.077664] [<801021c8>] (bcm2836_arm_irqchip_handle_irq) from [<80101a3c>] (__irq_svc+0x5c/0x7c)
kernel: [  336.077670] Exception stack(0x80d01ed8 to 0x80d01f20)
kernel: [  336.077679] 1ec0:                                                       80109b94 00000000
kernel: [  336.077690] 1ee0: 60000093 8011a520 80d04f68 80d00000 80d04fb0 00000001 80d9b18d 80c69a40
kernel: [  336.077701] 1f00: b73ff900 80d01f34 80d0535c 80d01f28 00000000 80109b98 60000013 ffffffff
kernel: [  336.077708]  r4:80109b98 r3:96b3c785
kernel: [  336.077723] [<80109b64>] (arch_cpu_idle) from [<808eea44>] (default_idle_call+0x34/0x48)
kernel: [  336.077737] [<808eea10>] (default_idle_call) from [<80153bc0>] (do_idle+0xec/0x170)
kernel: [  336.077747] [<80153ad4>] (do_idle) from [<80153f20>] (cpu_startup_entry+0x28/0x2c)
kernel: [  336.077759]  r8:00000000 r7:80c69a40 r6:80dac5c0 r5:80dac618 r4:000000ce r3:80d01f50
kernel: [  336.077771] [<80153ef8>] (cpu_startup_entry) from [<808e7e68>] (rest_init+0xbc/0xc0)
kernel: [  336.077786] [<808e7dac>] (rest_init) from [<80c00bd8>] (arch_call_rest_init+0x18/0x1c)
kernel: [  336.077792]  r4:80d04f48 r3:80c99260
kernel: [  336.077803] [<80c00bc0>] (arch_call_rest_init) from [<80c010a8>] (start_kernel+0x4cc/0x500)
kernel: [  336.077815] [<80c00bdc>] (start_kernel) from [<00000000>] (0x0)
kernel: [  336.077827] ---[ end trace 24a6d02eeb0175e0 ]---
```

dwrobel, 2020-09-14:
For the record on the latest `5.4.64` on `64-bit` version this was also much frequently reproducible in comparison to  `4.19.y` `32-bit` version _(with the same headless configuration as I mentioned [here](https://github.com/raspberrypi/linux/issues/3401#issuecomment-575915860))_:
```
kernel: ------------[ cut here ]------------
kernel: NETDEV WATCHDOG: enp1s0u1u1 (ax88179_178a): transmit queue 0 timed out
kernel: WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:447 dev_watchdog+0x394/0x3a0
kernel: Modules linked in: tun nft_nat nft_masq nft_objref nf_conntrack_netbios_ns nf_conntrack_broadcast nft_fib_inet nft_fib_ipv4 nft_fib_ipv6 nft_fib nft_reject_inet nf_reject_ipv4 nf_reject_ipv6 n
ft_reject nft_ct nf_tables_set nft_chain_nat ip6table_nat ip6table_mangle ip6table_raw ip6table_security iptable_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 iptable_mangle iptable_raw iptable_security ip_set nf_tables nfnetlink 
ip6table_filter ip6_tables iptable_filter dm_mirror dm_region_hash dm_log dm_mod brcmfmac brcmutil cfg80211 bcm2835_codec(C) bcm2835_isp(C) vc4 v4l2_mem2mem bcm2835_v4l2(C) videobuf2_dma_contig cec bcm2835_mmal_vchiq(C) ax88179_178a rfkil
l drm_kms_helper videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 videobuf2_common snd_soc_core videodev snd_compress snd_pcm_dmaengine snd_bcm2835(C) v3d mc dwc2 vc_sm_cma(C) snd_pcm snd_timer gpu_sched raspberrypi_hwmon snd syscopyarea
 rpivid_mem sysfillrect sysimgblt fb_sys_fops uio_pdrv_genirq uio gpio_fan
kernel:  sch_fq_codel zram zsmalloc drm fuse drm_panel_orientation_quirks
kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.4.64-1.rpi4.fc32.aarch64 #1
kernel: Hardware name: Raspberry Pi 4 Model B Rev 1.2 (DT)
kernel: pstate: 00000005 (nzcv daif -PAN -UAO)
kernel: pc : dev_watchdog+0x394/0x3a0
kernel: lr : dev_watchdog+0x394/0x3a0
kernel: sp : ffffffc010003d00
kernel: x29: ffffffc010003d00 x28: ffffff80f2631a80 
kernel: x27: 0000000000000004 x26: ffffff80ee1df480 
kernel: x25: 0000000000000140 x24: 00000000ffffffff 
kernel: x23: ffffff80ee1df45c x22: 0000000000000000 
kernel: x21: ffffffc010f56000 x20: ffffff80ee1df000 
kernel: x19: 0000000000000000 x18: 0000000000000030 
kernel: x17: 0000000000000000 x16: 0000000000000000 
kernel: x15: ffffffc010f60478 x14: ffffffffffffffff 
kernel: x13: ffffffc090003a07 x12: ffffffc010003a0f 
kernel: x11: 7274203a29613837 x10: 315f393731383878 
kernel: x9 : 0000000000000000 x8 : 3020657565757120 
kernel: x7 : ffffffc0106566a0 x6 : ffffffc010f59000 
kernel: x5 : 0000000000000000 x4 : ffffffc010ae44d0 
kernel: x3 : fffffffffffffffe x2 : 0000000000000004 
kernel: x1 : a6a675dc55bf7e00 x0 : 0000000000000000 
kernel: Call trace:
kernel:  dev_watchdog+0x394/0x3a0
kernel:  call_timer_fn+0x40/0x1e4
kernel:  expire_timers+0xc8/0x18c
kernel:  run_timer_softirq+0xdc/0x210
kernel:  __do_softirq+0x1a0/0x488
kernel:  irq_exit+0x118/0x120
kernel:  __handle_domain_irq+0xb4/0x130
kernel:  gic_handle_irq+0x5c/0xb0
kernel:  el1_irq+0xbc/0x180
kernel:  arch_cpu_idle+0x38/0x220
kernel:  do_idle+0x234/0x2b0
kernel:  cpu_startup_entry+0x2c/0x90
kernel:  rest_init+0xdc/0xe8
kernel:  arch_call_rest_init+0x18/0x20
kernel:  start_kernel+0x490/0x4c4
kernel: ---[ end trace 52602d8ce620bd83 ]---
```

So I switched to use vendor driver [v1.20.0](https://www.asix.com.tw/FrootAttach/driver/AX88179_178A_LINUX_DRIVER_v1.20.0_SOURCE.tar.bz2) with `dtoverlay=dwc2` and the following cmdline.txt:

ProactiveServices, 2021-07-06:
Seeing the same behaviour on a Pi 4b 1.2, Raspbian OS 10.10, kernel 5.10.17. Using the onboard wired Ethernet and official PSU. Headless device running from a small SSD in a USB3 enclosure plugged into a USB3 port. It's currently pinging about a dozen hosts every 15s and logging (via telegraf to influxdb) and doing nothing else. The hangs are happening multiple times a day, usually only for a few minutes but have seen one occasion of about half an hour.
[stacktrace.txt](https://github.com/raspberrypi/linux/files/6772908/stacktrace.txt)

teewhey, 2022-11-28:
Same thing for me on my Pi4. It happens as soon as I plugged in the LAN cable and all networks just unresponsive till I unplug it and use wlan instead.

Linux pipipi 5.15.76-v7l+ #1597 SMP Fri Nov 4 12:14:58 GMT 2022 armv7l GNU/Linux

[Mon Nov 28 17:19:34 2022] WARNING: CPU: 2 PID: 0 at net/sched/sch_generic.c:478 dev_watchdog+0x304/0x308
[Mon Nov 28 17:19:34 2022] NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 2 timed out
[Mon Nov 28 17:19:34 2022] Modules linked in: xt_nat xt_tcpudp veth xt_conntrack nft_chain_nat xt_MASQUERADE nf_nat nf_conntrack_netlink nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 nft_counter xt_addrtype nft_compat nf_tables nfnetlink br_netfilter bridge rfcomm cmac algif_hash aes_arm_bs crypto_simd cryptd algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc cp210x usbserial overlay 8021q garp stp llc vc4 sg snd_soc_hdmi_codec cec brcmfmac drm_kms_helper bcm2835_v4l2(C) bcm2835_codec(C) brcmutil bcm2835_isp(C) snd_soc_core cfg80211 bcm2835_mmal_vchiq(C) rpivid_hevc(C) v4l2_mem2mem videobuf2_dma_contig videobuf2_vmalloc v3d videobuf2_memops snd_compress gpu_sched snd_pcm_dmaengine videobuf2_v4l2 videobuf2_common raspberrypi_hwmon rfkill snd_bcm2835(C) snd_pcm videodev snd_timer i2c_brcmstb vc_sm_cma(C) mc snd syscopyarea sysfillrect sysimgblt uio_pdrv_genirq fb_sys_fops nvmem_rmem uio drm fuse i2c_dev drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
[Mon Nov 28 17:19:34 2022] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G         C        5.15.76-v7l+ #1597
[Mon Nov 28 17:19:34 2022] Hardware name: BCM2711
[Mon Nov 28 17:19:34 2022] Backtrace:
[Mon Nov 28 17:19:34 2022] [<c0bd7354>] (dump_backtrace) from [<c0bd75a0>] (show_stack+0x20/0x24)
[Mon Nov 28 17:19:34 2022]  r7:000001de r6:c0e3f708 r5:00000000 r4:60000113
[Mon Nov 28 17:19:34 2022] [<c0bd7580>] (show_stack) from [<c0bdbcb0>] (dump_stack_lvl+0x70/0x94)
[Mon Nov 28 17:19:34 2022] [<c0bdbc40>] (dump_stack_lvl) from [<c0bdbcec>] (dump_stack+0x18/0x1c)
[Mon Nov 28 17:19:34 2022]  r7:000001de r6:00000009 r5:c0aac28c r4:c0ec6d04
[Mon Nov 28 17:19:34 2022] [<c0bdbcd4>] (dump_stack) from [<c02226c0>] (__warn+0xfc/0x114)
[Mon Nov 28 17:19:34 2022] [<c02225c4>] (__warn) from [<c0bd7c94>] (warn_slowpath_fmt+0xa4/0xd8)
[Mon Nov 28 17:19:34 2022]  r7:000001de r6:c0ec6d04 r5:c1205048 r4:c0ec6cc8
[Mon Nov 28 17:19:34 2022] [<c0bd7bf4>] (warn_slowpath_fmt) from [<c0aac28c>] (dev_watchdog+0x304/0x308)
[Mon Nov 28 17:19:34 2022]  r9:00000005 r8:c1203d00 r7:c2390000 r6:c2378a00 r5:c23902e8 r4:00000002
[Mon Nov 28 17:19:34 2022] [<c0aabf88>] (dev_watchdog) from [<c02b05c0>] (call_timer_fn+0x40/0x198)
[Mon Nov 28 17:19:34 2022]  r9:eff12540 r8:fffffaf0 r7:c0aabf88 r6:00000100 r5:c23902e8 r4:fffffaf0
[Mon Nov 28 17:19:34 2022] [<c02b0580>] (call_timer_fn) from [<c02b1c10>] (run_timer_softirq+0x5e8/0x658)
[Mon Nov 28 17:19:34 2022]  r8:c155de0c r7:00000000 r6:c23902e8 r5:00000000 r4:fffffaf0
[Mon Nov 28 17:19:34 2022] [<c02b1628>] (run_timer_softirq) from [<c0201488>] (__do_softirq+0x198/0x474)
[Mon Nov 28 17:19:34 2022]  r10:ffffe000 r9:c155c000 r8:00000082 r7:00000100 r6:00000001 r5:00000002
[Mon Nov 28 17:19:34 2022]  r4:c1203084
[Mon Nov 28 17:19:34 2022] [<c02012f0>] (__do_softirq) from [<c02293d0>] (irq_exit+0xd8/0x120)
[Mon Nov 28 17:19:34 2022]  r10:c0e35ae0 r9:c155c000 r8:c1097158 r7:0000001e r6:00000000 r5:00000000
[Mon Nov 28 17:19:34 2022]  r4:ffffe000
[Mon Nov 28 17:19:34 2022] [<c02292f8>] (irq_exit) from [<c028b6c8>] (handle_domain_irq+0x70/0x88)
[Mon Nov 28 17:19:34 2022]  r5:00000000 r4:c109714c
[Mon Nov 28 17:19:34 2022] [<c028b658>] (handle_domain_irq) from [<c02012dc>] (gic_handle_irq+0x8c/0xa0)
[Mon Nov 28 17:19:34 2022]  r7:c155df30 r6:f081400c r5:f0814000 r4:c1205b6c
[Mon Nov 28 17:19:34 2022] [<c0201250>] (gic_handle_irq) from [<c0200aec>] (__irq_svc+0x4c/0x64)
[Mon Nov 28 17:19:34 2022] Exception stack(0xc155df30 to 0xc155df78)
[Mon Nov 28 17:19:34 2022] df20:                                     00042744 00000000 00000001 c021af00
[Mon Nov 28 17:19:34 2022] df40: ffffe000 00000004 c120509c c12050ec 00000001 c134209f c0e35ae0 c155df8c
[Mon Nov 28 17:19:34 2022] df60: c155df90 c155df80 c0208860 c0208864 60000013 ffffffff
[Mon Nov 28 17:19:34 2022]  r9:c155c000 r8:00000001 r7:c155df64 r6:ffffffff r5:60000013 r4:c0208864
[Mon Nov 28 17:19:34 2022] [<c020881c>] (arch_cpu_idle) from [<c0be90a0>] (default_idle_call+0x4c/0xd8)
[Mon Nov 28 17:19:34 2022] [<c0be9054>] (default_idle_call) from [<c025dc70>] (do_idle+0x114/0x164)
[Mon Nov 28 17:19:34 2022] [<c025db5c>] (do_idle) from [<c025dffc>] (cpu_startup_entry+0x28/0x2c)
[Mon Nov 28 17:19:34 2022]  r10:00000000 r9:410fd083 r8:00007000 r7:c155dff8 r6:30c0387d r5:00000002
[Mon Nov 28 17:19:34 2022]  r4:00000097 r3:60000093
[Mon Nov 28 17:19:34 2022] [<c025dfd4>] (cpu_startup_entry) from [<c020faa4>] (secondary_start_kernel+0x170/0x17c)
[Mon Nov 28 17:19:34 2022] [<c020f934>] (secondary_start_kernel) from [<00201810>] (0x201810)
[Mon Nov 28 17:19:34 2022]  r5:00000000 r4:01506040
[Mon Nov 28 17:19:34 2022] ---[ end trace 38ff62284e3b6609 ]---
[Mon Nov 28 17:19:42 2022] bcmgenet fd580000.ethernet eth0: Link is Down

kindacute, 2023-02-17:
Same thing here:
Feb 16 22:17:29  kernel: ------------[ cut here ]------------
Feb 16 22:17:29  kernel: WARNING: CPU: 0 PID: 0 at net/sched/sch_generic.c:478 dev_watchdog+0x304/0x308
Feb 16 22:17:29  kernel: NETDEV WATCHDOG: eth_lpd (cdc_ether): transmit queue 0 timed out
Feb 16 22:17:29  kernel: Modules linked in: xt_MASQUERADE nft_counter nft_chain_nat xt_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 xt_tcpudp nft_compat nf_tables nfnetlink cmac algif_hash aes_arm_bs crypto_simd cryptd algif_skcipher af_alg bnep hci_uart btbcm bluetooth ecdh_generic ecc brcmfmac vc4 brcmutil 8021q garp stp llc snd_soc_hdmi_codec cfg80211 cec drm_kms_helper cdc_ether v3d gpu_sched cdc_acm rfkill raspberrypi_hwmon snd_soc_core i2c_brcmstb bcm2835_v4l2(C) bcm2835_codec(C) rpivid_hevc(C) bcm2835_isp(C) bcm2835_mmal_vchiq(C) v4l2_mem2mem videobuf2_vmalloc videobuf2_dma_contig videobuf2_memops videobuf2_v4l2 videobuf2_common snd_compress snd_pcm_dmaengine snd_bcm2835(C) snd_pcm videodev snd_timer vc_sm_cma(C) snd mc syscopyarea sysfillrect sysimgblt fb_sys_fops nvmem_rmem uio_pdrv_genirq uio drm fuse drm_panel_orientation_quirks backlight ip_tables x_tables ipv6
Feb 16 22:17:29  kernel: CPU: 0 PID: 0 Comm: swapper/0 Tainted: G         C        5.15.72-v7l+ #1591
Feb 16 22:17:29  kernel: Hardware name: BCM2711
Feb 16 22:17:29  kernel: Backtrace: 
Feb 16 22:17:29  kernel: [<c0bd61d4>] (dump_backtrace) from [<c0bd6420>] (show_stack+0x20/0x24)
Feb 16 22:17:29  kernel:  r7:000001de r6:c0e3f4b8 r5:00000000 r4:60000113
Feb 16 22:17:29  kernel: [<c0bd6400>] (show_stack) from [<c0bdab30>] (dump_stack_lvl+0x70/0x94)
Feb 16 22:17:29  kernel: [<c0bdaac0>] (dump_stack_lvl) from [<c0bdab6c>] (dump_stack+0x18/0x1c)
Feb 16 22:17:29  kernel:  r7:000001de r6:00000009 r5:c0aab250 r4:c0ec69f0
Feb 16 22:17:29  kernel: [<c0bdab54>] (dump_stack) from [<c02226c0>] (__warn+0xfc/0x114)
Feb 16 22:17:29  kernel: [<c02225c4>] (__warn) from [<c0bd6b14>] (warn_slowpath_fmt+0xa4/0xd8)
Feb 16 22:17:29  kernel:  r7:000001de r6:c0ec69f0 r5:c1205048 r4:c0ec69b4
Feb 16 22:17:29  kernel: [<c0bd6a74>] (warn_slowpath_fmt) from [<c0aab250>] (dev_watchdog+0x304/0x308)
Feb 16 22:17:29  kernel:  r9:00000001 r8:c1203d00 r7:c1792800 r6:c1722a00 r5:c1792ae8 r4:00000000
Feb 16 22:17:29  kernel: [<c0aaaf4c>] (dev_watchdog) from [<c02b0598>] (call_timer_fn+0x40/0x198)
Feb 16 22:17:29  kernel:  r9:efef2540 r8:1cf4fca0 r7:c0aaaf4c r6:00000100 r5:c1792ae8 r4:1cf4fca0
Feb 16 22:17:29  kernel: [<c02b0558>] (call_timer_fn) from [<c02b1be8>] (run_timer_softirq+0x5e8/0x658)
Feb 16 22:17:29  kernel:  r8:c1201d94 r7:00000000 r6:c1792ae8 r5:00000000 r4:1cf4fca0
Feb 16 22:17:29  kernel: [<c02b1600>] (run_timer_softirq) from [<c0201488>] (__do_softirq+0x198/0x474)
Feb 16 22:17:29  kernel:  r10:ffffe000 r9:c1200000 r8:00000082 r7:00000100 r6:00000001 r5:00000002
Feb 16 22:17:29  kernel:  r4:c1203084
Feb 16 22:17:29  kernel: [<c02012f0>] (__do_softirq) from [<c02293d0>] (irq_exit+0xd8/0x120)
Feb 16 22:17:29  kernel:  r10:c0e358b8 r9:c1200000 r8:c1096158 r7:0000001e r6:00000000 r5:00000000
Feb 16 22:17:29  kernel:  r4:ffffe000
Feb 16 22:17:29  kernel: [<c02292f8>] (irq_exit) from [<c028b6c8>] (handle_domain_irq+0x70/0x88)
Feb 16 22:17:29  kernel:  r5:00000000 r4:c109614c
Feb 16 22:17:29  kernel: [<c028b658>] (handle_domain_irq) from [<c02012dc>] (gic_handle_irq+0x8c/0xa0)
Feb 16 22:17:29  kernel:  r7:c1201eb8 r6:f081400c r5:f0814000 r4:c1205b6c
Feb 16 22:17:29  kernel: [<c0201250>] (gic_handle_irq) from [<c0200aec>] (__irq_svc+0x4c/0x64)
Feb 16 22:17:29  kernel: Exception stack(0xc1201eb8 to 0xc1201f00)
Feb 16 22:17:29  kernel: 1ea0:                                                       8f8c5e64 00000000
Feb 16 22:17:29  kernel: 1ec0: 00000001 c021af00 ffffe000 00000001 c120509c c12050ec 00000001 c134201f
Feb 16 22:17:29  kernel: 1ee0: c0e358b8 c1201f14 c1201f18 c1201f08 c0208860 c0208864 60000013 ffffffff
Feb 16 22:17:29  kernel:  r9:c1200000 r8:00000001 r7:c1201eec r6:ffffffff r5:60000013 r4:c0208864
Feb 16 22:17:29  kernel: [<c020881c>] (arch_cpu_idle) from [<c0be7f20>] (default_idle_call+0x4c/0xd8)
Feb 16 22:17:29  kernel: [<c0be7ed4>] (default_idle_call) from [<c025dc70>] (do_idle+0x114/0x164)
Feb 16 22:17:29  kernel: [<c025db5c>] (do_idle) from [<c025dffc>] (cpu_startup_entry+0x28/0x2c)
Feb 16 22:17:29  kernel:  r10:effff900 r9:00000000 r8:00000000 r7:00000000 r6:ffffffff r5:c135b000
Feb 16 22:17:29  kernel:  r4:000000e6 r3:c108f294
Feb 16 22:17:29  kernel: [<c025dfd4>] (cpu_startup_entry) from [<c0bdf7d4>] (rest_init+0xc8/0xd0)
Feb 16 22:17:29  kernel: [<c0bdf70c>] (rest_init) from [<c1000c60>] (arch_call_rest_init+0x18/0x1c)
Feb 16 22:17:29  kernel:  r5:c135b000 r4:c135b068
Feb 16 22:17:29  kernel: [<c1000c48>] (arch_call_rest_init) from [<c10013fc>] (start_kernel+0x708/0x74c)
Feb 16 22:17:29  kernel: [<c1000cf4>] (start_kernel) from [<00000000>] (0x0)
Feb 16 22:17:29  kernel: ---[ end trace ce6c43b87da596b2 ]---

pelwell, 2023-02-20:
@kindacute That's not the same - you seem to be using your Pi 4 as an Ethernet gadget (notice the `cdc_ether`). It's not even clear if you are using the onboard Ethernet at all.

To everyone else:

This issue clearly persistently affects a small number of users, while most have no problems.

Please report what the other end of the Ethernet cable is plugged into (dedicated Ethernet switch, Wireless AP etc., specifying the model). As long as the kernel message is `NETDEV WATCHDOG: eth0 (bcmgenet): transmit queue 2 timed out` there is no need to post the stack traces as above.

teewhey, 2023-02-21:
> @kindacute That's not the same - you seem to be using your Pi 4 as an Ethernet gadget (notice the `cdc_ether`). It's not even clear if you are using the onboard Ethernet at all.
>
> To everyone else: [...]

For my case, the other end is my main wireless router (Asus RT-AX55)

pelwell, 2023-02-21:
There have been numerous reports of the power-saving Energy Efficient Ethernet setting causing problems when using BCM2711-based device with certain switches and routers. Provided you are running at least the most recent stable kernel (5.15.84) or later, try adding `dtparam=eee=off` to config.txt to see if that helps.