raspberrypi/linux #3404: [RPi4 4GB] xHCI host controller not responding, assume dead

JANogueira, 2020-01-13:
**Describe the bug**
After boot-up, an when activating a service that uses USB interface (Network UPS tools, as an example) xHCI interface crashs and the USB devices get disconnected. Recovery only possible after system reboot.

List of USB devices when the system boots up:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 003: ID 0463:ffff MGE UPS Systems UPS
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

After enabling a:
```
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
```

**To reproduce**
After system boot, start Network UPS tools (UPS connected through USB, having a poll frequency of 30 seconds), and the xHCI interface will crash shortly after, leading to all USB devices being disconnected.

**System**

Details:
`Raspberry Pi 4 Model B Rev 1.1`

```
# cat /etc/os-release | head -4
NAME=HassOS
VERSION="3.8 (RaspberryPi 4 64bit)"
ID=hassos
VERSION_ID=3.8
```

```
# uname -a
Linux hassio 4.19.93-v8 #1 SMP PREEMPT Sun Jan 12 18:33:10 UTC 2020 aarch64 Hassio/OS
```

```
# cat /proc/cpuinfo | tail -3
Revision        : c03111
Serial          : 1000000013df512e
Model           : Raspberry Pi 4 Model B Rev 1.1
#
```

```
# df                                                                                                                                                                                                                                     
Filesystem           1K-blocks      Used Available Use% Mounted on                                                                                                                                                                       
/dev/root                92032     92032         0 100% /                                                                                                                                                                                
devtmpfs               1915496         0   1915496   0% /dev                                                                                                                                                                             
tmpfs                  1948808         0   1948808   0% /dev/shm                                                                                                                                                                         
tmpfs                  1948808       708   1948100   0% /run                                                                                                                                                                             
tmpfs                  1948808         0   1948808   0% /sys/fs/cgroup                                                                                                                                                                   
tmpfs                  1948808       708   1948100   0% /etc/machine-id                                                                                                                                                                  
/dev/mmcblk0p7           91099     18656     65562  22% /mnt/overlay                                                                                                                                                                     
/dev/mmcblk0p7           91099     18656     65562  22% /root/.docker                                                                                                                                                                    
/dev/mmcblk0p7           91099     18656     65562  22% /etc/modprobe.d                                                                                                                                                                  
/dev/mmcblk0p7           91099     18656     65562  22% /etc/modules-load.d                                                                                                                                                              
/dev/mmcblk0p7           91099     18656     65562  22% /etc/docker                                                                                                                                                                      
/dev/mmcblk0p7           91099     18656     65562  22% /etc/dropbear                                                                                                                                                                    
/dev/mmcblk0p7           91099     18656     65562  22% /etc/udev/rules.d
/dev/mmcblk0p7           91099     18656     65562  22% /root/.ssh
/dev/mmcblk0p1           32686      3650     29036  11% /mnt/boot
/dev/mmcblk0p7           91099     18656     65562  22% /etc/hostname
/dev/mmcblk0p7           91099     18656     65562  22% /etc/systemd/timesyncd.conf
/dev/mmcblk0p7           91099     18656     65562  22% /etc/NetworkManager/system-connections
/dev/mmcblk0p7           91099     18656     65562  22% /etc/hosts
/dev/mmcblk0p8       122172044   6834280 109113732   6% /mnt/data
/dev/zram2               15856        40     14676   0% /tmp
[... 88 lines cut ...]
[    0.981020] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    0.981041] usb 1-1: Product: USB2.0 Hub
[    0.982796] hub 1-1:1.0: USB hub found
[    1.278306] usb 1-1.3: new full-speed USB device number 3 using xhci_hcd
[    1.615672] usbcore: registered new interface driver brcmfmac
[    1.999992] usb 1-1.3: New USB device found, idVendor=0463, idProduct=ffff, bcdDevice= 0.01
[    2.000016] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[    2.000027] usb 1-1.3: Product: 5E
[    2.000036] usb 1-1.3: Manufacturer: EATON
[    3.967565] hid-generic 0003:0463:FFFF.0001: hiddev96,hidraw0: USB HID v1.10 Device [EATON 5E] on usb-0000:01:00.0-1.3/input0
[  172.256228] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
[  172.272306] xhci_hcd 0000:01:00.0: Host halt failed, -110
[  172.272321] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[  172.272369] xhci_hcd 0000:01:00.0: HC died; cleaning up [  172.272442] usb 1-1: USB disconnect, device number 2
[  172.272467] usb 1-1.3: USB disconnect, device number 3
```

**Additional context**
This bug report was opened here because HassOS3.8 rus default raspbian Kernel.
This issue was initially reported here:
https://github.com/home-assistant/hassos/issues/526

USB cable was replaced to ensure that it was not faulty.

If more logs are needed, please let me know

timg236, 2020-01-13:
It might be worth trying different versions of the VL805 firmware. There’s a thread about it here 
https://www.raspberrypi.org/forums/viewtopic.php?t=260879

JANogueira, 2020-01-13:
Thank you. I will give it a try and share the outcome.

JANogueira, 2020-01-13:
Well, on raspbian all went well, but not in hassio.
USB crash is still happening in hassio.

Before shutting down Raspbian I have ensured that the latest vl805 firmware was properly installed (using ´sudo vl805´ and have shown 0137ad).

Do you know if the EEPROM bin file deppends on any other optimization on the linux kernel side? or it should fix the USB behaviour by itself?

Thanks in advance.

JANogueira, 2020-01-13:
Please, disregard my last post.
I was doing it wrong while trying to flash the EEPROM.

Took additional actions and all good apparently. 3 hours have passed and all is stable.
I will observe this for a couple of days more just to be sure, and share the outcome.

Many thanks once again!

JANogueira, 2020-01-15:
Just to confirm that everything is still working perfectly.
0137ad solved the issue. USB devices are all stable and I had a small increase on the sdcard I/O.

hardwareadictos, 2020-01-29:
Hey!! Having exactly the same issue on VL805 FW version: 000137ad

Are you still having issues? Can you provide more info about what you did to solve it?

hardwareadictos, 2020-01-29:
Here some logs:

[log_xhci_rpi4.txt](https://github.com/raspberrypi/linux/files/4130347/log_xhci_rpi4.txt)

JANogueira, 2020-01-29:
Hi @hardwareadictos,

It worked for about 1 week with VL805 FW 0137ad and then I am facing the same issue again.
Not fully solved unfortunately...

hardwareadictos, 2020-01-29:
I opened an issue for that because I don't know if we have the same issue (I suspect we do):

https://github.com/raspberrypi/linux/issues/3438

Don't doubt on participating. Thanks for answering 😊

hardwareadictos, 2020-01-30:
This could explain some things: https://github.com/raspberrypi/linux/commit/c74b1b53254016fd83b580b8d49bb02d72ce4836

Dont know if that change is already implemented on last kernel...

pelwell, 2020-01-30:
That patch allows for the possibility of loading VL805 firmware from system RAM rather than EEPROM, which has to be re-enabled after a PCI reset. The VPU firmware is not making use of that facility, so when it comes to your problem that commit is a [red herring](https://en.wikipedia.org/wiki/Red_herring).

hardwareadictos, 2020-01-30:
> That patch allows for the possibility of loading VL805 firmware from system RAM rather than EEPROM, which has to be re-enabled after a PCI reset. The VPU firmware is not making use of that facility, [...]

Im not an expert, that's why i said "could" and not "actually". Was also the only recent commit referencing to xHCI.

That issue isnt happening on previous releases.

What's you opinion on that?

Thanks in advance :)

pelwell, 2020-01-30:
Perhaps the on-board voltage is a marginal - try adding `over_voltage=1` to config.txt and rebooting.

hardwareadictos, 2020-01-30:
> Perhaps the on-board voltage is a marginal - try adding `over_voltage=1` to config.txt and rebooting.

Thank you! Applied. Will test it some days and i will report back.

JANogueira, 2020-01-30:
thank you @pelwell 
Also applied this to my hassio installation. I will test it as well for some days and report back.

Cheers

hardwareadictos, 2020-01-31:
24 hours. No issues so far. Next report on Monday, but seems that was the issue...

hardwareadictos, 2020-02-07:
One week without issues, i consider this issue fixed :)

JANogueira, 2020-02-07:
I also confirm that after 1 week, all is ok.
But not sure if over voltage should be the way to go. This is a workaround and not a fix imho.

hardwareadictos, 2020-02-07:
Agree. It was working before that voltage related changes on kernel. Maybe someone can give more feedback about that. But for the moment all is working as always

JamesH65, 2020-02-08:
> I also confirm that after 1 week, all is ok.
> But not sure if over voltage should be the way to go. This is a workaround and not a fix imho.

Indeed, I expect a future firmware release will fix this, if the most recent one hasn't already done so.

JANogueira, 2020-02-08:
I am currently running on 0137ad

timg236, 2020-02-08:
The VideoCore firmware controls DVFS so the update would be there rather than the XHCI firmware. Might be worth trying rpi-update (assuming you are familiar with rpi-update / risks of using early firmware)

DrJohnM61, 2020-02-11:
I am having the same issue after updating to the latest release. If I disconnect the [powered] USB3 hub and boot and then connect the hub, I can see the disks for a few minutes (they show in 'blkid' and 'df' and I can also 'ls' the mount points and see my files). They then they disconnect. Looking at dmesg, the same error as others is being produced:
[   77.881799] xhci_hcd 0000:01:00.0: WARNING: Host System Error
[   82.887053] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
[   82.887070] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[   82.887236] xhci_hcd 0000:01:00.0: HC died; cleaning up
[   82.887275] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.

I am currently running on 0137ad

I have tried over_voltage=1 (in fact I have tried a number of over_voltage numbers or 1,2 and 3)
I have performed a rpi-update
CPU/GPU is not overclocked

$ cat /proc/cpuinfo | tail -3
Revision	: c03112
Serial		: 10000000b9ae9a7d
Model		: Raspberry Pi 4 Model B Rev 1.2

$ cat /etc/os-release | head -4
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"

$ uname -a
Linux PiHoleNAS 4.19.102-v7l+ #1295 SMP Thu Feb 6 15:49:36 GMT 2020 armv7l GNU/Linux

[dmesg2.txt](https://github.com/raspberrypi/linux/files/4185010/dmesg2.txt)

I am an old DEC 10/TOPS-10, VAX/VMS and PDP11/RSX11M+ assembler programmer. Unix is rather new to me. I apologies in advance if I come over as a noob

popcornmix, 2020-02-11:
@DrJohnM61 do you know what firmware/kernel you were on previously?
Does force_turbo=1 help?
Does switching to firmware from [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262649&start=100#p1610257) help?

DrJohnM61, 2020-02-11:
> @DrJohnM61 do you know what firmware/kernel you were on previously?
> Does force_turbo=1 help?
> Does switching to firmware from [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262649&start=100#p1610257) help?

IIRC the prior version was the 5th Feb release

I tried the force_turbo=1 but does not seem to make any difference.
[   13.045392] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[   13.045464] xhci_hcd 0000:01:00.0: HC died; cleaning up

Will try the firmware alternative in the morning (its late now here in London)

DrJohnM61, 2020-02-12:
> @DrJohnM61 do you know what firmware/kernel you were on previously?
> Does force_turbo=1 help?
> Does switching to firmware from [here](https://www.raspberrypi.org/forums/viewtopic.php?f=28&t=262649&start=100#p1610257) help?

I copied the files that your link pointed to into the /boot directory. Not sure if I was supposed to do anything after that (I am a unix noob). Rebooted and the problem with USB 3 connected hub and devices is still there. I note that the 'name -a' showed no difference in firmware version, so I suspect I am missing a step.

Went back to page one of the blog that you direct me to and executed the 'sudo rpi-update 4b2c270' command. Rebooted and then power cycled and the problem has gone away, so clearly (IMHO), this seems to be an issue that has been introduced in a later firmware version.

I will note that somewhere between the 4b2c270 version and the latest, I could not boot the RP4 with the connected USB hub/disks and see them but would have to boot without the hub connected and then plug it in after the boot had completed. This change happened after implementing the firmware change that allowed overclocking. I had for some time looked at forums to find a combination of voltage and wait times to try and fix the issue. However, the latest version of the firmware  (4.19.102-v7l+ #1295 SMP) just did not allow the external powered USB hub or disks to stably connect (and even if the USB HUB was connected after boot, the USB ports would successively shut down, causing problems with my RAID sets, until the whole HUB disconnected).

Here is a snip from the log showing the USB disconnects (post force_turbo setting):
[   14.305428] sd 2:0:0:0: [sdc] tag#20 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.305433] sd 2:0:0:0: [sdc] tag#20 CDB: opcode=0x41 41 00 89 e4 81 00 00 03 00 00
[   14.335885] xhci_hcd 0000:01:00.0: WARN Can't disable streams for endpoint 0x81, streams are being disabled already
[   14.337628] usb 2-1.4.3: USB disconnect, device number 6
[   14.338786] sd 2:0:0:0: [sdc] Synchronizing SCSI cache
[   14.355410] print_req_error: I/O error, dev sdc, sector 2313453824
[   14.375412] sd 3:0:0:0: [sdd] tag#25 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.375420] sd 3:0:0:0: [sdd] tag#25 CDB: opcode=0x41 41 00 89 e4 80 00 00 01 00 00
[   14.445403] sd 3:0:0:0: [sdd] tag#25 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.445412] sd 3:0:0:0: [sdd] tag#25 CDB: opcode=0x41 41 00 89 e4 80 00 00 01 00 00
[   14.455411] print_req_error: I/O error, dev sdc, sector 2313454592
[   14.455425] print_req_error: I/O error, dev sdc, sector 2313454848
[   14.455434] print_req_error: I/O error, dev sdc, sector 2313455616
[   14.525404] sd 3:0:0:0: [sdd] tag#25 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.525412] sd 3:0:0:0: [sdd] tag#25 CDB: opcode=0x41 41 00 89 e4 80 00 00 01 00 00
[   14.595450] sd 3:0:0:0: [sdd] tag#25 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.595457] sd 3:0:0:0: [sdd] tag#25 CDB: opcode=0x41 41 00 89 e4 80 00 00 01 00 00
[   14.675454] sd 3:0:0:0: [sdd] tag#25 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.675461] sd 3:0:0:0: [sdd] tag#25 CDB: opcode=0x41 41 00 89 e4 80 00 00 01 00 00
[   14.745449] sd 3:0:0:0: [sdd] tag#25 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.745458] sd 3:0:0:0: [sdd] tag#25 CDB: opcode=0x41 41 00 89 e4 80 00 00 01 00 00
[   14.805398] sd 3:0:0:0: [sdd] tag#25 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.805405] sd 3:0:0:0: [sdd] tag#25 CDB: opcode=0x41 41 00 89 e4 80 00 00 01 00 00
[   14.855429] sd 2:0:0:0: [sdc] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=0x00
[   14.865412] sd 3:0:0:0: [sdd] tag#25 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.865419] sd 3:0:0:0: [sdd] tag#25 CDB: opcode=0x41 41 00 89 e4 80 00 00 01 00 00
[   14.925448] sd 3:0:0:0: [sdd] tag#25 sense submit err -19 uas-tag 1 inflight: s-st a-out s-out a-cmd s-cmd 
[   14.925457] sd 3:0:0:0: [sdd] tag#25 CDB: opcode=0x41 41 00 89 e4 80 00 00 01 00 00
[   14.975986] xhci_hcd 0000:01:00.0: WARN Can't disable streams for endpoint 0x81, streams are being disabled already
[   14.977293] usb 2-1.4.4: USB disconnect, device number 7
[   14.978401] sd 3:0:0:0: [sdd] Synchronizing SCSI cache
[   14.978422] print_req_error: I/O error, dev sdd, sector 2313453568
[   15.075452] print_req_error: I/O error, dev sdd, sector 2313453824
[   15.485545] sd 3:0:0:0: [sdd] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=0x00
[   15.605939] xhci_hcd 0000:01:00.0: WARN Can't disable streams for endpoint 0x81, streams are being disabled already
[   17.785614] Buffer I/O error on dev md0, logical block 0, lost sync page write
[   17.785622] EXT4-fs (md0): I/O error while writing superblock
[   17.866157] Buffer I/O error on dev md0, logical block 976690672, async page read
[   17.887111] Buffer I/O error on dev md0, logical block 976690672, async page read
[   18.037901] md0: detected capacity change from 4000525058048 to 0
[   18.037951] md: md0 stopped.
[   18.225997] bridge: filtering via arp/ip/ip6tables is no longer available by default. Update your scripts to load br_netfilter if you need this.

Happy to try different firmware if someone can explain how to install it after pulling from a file share.

hardwareadictos, 2020-02-13:
> CPU/GPU is not overclocked

Maybe your Power supply isnt giving you enough power then

DrJohnM61, 2020-02-13:
> > CPU/GPU is not overclocked
> 
> Maybe your Power supply isnt giving you enough power then

The hub is powered, so there is 'no' drain for the external hard disks. The power supply for the Pi4 says that it is 5V/3A and as noted, it works fine with prior versions of the firmware.

I will swap out the power supply for the official raspberry pi version, upgrade the firmware and retry.

Thanks for the suggestion.

DrJohnM61, 2020-02-13:
Latests tests...
Updated back to 4.19.97-v7l+ with swapped out power supply. Same issues - Hard disks are not mounted (for any length of time) with a xHCI host controller not responding, assume dead. Log attached.

[system-fail-firmware.log](https://github.com/raspberrypi/linux/files/4198054/system-fail-firmware.log)

I have also upgraded to 4.19.102-v7l+ with no improvement.

Downgraded back to 4.19.83-v7l+ and everything is working again.

did a 'sudo apt-get install --reinstall raspberrypi-kernel'
This took me back up to 4.19.97-v7l+

Reboot - everything now working OK!
set:
over_voltage=5
arm_freq=2140

Does not boot at this frequency. Changed arm_freq=2000 and boots fine and disks are visible.

So at this stage, I have no idea why the error has stopped being shown except I did a 'sudo apt-get install --reinstall raspberrypi-kernel'

JamesH65, 2020-02-14:
Can I confirm that the latest software (sudo apt update; sudo apt full-upgrade) is working fine at standard clocks? There have been changes in here to help with booting. Also, we cannot guarantee overclocks, so expecting 2140 to work is very ambitious-  that is a huge overclock. Even 2000 is pretty big - 33% over stock. 

My thoughts are that this issue has been solved (at standard clocks, and sensible overclocks) and can be closed, but just need to confirm.

hardwareadictos, 2020-02-14:
We are missing the main goal of the issue. Stability issues with peripherals connected ....

No overclocks here

JamesH65, 2020-02-14:
> We are missing the main goal of the issue. Stability issues with peripherals connected ....
> 
> No overclocks here

And various people has said they are fixed with the latest software. Just trying to confirm whether that is the case.

DrJohnM61, 2020-02-14:
My RP4b has been working fine for the last 24hrs and I have transferred over 3tb to my raid array without any issue, so closed for me. I will note, that the issue was being seen on 4.19.97-v7l+ without overvolt or overclock and just seemed to go away after several up/downgrades followed by a 'sudo apt-get install --reinstall raspberrypi-kernel'.

Thanks everyone for the assistance/advice.

hardwareadictos, 2020-02-14:
I updated to last kernel 4.19.97-v7l+ and i uncommented over_voltage parameter on kernel config. No issues so far, will report back on 24 hours.

JANogueira, 2020-02-14:
I will allow some more time so other people can report similar feedback before closing this.

SemMulder, 2020-02-17:
I just encountered this again today on a fully up-to-date RPi 4. I added the suggested `over_voltage=1` flag and that seems to have made it at least somewhat more stable. I am still getting the following logs for the attached HDD though:
```
Feb 17 21:06:20 raspberrypi kernel: sd 0:0:0:0: [sda] tag#7 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
Feb 17 21:06:20 raspberrypi kernel: sd 0:0:0:0: [sda] tag#7 Sense Key : 0xb [current] 
Feb 17 21:06:20 raspberrypi kernel: sd 0:0:0:0: [sda] tag#7 ASC=0x47 ASCQ=0x1 
Feb 17 21:06:20 raspberrypi kernel: sd 0:0:0:0: [sda] tag#7 CDB: opcode=0x28 28 00 2c 0c a3 10 00 04 00 00
Feb 17 21:06:20 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 739025680
Feb 17 21:06:21 raspberrypi kernel: sd 0:0:0:0: [sda] tag#8 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
Feb 17 21:06:21 raspberrypi kernel: sd 0:0:0:0: [sda] tag#8 Sense Key : 0xb [current] 
Feb 17 21:06:21 raspberrypi kernel: sd 0:0:0:0: [sda] tag#8 ASC=0x47 ASCQ=0x1 
Feb 17 21:06:21 raspberrypi kernel: sd 0:0:0:0: [sda] tag#8 CDB: opcode=0x28 28 00 2c 0c a7 10 00 04 00 00
Feb 17 21:06:21 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 739026704
Feb 17 21:08:55 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
Feb 17 21:08:55 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 Sense Key : 0xb [current] 
Feb 17 21:08:55 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 ASC=0x47 ASCQ=0x1 
Feb 17 21:08:55 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 CDB: opcode=0x28 28 00 15 4f 74 00 00 04 00 00
Feb 17 21:08:55 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 357528576
Feb 17 21:08:56 raspberrypi kernel: sd 0:0:0:0: [sda] tag#23 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
Feb 17 21:08:56 raspberrypi kernel: sd 0:0:0:0: [sda] tag#23 Sense Key : 0xb [current] 
Feb 17 21:08:56 raspberrypi kernel: sd 0:0:0:0: [sda] tag#23 ASC=0x47 ASCQ=0x1 
Feb 17 21:08:56 raspberrypi kernel: sd 0:0:0:0: [sda] tag#23 CDB: opcode=0x28 28 00 15 4f 78 00 00 04 00 00
Feb 17 21:08:56 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 357529600
```

Will let it run and report back soon.

hardwareadictos, 2020-02-17:
48 hours with latest kernel and without over_voltage overlay on kernel. Seems stable for the moment.

SemMulder, 2020-02-17:
Still hitting it, even with `over_voltage=1`:
```
Feb 17 22:19:51 raspberrypi kernel: sd 0:0:0:0: [sda] tag#14 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
Feb 17 22:19:51 raspberrypi kernel: sd 0:0:0:0: [sda] tag#14 Sense Key : 0xb [current] 
Feb 17 22:19:51 raspberrypi kernel: sd 0:0:0:0: [sda] tag#14 ASC=0x47 ASCQ=0x1 
Feb 17 22:19:51 raspberrypi kernel: sd 0:0:0:0: [sda] tag#14 CDB: opcode=0x28 28 00 2c 58 20 00 00 04 00 00
Feb 17 22:19:51 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 743972864
Feb 17 22:19:52 raspberrypi kernel: sd 0:0:0:0: [sda] tag#13 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
Feb 17 22:19:52 raspberrypi kernel: sd 0:0:0:0: [sda] tag#13 Sense Key : 0xb [current] 
Feb 17 22:19:52 raspberrypi kernel: sd 0:0:0:0: [sda] tag#13 ASC=0x47 ASCQ=0x1 
Feb 17 22:19:52 raspberrypi kernel: sd 0:0:0:0: [sda] tag#13 CDB: opcode=0x28 28 00 2c 58 24 00 00 04 00 00
Feb 17 22:19:52 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 743973888
Feb 17 22:20:32 raspberrypi kernel: sd 0:0:0:0: [sda] tag#23 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
Feb 17 22:20:32 raspberrypi kernel: sd 0:0:0:0: [sda] tag#23 Sense Key : 0xb [current] 
Feb 17 22:20:32 raspberrypi kernel: sd 0:0:0:0: [sda] tag#23 ASC=0x47 ASCQ=0x1 
Feb 17 22:20:32 raspberrypi kernel: sd 0:0:0:0: [sda] tag#23 CDB: opcode=0x28 28 00 2c 58 77 40 00 04 00 00
Feb 17 22:20:32 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 743995200
Feb 17 22:20:33 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 UNKNOWN(0x2003) Result: hostbyte=0x00 driverbyte=0x08
Feb 17 22:20:33 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 Sense Key : 0xb [current] 
Feb 17 22:20:33 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 ASC=0x47 ASCQ=0x1 
Feb 17 22:20:33 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 CDB: opcode=0x28 28 00 2c 58 7b 40 00 04 00 00
Feb 17 22:20:33 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 743996224
Feb 17 22:20:58 raspberrypi kernel: xhci_hcd 0000:01:00.0: WARNING: Host System Error
Feb 17 22:20:58 raspberrypi kernel: xhci_hcd 0000:01:00.0: Host halt failed, -110
Feb 17 22:21:22 raspberrypi kernel: sd 0:0:0:0: [sda] tag#25 uas_eh_abort_handler 0 uas-tag 2 inflight: CMD IN 
Feb 17 22:21:22 raspberrypi kernel: sd 0:0:0:0: [sda] tag#25 CDB: opcode=0x28 28 00 15 63 b4 00 00 04 00 00
Feb 17 22:21:27 raspberrypi kernel: xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
Feb 17 22:21:27 raspberrypi kernel: xhci_hcd 0000:01:00.0: Host halt failed, -110
Feb 17 22:21:27 raspberrypi kernel: xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
Feb 17 22:21:27 raspberrypi kernel: xhci_hcd 0000:01:00.0: HC died; cleaning up
Feb 17 22:21:27 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD 
Feb 17 22:21:27 raspberrypi kernel: usb 1-1: USB disconnect, device number 2
Feb 17 22:21:27 raspberrypi kernel: sd 0:0:0:0: [sda] tag#24 CDB: opcode=0x28 28 00 15 63 b0 00 00 04 00 00
Feb 17 22:21:27 raspberrypi kernel: usb 2-1: USB disconnect, device number 2
Feb 17 22:21:27 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 0
Feb 17 22:21:27 raspberrypi kernel: sd 0:0:0:0: [sda] Synchronizing SCSI cache
Feb 17 22:21:27 raspberrypi kernel: sd 0:0:0:0: Device offlined - not ready after error recovery
Feb 17 22:21:27 raspberrypi kernel: sd 0:0:0:0: Device offlined - not ready after error recovery
Feb 17 22:21:27 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 358855680
Feb 17 22:21:27 raspberrypi kernel: print_req_error: I/O error, dev sda, sector 358854656
```

popcornmix, 2020-02-17:
@SemMulder what do:
```
vcgencmd version
vcgencmd get_config over_voltage_avs
```
report? Does over_voltage=2 help? Was this reliable with an older firmware?

SemMulder, 2020-02-17:
```
pi@raspberrypi:~ $ vcgencmd version
Feb 12 2020 12:36:21 
Copyright (c) 2012 Broadcom
version c3c8dbdf147686fb0c3f32aece709d0653368810 (clean) (release) (start)
pi@raspberrypi:~ $ vcgencmd get_config over_voltage_avs
over_voltage_avs=-36250
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.97-v7l+ #1294 SMP Thu Jan 30 13:21:14 GMT 2020 armv7l GNU/Linux
pi@raspberrypi:~ $ sudo vl805 
VL805 FW version: 000137ad
```

> Does over_voltage=2 help? 

Will try that next together with  `sudo apt-get install --reinstall raspberrypi-kernel`.

> Was this reliable with an older firmware?

Not sure, since I just got this one. Did also have issues with underpowered HDMI though...

popcornmix, 2020-02-17:
```sudo rpi-update afbea380```
would get you to a firmware/kernel before the DVFS changes.
I would expect that would only help if over_voltage=<n> helps for some value of n (1 seems to have been enough in other cases I've heard of, but you could try 2 or 3).

But the issue could be unrelated to that.

SemMulder, 2020-02-17:
Is the output of the `vcgencmd get_config over_voltage_avs` command supposed to change depending on `over_voltage=...`? Because I am not getting any changes switching between not setting `over_voltage` and setting it to either 1 or 2.

The flag is supposed to be set in `/boot/config.txt`, correct?

popcornmix, 2020-02-18:
No over_voltage_avs is purely internal, based on measurements of the ring oscillators of the specific chip. It's not something the user can adjust, but it gives some info to me about the type of silicon we are dealing with.

SemMulder, 2020-02-18:
Running with `rpi-update afbea380` (i.e. `Linux raspberrypi 4.19.83-v7l+`) and no `over_voltage` flags I get the following:
```
Feb 18 18:15:57 raspberrypi kernel: usb usb2-port1: unable to enumerate USB device
Feb 18 18:16:08 raspberrypi kernel: usb 2-1: new SuperSpeed Gen 1 USB device number 23 using xhci_hcd
Feb 18 18:16:13 raspberrypi kernel: usb 2-1: device descriptor read/8, error -110
Feb 18 18:16:13 raspberrypi kernel: usb 2-1: new SuperSpeed Gen 1 USB device number 23 using xhci_hcd
Feb 18 18:16:18 raspberrypi kernel: usb 2-1: device descriptor read/8, error -110
Feb 18 18:16:19 raspberrypi kernel: usb 2-1: new SuperSpeed Gen 1 USB device number 24 using xhci_hcd
Feb 18 18:16:24 raspberrypi kernel: usb 2-1: device descriptor read/8, error -110
Feb 18 18:16:24 raspberrypi kernel: usb 2-1: new SuperSpeed Gen 1 USB device number 24 using xhci_hcd
Feb 18 18:16:29 raspberrypi kernel: usb 2-1: device descriptor read/8, error -110
Feb 18 18:16:29 raspberrypi kernel: usb usb2-port1: attempt power cycle
Feb 18 18:16:29 raspberrypi kernel: usb 2-1: new SuperSpeed Gen 1 USB device number 25 using xhci_hcd
Feb 18 18:16:35 raspberrypi kernel: usb 2-1: device descriptor read/8, error -110
Feb 18 18:16:35 raspberrypi kernel: usb 2-1: new SuperSpeed Gen 1 USB device number 25 using xhci_hcd
Feb 18 18:16:40 raspberrypi kernel: usb 2-1: device descriptor read/8, error -110
Feb 18 18:16:40 raspberrypi kernel: usb 2-1: new SuperSpeed Gen 1 USB device number 26 using xhci_hcd
Feb 18 18:16:45 raspberrypi kernel: usb 2-1: device descriptor read/8, error -110
Feb 18 18:16:45 raspberrypi kernel: usb 2-1: new SuperSpeed Gen 1 USB device number 26 using xhci_hcd
Feb 18 18:16:50 raspberrypi kernel: usb 2-1: device descriptor read/8, error -110
```

Error code 110 is `ETIMEDOUT` I believe. The behavior seems slightly better though since the xHCI host controller keeps running, but ultimately the USB connection is still lost.

Maybe the `xHCI host controller not responding, assume dead` error is because of a bug when handling underlying USB issues?

I am not sure if my issues are still related to the current ticket now. Should I open a new ticket? Is there anything I can try to narrow this down?

SemMulder, 2020-02-18:
After researching my USB-SATA device:
```
Bus 002 Device 002: ID 152d:0578 JMicron Technology Corp. / JMicron USA Technology Corp. JMS567 SATA 6Gb/s bridge
```
I stumbled on the following: https://www.smartmontools.org/wiki/SAT-with-UAS-Linux. So I switched to the `usb-storage` driver now. So far this seems to be stable, but I will let it run for a while and report back.

SemMulder, 2020-02-22:
Just FYI, I this issue is no longer occurring after disabling `uas`.

Pillendreher, 2020-06-11:
Seems like I have encountered something similar to this. I have an SSD attached to the Rasperry Pi 4 and have not encountered any problems writing to it or reading from it directly via the Pi. Yet sometimes the whole drive crashes when I access it remotely. It looks like sharing the SSD via SAMBA makes matters worse, yet I have encountered the same issue while transfering a file from the SSD to my PC as well (transfer rate drops and suddenly the drive is gone). On my latest crash the log told me this:

```
root@DietPi:~# dmesg -l err,crit,alert,emerg
[ 5335.445477] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[ 5335.447503] xhci_hcd 0000:01:00.0: HC died; cleaning up
[ 5335.525443] print_req_error: I/O error, dev sda, sector 367394728
[ 5335.525497] print_req_error: I/O error, dev sda, sector 367390632
[ 5335.525537] print_req_error: I/O error, dev sda, sector 367387560
[ 5335.525576] print_req_error: I/O error, dev sda, sector 367388584
[ 5335.525615] print_req_error: I/O error, dev sda, sector 367392680
[ 5335.525651] print_req_error: I/O error, dev sda, sector 361440792
[ 5335.525660] print_req_error: I/O error, dev sda, sector 361439768
[ 5335.525673] print_req_error: I/O error, dev sda, sector 367396776
[ 5335.525686] print_req_error: I/O error, dev sda, sector 361455128
[ 5335.525701] print_req_error: I/O error, dev sda, sector 361456152
[ 5335.533666] XFS (sda1): writeback error on sector 361437720
[ 5335.537049] XFS (sda1): writeback error on sector 361486872
[ 5335.539202] XFS (sda1): writeback error on sector 361462296
[ 5335.539232] XFS (sda1): metadata I/O error in "xlog_iodone" at daddr 0xee7b080 len 64 error 5
[ 5335.539302] XFS (sda1): Log I/O Error Detected.  Shutting down filesystem
[ 5335.539313] XFS (sda1): Please umount the filesystem and rectify the problem(s)
[ 5335.583470] XFS (sda1): writeback error on sector 330838608
```

Both the Raspberry Pi and the SSD care are brand new. The SSD isn't, but running smartctl on it tells me there are neither "critical warnings" nor "media and data integrity errors".

hardwareadictos, 2020-06-11:
> Seems like I have encountered something similar to this. I have an SSD attached to the Rasperry Pi 4 and have not encountered any problems writing to it or reading from it directly via the Pi. Yet [...]

Did you try to blacklist uas for this drive via quirks on cmdline.txt?

Pillendreher, 2020-06-11:
> Did you try to blacklist uas for this drive via quirks on cmdline.txt?

Not yet, I have just started doing some research on the whole thing. I will give that a try, thank you for pointing it out!

Pillendreher, 2020-06-27:
Just a quick follow-up: Since moving away from uas, I have not encountered a single error. Thanks for the advice!

fat0troll, 2021-12-04:
Facing the same issue. Official PSU, externally powered hub for HDD. Almost every hour or two it just disappears.

```
[35045.287776] xhci_hcd 0000:01:00.0: WARNING: Host System Error
[35048.867773] sd 0:0:0:0: [sda] tag#27 sense submit err -22 uas-tag 2 inflight: s-st a-out s-out a-cmd s-cmd
[35048.867796] sd 0:0:0:0: [sda] tag#27 CDB: opcode=0x8a 8a 00 00 00 00 00 f5 cf 18 10 00 00 00 10 00 00
[35050.307632] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
[35050.307649] xhci_hcd 0000:01:00.0: USBSTS: HCHalted HSE EINT
[35050.307695] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[35050.307822] xhci_hcd 0000:01:00.0: HC died; cleaning up
```

JANogueira, 2021-12-04:
Guys, please share your current firmware version and all the details existing on the first post so you can help the devs with the right info on the first shot.

malidong, 2022-02-13:
I got a same issue here, and it happened on a WD 6TB HDD via USB3.

```
pi@raspberrypi:~ $ lsusb                                                                                                                                                                                                                                                                                             
Bus 002 Device 003: ID 152d:9561 JMicron Technology Corp. / JMicron USA Technology Corp. 
Bus 002 Device 002: ID 0411:0286 BUFFALO INC. (formerly MelCo., Inc.) 
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 001 Device 002: ID 2109:3431 VIA Labs, Inc. Hub
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
pi@raspberrypi:~ $ lsusb -t
/:  Bus 02.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/4p, 5000M
    |__ Port 1: Dev 2, If 0, Class=Mass Storage, Driver=usb-storage, 5000M
    |__ Port 2: Dev 3, If 0, Class=Mass Storage, Driver=uas, 5000M
/:  Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
pi@raspberrypi:~ $ uname -a                                                                                                                                                                                                                                                                                                  
Linux raspberrypi 5.10.63-v7l+ #1496 SMP Wed Dec 1 15:58:56 GMT 2021 armv7l GNU/Linux
```

The error logs are here.
```
Feb 13 08:32:56 raspberrypi kernel: [74601.333202] xhci_hcd 0000:01:00.0: WARNING: Host System Error
Feb 13 08:33:01 raspberrypi kernel: [74606.408786] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
Feb 13 08:33:01 raspberrypi kernel: [74606.408803] xhci_hcd 0000:01:00.0: USBSTS: HCHalted HSE EINT
Feb 13 08:33:01 raspberrypi kernel: [74606.408969] usb 1-1: USB disconnect, device number 2
Feb 13 08:33:01 raspberrypi kernel: [74606.411156] usb 2-1: USB disconnect, device number 2
...
Feb 13 08:33:01 raspberrypi kernel: [74606.411923] sd 1:0:0:1: Device offlined - not ready after error recovery
...
Feb 13 08:33:01 raspberrypi kernel: [74606.415336] sd 0:0:0:0: [sda] Synchronizing SCSI cache
Feb 13 08:33:01 raspberrypi kernel: [74606.415526] sd 0:0:0:0: [sda] Synchronize Cache(10) failed: Result: hostbyte=0x01 driverbyte=0x00
Feb 13 08:33:01 raspberrypi kernel: [74606.432049] BTRFS info (device sda): forced readonly
Feb 13 08:33:01 raspberrypi kernel: [74606.461166] usb 2-2: USB disconnect, device number 3
Feb 13 08:33:01 raspberrypi kernel: [74606.463440] sd 1:0:0:0: [sdb] Synchronizing SCSI cache
Feb 13 08:33:02 raspberrypi kernel: [74606.768917] sd 1:0:0:0: [sdb] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=0x00
Feb 13 08:33:02 raspberrypi kernel: [74606.950797] sd 1:0:0:1: [sdc] Synchronizing SCSI cache
Feb 13 08:33:02 raspberrypi kernel: [74607.198954] sd 1:0:0:1: [sdc] Synchronize Cache(10) failed: Result: hostbyte=0x07 driverbyte=0x00
Feb 13 08:33:02 raspberrypi kernel: [74607.269508] xhci_hcd 0000:01:00.0: WARN Can't disable streams for endpoint 0x82, streams are being disabled already
Feb 13 08:33:08 raspberrypi kernel: [74613.287026] btrfs_dev_stat_print_on_error: 4153 callbacks suppressed
Feb 13 08:33:13 raspberrypi kernel: [74618.331792] btrfs_dev_stat_print_on_error: 4871 callbacks suppressed
Feb 13 08:33:20 raspberrypi kernel: [74624.806919] btrfs_dev_stat_print_on_error: 5846 callbacks suppressed
Feb 13 08:33:25 raspberrypi kernel: [74629.820242] btrfs_dev_stat_print_on_error: 6337 callbacks suppressed
Feb 13 08:33:30 raspberrypi kernel: [74635.295587] btrfs_dev_stat_print_on_error: 4380 callbacks suppressed
```

>Just FYI, this issue is no longer occurring after disabling uas.
But by following the keyword `uas` here, I found some information.
`https://forums.raspberrypi.com/viewtopic.php?t=245931`
and of course `https://www.smartmontools.org/wiki/SAT-with-UAS-Linux`

I changed the usb driver to usb-storage.quirks. Now waiting and see if the issue will happens again.

wyuenho, 2022-03-29:
Exact same issue.

`Linux home 5.4.0-1056-raspi #63-Ubuntu SMP PREEMPT Mon Mar 14 07:48:06 UTC 2022 aarch64 aarch64 aarch64 GNU/Linux`

```
sudo rpi-eeprom-config 
[all]
BOOT_UART=0
WAKE_ON_GPIO=0
POWER_OFF_ON_HALT=1
DHCP_TIMEOUT=45000
DHCP_REQ_TIMEOUT=4000
TFTP_FILE_TIMEOUT=30000
ENABLE_SELF_UPDATE=1
DISABLE_HDMI=0
BOOT_ORDER=0xf41
```

```
sudo rpi-eeprom-update 
BCM2711 detected
VL805 firmware in bootloader EEPROM
BOOTLOADER: up-to-date
CURRENT: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 LATEST: Thu 03 Sep 2020 12:11:43 PM UTC (1599135103)
 FW DIR: /lib/firmware/raspberrypi/bootloader/default
VL805: up-to-date
CURRENT: 000138a1
 LATEST: 000138a1
```

nikita-fuchs, 2022-05-03:
This issue is almost 2 and a half years old now 🏝

After a short while, my external SSD also vanishes, RPi 4.

`dmesg`: 

```
809.987218] xhci_hcd 0000:01:00.0: WARNING: Host System Error
[  810.019230] xhci_hcd 0000:01:00.0: Host halt failed, -110
[  815.069471] xhci_hcd 0000:01:00.0: xHCI host not responding to stop endpoint command.
[  815.069494] xhci_hcd 0000:01:00.0: USBSTS: HSE EINT
[  815.101508] xhci_hcd 0000:01:00.0: Host halt failed, -110
[  815.101521] xhci_hcd 0000:01:00.0: xHCI host controller not responding, assume dead
[  815.101661] xhci_hcd 0000:01:00.0: HC died; cleaning up
[  815.102216] sd 0:0:0:0: [sda] tag#0 uas_eh_abort_handler 0 uas-tag 1 inflight: CMD
[  815.102238] sd 0:0:0:0: [sda] tag#0 CDB: opcode=0x28 28 00 00 09 43 1a 00 00 e6 00
[  815.102284] usb 1-1: USB disconnect, device number 2
[  815.102307] usb 1-1.3: USB disconnect, device number 3
[  815.102549] usb 2-1: USB disconnect, device number 2
[  815.103468] sd 0:0:0:0: [sda] tag#2 uas_zap_pending 0 uas-tag 3 inflight: CMD
[  815.103490] sd 0:0:0:0: [sda] tag#2 CDB: opcode=0x28 28 00 00 09 65 00 00 00 20 00
[  815.103504] sd 0:0:0:0: [sda] tag#3 uas_zap_pending 0 uas-tag 4 inflight: CMD
[  815.103515] sd 0:0:0:0: [sda] tag#3 CDB: opcode=0x28 28 00 00 09 43 03 00 00 01 00
[  815.103528] sd 0:0:0:0: [sda] tag#4 uas_zap_pending 0 uas-tag 5 inflight: CMD
[  815.103539] sd 0:0:0:0: [sda] tag#4 CDB: opcode=0x28 28 00 00 09 43 01 00 00 01 00
[  815.103551] sd 0:0:0:0: [sda] tag#5 uas_zap_pending 0 uas-tag 6 inflight: CMD
[  815.103561] sd 0:0:0:0: [sda] tag#5 CDB: opcode=0x28 28 00 00 09 43 00 00 00 01 00
[  815.103573] sd 0:0:0:0: [sda] tag#6 uas_zap_pending 0 uas-tag 7 inflight: CMD
[  815.103583] sd 0:0:0:0: [sda] tag#6 CDB: opcode=0x28 28 00 00 09 43 02 00 00 01 00
[  815.103595] sd 0:0:0:0: [sda] tag#7 uas_zap_pending 0 uas-tag 8 inflight: CMD
[  815.103604] sd 0:0:0:0: [sda] tag#7 CDB: opcode=0x28 28 00 00 09 43 19 00 00 01 00
[  815.103616] sd 0:0:0:0: [sda] tag#8 uas_zap_pending 0 uas-tag 9 inflight: CMD
[  815.103626] sd 0:0:0:0: [sda] tag#8 CDB: opcode=0x28 28 00 00 09 43 18 00 00 01 00
[  815.103640] sd 0:0:0:0: [sda] tag#11 uas_zap_pending 0 uas-tag 12 inflight: CMD
[  815.103650] sd 0:0:0:0: [sda] tag#11 CDB: opcode=0x28 28 00 00 09 43 15 00 00 01 00
[  815.103663] sd 0:0:0:0: [sda] tag#12 uas_zap_pending 0 uas-tag 13 inflight: CMD
[  815.103673] sd 0:0:0:0: [sda] tag#12 CDB: opcode=0x28 28 00 00 09 43 14 00 00 01 00
[  815.103686] sd 0:0:0:0: [sda] tag#13 uas_zap_pending 0 uas-tag 14 inflight: CMD
[  815.103695] sd 0:0:0:0: [sda] tag#13 CDB: opcode=0x28 28 00 00 09 43 13 00 00 01 00
[  815.103707] sd 0:0:0:0: [sda] tag#14 uas_zap_pending 0 uas-tag 15 inflight: CMD
[  815.103717] sd 0:0:0:0: [sda] tag#14 CDB: opcode=0x28 28 00 00 09 43 12 00 00 01 00
[  815.103730] sd 0:0:0:0: [sda] tag#15 uas_zap_pending 0 uas-tag 16 inflight: CMD
[  815.103739] sd 0:0:0:0: [sda] tag#15 CDB: opcode=0x28 28 00 00 09 43 11 00 00 01 00
[  815.103751] sd 0:0:0:0: [sda] tag#16 uas_zap_pending 0 uas-tag 17 inflight: CMD
[  815.103762] sd 0:0:0:0: [sda] tag#16 CDB: opcode=0x28 28 00 00 09 43 10 00 00 01 00
[  815.103774] sd 0:0:0:0: [sda] tag#17 uas_zap_pending 0 uas-tag 18 inflight: CMD
[  815.103784] sd 0:0:0:0: [sda] tag#17 CDB: opcode=0x28 28 00 00 09 43 0f 00 00 01 00
[  815.103796] sd 0:0:0:0: [sda] tag#18 uas_zap_pending 0 uas-tag 19 inflight: CMD
[  815.103806] sd 0:0:0:0: [sda] tag#18 CDB: opcode=0x28 28 00 00 09 43 0e 00 00 01 00
[  815.103819] sd 0:0:0:0: [sda] tag#19 uas_zap_pending 0 uas-tag 20 inflight: CMD
[  815.103828] sd 0:0:0:0: [sda] tag#19 CDB: opcode=0x28 28 00 00 09 43 0d 00 00 01 00
[  815.103841] sd 0:0:0:0: [sda] tag#20 uas_zap_pending 0 uas-tag 21 inflight: CMD
[  815.103851] sd 0:0:0:0: [sda] tag#20 CDB: opcode=0x28 28 00 00 09 43 0c 00 00 01 00
[  815.103863] sd 0:0:0:0: [sda] tag#21 uas_zap_pending 0 uas-tag 22 inflight: CMD
[  815.103873] sd 0:0:0:0: [sda] tag#21 CDB: opcode=0x28 28 00 00 09 43 0b 00 00 01 00
[  815.103885] sd 0:0:0:0: [sda] tag#22 uas_zap_pending 0 uas-tag 23 inflight: CMD
[  815.103895] sd 0:0:0:0: [sda] tag#22 CDB: opcode=0x28 28 00 00 09 43 0a 00 00 01 00
[  815.103908] sd 0:0:0:0: [sda] tag#23 uas_zap_pending 0 uas-tag 24 inflight: CMD
[  815.103917] sd 0:0:0:0: [sda] tag#23 CDB: opcode=0x28 28 00 00 09 43 08 00 00 01 00
[  815.103929] sd 0:0:0:0: [sda] tag#24 uas_zap_pending 0 uas-tag 25 inflight: CMD
[  815.103939] sd 0:0:0:0: [sda] tag#24 CDB: opcode=0x28 28 00 00 09 43 07 00 00 01 00
[  815.103952] sd 0:0:0:0: [sda] tag#25 uas_zap_pending 0 uas-tag 26 inflight: CMD
[  815.103961] sd 0:0:0:0: [sda] tag#25 CDB: opcode=0x28 28 00 00 09 43 06 00 00 01 00
[  815.103973] sd 0:0:0:0: [sda] tag#26 uas_zap_pending 0 uas-tag 27 inflight: CMD
[  815.103983] sd 0:0:0:0: [sda] tag#26 CDB: opcode=0x28 28 00 00 09 43 09 00 00 01 00
[  815.103994] sd 0:0:0:0: [sda] tag#27 uas_zap_pending 0 uas-tag 28 inflight: CMD
[  815.104004] sd 0:0:0:0: [sda] tag#27 CDB: opcode=0x28 28 00 00 09 43 05 00 00 01 00
[  815.104016] sd 0:0:0:0: [sda] tag#28 uas_zap_pending 0 uas-tag 29 inflight: CMD
[  815.104026] sd 0:0:0:0: [sda] tag#28 CDB: opcode=0x28 28 00 00 09 43 04 00 00 01 00
[  815.121722] sd 0:0:0:0: Device offlined - not ready after error recovery
[... 62 lines cut ...]
VERSION_ID="11"
VERSION="11 (bullseye)"
```
```
$ uname -a
Linux raspberrypi 5.15.32-v8+ #1538 SMP PREEMPT Thu Mar 31 19:40:39 BST 2022 aarch64 GNU/Linux
```
```
$ cat /proc/cpuinfo | tail -3
Revision	: c03114
[... cut ...]
Model		: Raspberry Pi 4 Model B Rev 1.4
```
```
(after the USB drive disappeared)
$ df
Filesystem     1K-blocks    Used Available Use% Mounted on
/dev/root       12532936 2535188   9338064  22% /
devtmpfs         1777048       0   1777048   0% /dev
tmpfs            1942776       0   1942776   0% /dev/shm
tmpfs             777112     864    776248   1% /run
tmpfs               5120       4      5116   1% /run/lock
/dev/mmcblk0p8    258094   31995    226100  13% /boot
tmpfs             388552       0    388552   0% /run/user/1000
```

EliaTolin, 2022-08-26:
News?
I have same issue

DerKleinePunk, 2022-09-10:
I have the same Problem!

percysnoodle, 2022-11-26:
I'm seeing the same problem.

Nazgile94, 2022-12-01:
me ² external hdd enclosure 

kernel: [ 447.129915] xhci_hcd 0000:01:00.0: WARN Successful completion on short TX
Raspberry pi4 4gb 
raspbian 64bit

openmediavault 6
Version
6.0.46-5 (Shaitan)
Prozessor
BCM2835
Kernel
Linux 5.15.76-v8+

device disconnects, share disaspears - need to reset enclosure + pi
on a x86 linux machine , all working.

EliaTolin, 2022-12-01:
@Nazgile94 same problem.

Without rebooting not works if i umount and mount again.

[17 more replies]

closed 2022-12-01 by P33M