raspberrypi/linux #3406: RPi4: Cannot use multiple USB webcams at higher resolutions

maximumsomething, 2020-01-16:
**Describe the bug**
A single usb camera works fine at high resolutions, but an additional camera on a different USB port will not work if they are both running at the same time. Lowering the resolution of both cameras is a workaround.

This seems remarkably like the issues with the old Pis where all USB ports were connected to an internal hub and shared only 480 mbps of bandwidth total, but this is not the case on the Raspberry Pi 4.

I tried enabling `UVC_QUIRK_FIX_BANDWIDTH`, but the result was the same as before.

**To reproduce**
Connect two USB 2.0 webcams to a Raspberry Pi 4 and make them both capture raw (YUYV) video at 800x448 resolution.

**Expected behaviour**
Both cameras work.

**Actual behaviour**
The first camera will work fine, but the second will hang on the `ioctl(VIDIOC_DQBUF)` call (which gets called every frame).

**System**
It is a Raspberry Pi 4 2GB with the September 2019 Raspbian Buster image, and it has never been updated to my knowledge.

**Logs**
Here is the dmesg log when `echo 0xffff | sudo tee /sys/module/uvcvideo/parameters/trace` is ran then the cameras are activated:
```
[  625.381174] uvcvideo: uvc_v4l2_open
[  625.621027] uvcvideo: Resuming interface 0
[  625.621033] uvcvideo: Resuming interface 1
[  625.621109] uvcvideo: Trying format 0x56595559 (YUYV): 800x448.
[  625.621115] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
[  625.623883] uvcvideo: uvc_v4l2_mmap
[  625.624995] uvcvideo: uvc_v4l2_mmap
[  625.626018] uvcvideo: uvc_v4l2_mmap
[  625.627056] uvcvideo: uvc_v4l2_mmap
[  625.628266] uvcvideo: Device requested 3060 B/frame bandwidth.
[  625.628275] uvcvideo: Selecting alternate setting 11 (3060 B/frame bandwidth).
[  625.802845] uvcvideo: Allocated 5 URB buffers of 32x3060 bytes each.
[  625.845977] uvcvideo: Control 3/4 value change len 7.
[  625.936218] uvcvideo: Frame complete (EOF found).
[  625.936227] uvcvideo: EOF in empty payload.
[  625.936824] uvcvideo: uvc_v4l2_open
[  625.947935] uvcvideo: frame 1 stats: 0/236/237 packets, 0/0/237 pts (!early initial), 236/237 scr, last pts/stc/sof 620271536/639183936/43843
[  625.979906] uvcvideo: Frame complete (EOF found).
[  625.979916] uvcvideo: EOF in empty payload.
[  625.991933] uvcvideo: frame 2 stats: 0/236/237 packets, 0/0/237 pts (!early initial), 236/237 scr, last pts/stc/sof 632703536/651410136/43879
[  626.020081] uvcvideo: Frame complete (EOF found).
[  626.020091] uvcvideo: EOF in empty payload.
[  626.022181] uvcvideo: Resuming interface 2
[  626.022193] uvcvideo: Resuming interface 3
[  626.022267] uvcvideo: Trying format 0x56595559 (YUYV): 800x448.
[  626.022279] uvcvideo: Using default frame interval 33333.3 us (30.0 fps).
[  626.031949] uvcvideo: frame 3 stats: 0/236/237 packets, 0/0/237 pts (!early initial), 236/237 scr, last pts/stc/sof 645202936/664011736/43916
[  626.050180] uvcvideo: uvc_v4l2_mmap
[  626.053405] uvcvideo: uvc_v4l2_mmap
[  626.054085] uvcvideo: Control 3/4 value change len 7.
[  626.055709] uvcvideo: uvc_v4l2_mmap
[  626.058676] uvcvideo: uvc_v4l2_mmap
[  626.060225] uvcvideo: Frame complete (EOF found).
[  626.060234] uvcvideo: EOF in empty payload.
[  626.066392] uvcvideo: Device requested 3060 B/frame bandwidth.
[  626.066409] uvcvideo: Selecting alternate setting 11 (3060 B/frame bandwidth).
[  626.076003] uvcvideo: frame 4 stats: 0/236/237 packets, 0/0/237 pts (!early initial), 236/237 scr, last pts/stc/sof 657703136/676425736/43953
[  626.079445] uvcvideo: Allocated 5 URB buffers of 32x3060 bytes each.
[  626.115965] uvcvideo: frame 5 stats: 0/236/237 packets, 0/0/237 pts (!early initial), 236/237 scr, last pts/stc/sof 670203336/688989736/43990
[  626.155972] uvcvideo: frame 6 stats: 0/236/237 packets, 0/0/237 pts (!early initial), 236/237 scr, last pts/stc/sof 682703336/701441336/44027
[  626.199975] uvcvideo: frame 7 stats: 0/236/237 packets, 0/0/237 pts (!early initial), 236/237 scr, last pts/stc/sof 695202936/714005336/44064
[  626.240069] uvcvideo: frame 8 stats: 0/236/237 packets, 0/0/237 pts (!early initial), 236/237 scr, last pts/stc/sof 707703336/726419336/44101
```
After that, it spams `uvcvideo: USB isochronous frame lost (-18).` Between those messages, it will report every a message like this one:

```
Marking buffer as bad (error bit set).
uvcvideo: Frame complete (EOF found).
uvcvideo: frame 72 stats: 108/1/109 packets, 1/103/109 pts (early initial), 108/109 scr, last pts/stc/sof 3923567593/3923957787/1908
uvcvideo: frame 65 stats: 0/236/237 packets, 0/0/237 pts (!early initial), 236/237 scr, last pts/stc/sof 1406594448/1425309848/18759
```

timg236, 2020-01-16:
Try updating the VL805 (USB controller) firmware to 0137ad. See this forum thread
https://www.raspberrypi.org/forums/viewtopic.php?t=260879

maximumsomething, 2020-01-16:
I'm closing this issue, I will reopen it if it still occurs after the firmware update.

maximumsomething, 2020-01-19:
I'm experiencing the same problem after the firmware update.

P33M, 2020-01-19:
I wouldn't expect that configuration to work with either firmware.

>This seems remarkably like the issues with the old Pis where all USB ports were connected to an internal hub and shared only 480 mbps of bandwidth total, but this is not the case on the Raspberry Pi 4.

This is also the case on Pi 4. There is an (internal) USB2.0 hub and all ports share the same upstream USB2.0 link.

Add up the bandwidth consumed by each endpoint and you'll find that it exceeds the maximum allowed by the USB spec - 80% of the total is the maximum for all aggregate periodic traffic.

3060bytes per USB microframe * 2 devices = 391mbit/s, 81% of 480mbit/s. Peak bandwidth is what matters, not the average output bitrate, because if it doesn't fit in a microframe then the transfer is terminated with a babble error.

I recommend either using USB3.0 cameras, or cameras that do mjpeg/h264 compression.

maximumsomething, 2020-01-20:
> This is also the case on Pi 4. There is an (internal) USB2.0 hub and all ports share the same upstream USB2.0 link.

The official docs say that's not the case:

https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md#overview_pi4

> For the Pi 4, a fully-featured host controller drives the downstream USB ports. Downstream USB is provided by a Via Labs VL805 chip - that supports two USB 2.0 ports and two USB 3.0 ports. This is connected to the BCM2711 SoC using a PCIe link, which is extremely fast. Therefore, the Pi 4 does not have the speed constraints of previous models.

P33M, 2020-01-20:
All *USB2.0* ports on Pi 4 share the same 480mbit/s of bandwidth, because they are connected by an internal hub. Note that USB3.0 ports contain USB2.0 connections.

maximumsomething, 2020-01-20:
> All _USB2.0_ ports on Pi 4 share the same 480mbit/s of bandwidth, because they are connected by an internal hub. Note that USB3.0 ports contain USB2.0 connections.

Are you sure? I can find nothing on the internet suggesting USB 2.0 bandwidth is shared through one internal port. Can you point me to an article with more details on the Pi 4's USB controller?

pelwell, 2020-01-20:
You can treat @P33M's comments on the matter as gospel.

P33M, 2020-01-20:
Look at your lsusb output above: all your connected USB devices are connected via 
```
 Bus 01.Port 1: Dev 1, Class=root_hub, Driver=xhci_hcd/1p, 480M
    |__ Port 1: Dev 2, If 0, Class=Hub, Driver=hub/4p, 480M
```
That is to say, a single 480mbit root port with a downstream 4-port hub.

JamesH65, 2020-01-20:
> The official docs say that's not the case:
>
> https://www.raspberrypi.org/documentation/hardware/raspberrypi/usb/README.md#overview_pi4

Docs appear to be 'wrong', or at least misleading - I'll update them.

JamesH65, 2020-01-20:
See https://github.com/raspberrypi/documentation/pull/1390 for proposed change to docs.

closed 2020-01-20 by maximumsomething