raspberrypi/linux #3407: Implicit declaration of function `__irq_set_affinity` error while building kernel

mcemilg, 2020-01-16:
I am following the kernel building [post](https://www.raspberrypi.org/documentation/linux/kernel/building.md) to build a custom configured kernel on Raspberry Pi 4.  I am getting this error while building the kernel:
```make
  CC      drivers/pci/pcie/portdrv_core.o
drivers/pci/controller/pcie-brcmstb.c: In function ‘brcm_msi_set_affinity’:
drivers/pci/controller/pcie-brcmstb.c:962:9: error: implicit declaration of function ‘__irq_set_affinity’; did you mean ‘irq_set_affinity’? [-Werror=implicit-function-declaration]
  return __irq_set_affinity(msi->irq, mask, force);
         ^~~~~~~~~~~~~~~~~~
         irq_set_affinity
  CC      drivers/pci/pcie/portdrv_pci.o
```

I did not touch any config for `IRQ`, all default.

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Raspberry Pi 4 / 4 gb
* Which OS and version: 
```
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage5
```
* Which firmware version (`vcgencmd version`):
```
Sep 24 2019 17:34:30 
Copyright (c) 2012 Broadcom
version cd3add54955f8fa065b414d8fc07c525e7ddffc8 (clean) (release) (start)
```
* Which kernel version (`uname -a`):
```
Linux raspberrypi 4.19.75-v7l+ #1270 SMP Tue Sep 24 18:51:41 BST 2019 armv7l GNU/Linux
```

pelwell, 2020-01-16:
1. Which branch are you on?
2. How did you download the tree?
3. Please post the relevant parts of the `history` from the shell where you are building the kernel.

mcemilg, 2020-01-16:
>     1. Which branch are you on?
The current active branch `rpi-4.19.y`.
>     2. How did you download the tree?
I did not understand what you mean. But if you are talking about source tree I just cloned from here the github repo.
>     3. Please post the relevant parts of the `history` from the shell where you are building the kernel.
I just installed dependencies and followed the building commands I mentioned about it, here is the list:
```
KERNEL=kernel7l
make bcm2711_defconfig
make -j4 zImage modules dtbs
```

pelwell, 2020-01-17:
It just built for me with no errors - head of tree was ac6064a0. What does `git status` report?

ED6E0F17, 2020-01-17:
"irq_set_affinity" depends on CONFIG_SMP - I don`t know why you would want to disable that, but I think it does count as a bug, and looks like an easy fix.

added in:
https://github.com/raspberrypi/linux/commit/7b0af0925c8cc8b1773365ec066c07049aea4bf3

````
static int brcm_msi_set_affinity(struct irq_data *irq_data,
				 const struct cpumask *mask, bool force)
{
-	return -EINVAL;
+	struct brcm_msi *msi = irq_data_get_irq_chip_data(irq_data);
+	return __irq_set_affinity(msi->irq, mask, force);
}

pelwell, 2020-01-17:
Yes, I saw that - we should fix it (although the upstream driver should be in 5.6), but the OP claims to be using bcm2711_defconfig...

mcemilg, 2020-01-17:
Hey guys, actually I changed the config after command `make bcm2711_defconfig`, sorry for misrepresentation. Here is my `.config` file [config.txt](https://github.com/raspberrypi/linux/files/4077435/config.txt). I was trying to create a minimal kernel config. I disabled SMP because the help page says something like this "If you have a one CPU say no". I change [this line](https://github.com/raspberrypi/linux/blob/rpi-4.19.y/drivers/pci/controller/pcie-brcmstb.c#L962 ) to this:
```
static int brcm_msi_set_affinity(struct irq_data *irq_data,
				 const struct cpumask *mask, bool force)
{
	struct brcm_msi *msi = irq_data_get_irq_chip_data(irq_data);
- 	return __irq_set_affinity(msi->irq, mask, force);
+       return irq_set_affinity(msi->irq, mask);
}
```

I didn't know what I did but I just trying to overcome this issue to continue. It did work, I built the kernel and reboot with it. I noticed that the kernel did not see all cores of CPU. I got this:
```
From File /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3
```
But the default kernel config shows all of them:
```
From File /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 1
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 2
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd08
CPU revision	: 3

processor	: 3
model name	: ARMv7 Processor rev 3 (v7l)
BogoMIPS	: 108.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CLinux version 4.19.75-v7l+ (dom@buildbot) (gcc version 4.9.3 (crosstool-NG crosstool-ng-1.22.0-88-g8460611)) #1270 SMP Tue Sep 24 18:51:41 BST 2019
```
Is it because the SMP disabled or the line I've changed?

pelwell, 2020-01-17:
> Is it because the SMP disabled or the line I've changed?

Yes. There is a confusion about CPUs vs cores - the A73 as 4 cores, so SMP should have been enabled (we try not to enable kernel features that we don't need).

JamesH65, 2020-02-14:
Closing this issue as questions answered/issue resolved.

closed 2020-02-14 by JamesH65