raspberrypi/linux #3408: RPI 4 AP MODE wlan stuck

figeac888, 2020-01-16:
Hello,

See also https://github.com/raspberrypi/linux/issues/2453.

Reproduced with RPI 4 running on raspbian buster : 

Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

Linux USA 4.19.95-v7l+ #1291 SMP Tue Jan 14 14:54:26 GMT 2020 armv7l GNU/Linux

/opt/vc/bin/vcgencmd version
Jan 14 2020 15:01:27
Copyright (c) 2012 Broadcom
version 8da878341f0c4a12f73d227c0c5b6ca99219292e (clean) (release) (start)

[76229.046423] brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted
[76261.584586] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76264.144636] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76264.144650] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76275.984804] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76278.544850] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76278.544863] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76292.865062] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76295.425149] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76295.425162] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76310.385333] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76312.945373] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76312.945386] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76336.625740] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76339.185805] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76339.185818] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76348.945941] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76351.505972] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76351.505985] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76354.066015] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76356.626053] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76356.626066] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76365.106219] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76367.666231] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76367.666246] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76373.666348] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76376.226422] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76376.226437] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76378.786455] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76381.346478] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76381.346493] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
[76383.906532] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76386.466535] brcmfmac: brcmf_proto_bcdc_query_dcmd: brcmf_proto_bcdc_msg failed w/status -110
[76386.466550] brcmfmac: brcmf_cfg80211_get_station: GET STA INFO failed, -110
**Describe the bug**

WLAN stuck, no more wifi in ap mode

**To reproduce**
Just let the rpi running in ap mode with traffic

**Expected behaviour**
AP MODE in good health

JamesH65, 2020-02-14:
Hmm, that is a wireless firmware crash issue. So this just happens whilst running as an AP, nothing special going on? How long does it take before it fails?

figeac888, 2020-02-14:
Yes just run as an AP with 1 or more client connected. Failure is quite random, here are the history of fails  detected by my watchdog:

01/19/20--02:35:01 : rpi rebooted by cron
01/19/20--02:36:44 : rpi rebooted by cron
01/20/20--10:35:01 : rpi rebooted by cron
01/20/20--10:40:01 : rpi rebooted by cron
01/20/20--10:44:01 : rpi rebooted by cron
01/20/20--10:45:01 : rpi rebooted by cron
01/20/20--10:46:02 : rpi rebooted by cron
01/20/20--10:47:01 : rpi rebooted by cron
01/20/20--10:48:01 : rpi rebooted by cron
01/20/20--10:49:01 : rpi rebooted by cron
01/20/20--13:42:01 : rpi rebooted by cron
01/25/20--19:48:02 : rpi rebooted by cron
01/27/20--16:52:01 : rpi rebooted by cron
01/28/20--17:06:01 : rpi rebooted by cron
01/31/20--08:29:01 : rpi rebooted by cron
02/01/20--14:16:01 : rpi rebooted by cron
02/02/20--15:04:01 : rpi rebooted by cron
02/02/20--15:55:01 : rpi rebooted by cron
02/02/20--18:55:01 : rpi rebooted by cron
02/03/20--08:38:01 : rpi rebooted by cron

vicuvic, 2020-03-26:
Have similar issues when running airodump-ng.
Kali Linux fresh installed, after few minutes got this message:
brcmfmac: brcmf_sdio_hostmail: mailbox indicates firmware halted

After rebooting started to work, for few minutes and again the same error message.

chrisaljoudi, 2020-03-29:
Also encountering this issue on a Pi 4 setup as an AP with `hostapd` @JamesH65—`brcmfmac` seems to crash/halt. Any details I can provide to help track this down?

JamesH65, 2020-03-30:
@pelwell Is this something else to send off to Cypress?

pelwell, 2020-03-30:
Clear step-by-step instructions starting from a stock Raspbian always help, as does anything that increases the chance of failure.