raspberrypi/linux #3409: Moode Audio - HDD issue

dwadi, 2020-01-17:
Hi I have problem with external HDD on my Pi 4B 4GB. I testes 350gb 2.5" hdd with usb 2.0 enclosure and drive show up on list only during scan after that gone. I use Chinese DAC with Pi. Second I use USB 3.0 enclosure - same thing. Even after using powered hub (5V, 1A) drive is gone after scanning the files.

pelwell, 2020-01-17:
Can you clarify what you mean when you say "drive is gone"? Does it appear in lsusb or lsblk?

pelwell, 2020-01-17:
Also, please test with a stock Raspbian image with no other peripherals attached (except for keyboard, mouse, display and possibly network).

JamesH65, 2020-02-14:
@dwadi Have you been able to make the tests as outlined above? Or have you fixed the issue? In which case please close it.