raspberrypi/linux #3415: Error initializing Wifi 40Mhz channel width in drivers/net/wireless/broadcom/brcm80211/brcmfmac

bnicolae, 2020-01-20:
**Describe the bug**
The wifi driver does not intialize 802.11n mode correctly. Specifically, it lacks support for 40Mhz channel width, limiting wifi to 72 Mbps.

**To reproduce**
Run hostapd using the following configuration:

interface=wlan0
driver=nl80211
ssid=XXXX
hw_mode=g
ieee80211n=1
ht_capab=[SHORT-GI-40][HT40+][HT40-][DSSS_CCK-40]
channel=6
wme_enabled=1
macaddr_acl=0
auth_algs=1
ignore_broadcast_ssid=0
wpa=2
wpa_passphrase=YYYY
wpa_key_mgmt=WPA-PSK
wpa_pairwise=TKIP
rsn_pairwise=CCMP

**Expected behaviour**
40 Mhz channel width, which is supported by the antenna should work correctly. Commenting out the "ht_capab" option works just fine for 20Mhz channel width.

**Actual behaviour**
The hostapd service cannot start correctly due to a problem with the kernel driver.

**System**
Raspberry Pi 3 v1.2 running reference OS 2019-07-10, firmware version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (Sep 24 2019 17:39:25) and kernel 4.19.75-v7+.

**Logs**
[    6.192781] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
[    6.202134] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.202476] usbcore: registered new interface driver brcmfmac
[    6.409496] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    6.409604] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
[    6.410368] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f

**Additional context**

bnicolae, 2020-01-20:
Issue still present after upgrading to kernel 4.19.93-v7+

dmesg log:

```
 [    5.869624] cfg80211: Loading compiled-in X.509 certificates for regulatory database
 [    6.006217] cfg80211: Loaded X.509 cert 'sforshee: 00b28ddf47aef9cea7'
 [    6.070648] brcmfmac: F1 signature read @0x18000000=0x1541a9a6
 [    6.082457] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
 [    6.082767] usbcore: registered new interface driver brcmfmac
 [    6.289024] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
 [    6.289154] brcmfmac: brcmf_c_process_clm_blob: no clm_blob available (err=-2), device may have limited channels available
 [    6.290004] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Oct 23 2017 03:55:53 version 7.45.98.38 (r674442 CY) FWID 01-e58d219f
```

Ted-Wang, 2020-03-26:
I got the same issue on 4.19.66-v7+
and sometime, an under-voltage error is reported.
But I can still connect to my wifi.

burnbabyburn, 2020-05-13:
See linked topic. Get the full firmware with clm_blob to get missing ht capabilities.

bnicolae, 2020-05-15:
Indeed, the clm_blob is missing in the firmware. However, even after adding the clm_blob (which makes the kernel error message disappear), the chip (BCM43430 for Pi 3 v1.2) is still not HT40 capable. Apparently the kernel module can be patched to support it, but the sdio.txt needs to be adjusted accordingly too: 
https://www.raspberrypi.org/forums/viewtopic.php?t=144058

burnbabyburn, 2020-05-18:
Thanks for pointing that out!
And while 43430-sdio.txt represent board specific "nvram like" parameters we'll need to reach cypress/Broadcom? Everyone is using "default 20Mhz channel with sdio.txt" as it's the only one supplied by nonfree binary. Even Revision A0 of this chip just uses 20MHz

bnicolae, 2020-06-01:
I'm not sure. Unfortunately, I don't have the time to do hacking. This is something the Pi devs should take a look at, they are more familiar with the hardware.

bnicolae, 2020-07-20:
This issue persists for the newest kernel:
`Linux raspberrypi 5.4.51-v8+ #1326 SMP PREEMPT Fri Jul 17 10:58:13 BST 2020 aarch64 GNU/Linux`
and firmware:
`[    8.537721] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43430-sdio for chip BCM43430/1
[    8.551253] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM43430/1 wl0: Feb 16 2020 22:39:24 version 7.45.98.97 (r724416 CY) FWID 01-bf41ed64`

Output of `iw list` (no HT40 capabilities):
`Capabilities: 0x1020
                        HT20
                        Static SM Power Save
                        RX HT20 SGI
                        No RX STBC
                        Max AMSDU length: 3839 bytes
                        DSSS/CCK HT40
`

edgardmessias, 2020-07-20:
Wireless not work, but I can get HT40 in `iw list` if do:
```shell
sudo ./wl down
sudo ./wl mimo_bw_cap 1
sudo ./wl up
```

To get the `wl` binary: https://community.cypress.com/thread/44416

bnicolae, 2020-10-13:
Thanks @edgardmessias, we probably need some feedback on this topic from the kernel maintainers. Unfortunately it seems this is not a priority for them at the moment.