raspberrypi/linux #3416: v3d: swiotlb buffer is full

hvenev, 2020-01-20:
This error is shown when a wayland compositor is started (gnome-shell, weston, sway) on a pi 4 running 64-bit linux 5.4. It only happens when `total_mem > 1024`.

Kernel log:
```
[   17.606731] v3d fec00000.v3d: swiotlb buffer is full (sz: 536576 bytes), total 32768 (slots), used 3487 (slots)
[   17.606812] v3d fec00000.v3d: overflow 0x0000000040898000+536576 of DMA mask 7ffffffff bus mask 3fffffff
[   17.606867] ------------[ cut here ]------------
[   17.606903] WARNING: CPU: 2 PID: 500 at kernel/dma/direct.c:35 report_addr+0x38/0xa0
[   17.606936] Modules linked in: spidev mdio_bcm_unimac brcmfmac vc4 v3d snd_bcm2835(C) bcm2835_codec(C) snd_soc_core gpu_sched cfg80211 snd_pcm_dmaengine videobuf2_dma_contig bcm2835_v4l2(C) snd_pcm brcmutil v4l2_mem2mem videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 cec videobuf2_common xhci_pci xhci_hcd videodev bcm2835_mmc spi_bcm2835 mc rpivid_mem genet gpio_fan leds_gpio
[   17.607096] CPU: 2 PID: 500 Comm: gnome-shell Tainted: G         C        5.4.11-1-hvenev-pi4.aarch64 #1
[   17.607136] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   17.607163] pstate: 40000005 (nZcv daif -PAN -UAO)
[   17.607186] pc : report_addr+0x38/0xa0
[   17.607204] lr : report_addr+0x98/0xa0
[   17.607221] sp : ffffffc0113abb20
[   17.607238] x29: ffffffc0113abb20 x28: ffffffc0113abd40 
[   17.607262] x27: 0000000000000010 x26: ffffffc0113abd40 
[   17.607287] x25: ffffff804cfdb200 x24: 0000000000000000 
[   17.607311] x23: 0000000000000000 x22: ffffff80516a0410 
[   17.607335] x21: 0000000000000001 x20: 0000000000083000 
[   17.607359] x19: ffffff80516a0410 x18: 0000000000000000 
[   17.607382] x17: 0000000000000000 x16: 0000000000000000 
[   17.607406] x15: 000000000000ad55 x14: 0000000000000010 
[   17.607430] x13: 00000000ffffffff x12: ffffffd57b8280b8 
[   17.607454] x11: 0000000000000001 x10: 0000000000000000 
[   17.607478] x9 : 0000000000000003 x8 : 000000000000016d 
[   17.607502] x7 : 0000000000000000 x6 : 0000000000000001 
[   17.607525] x5 : ffffff8053f98808 x4 : 0000000000000001 
[   17.607549] x3 : ffffff8053f98808 x2 : 0000000000000007 
[   17.607573] x1 : 0000000000000006 x0 : 000000000000005c 
[   17.607597] Call trace:
[   17.607612]  report_addr+0x38/0xa0
[   17.607630]  dma_direct_map_page+0xe8/0x100
[   17.607651]  dma_direct_map_sg+0x78/0xe0
[   17.607675]  drm_gem_shmem_get_pages_sgt+0x7c/0xc8
[   17.607704]  v3d_bo_create_finish+0x20/0x108 [v3d]
[   17.607727]  v3d_bo_create+0x24/0x58 [v3d]
[   17.607748]  v3d_create_bo_ioctl+0x34/0xa8 [v3d]
[   17.607771]  drm_ioctl_kernel+0x94/0xf0
[   17.607789]  drm_ioctl+0x1ec/0x420
[   17.607807]  do_vfs_ioctl+0x864/0xb40
[   17.607824]  ksys_ioctl+0x44/0x90
[   17.607841]  __arm64_sys_ioctl+0x1c/0x48
[   17.607863]  el0_svc_handler+0x74/0x140
[   17.607882]  el0_svc+0x8/0xc
[   17.607897] ---[ end trace b9055b2a98febd2d ]---
[   17.607993] ------------[ cut here ]------------
[   17.609092] kernel BUG at drivers/gpu/drm/v3d/v3d_mmu.c:103!
[   17.609763] Internal error: Oops - BUG: 0 [#1] SMP
[   17.610408] Modules linked in: spidev mdio_bcm_unimac brcmfmac vc4 v3d snd_bcm2835(C) bcm2835_codec(C) snd_soc_core gpu_sched cfg80211 snd_pcm_dmaengine videobuf2_dma_contig bcm2835_v4l2(C) snd_pcm brcmutil v4l2_mem2mem videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 cec videobuf2_common xhci_pci xhci_hcd videodev bcm2835_mmc spi_bcm2835 mc rpivid_mem genet gpio_fan leds_gpio
[   17.612503] CPU: 2 PID: 500 Comm: gnome-shell Tainted: G        WC        5.4.11-1-hvenev-pi4.aarch64 #1
[   17.613171] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   17.613827] pstate: a0000005 (NzCv daif -PAN -UAO)
[   17.614474] pc : v3d_mmu_insert_ptes+0xec/0x110 [v3d]
[   17.615115] lr : v3d_bo_create_finish+0xd0/0x108 [v3d]
[   17.615752] sp : ffffffc0113abbb0
[   17.616394] x29: ffffffc0113abbb0 x28: ffffffc0113abd40 
[   17.617045] x27: 0000000000000010 x26: ffffffc0113abd40 
[   17.617695] x25: ffffff8046c52200 x24: 0000000000000042 
[   17.618349] x23: ffffff804df39800 x22: 0000000000000000 
[   17.618997] x21: 0000000000001ce0 x20: ffffff804c070000 
[   17.619645] x19: 0000000000ffffff x18: 0000000000000000 
[   17.620294] x17: 0000000000000000 x16: ffffffd57b7bf288 
[   17.620945] x15: 000000000000ad55 x14: 0000000000000010 
[   17.621599] x13: 00000000ffffffff x12: ffffffd57b8280b8 
[   17.622258] x11: 0000000000000001 x10: 0000000000000000 
[   17.622918] x9 : ffffff804cdabe00 x8 : ffffff804c070718 
[   17.623582] x7 : ffffff804cdabdc0 x6 : 00000000ffffffff 
[   17.624250] x5 : ffffff8043264a30 x4 : 0000000000000000 
[   17.624920] x3 : ffffff804c070000 x2 : 0000000000000000 
[   17.625588] x1 : 00000000ffffffff x0 : ffffff804cfdb200 
[   17.626267] Call trace:
[   17.626945]  v3d_mmu_insert_ptes+0xec/0x110 [v3d]
[   17.627630]  v3d_bo_create_finish+0xd0/0x108 [v3d]
[   17.628317]  v3d_bo_create+0x24/0x58 [v3d]
[   17.628999]  v3d_create_bo_ioctl+0x34/0xa8 [v3d]
[   17.629697]  drm_ioctl_kernel+0x94/0xf0
[   17.630388]  drm_ioctl+0x1ec/0x420
[   17.631068]  do_vfs_ioctl+0x864/0xb40
[   17.631871]  ksys_ioctl+0x44/0x90
[   17.632549]  __arm64_sys_ioctl+0x1c/0x48
[   17.633236]  el0_svc_handler+0x74/0x140
[   17.633921]  el0_svc+0x8/0xc
[   17.634600] Code: a9425bf5 f9401bf7 a8c47bfd d65f03c0 (d4210000) 
[   17.635299] ---[ end trace b9055b2a98febd2e ]---
[   17.636260] ------------[ cut here ]------------
[   17.637094] WARNING: CPU: 2 PID: 0 at kernel/rcu/tree.c:569 rcu_idle_enter+0x64/0x70
[   17.637929] Modules linked in: spidev mdio_bcm_unimac brcmfmac vc4 v3d snd_bcm2835(C) bcm2835_codec(C) snd_soc_core gpu_sched cfg80211 snd_pcm_dmaengine videobuf2_dma_contig bcm2835_v4l2(C) snd_pcm brcmutil v4l2_mem2mem videobuf2_vmalloc videobuf2_memops videobuf2_v4l2 cec videobuf2_common xhci_pci xhci_hcd videodev bcm2835_mmc spi_bcm2835 mc rpivid_mem genet gpio_fan leds_gpio
[   17.640516] CPU: 2 PID: 0 Comm: swapper/2 Tainted: G      D WC        5.4.11-1-hvenev-pi4.aarch64 #1
[   17.641415] Hardware name: Raspberry Pi 4 Model B Rev 1.1 (DT)
[   17.642257] pstate: 200003c5 (nzCv DAIF -PAN -UAO)
[   17.643216] pc : rcu_idle_enter+0x64/0x70
[   17.644087] lr : do_idle+0x1d4/0x208
[   17.644815] sp : ffffffc01008bf40
[   17.645542] x29: ffffffc01008bf40 x28: a291061842cd7893 
[   17.646372] x27: d6aec3c011e307c5 x26: 92915a3018041801 
[   17.647106] x25: 0000000000000000 x24: 0000000000000000 
[   17.647873] x23: ffffffd57bae3770 x22: 0000000000000060 
[   17.648575] x21: ffffffd57bbc9898 x20: 0000000000000002 
[   17.649278] x19: ffffffd57bbc9820 x18: 00000000fffffffc 
[   17.649981] x17: 00e800001de42f53 x16: 0000000000000000 
[   17.650677] x15: 0000000000000010 x14: 0000000000000003 
[   17.651372] x13: 0000000000000000 x12: 0000000000000000 
[   17.652065] x11: 0000000000000000 x10: 00000000000008a0 
[   17.652776] x9 : ffffffc01008beb0 x8 : ffffff80514a2200 
[   17.653483] x7 : 0000000000000400 x6 : 00000000393549c2 
[   17.654168] x5 : 00ffffffffffffff x4 : ffffffaad84bd000 
[   17.654856] x3 : 4000000000000002 x2 : 4000000000000000 
[   17.655544] x1 : ffffffd57bae5380 x0 : ffffff8053fa2380 
[   17.656231] Call trace:
[   17.656922]  rcu_idle_enter+0x64/0x70
[   17.657614]  do_idle+0x1d4/0x208
[   17.658306]  cpu_startup_entry+0x20/0x60
[   17.659002]  secondary_start_kernel+0x138/0x170
[   17.659693] ---[ end trace b9055b2a98febd2f ]---
```

pelwell, 2020-01-20:
You must be using an incorrect DTB - v3d should not require any swiotlb (software bounce buffers).

The v3d node in bcm2711-rpi-4-b.dtb should end up looking like this:
```
	v3dbus {
		compatible = "simple-bus";
		#address-cells = <1>;
		#size-cells = <2>;
		ranges = <0x7c500000 0x0 0xfc500000 0x0 0x03300000>, <0x40000000 0x0 0xff800000 0x0 0x00800000>;
		dma-ranges = <0x00000000 0x0 0x00000000 0x4 0x00000000>;
		v3d: v3d@7ec04000 {
			compatible = "brcm,2711-v3d";
			reg = <0x7ec00000 0x0 0x4000>, <0x7ec04000 0x0 0x4000>;
			reg-names = "hub", "core0";
			power-domains = <&pm BCM2835_POWER_DOMAIN_GRAFX_V3D>;
			resets = <&pm BCM2835_RESET_V3D>;
			clocks = <&clocks BCM2835_CLOCK_V3D>;
			interrupts = <GIC_SPI 74 IRQ_TYPE_LEVEL_HIGH>;
			status = "disabled";
		};
	};
```

hvenev, 2020-01-20:
I was using the device tree from raspberrypi/firmware. Building the one that comes with new kernel from source fixed the issue.

closed 2020-01-20 by hvenev

xyklex, 2020-08-03:
Hi @pelwell, @hvenev is this only working with 64 bits?

I was not able to make it works with this commit https://github.com/raspberrypi/linux/commit/0fe13a5838b20d5078c46ef4f132f02aa3a06641

I'm building a Yocto image and I updated that commit to the source version, but I'm building it on 32 bits.

dominikschnitzer, 2020-08-27:
I also see this error with 5.4 on a yocto dunfell build

@hvenev which device tree were you using.

Also, is this related to #3411?

hvenev, 2020-08-27:
I've only ever used a 64-bit kernel. I haven't seen this error since I switched to >= 5.4 device trees. Over the last few months I've used 5.4 to 5.7 and I haven't seen any swiotlb-related errors.

nicoske, 2022-11-15:
I have the exact same issue with a buildroot build on PI4 64 bits using kernel 5.15.76-v8 with DTB built from kernel and same version of DTBO coming from rpi-firmware package.

pelwell, 2022-11-15:
The overlays are meant to be kept in step with the kernel and base DTBs - mixing and matching is a recipe for disaster.

nicoske, 2022-11-15:
I've just copied the DTB + DTBO generated by the kernel build to the Pi4 and have the same issue unfortunately.

nicoske, 2022-11-15:
Forgot to mention that those messages only appears after starting the webkit instance using cog!

pelwell, 2022-11-15:
Do you have a `gpu_mem` setting in config.txt?

nicoske, 2022-11-15:
Nope, this is my config:

```
disable_splash=1
kernel=u-boot.bin
arm_64bit=1
disable_overscan=1
dtparam="audio=on"

[pi4]
dtoverlay=vc4-kms-v3d
max_framebuffers=2
```

timothymawhinney, 2022-11-15:
Also getting this in dmesg on Gentoo with CM4 running kernel 6.0 under wayland and sway;

vc4-drm gpu: swiotlb buffer is full (sz: 8093696 bytes), total 32768 (slots), used 68 (slots)

pelwell, 2022-11-15:
Is U-boot configured to use the DTB loaded and modified by the firmware? If not, it could be breaking all sorts of things.

nicoske, 2022-11-15:
Yes, in fact I also tested by loading the kernel directly without using U-Boot and I still have those errors in kernel log and degraded performance.
I'm also using Sway 1.7

nicoske, 2022-11-16:
So today I've recompiled my entire project using Buildroot 2022.05 instead of 2022.08 and.. no errors anymore !
I'm still using the same kernel and DTB/DTBO..