raspberrypi/linux #3417: genet on 5.4 does not work phy-mode=rgmii-rxid

hvenev, 2020-01-20:
My Pi 4's ethernet connection stopped working when I updated the device tree to 5.4. It was working fine using the dtb from raspberrypi/firmware.

I am using the latest 5.4 kernel (cf75e7063b9d5f3bc3d8d9345a0450ce325e9e87), which should support RGMII_RXID.

The link is set up and configured successfully, and it is receiving packets. However, I don't think anything it sends is being received on the other side.

Reverting the change to phy-mode fixed it for me:

```patch
diff --git a/arch/arm/boot/dts/bcm2711-rpi-4-b.dts b/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
index 78273ec873ff..348765be064f 100644
--- a/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
+++ b/arch/arm/boot/dts/bcm2711-rpi-4-b.dts
@@ -136,7 +136,7 @@
 
 &genet {
        phy-handle = <&phy1>;
-       phy-mode = "rgmii-rxid";
+       phy-mode = "rgmii";
        status = "okay";
 };
 ```

pelwell, 2020-01-21:
It's working for me with https://github.com/raspberrypi/linux/commit/cf75e7063b9d5f3bc3d8d9345a0450ce325e9e87, with fresh builds of both 32- and 64-bit kernels. Are there any relevant errors (`dmesg -l err`) or PHY-related logging (`dmesg | grep -i phy`)? You should compare the results in the success and failure cases.

hvenev, 2020-01-22:
Here is the relevant part of the kernel log (errors or lines mentioning phy or genet):

```
[    0.348145] libphy: Fixed MDIO Bus: probed
[    3.246455] bcmgenet fd580000.ethernet: failed to get enet clock
[    3.246491] bcmgenet fd580000.ethernet: GENET 5.0 EPHY: 0x0000
[    3.246511] bcmgenet fd580000.ethernet: failed to get enet-wol clock
[    3.246527] bcmgenet fd580000.ethernet: failed to get enet-eee clock
[    3.246552] bcmgenet: Skipping UMAC reset
[    3.644304] bcm2835_audio bcm2835_audio: Failed to get DT property 'brcm,pwm-channels'
[    3.697928] bcmgenet fd580000.ethernet eth: renamed from eth0
[    3.815666] libphy: bcmgenet MII bus: probed
[    6.087502] bcmgenet: Skipping UMAC reset
[    6.094001] bcmgenet fd580000.ethernet: configuring instance for external RGMII
[    6.096382] bcmgenet fd580000.ethernet eth: Link is Down
[   11.175870] bcmgenet fd580000.ethernet eth: Link is Up - 1Gbps/Full - flow control off
[  305.042501] bcmgenet fd580000.ethernet eth: Link is Down
[  309.095905] bcmgenet fd580000.ethernet eth: Link is Up - 1Gbps/Full - flow control off
```

lategoodbye, 2020-01-25:
Based on this kernel log, there are not critical errors.

Please provide more information:
What is your exact network setup?
What kind of network protocol do you use?

lategoodbye, 2020-01-26:
I think i've found the reason for this issue. There is a patch for the PHY driver, which has just been backport to 5.4.15:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=v5.4.15&id=52fa627e20df2ea352bf1ecec636d8f81a196bbd

hvenev, 2020-01-26:
There is an ethernet cable between the pi and the router.

Setup 1 (what I normally use): There is a bridge interface and the ethernet interface is enslaved to it. The bridge has a /23 IPv4 address. The default gateway is in the same subnet.
Setup 2 (what I tried once): There is no bridge and the ethernet interface does not have a master. Same network configuration otherwise.

In both setups, the command

```
# arping -I INTERFACE ROUTER_IP -s PI_IP
```

only works with rgmii, not rgmii-rxid. I'm running the same kernel and the same DTB, with an overlay that I conditionally enable to toggle between the two modes.

`tcpdump` on the pi confirms that packets are being received on the interface. However, ARP requests don't get any replies.

I've merged 5.4.15 into the 5.4 branch from this repository and it did not seem to help.

lategoodbye, 2020-03-06:
There is another patch:
https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v5.6-rc4&id=402482a6a78e5c61d8a2ec6311fc5b4aca392cd6

hvenev, 2020-03-12:
The patch doesn't seem to help.

vianpl, 2020-03-13:
Sorry I'm late to the party. So I get that you're using the rpi-5.4.y, as opposed to the official kernel stable branch right?

hvenev, 2020-03-13:
I upgraded to 5.5 a while ago, so I'm using the rpi-5.5.y branch.

lategoodbye, 2020-03-13:
Is it the same kernel which has USB broken #3501 ?

hvenev, 2020-03-13:
No, this also happens on an unmodified kernel.

lategoodbye, 2020-05-08:
@hvenev Can you please check your kernel config if CONFIG_BROADCOM_PHY is enabled?

hvenev, 2020-05-08:
Yes, I had also only enabled CONFIG_BCM7XXX_PHY but not CONFIG_BROADCOM_PHY.

I have already added the config fix for the next kernel update. Once linux 5.6.12 is out (plus another 24 hours or so), I will share results from 3 independently acquired Pi4s.

Thanks everyone for finding the problem!

hvenev, 2020-05-10:
Fixed.

closed 2020-05-10 by hvenev