raspberrypi/linux #3426: systemd[1]: Failed to bump fs.file-max, ignoring: Invalid argument

slrslr, 2020-01-24:
Hello, on Raspberry Pi 4, Raspbian buster 10 full installed from NOOBS, kernel 4.19.93-v7l+

> \# vcgencmd version
> Jan 10 2020 ...

> \# cat /etc/rpi-issue
> Raspberry Pi reference 2019-09-26

"dmesg" shows:

> systemd[1]: Failed to bump fs.file-max, ignoring: Invalid argument

i tried to manually increase the open files limit:

> \# cat /proc/sys/fs/file-max
> 65536

> \# grep fs.file /etc/sysctl.conf
> fs.file-max=65536

> \# ulimit -a
> core file size          (blocks, -c) 0
> data seg size           (kbytes, -d) unlimited
> scheduling priority             (-e) 0
> file size               (blocks, -f) unlimited
> pending signals                 (-i) 27756
> max locked memory       (kbytes, -l) 65536
> max memory size         (kbytes, -m) unlimited
> open files                      (-n) 1024
> pipe size            (512 bytes, -p) 8
> POSIX message queues     (bytes, -q) 819200
> real-time priority              (-r) 0
> stack size              (kbytes, -s) 8192
> cpu time               (seconds, -t) unlimited
> max user processes              (-u) 27756
> virtual memory          (kbytes, -v) unlimited
> file locks                      (-x) unlimited

I did it wrong? How to do it well or is this something that You can tweak so it works?

XECDesign, 2020-01-24:
IIRC, it's just a systemd issue where they're using a wrong data type. The fix should eventually make it into Debian, but since it doesn't cause problems unless you actually run into the limit, I don't think that will happen any time soon.

JamesH65, 2020-02-14:
This issue will be closed within 30 days unless further interactions are posted. If you wish this issue to remain open, please add a comment. A closed issue may be reopened if requested.

XECDesign, 2020-02-19:
Should be fixed in the latest version of systemd in apt:

```
systemd (241-7~deb10u3) buster; urgency=medium

  * core: set fs.file-max sysctl to LONG_MAX rather than ULONG_MAX.
    Since kernel 5.2 (but also stable kernels like 4.19.53) the kernel
    thankfully returns proper errors when we write a value out of range to
    the sysctl. Which however breaks writing ULONG_MAX to request the
    maximum value. Hence let's write the new maximum value instead,
    LONG_MAX. (Closes: #945018)
  * core: change ownership/mode of the execution directories also for static
    users.
    This ensures that execution directories like CacheDirectory and
    StateDirectory are properly chowned to the user specified in User= before
    launching the service. (Closes: #919231)

 -- Michael Biebl <biebl@debian.org>  Wed, 29 Jan 2020 19:07:53 +0100
```

closed 2020-11-18 by pelwell