raspberrypi/linux #3428: lib/ts_kmp.c typo

miroslav-beranic, 2020-01-26:
In source file lib/ts_kmp.c there is a typo:

`#include <linux/module.h>
#include <linux/types.h>
#include <lifux/string.h> // lifux should be linux
#include <linux/ctype.h>
#include <linux/textsearch.h>
`

6by9, 2020-01-26:
https://github.com/raspberrypi/linux/blob/rpi-4.19.y/lib/ts_kmp.c#L35 says linux, as do the rpi-5.4.y and rpi-5.5.y branches. This sounds like a cloning error.

pelwell, 2020-01-26:
Agreed - this error is not present in our repo or upstream.

closed 2020-01-26 by pelwell

miroslav-beranic, 2020-01-26:
OK. I agree. But, what does the git status say - up to date, but ( local ) file has different content that when I look at the online "version". Any idea?

If this "This sounds like a cloning error.", how can this be? How can this be fixed? What does not git knows the difference? ... out of this world ...

pelwell, 2020-01-26:
Does `git show master:lib/ts_kmp.c` show the correct version?
Did you clone with depth=1?

miroslav-beranic, 2020-01-26:
Hi,

well, it looks like I did not clone this repo manualy, but using scripts. Here is my history output:

```
$ git clone git://git.yoctoproject.org/poky
$ cd poky/
$ git clone git://git.yoctoproject.org/meta-raspberrypi
$ source oe-init-build-env 
$ geany conf/bblayers.conf 
$ bitbake rpi-basic-image
```

Where the output from `bitbake rpi-basic-image` is ( this is the 2nd run, as I thought network failure -- it was building over the night 1st time ):
```
[miroslav@w520 build]$ bitbake rpi-basic-image
WARNING: Host distribution "fedora-31" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Loading cache: 100% |##########################################################################################################################################################################################################| Time: 0:00:00
Loaded 1324 entries from dependency cache.
Parsing recipes: 100% |########################################################################################################################################################################################################| Time: 0:00:00
Parsing of 797 .bb files complete (796 cached, 1 parsed). 1325 targets, 63 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "universal"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi"
DISTRO               = "poky"
DISTRO_VERSION       = "3.0+snapshot-20200126"
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "master:ca3993cc4b13d4e661228cee6fb9448adfd0a4ba"
meta-raspberrypi     = "master:37d3dd955d34eab5f4e2c924e8944ac0d15be1c1"

Initialising tasks: 100% |#####################################################################################################################################################################################################| Time: 0:00:02
Sstate summary: Wanted 478 Found 0 Missed 478 Current 923 (0% match, 65% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
ERROR: linux-raspberrypi-1_4.19.97+gitAUTOINC+c8fbe9c63c-r0 do_compile_kernelmodules: oe_runmake failed
ERROR: linux-raspberrypi-1_4.19.97+gitAUTOINC+c8fbe9c63c-r0 do_compile_kernelmodules: Execution of '/path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/temp/run.do_compile_kernelmodules.1757828' failed with exit code 1:
make: Entering directory '/path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/linux-raspberrypi-standard-build'
  GEN     ./Makefile
  Using /path/to/yocto/poky/build/tmp/work-shared/raspberrypi/kernel-source as source for kernel
  CALL    /path/to/yocto/poky/build/tmp/work-shared/raspberrypi/kernel-source/scripts/checksyscalls.sh
  CC [M]  lib/ts_kmp.o
/path/to/yocto/poky/build/tmp/work-shared/raspberrypi/kernel-source/lib/ts_kmp.c:35:10: fatal error: lifux/string.h: No such file or directory
   35 | #include <lifux/string.h>
      |          ^~~~~~~~~~~~~~~~
compilation terminated.
make[3]: *** [/path/to/yocto/poky/build/tmp/work-shared/raspberrypi/kernel-source/scripts/Makefile.build:309: lib/ts_kmp.o] Error 1
make[2]: *** [/path/to/yocto/poky/build/tmp/work-shared/raspberrypi/kernel-source/Makefile:1052: lib] Error 2
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [Makefile:146: sub-make] Error 2
make: *** [Makefile:24: __sub-make] Error 2
make: Leaving directory '/path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/linux-raspberrypi-standard-build'
WARNING: /path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/temp/run.do_compile_kernelmodules.1757828:1 exit 1 from 'exit 1'

ERROR: Logfile of failure stored in: /path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/temp/log.do_compile_kernelmodules.1757828
Log data follows:
| DEBUG: Executing shell function do_compile_kernelmodules
| NOTE: make HOSTCC=gcc  -isystem/path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/recipe-sysroot-native/usr/include -O2 -pipe -L/path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/recipe-sysroot-native/usr/lib                         -L/path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/recipe-sysroot-native/lib                         -Wl,--enable-new-dtags                         -Wl,-rpath-link,/path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/recipe-sysroot-native/usr/lib                         -Wl,-rpath-link,/path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8fbe9c63c-r0/recipe-sysroot-native/lib                         -Wl,-rpath,/path/to/yocto/poky/build/tmp/work/raspberrypi-poky-linux-gnueabi/linux-raspberrypi/1_4.19.97+gitAUTOINC+c8 [... cut ...]
ERROR: Task (/path/to/yocto/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.19.bb:do_compile_kernelmodules) failed with exit code '1'
NOTE: Tasks Summary: Attempted 1867 tasks of which 1772 didn't need to be rerun and 1 failed.

Summary: 1 task failed:
  /path/to/yocto/poky/meta-raspberrypi/recipes-kernel/linux/linux-raspberrypi_4.19.bb:do_compile_kernelmodules
Summary: There was 1 WARNING message shown.
Summary: There were 2 ERROR messages shown, returning a non-zero exit code.
```
I've deleted the `/path/to/yocto/poky/build/tmp` and will start over now.

miroslav-beranic, 2020-01-26:
Git commands -- status checks -- where done from "generated" folder: `/path/to/yocto/poky/build/tmp/work-shared/raspberrypi/kernel-source`.

miroslav-beranic, 2020-01-26:
Hi,

Does git show master:lib/ts_kmp.c show the correct version? -- I've deleted the tmp folder, so can not tell.

Did you clone with depth=1? -- the commands above are the exact ones, so no I did not; but I guess as I understand now, I did not clone the repos ( with errors ) manually.

miroslav-beranic, 2020-01-26:
After I've deleted `tmp` folder, I re-ran the `bitbake rpi-basic-image`, here is the output:

```
$ bitbake rpi-basic-image
WARNING: Host distribution "fedora-31" has not been validated with this version of the build system; you may possibly experience unexpected failures. It is recommended that you use a tested distribution.
Parsing recipes: 100% |########################################################################################################################################################################################################| Time: 0:00:26
Parsing of 797 .bb files complete (0 cached, 797 parsed). 1325 targets, 63 skipped, 0 masked, 0 errors.
NOTE: Resolving any missing task queue dependencies

Build Configuration:
BB_VERSION           = "1.44.0"
BUILD_SYS            = "x86_64-linux"
NATIVELSBSTRING      = "fedora-31"
TARGET_SYS           = "arm-poky-linux-gnueabi"
MACHINE              = "raspberrypi"
DISTRO               = "poky"
DISTRO_VERSION       = "3.0+snapshot-20200126"
TUNE_FEATURES        = "arm armv6 vfp arm1176jzfs callconvention-hard"
TARGET_FPU           = "hard"
meta                 
meta-poky            
meta-yocto-bsp       = "master:ca3993cc4b13d4e661228cee6fb9448adfd0a4ba"
meta-raspberrypi     = "master:37d3dd955d34eab5f4e2c924e8944ac0d15be1c1"

Initialising tasks: 100% |#####################################################################################################################################################################################################| Time: 0:00:02
Sstate summary: Wanted 1401 Found 963 Missed 438 Current 0 (68% match, 0% complete)
NOTE: Executing Tasks
NOTE: Setscene tasks completed
WARNING: rpi-basic-image-1.0-r0 do_image: The image 'rpi-basic-image' is deprecated, please use 'core-image-base' instead
NOTE: Tasks Summary: Attempted 3694 tasks of which 2467 didn't need to be rerun and all succeeded.

Summary: There were 2 WARNING messages shown.
```

It loolks like, it works; but -- how can I be sure this it OK, and that there was no changed chars somewhere inside?

Btw: Next time, I will build with `core-image-base`.