raspberrypi/linux #3430: Possible bug with cdc_ether, triggers NETDEV WATCHDOG

sim0nx, 2020-01-26:
**Describe the bug**
I am using a RPI zero, flashed with the latest raspbian lite (buster) and trying to get usb-otg-ethernet working.

In the config.txt I have added:
```
dtoverlay=dwc2
```

in the cmdline.txt:
```
modules-load=dwc2,g_ether g_ether.host_addr=00:22:82:ff:ff:01 g_ether.dev_addr=00:22:82:ff:ff:02
```

**Expected behaviour**
The host should add a usb network interface with the MAC address "00:22:82:ff:ff:01" and networking should just work.

**Actual behaviour**
Most of the time the interface is correctly added with that MAC address, though the host is not able to send any packets to the RPI.
Using tcpdump on the interface I can clearly see the RPI sending ARP packets looking for other systems, but any attempt to establish a connection from the host fails.

Now traing to reboot the RPI several times, once in a while the connection does work. In that case there are no problems at all.
Note that there is no firewal causing issues or something else. This is really a random behavior.

The issue is also not related to a specific host, but this issues shows on all tested hosts (debian buster, ubuntu 19.10).

Here is a dmesg output of when the connection fails:
```
[324250.365696] usb 5-2: new high-speed USB device number 16 using ehci-pci
[324251.061798] usb 5-2: device descriptor read/64, error -71
[324251.326832] usb 5-2: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 4.19
[324251.326837] usb 5-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[324251.326840] usb 5-2: Product: RNDIS/Ethernet Gadget
[324251.326842] usb 5-2: Manufacturer: Linux 4.19.75+ with 20980000.usb
[324251.334188] cdc_subset: probe of 5-2:1.0 failed with error -22
[324251.335146] cdc_ether 5-2:1.0 eth1: register 'cdc_ether' at usb-0000:00:16.2-2, CDC Ethernet Device, 00:22:82:ff:ff:01
[324251.368094] cdc_ether 5-2:1.0 enx002282ffff01: renamed from eth1
[324267.793849] audit: type=1400 audit(1580062868.929:395): apparmor="DENIED" operation="open" profile="/usr/bin/akonadiserver" name="/usr/local/share/mime/packages/" pid=23761 comm=5468726561642028706F6F6C656429 requested_mask="r" denied_mask="r" fsuid=500 ouid=0
[324292.627110] device enx002282ffff01 entered promiscuous mode
[324301.271192] device enx002282ffff01 left promiscuous mode
[324315.836194] audit: type=1400 audit(1580062916.972:396): apparmor="DENIED" operation="open" profile="/usr/bin/akonadiserver" name="/usr/local/share/mime/packages/" pid=23761 comm=5468726561642028706F6F6C656429 requested_mask="r" denied_mask="r" fsuid=500 ouid=0
[324317.427898] device enx002282ffff01 entered promiscuous mode
[324320.203948] ------------[ cut here ]------------
[324320.203994] NETDEV WATCHDOG: enx002282ffff01 (cdc_ether): transmit queue 0 timed out
[324320.204048] WARNING: CPU: 4 PID: 0 at net/sched/sch_generic.c:448 dev_watchdog+0x253/0x260
[324320.204057] Modules linked in: cdc_ether cdc_subset usbnet mii vboxnetadp(OE) vboxnetflt(OE) vboxdrv(OE) btrfs zstd_compress zstd_decompress ufs qnx4 hfsplus hfs minix msdos jfs cpuid xt_conntrack xt_MASQUERADE nf_conntrack_netlink xfrm_user xfrm_algo nft_counter xt_addrtype nft_compat nft_chain_nat nf_nat nf_conntrack nf_defrag_ipv6 nf_defrag_ipv4 br_netfilter bridge stp llc md4 sha512_ssse3 sha512_generic cmac nls_utf8 cifs gcm libarc4 dns_resolver fscache fuse nls_ascii nls_cp437 vfat fat uas usb_storage bnep nf_tables nfnetlink bluetooth overlay drbg ansi_cprng ecdh_generic ecc rfkill binfmt_misc ext4 snd_hda_codec_realtek crc16 mbcache amdgpu jbd2 snd_hda_codec_generic snd_hda_codec_hdmi ledtrig_audio snd_hda_intel snd_hda_codec edac_mce_amd gpu_sched kvm_amd snd_hda_core ccp ttm snd_hwdep rng_core joydev evdev snd_pcm_oss kvm snd_mixer_oss drm_kms_helper irqbypass pcspkr serio_raw wmi_bmof mxm_wmi snd_pcm fam15h_power k10temp drm snd_timer snd sp5100_tco i2c_algo_bit watchdog
[324320.204129]  soundcore mfd_core sg button acpi_cpufreq loop ecryptfs parport_pc ppdev lp parport sunrpc ip_tables x_tables autofs4 algif_skcipher af_alg dm_crypt dm_mod raid10 raid456 async_raid6_recov async_memcpy async_pq async_xor async_tx xor raid6_pq raid1 raid0 multipath linear md_mod xfs libcrc32c crc32c_generic hid_logitech_hidpp hid_logitech_dj hid_generic usbhid hid sd_mod crct10dif_pclmul crc32_pclmul crc32c_intel ghash_clmulni_intel ohci_pci ahci r8169 libahci aesni_intel realtek xhci_pci libata libphy xhci_hcd aes_x86_64 crypto_simd ohci_hcd ehci_pci ehci_hcd cryptd glue_helper i2c_piix4 usbcore scsi_mod usb_common wmi [last unloaded: vboxdrv]
[324320.204179] CPU: 4 PID: 0 Comm: swapper/4 Tainted: G           OE     5.3.0-0.bpo.2-amd64 #1 Debian 5.3.9-2~bpo10+1
[324320.204181] Hardware name: Gigabyte Technology Co., Ltd. GA-990FXA-D3/GA-990FXA-D3, BIOS F8b 07/11/2012
[324320.204186] RIP: 0010:dev_watchdog+0x253/0x260
[324320.204190] Code: 48 85 c0 75 e4 eb 9d 4c 89 ef c6 05 c6 76 a9 00 01 e8 d1 1f fb ff 89 d9 4c 89 ee 48 c7 c7 a0 d4 90 b8 48 89 c2 e8 f6 bc a1 ff <0f> 0b e9 7c ff ff ff 66 0f 1f 44 00 00 0f 1f 44 00 00 41 57 41 56
[324320.204192] RSP: 0018:ffffbdae80174e88 EFLAGS: 00010282
[324320.204194] RAX: 0000000000000000 RBX: 0000000000000000 RCX: 0000000000000000
[324320.204196] RDX: 0000000000040400 RSI: 00000000000000f6 RDI: 0000000000000300
[324320.204198] RBP: ffff99fa4e37345c R08: 0000000000001407 R09: 0000000000000004
[324320.204199] R10: 0000000000000000 R11: 0000000000000001 R12: ffff99f97bbb0e80
[324320.204201] R13: ffff99fa4e373000 R14: ffff99fa4e373480 R15: 0000000000000001
[324320.204204] FS:  0000000000000000(0000) GS:ffff99fceeb00000(0000) knlGS:0000000000000000
[324320.204206] CS:  0010 DS: 0000 ES: 0000 CR0: 0000000080050033
[324320.204208] CR2: 00007f55707aee90 CR3: 00000003f1a8e000 CR4: 00000000000406e0
[324320.204210] Call Trace:
[324320.204213]  <IRQ>
[324320.204220]  ? pfifo_fast_enqueue+0x140/0x140
[324320.204224]  call_timer_fn+0x2d/0x130
[324320.204228]  run_timer_softirq+0x19e/0x410
[324320.204232]  ? tick_sched_timer+0x37/0x70
[324320.204235]  ? __hrtimer_run_queues+0x110/0x280
[324320.204238]  ? ktime_get+0x3a/0xa0
[324320.204243]  __do_softirq+0xdf/0x2e5
[324320.204249]  irq_exit+0xa3/0xb0
[324320.204252]  smp_apic_timer_interrupt+0x74/0x130
[... 51 lines cut ...]
Sep 26 01:24:40 raspberrypi kernel: [    7.409140] dwc2 20980000.usb: new device is high-speed
Sep 26 01:24:40 raspberrypi kernel: [    7.617156] dwc2 20980000.usb: new device is high-speed
Sep 26 01:24:40 raspberrypi kernel: [    7.853117] dwc2 20980000.usb: new device is high-speed
Sep 26 01:24:40 raspberrypi kernel: [    7.981253] dwc2 20980000.usb: new device is high-speed
Sep 26 01:24:40 raspberrypi kernel: [    8.041153] dwc2 20980000.usb: new address 15
Sep 26 01:24:40 raspberrypi kernel: [    8.074309] g_ether gadget: high-speed config #1: CDC Ethernet (ECM)

```

**System**
 Copy and paste the results of the raspinfo command in to this section. Alternatively, copy and paste a pastebin link, or add answers to the following questions:

* Which model of Raspberry Pi? PiZeroW

* Which OS and version (`cat /etc/rpi-issue`)? 
Raspberry Pi reference 2019-09-26
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 80d486687ea77d31fc3fc13cf3a2f8b464e129be, stage2

* Which firmware version (`vcgencmd version`)?
Sep 24 2019 17:37:19 
Copyright (c) 2012 Broadcom
version 6820edeee4ef3891b95fc01cf02a7abd7ca52f17 (clean) (release) (start)

* Which kernel version (`uname -a`)?
Linux raspberrypi 4.19.75+ #1270 Tue Sep 24 18:38:54 BST 2019 armv6l GNU/Linux

pelwell, 2020-01-26:
Am I right in thinking the second kernel log is actually for a successful connection?

ED6E0F17, 2020-01-26:
sim0nx appears to be using Linux 5.3 on the host. The linked bug report suggests that the bug is in the host, not the gadget. There have been some very recent commits in Linux 5.4 that may resolve this issue.

````
> The issue is reported on the host, but it's really a problem with the
> gadget.  You'll probably have the same issues with any host, including
> hosts running something other than Linux.  They just won't be as verbose
> about it.
>
That was my original assumption also.  But I tested the Pi Zero on 
Windows 10 (build 1903) and latest MacOS, and it works fine on those 
machines.

pelwell, 2020-01-26:
Sorry, I get it now - you are reporting that a Pi can crash another host? If the Pi doesn't crash then it seems odd to report it here. Occam's razor says it's more likely that there is a bug in the host side that can cause failed connections and crashes, rather than a bug in the Pi gadget that can cause connection failures, that in turn causes the host to crash.

sim0nx, 2020-01-26:
> Am I right in thinking the second kernel log is actually for a successful connection?

Nope, that log is from when the connection did not work.

pelwell, 2020-01-26:
See later comment.

sim0nx, 2020-01-26:
There is no crash on the host, neither on the Pi.
The only thing I observe is that the usb-ethernet connection from the host to the Pi sometimes works, and sometimes doesn't, for no apparent reason.

When it does not work, I can still see packets from the Pi (using tcpdump), but I can't e.g. ping or SSH, the Pi end.

pelwell, 2020-01-26:
As already quoted by @ED6E0F17, the linked bug report said:
> But I tested the Pi Zero on 
Windows 10 (build 1903) and latest MacOS, and it works fine on those 
machines.

I still fail to see what leads you to conclude that the fault lies in the Pi.

sim0nx, 2020-01-26:
I just thought it was related as what he describes seems to be close to what I am seeing. Please disregard it then.

I am having the issue right now, here is a packet trace:
```
20:59:37.525477 ARP, Request who-has 10.245.0.1 tell 10.245.0.2, length 28
20:59:37.525514 ARP, Reply 10.245.0.1 is-at 00:22:82:ff:ff:01, length 28
20:59:37.600402 ARP, Request who-has 10.245.0.2 tell 10.245.0.1, length 28
```

.1 is the host, .2 is the Pi.
So everything looks, fine ... though it just continues like that and I am never able to simply ping the Pi.

Kernel log from attaching the Pi:
```
[330146.570890] usb 5-2: new high-speed USB device number 33 using ehci-pci
[330147.010892] usb 5-2: device descriptor read/64, error -71
[330147.850904] usb 5-2: device descriptor read/64, error -71
[330148.086931] usb 5-2: new high-speed USB device number 34 using ehci-pci
[330148.248166] usb 5-2: New USB device found, idVendor=0525, idProduct=a4a2, bcdDevice= 4.19
[330148.248170] usb 5-2: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[330148.248174] usb 5-2: Product: RNDIS/Ethernet Gadget
[330148.248176] usb 5-2: Manufacturer: Linux 4.19.97+ with 20980000.usb
[330148.259278] cdc_subset: probe of 5-2:1.0 failed with error -22
[330148.260212] cdc_ether 5-2:1.0 eth1: register 'cdc_ether' at usb-0000:00:16.2-2, CDC Ethernet Device, 00:22:82:ff:ff:01
[330148.293765] cdc_ether 5-2:1.0 enx002282ffff01: renamed from eth1
```

I am not sure if it is the Pi or the host, but as I said I am seeing this on various kernel versions (4.19 / 5.3) and various hosts I have tested it on.

Is there anything I could do to debug this ?

sim0nx, 2020-01-26:
I just discovered UDC for configuring gadget ethernet mode, which gives consistent good results.
I am thus closing this issue.

Thanks for your looking into this.

closed 2020-01-26 by sim0nx

Allarious, 2020-07-09:
I had the same problem with raspberry pi zero, it seems that when I load g_ether via cmdline.txt it leads to instability but when I load g_ether after the boot is complete, it works fine. loading g_ether in cmdline.txt might trigger some kind of race condition.
I solved it by adding this line of code to /etc/rc.local which runs after most services are up:
```
sudo modprobe g_ether
```
this way you can set fixed mac addresses for your pi by adding this to /etc/modprobe.d/g_ether.conf:
```
options g_ether host_addr=02:11:22:33:44:55 dev_addr=02:11:22:33:44:56
```

PaulCrz, 2021-03-17:
> I had the same problem with raspberry pi zero, it seems that when I load g_ether via cmdline.txt it leads to instability but when I load g_ether after the boot is complete, it works fine. loading [...]

Thank you ! I confirm I had exactly the same problem and it comes from the fact that you have to wait for the raspberry to be initialized before launching the driver, otherwise it can work but randomly