raspberrypi/linux #3439: bug when systemd schedule a fsck of a partition on a USB HDD at boot

solsticedhiver, 2020-01-30:
When booting a rpi3-b (running archlinux-arm/armv7 kernel 4.19.97-1-ARCH), the kernel crash with a 

    BUG: scheduling while atomic: kauditd/37/0x00000003

when _systemd_ issue a **fsck** on a fs on a partition on a **USB HDD**

If you either remove the `/etc/fstab` line or set `fs_passno` to 0 (instead of 2) to disable any fsck, then the boot is normal.

The partition is a _non system partition_ (no root or boot).

It's worse if you have root on the USB HDD (boot still on SD card), because the kernel throws an almost endless list of stack trace, and might stops after some minutes

[image]

P33M, 2020-01-30:
Arch Linux build their own kernels for Raspberry Pi. I suggest you seek support from the maintainers of that distribution.

solsticedhiver, 2020-01-31:
They build but they do not write one. So the bug comes more from your code that their build

pelwell, 2020-01-31:
Your stack trace is incomplete - we need to know what is blitting to the display with interrupts disabled. If you have an image where blt_move is further up the screen then that would be very helpful.

pelwell, 2020-01-31:
Adding `framebuffer_height=2160` to config.txt will fit more lines on the screen.

solsticedhiver, 2020-01-31:
sorry. I can't reproduce it anymore. That's weird. may be because I used a 2.0A power supply not recommended for a rpi3) during my test ?

It was during a bug hunting session because I have a another system (aka same OS and hardware but different setup) that throws an almost endless list of stack trace at boot, I don't know why. but it happens only on that system.

May be I should open a bug for that instead ?

closed 2020-01-31 by solsticedhiver